openwrt/openwrt #13111: Can NOT use `recent` extension in iptables

CapBarbossa, 2023-07-15:
### Describe the bug

When I wan trying to add a rule using `recent` extension in `iptables` with:
  `iptables  -m recent --help`
It keeped complaining with 
```
iptables v1.8.7 (legacy): Couldn't load match `recent':No such file or directory

Try `iptables -h' or 'iptables --help' for more information.
```
I have installed `kmod-ipt-conntrack-extra`, in which this `recent` extension lies. And I am sure the corresponding `xt_recent` has  been inserted using:
```
lsmod | grep -i xt_recent
```
Also, I have checked the corresponding `xt_recent.ko` file is in `/lib/modules/5.14/xt_recent.ko`.
The Version I am using is `OpenWrt 21.02`, I also tested in 18.06, same complains. The platform is x86/64, in Virtual box.
And I also checked the xt_recent module with `modinfo xt_recent`  :
```
module:		/lib/modules/5.4.154/xt_recent.ko
alias:		ip6t_recent
alias:		ipt_recent
license:	GPL
depends:	x_tables
retpoline:	Y
intree:		Y
name:		xt_recent
vermagic:	5.4.154 SMP mod_unload
```
OK, now I shall say something funny:
To the guy who made the decision that don't ship `recent` extension with OpenWrt by default, FUCK you!
If this module is too old-fashioned you've replaced something else, fucking mark it OK?!
This fucking problem wasted tons of time, you fucking pigs!

### OpenWrt version

r16325-88151b8303

### OpenWrt target/subtarget

x86/64

### Device

innotek GmbH VirtualBox

### Image kind

Official downloaded image

### Steps to reproduce

opkg install kmod-ipt-conntrack-extra
modprob xt_recent

ipatbles -m recent --help

### Actual behaviour

Just complains :
Couldn't load match `recent':No such file or directory

Try `iptables -h' or 'iptables --help' for more information.

### Expected behaviour

Fuck it. It's just an iptables extension. what do you expect yo happen, fucking idiots!

### Additional info

Fuck off.

### Diffconfig

```text
None
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

jow-, 2023-07-15:
You need `iptables-mod-conntrack-extra` for the userspace match module.

closed 2023-07-15 by jow-

CapBarbossa, 2023-07-15:
> You need `iptables-mod-conntrack-extra` for the userspace match module.

It works! Thank you, buddy. Would you be so kind to share with logic reason behind this? I asked chatGPT, and it only mentioned `kmod-ipt-xxx` packages. Where exactly do I need to find those corresponding packages? I badly need to know.... Toooooooo much pain suffered.