openwrt/openwrt #13112: [23.05-rc2, snapshot] mediatek/filogic: 2.4 GHz: HE40/HT40 doesn't work if 802.11s (mesh) is configured

csharper2005, 2023-07-15:
### Describe the bug

2.4 GHz: HE40/HT40 doesn't work if 802.11s (mesh) is configured. Device uses 20 MHz BW.

### OpenWrt version

r23228-cd17d8df2a, r23567-9236757a99

### OpenWrt target/subtarget

mediatek/filogic

### Device

Mercusys MR90X v1

### Image kind

Official downloaded image

### Steps to reproduce

1. Configure 40 MHz BW for 2.4:
```
	option htmode 'HE40'
	option noscan '1'
```
2. Make sure that clients connects using 40 MHz BW
3. Configure 802.11s:
```
config wifi-iface 'wifinet4'
	option device 'radio0'
	option mode 'mesh'
	option encryption 'sae'
	option mesh_id 'xxxx'
	option mesh_fwding '1'
	option mesh_rssi_threshold '0'
	option key '********************'
	option network 'lan'
	option skip_inactivity_poll '1'
	option disassoc_low_ack '0'
```

### Actual behaviour

Clients connect using only 20 MHz BW if 802.11s node is enabled.

### Expected behaviour

Clients can connect using 40 MHz BW even if 802.11s node is enabled.

### Additional info

802.11s node is enabled:
```
# iwinfo phy0-mesh0
phy0-mesh0 ESSID: "xxxx"
          Access Point: xx:xx:xx:xx:xx:xx
          Mode: Mesh Point  Channel: 1 (2.412 GHz)  HT Mode: HT20
          Center Channel 1: 1 2: unknown
          Tx-Power: 27 dBm  Link Quality: 59/70
          Signal: -51 dBm  Noise: -92 dBm
          Bit Rate: 1.0 MBit/s
          Encryption: WPA3 SAE (CCMP)
          Type: nl80211  HW Mode(s): 802.11ax/b/g/n
          Hardware: embedded [MediaTek MT7986]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy0

# iwinfo phy0-ap0 info
phy0-ap0  ESSID: "xxxx"
          Access Point: xx:xx:xx:xx:xx:xx
          Mode: Master  Channel: 1 (2.412 GHz)  HT Mode: HE20
          Center Channel 1: 1 2: unknown
          Tx-Power: 27 dBm  Link Quality: unknown/70
          Signal: unknown  Noise: -92 dBm
          Bit Rate: unknown
          Encryption: WPA3 SAE (CCMP)
          Type: nl80211  HW Mode(s): 802.11ax/b/g/n
          Hardware: embedded [MediaTek MT7986]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy0
```
802.11s node is disabled:
```
phy0-ap0  ESSID: "xxxx"
          Access Point: xx:xx:xx:xx:xx:xx
          Mode: Master  Channel: 1 (2.412 GHz)  HT Mode: HE40
          Center Channel 1: 3 2: unknown
          Tx-Power: 27 dBm  Link Quality: unknown/70
          Signal: unknown  Noise: -92 dBm
          Bit Rate: unknown
          Encryption: WPA3 SAE (CCMP)
          Type: nl80211  HW Mode(s): 802.11ax/b/g/n
          Hardware: embedded [MediaTek MT7986]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy0
```

Corresponding router (TP-Link EC330-G5u v1, ramips/mt7621) is not affected if 802.11s is enabled. HT40 works fine:
```
# iwinfo phy0-mesh0 info
phy0-mesh0 ESSID: "xxxx"
          Access Point: xx:xx:xx:xx:xx:xx
          Mode: Mesh Point  Channel: 1 (2.412 GHz)  HT Mode: HT20
          Center Channel 1: 1 2: unknown
          Tx-Power: 20 dBm  Link Quality: 66/70
          Signal: -44 dBm  Noise: -84 dBm
          Bit Rate: 288.9 MBit/s
          Encryption: WPA3 SAE (CCMP)
          Type: nl80211  HW Mode(s): 802.11b/g/n
          Hardware: 14C3:7615 7615:14C3 [MediaTek MT7615E]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy0

# iwinfo phy0-ap0 info
phy0-ap0  ESSID: "xxxx"
          Access Point: xx:xx:xx:xx:xx:xx
          Mode: Master  Channel: 1 (2.412 GHz)  HT Mode: HT40
          Center Channel 1: 3 2: unknown
          Tx-Power: 20 dBm  Link Quality: 70/70
          Signal: -32 dBm  Noise: -83 dBm
          Bit Rate: 300.0 MBit/s
          Encryption: WPA3 SAE (CCMP)
          Type: nl80211  HW Mode(s): 802.11b/g/n
          Hardware: 14C3:7615 7615:14C3 [MediaTek MT7615E]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy0
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

hauke, 2023-07-15:
Please try a openwrt 23.05 snapshot build, multiple fixes were backported for mt76 after 23.05.0-rc2, see 649d9e359005377b9efc445466db58f54a547875 and a70d030fb33c246c3eb222beccbd69c427b3a66f

csharper2005, 2023-07-15:
@hauke https://github.com/openwrt/openwrt/commit/649d9e359005377b9efc445466db58f54a547875 and https://github.com/openwrt/openwrt/commit/a70d030fb33c246c3eb222beccbd69c427b3a66f were cherry-picked from the main branch. I tried the latest snapshot r23566-37ff916af7 (Sat Jul 15 09:58:38 2023 @downloads.openwrt.org). 
Clients can use 40MHz:
[image]
Mesh are 20MHz only:
[image]
Does it make sense to check the 23.05-snapshot if the issue shows up in the main too?

csharper2005, 2023-07-15:
I noticed that 5 GHz mesh is also 20 MHz only. So, the issue is band independent.

csharper2005, 2023-07-15:
Some additional cases when 802.11s (mesh) use only 20MHz BW:
- mediatek/filogic (Mercusys MR90X v1, MediaTek MT7986) - ramips/mt7621 (WiFire S1500.NBN, MediaTek MT7602E)
- ramips/mt7621 (TP-Link EC330-G5u v1, MediaTek MT7615E) - ramips/mt7621 (WiFire S1500.NBN, MediaTek MT7602E)

xabolcs, 2023-07-15:
I'm unable to reproduce with D-Link DIR-860L B1 (with 22.03.5 and 23.05-SNAPSHOT at `r23274-b32a4bc33f`) + Xiaomi AX3200 (with 23.05-SNAPSHOT, `r23273-50507f5af3`,MT7622, MT7915E) on 5 GHz mesh at channel 36.

I know, it's not `mediatek/filogic` but still `mt76`. I could change the DIR-860L to Dynalink DL-WRX36 in the 5 GHz mesh with AX3200, if it makes sense. Should I give it a try?

(Offtopic, but trying to reproduce this issue, I got a some `MAC error detected` errors on the DIR-860L. :shrug: 
```
Sat Jul 15 20:14:13 2023 kern.info kernel: [ 2483.547746] mt76x2e 0000:01:00.0: Firmware Version: 0.0.00
Sat Jul 15 20:14:13 2023 kern.info kernel: [ 2483.558778] mt76x2e 0000:01:00.0: Build: 1
Sat Jul 15 20:14:13 2023 kern.info kernel: [ 2483.566991] mt76x2e 0000:01:00.0: Build Time: 201607111443____
Sat Jul 15 20:14:13 2023 kern.info kernel: [ 2483.596554] mt76x2e 0000:01:00.0: Firmware running!
Sat Jul 15 20:14:13 2023 kern.info kernel: [ 2483.608327] ieee80211 phy0: Hardware restart was requested
```
)

csharper2005, 2023-07-15:
> I'm unable to reproduce with D-Link DIR-860L B1 (with 22.03.5 and 23.05-SNAPSHOT at `r23274-b32a4bc33f`) + Xiaomi AX3200 (with 23.05-SNAPSHOT, `r23273-50507f5af3`,MT7622, MT7915E) on 5 GHz mesh at channel 36.

Thanks for the test! Did you reach an 80 MHz BW 802.11s connection between the routers?

> I know, it's not mediatek/filogic but still mt76. I could change the DIR-860L to Dynalink DL-WRX36 in the 5 GHz mesh with AX3200, if it makes sense. Should I give it a try?

Yes, please. I have only 20 MHz link even between 2 mt7621 routers.

> (Offtopic, but trying to reproduce this issue, I got a some `MAC error detected` errors on the DIR-860L.

Oh, mt7602? How familiar. :) The possible solution - https://github.com/openwrt/openwrt/blob/main/target/linux/ramips/dts/mt7621_sercomm_s1500.dtsi#L214

xabolcs, 2023-07-15:
> Did you reach the 80 MHz BW 802.11s connection between the routers?

Sure, without problem.
From DIR-860L point of view:
```
mesh0     ESSID: "mesh"
          Access Point: XX:XX:XX:XX:XX:XX
          Mode: Mesh Point  Channel: 36 (5.180 GHz)  HT Mode: VHT80
          Center Channel 1: 42 2: unknown
          Tx-Power: 20 dBm  Link Quality: 39/70
          Signal: -71 dBm  Noise: unknown
          Bit Rate: 390.0 MBit/s
          Encryption: WPA3 SAE (CCMP)
          Type: nl80211  HW Mode(s): 802.11ac/n
          Hardware: 14C3:7662 14C3:7662 [MediaTek MT76x2E]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy0

phy1-ap0  ESSID: "ap"
          Access Point: XX:XX:XX:XX:XX:XX
          Mode: Master  Channel: 3 (2.422 GHz)  HT Mode: HT40
          Center Channel 1: 5 2: unknown
          Tx-Power: 10 dBm  Link Quality: 70/70
          Signal: -37 dBm  Noise: unknown
          Bit Rate: 132.7 MBit/s
          Encryption: WPA2 PSK (CCMP)
          Type: nl80211  HW Mode(s): 802.11b/g/n
          Hardware: 14C3:7662 14C3:7662 [MediaTek MT76x2E]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy1

```

The same mesh part from AX3200:
```
mesh0     ESSID: "mesh"
          Access Point: XX:XX:XX:XX:XX:XX
          Mode: Mesh Point  Channel: 36 (5.180 GHz)  HT Mode: VHT80
          Center Channel 1: 42 2: unknown
          Tx-Power: 20 dBm  Link Quality: 38/70
          Signal: -72 dBm  Noise: unknown
          Bit Rate: 117.0 MBit/s
          Encryption: WPA3 SAE (CCMP)
          Type: nl80211  HW Mode(s): 802.11ac/ax/n
          Hardware: 14C3:7915 14C3:7915 [MediaTek MT7915E]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: wl1

```

And here is the link from Dynalink DL-WRX36 - while `iperf3`-ing:
```
mesh0     ESSID: "mesh"
          Access Point: DE:AD:BE:EF:XX:XX
          Mode: Mesh Point  Channel: 36 (5.180 GHz)  HT Mode: VHT80
          Center Channel 1: 42 2: unknown
          Tx-Power: 18 dBm  Link Quality: 30/70
          Signal: -80 dBm  Noise: -110 dBm
          Bit Rate: 324.2 MBit/s
          Encryption: WPA3 SAE (CCMP)
          Type: nl80211  HW Mode(s): 802.11ac/ax/n
          Hardware: embedded [Qualcomm Atheros IPQ8074]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy0

```
Without stress it reports 6 mbps for `iwinfo`, also on LuCI. :roll_eyes: 
[image]

From AX3200 point of view:
```
mesh0     ESSID: "mesh"
          Access Point: XX:XX:XX:XX:XX:XX
          Mode: Mesh Point  Channel: 36 (5.180 GHz)  HT Mode: VHT80
          Center Channel 1: 42 2: unknown
          Tx-Power: 20 dBm  Link Quality: 34/70
          Signal: -76 dBm  Noise: unknown
          Bit Rate: 408.3 MBit/s
          Encryption: WPA3 SAE (CCMP)
          Type: nl80211  HW Mode(s): 802.11ac/ax/n
          Hardware: 14C3:7915 14C3:7915 [MediaTek MT7915E]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: wl1

```
[image]

csharper2005, 2023-07-16:
@xabolcs thanks again. I added all test results to the issue description. And I tested again. Looks like it affects 2.4 GHz only. 20 MHz on 5 GHz in my previous tests was due to power saving when idle.

csharper2005, 2023-07-16:
> Please try a openwrt 23.05 snapshot build

23.05-SNAPSHOT r23274-b32a4bc33f. Issue exists with 2.4 band. I added this result to the issue description.

It would be good to test on non-mt76 platforms (e.g. Qualcomm).

xabolcs, 2023-07-16:
> It would be good to test on non-mt76 platforms (e.g. Qualcomm).

I tried mesh on 2.4 band on the Dynalink, but it didn't came up with 40 Mhz wide  (ie. `HT40`, `HE40`) band mode:
```
wpa_supplicant[1864]: Could not select hw_mode and channel. (-3)
```

Will check the logs from AX3200 too.

csharper2005, 2023-07-16:
> I tried mesh on 2.4 band on the Dynalink, but it didn't came up with 40 Mhz wide

Got the same error when selecting channel 13. You can try channel 1 or 9.

xabolcs, 2023-07-16:
Thanks for the hint! Far better now!

Dynalink:
```
mesh0     ESSID: "mesh"
          Access Point: XX:XX:XX:XX:XX:XX
          Mode: Mesh Point  Channel: 9 (2.452 GHz)  HT Mode: HT20
          Center Channel 1: 9 2: unknown
          Tx-Power: 14 dBm  Link Quality: 48/70
          Signal: -62 dBm  Noise: -104 dBm
          Bit Rate: 1.0 MBit/s
          Encryption: WPA3 SAE (CCMP)
          Type: nl80211  HW Mode(s): 802.11ax/b/g/n
          Hardware: embedded [Qualcomm Atheros IPQ8074]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy1

```

AX3200:
```
mesh0     ESSID: "mesh"
          Access Point: XX:XX:XX:XX:XX:XX
          Mode: Mesh Point  Channel: 9 (2.452 GHz)  HT Mode: HT20
          Center Channel 1: 9 2: unknown
          Tx-Power: 20 dBm  Link Quality: 52/70
          Signal: -58 dBm  Noise: -85 dBm
          Bit Rate: 288.9 MBit/s
          Encryption: WPA3 SAE (CCMP)
          Type: nl80211  HW Mode(s): 802.11b/g/n
          Hardware: embedded [MediaTek MT7622]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: wl0

```

Looks like it's really 2g only. And it's not only for `mt76`!

(Offtopic here, but with ath (from 9 to 11) I got only low transmission power in mesh mode. :shrug:  The max. is 14 dBm in 2g and 18 dBm in 5g mode ... Of course in `ap` mode it's all ok!)

csharper2005, 2023-07-17:
> Looks like it's really 2g only. And it's not only for `mt76`!

Yeah, the best driver in the world is not to blame. :) The issue exists between 2 Qualcomms:
Asus OnHub Qualcomm IPQ8064 | Netgear Nighthawk X4S R7800 Qualcomm IPQ8065
-- | --

> wpa_supplicant[1864]: Could not select hw_mode and channel. (-3)

I think we need open a separate issue for this error.

raenye, 2023-08-13:
Sharing my datapoint.

- 2.4GHz: mesh is HT20, AP is HT40
- 5GHz: both mesh and AP are VHT80

#### Device 1
Xiaomi Redmi AX6S (mt7622bv/mt7975an), identical to AX3200
snapshot r23682+8-1eb67cb070

```
# iwinfo
mesh0     ESSID: "xxxx"
          Access Point: xx:xx:xx:xx:xx:xx
          Mode: Mesh Point  Channel: 9 (2.452 GHz)  HT Mode: HT20
          Center Channel 1: 9 2: unknown
          Tx-Power: 1 dBm  Link Quality: 48/70
          Signal: -62 dBm  Noise: -88 dBm
          Bit Rate: 13.0 MBit/s
          Encryption: WPA3 SAE (CCMP)
          Type: nl80211  HW Mode(s): 802.11b/g/n
          Hardware: embedded [MediaTek MT7622]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: wl0

mesh1     ESSID: "xxxx-5"
          Access Point: xx:xx:xx:xx:xx:xx
          Mode: Mesh Point  Channel: 36 (5.180 GHz)  HT Mode: VHT80
          Center Channel 1: 42 2: unknown
          Tx-Power: 14 dBm  Link Quality: 45/70
          Signal: -65 dBm  Noise: -92 dBm
          Bit Rate: 195.1 MBit/s
          Encryption: WPA3 SAE (CCMP)
          Type: nl80211  HW Mode(s): 802.11ac/ax/n
          Hardware: 14C3:7915 14C3:7915 [MediaTek MT7915E]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: wl1
```
#### Device 2
D-Link DAP-1620-B1 (mt7621at/mt7615dn DBDC)
snapshot r23658+8-e3dd5ee02b

```
# iwinfo
mesh0     ESSID: "xxxx"
          Access Point: xx:xx:xx:xx:xx:xx
          Mode: Mesh Point  Channel: 9 (2.452 GHz)  HT Mode: HT20
          Center Channel 1: 9 2: unknown
          Tx-Power: 5 dBm  Link Quality: 41/70
          Signal: -69 dBm  Noise: unknown
          Bit Rate: 13.0 MBit/s
          Encryption: WPA3 SAE (CCMP)
          Type: nl80211  HW Mode(s): 802.11b/g/n
          Hardware: 14C3:7615 7615:14C3 [MediaTek MT7615E]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy0

mesh1     ESSID: "xxxx-5"
          Access Point: xx:xx:xx:xx:xx:xx
          Mode: Mesh Point  Channel: 36 (5.180 GHz)  HT Mode: VHT80
          Center Channel 1: 42 2: unknown
          Tx-Power: 14 dBm  Link Quality: 40/70
          Signal: -71 dBm  Noise: -85 dBm
          Bit Rate: 468.0 MBit/s
          Encryption: WPA3 SAE (CCMP)
          Type: nl80211  HW Mode(s): 802.11ac/n
          Hardware: 14C3:7615 7615:14C3 [MediaTek MT7615E]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy1
```

csharper2005, 2023-10-15:
23.05.0 - issue is still alive

Writter77, 2023-10-23:
In my Xiaomi AX3200 does not switch to 40MHz. If there is a force 40MHz also does not switch to 40MHz. I have 20MHz all the time. How to fix it?

raenye, 2023-10-23:
@Writter77 are you talking about mesh (802.11s)? If not, your issue is not related to this one.

Writter77, 2023-10-23:
I'm not talking about mesh Despite this, the problem also occurs on ordinary openwrt without mesh

raenye, 2023-10-23:
So please avoid writing about your problem in an issue specific to mesh points, even if the symptoms seem similar.
Your problem is probably due to misconfiguration, and it's better to ask about it in the forum or over discord.

exidler, 2023-10-30:
OpenWrt 23.05.0 release: `OpenWrt 23.05.0 r23497-6637af95aa` build with official builder with customized packages list
WPAD: wpad-wolfssl
HW: BPI-R3 (mediatek/filogic)

2.4 Ghz HT40 does not work in mesh mode.
Tried 1, 6 channel. WPA3-SAE and no-encryption too. 
No luck, only 20Mhz

5Ghz mesh - 80Mhz works fine!

```
mesh   ESSID: "xxxx"
          Access Point: C2:1B:30:B9:2B:D8
          Mode: Mesh Point  Channel: 6 (2.437 GHz)  HT Mode: HT20
          Center Channel 1: 6 2: unknown
          Tx-Power: 23 dBm  Link Quality: 56/70
          Signal: -54 dBm  Noise: -92 dBm
          Bit Rate: 258.0 MBit/s
          Encryption: WPA3 SAE (CCMP)
          Type: nl80211  HW Mode(s): 802.11ax/b/g/n
          Hardware: embedded [MediaTek MT7986]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy0
```

```
config wifi-device 'radio0'
	option type 'mac80211'
	option path 'platform/soc/18000000.wifi'
	option channel '6'
	option band '2g'
	option htmode 'HE40'
	option noscan '1'
	option cell_density '0'
	option txpower '23'

config wifi-iface 'wifinet5'
	option device 'radio0'
	option mode 'mesh'
	option encryption 'sae'
	option mesh_id 'xxxx'
	option mesh_fwding '1'
	option mesh_rssi_threshold '0'
	option network 'lan'
	option key 'xxxx'
```

csharper2005, 2023-11-06:
@Ansuel we found a root cause for 22.03.5 [here](https://forum.openwrt.org/t/openwrt-23-05-0-first-stable-release/174239/361). 

The `23.05` and `main` issue looks much more complex. `noscan=1` is in place. https://github.com/openwrt/openwrt/issues/13128 seems to be related. Likely this started after https://github.com/openwrt/openwrt/commit/304423a4ff841129f1b6b7504c83dd11c1f71c84.
```
cat wpa_supplicant-phy0-mesh0.conf
country=**
network={

        ssid="2400"
        key_mgmt=SAE
        mode=5
        mesh_fwding=1
        mesh_rssi_threshold=0
        fixed_freq=1
        frequency=2452
        ht40=1
        disable_vht=1
        noscan=1
        sae_password="*************"
        beacon_int=100
}
```

Ansuel, 2023-11-07:
@csharper2005 can you by chance provide the change done for 22.03?

Ansuel, 2023-11-08:
I found the problem! @csharper2005 can you compile a testing image?

csharper2005, 2023-11-08:
> I found the problem! @csharper2005 can you compile a testing image?

Great news! I can. Please let me know what should be patched and which branch.

commit 6c9ac57d (2023-11-09): hostapd: permit 40MHz in 802.1s only also for 2.4GHz g/n with noscan

Currently for 802.1s only, for wifi 2.4GHz in g/n mode, 40MHz is never
permitted.

This is probably due to the complexity of setting periodic check for the
intolerant bit. When noscan option is set, we ignore the presence of the
intoleran bit in near AP, so we can enable 40MHz and ignore any complex
logic for checking.

Fixes: #13112
commit 85d1b43b (2023-11-09): hostapd: permit 40MHz in 802.1s only also for 2.4GHz g/n with noscan

Currently for 802.1s only, for wifi 2.4GHz in g/n mode, 40MHz is never
permitted.

This is probably due to the complexity of setting periodic check for the
intolerant bit. When noscan option is set, we ignore the presence of the
intoleran bit in near AP, so we can enable 40MHz and ignore any complex
logic for checking.

Fixes: #13112

(cherry picked from commit 6c9ac57d58ab0d2c71fb51120a2166b7e0bd7183)
closed 2023-11-09 by openwrt-bot