openwrt/openwrt #13114: openwrt在宽带的IPV6的PD变动以后就不会给LAN分配IPV6地址了,希望能修复这个bug

chainofhonor, 2023-07-16:
### Describe the bug

国内宽带的IPV6地址是经常变的,一般来说每个星期会变一次
然后openwrt最开始是正常的,WAN口正常获取PD,然后LAN口正常的分配IPV6地址

但是每当电信重新分配IPV6以后
openwrt的WAN口会正常的变更IPV6地址,但是LAN口却不会,然后局域网内的设备就全部没有IPV6地址了

感觉像是openwrt并没有接收新的PD

这个问题我去年就提过了,但是不知道为什么,直到现在这个问题还是没修复

### OpenWrt version

r23228-cd17d8df2a

### OpenWrt target/subtarget

ramips/mt7621

### Device

Netgear R6220

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

danpawlik, 2023-07-16:
@chainofhonor please change issue title and description to English.

delgh1, 2023-07-17:
不是bug。
This is not a bug. IPv6 prefixes are supposed to be robust and rarely change (maybe once in a decade).
Ask your ISP to stop doing that. File a complain on the grounds of non-compliance of RIPE-690 and RFC8978.

Also enabling RA may fix this.

chainofhonor, 2023-07-18:
> 不是bug。 This is not a bug. IPv6 prefixes are supposed to be robust and rarely change (maybe once in a decade). Ask your ISP to stop doing that. File a complain on the grounds of non-compliance of RIPE-690 and RFC8978.
> 
> Also enabling RA may fix this.

我试试 默认的RA模式是auto 我改为forced试试 等过几天我看看是否会正常改变再来回帖

XiaoliChan, 2023-11-24:
> 我试试 默认的RA模式是auto 我改为forced试试 等过几天我看看是否会正常改变再来回帖

Did the `forced` mode in RA solve this problem?

chainofhonor, 2023-11-24:
> > 我试试 默认的RA模式是auto 我改为forced试试 等过几天我看看是否会正常改变再来回帖
> 
> Did the `forced` mode in RA solve this problem?

我没用openwrt了 我改用rockylinux当软路由了,然后我用IPV6 NAT模式  内网是fc01::/64的网段  这样也能用于上网

XiaoliChan, 2023-11-27:
> My ISP also changes the IPv6 prefix every 24 hours.
> 
> If you are using PPPoE have sure that `ipv6` setting is not set at all or set to `auto`, see https://openwrt.org/docs/guide-user/network/ipv6/configuration#upstream_configuration_for_wan_interfaces If you set it to `option ipv6 1` it will only configure it the first time correctly.

Thanks for your solution, it looks like my issue is gone after I decreased the RA lifetime to 5 mins.

Icemans007, 2024-09-26:
> > 不是bug。 This is not a bug. IPv6 prefixes are supposed to be robust and rarely change (maybe once in a decade). Ask your ISP to stop doing that. File a complain on the grounds of non-compliance of RIPE-690 and RFC8978.
> > Also enabling RA may fix this.
> 
> 我试试 默认的RA模式是auto 我改为forced试试 等过几天我看看是否会正常改变再来回帖

我也遇到这个问题，按你这个方式设置后，好似已经解决