openwrt/openwrt #13116: When setting DHCP Option 43 on OpenWrt, there is an extra 16 hexadecimal FF in the DHCP response.

acdxxx177, 2023-07-16:
### Describe the bug

I filled in "**vendor:ubnt,1,192.168.3.195**" in "**Network** » **Interfaces** » **LAN** » **DHCP Server** » **Advanced Settings.**" When I captured packets using Wireshark, I found that the value of Option 43 is "**0104c0a803c3ff**" but it should be "**0104c0a803c3**" without the extra 16 hexadecimal FF.

> cat  /etc/config/dhcp
```
# Omit default configuration
config dhcp 'lan'
        option interface 'lan'
        option start '100'
        option limit '150'
        option leasetime '12h'
        option dhcpv4 'server'
        option dhcpv6 'server'
        option ra 'server'
        list ra_flags 'managed-config'
        list ra_flags 'other-config'
        list dhcp_option 'vendor:ubnt,1,192.168.3.195'
```

> cat /var/etc/dnsmasq.conf.cfg01411c
```
# Omit default configuration
bogus-priv
conf-file=/usr/share/dnsmasq/rfc6761.conf
dhcp-range=set:lan,192.168.3.100,192.168.3.249,255.255.255.0,12h
dhcp-option=lan,vendor:ubnt,1,192.168.3.195
no-dhcp-interface=pppoe-wan
```

### OpenWrt version

r22037-b174de2c5f

### OpenWrt target/subtarget

x86/64

### Device

VMware, Inc. VMware Virtual Platform

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

Spudz76, 2023-07-16:
Unclear why there is a leading comma in `vendor:,`

john-tho, 2023-07-17:
Looks like this should be fine, according to the DHCP options RFC? Are you having issues with a client?
https://www.rfc-editor.org/rfc/rfc2132.html#section-8.4
>      3) Code 255 (END), if present, signifies the end of the
>         encapsulated vendor extensions, not the end of the vendor
>         extensions field. If no code 255 is present, then the end of
>         the enclosing vendor-specific information field is taken as the
>         end of the encapsulated vendor-specific extensions field.

If you want to look through why this happens, see `DHOPT_VENDOR` in `dnsmasq` source: https://thekelleys.org.uk/gitweb/?p=dnsmasq.git;a=blob;f=src/option.c;h=832272559d68fd93c8ab61605e2211a6826df07e;hb=HEAD

[dnsmasq man page](https://dnsmasq.org/docs/dnsmasq-man.html) examples:
> `--dhcp-option=vendor:PXEClient,1,0.0.0.0` sends the encapsulated vendor class-specific option "mftp-address=0.0.0.0" to any client whose vendor-class matches "PXEClient".
> It is possible to omit the vendorclass completely; `--dhcp-option=vendor:,1,0.0.0.0` in which case the encapsulated option is always sent.

These OpenWrt options are applied through the init script: https://github.com/openwrt/openwrt/blob/main/package/network/services/dnsmasq/files/dnsmasq.init

Not seeing an issue in any of this?

acdxxx177, 2023-07-18:
> Looks like this should be fine, according to the DHCP options RFC? Are you having issues with a client? https://www.rfc-editor.org/rfc/rfc2132.html#section-8.4
>
>
> If you want to look through why this [...]

Thank you for your response. I was just comparing it to other routers, why does this phenomenon occur? Now I understand the principle.

closed 2023-07-18 by acdxxx177