openwrt/openwrt #13117: eth0 and eth1

nightlyDuck, 2023-07-16:
### Describe the bug

When the OS is fully started, wan (eth0 port) and lan (eth1 port) do not work correctly, namely, traffic between these ports does not pass. Switching the port speed also does not work correctly.
Surprisingly, assemblies from the manufacturer work fine.

### OpenWrt version

r23579-1c56801dd2 (also applies to 23.05-rc2)

### OpenWrt target/subtarget

rockchip/armv8

### Device

Xunlong Orange Pi R1 Plus LTS

### Image kind

Self-built image

### Steps to reproduce

1. Flash 23.05-rc2 (or build from latest commit) image SD card.
2. Boot system
3. Plug rj-45 cable in wan and lan

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

RyanBlakeIT, 2023-09-19:
Having the same issue and all ETH ports are not working properly, not just ETH1.

danpawlik, 2023-09-20:
Could you add  `dmesg` or `logread` logs?

RyanBlakeIT, 2023-09-20:
> Replying to https://github.com/openwrt/openwrt/issues/13117#issuecomment-1727881187
I'm embarrassed to say that I should've created a separate bug report since this is impacting my TP-Link EAP615-Wall v1, not an OrangePi.

kristof-dragon, 2023-10-02:
> Could you add `dmesg` or `logread` logs?

Please see them attached.

I have been keeping an eye on the progress, as I have been eagerly waiting for some progress...
The LAN (eth1) passes traffic just fine, but the eth0 does not even receive a DHCP IP.

When running the board with the orangepi ubuntu image it is actually the eth0 interface that can be utilised (the eth1 is not detected...) just adding this as some context that might help

anyway, please see the dmesg and the kogread exports attached.
[logread_and_dmesg_openwrt_opiR1lts.zip](https://github.com/openwrt/openwrt/files/12786780/logread_and_dmesg_openwrt_opiR1lts.zip)

commit 3645ac8a (2024-01-04): rockchip: configure eth pad driver strength for orangepi r1 plus lts

The default strength is not enough to provide stable connection
under 3.3v LDO voltage.

Fixes: 32d5921b8b55 ("rockchip: add Orange Pi R1 Plus LTS support")
Fixes: #13117
Fixes: #13759
PR #14276 (2024-01-04): rockchip: configure eth pad driver strength for orangepi r1 plus lts
The default strength is not enough to provide stable connection under 3.3v LDO voltage.

Fixes: 32d5921b8b55 ("rockchip: add Orange Pi R1 Plus LTS support")
Fixes: #13117
Fixes: #13759
commit 80ef582d (2024-01-05): rockchip: configure eth pad driver strength for orangepi r1 plus lts

The default strength is not enough to provide stable connection
under 3.3v LDO voltage.

Fixes: 3f3586a06d27 ("rockchip: add Orange Pi R1 Plus LTS support")
Fixes: #13117
Fixes: #13759

(cherry picked from commit 3645ac8a10d9abb1451343beaf7d65b53eeecffd)
[rebased onto openwrt-23.05 branch]
closed 2024-01-04 by openwrt-bot