openwrt/openwrt #13119: Enabling KERNEL_DEVTMPFS has a new required symbol

graysky2, 2023-07-16:
### Describe the bug

Enabling `CONFIG_KERNEL_DEVTMPFS` needs to also add this symbol to the resulting `.config`: `CONFIG_DEVTMPFS_SAFE=y` or else the build will fail.

### OpenWrt version

N/A

### OpenWrt target/subtarget

x86/64

### Device

AMD Ryzen 7

### Image kind

Official downloaded image

### Steps to reproduce

1. `make nconfig`
2. navigate to Global build settings > Kernel build options and enable `Compile the kernel with device tmpfs enabled (KERNEL_DEVTMPFS [=y])` and `Compile the kernel with device tmpfs enabled (KERNEL_DEVTMPFS [=y])`
3. Try to build

### Actual behaviour

Build will error out due to missing symbol.

### Expected behaviour

Symbol should be added automatically.

### Additional info

```% make V=s
...
*
* Restart config...
*
*
* Generic Driver Options
*
Support for uevent helper (UEVENT_HELPER) [Y/n/?] y
  path to uevent helper (UEVENT_HELPER_PATH) [/sbin/hotplug] /sbin/hotplug
Maintain a devtmpfs filesystem to mount at /dev (DEVTMPFS) [Y/n/?] y
  Automount devtmpfs at /dev, after the kernel mounted the rootfs (DEVTMPFS_MOUNT) [Y/n/?] y
  Use nosuid,noexec mount options on devtmpfs (DEVTMPFS_SAFE) [N/y/?] (NEW) 
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brainstorm, 2023-09-22:
I can confirm this in other targets: BananaPi BPI-R3, every subsequent `make menuconfig` asks for that `DEVTMPFS_SAFE` setting.

brainstorm, 2023-10-02:
... and this seems to be fixed by https://github.com/openwrt/openwrt/commit/e26e088203303d79eb87056f8ca74b1a5f585206 ?

brainstorm, 2023-10-02:
Well, obviously it didn't fix it for Mediatek/Filogic until applying this:

```diff
diff --git a/target/linux/mediatek/filogic/config-6.1 b/target/linux/mediatek/filogic/config-6.1
index 36cb9c1e4e..0e4cd709fb 100644
--- a/target/linux/mediatek/filogic/config-6.1
+++ b/target/linux/mediatek/filogic/config-6.1
@@ -471,3 +471,5 @@ CONFIG_ZONE_DMA32=y
 CONFIG_ZSTD_COMMON=y
 CONFIG_ZSTD_COMPRESS=y
 CONFIG_ZSTD_DECOMPRESS=y
+# CONFIG_DEVTMPFS_SAFE is not set
```

Perhaps affects other architectures the same way?

M95D, 2023-11-17:
Probably fixed with https://github.com/openwrt/openwrt/issues/13119

hauke, 2023-11-26:
Fixed in 07f90910c6f176c6459225ed0953e6b040ced19e.

closed 2023-11-26 by hauke