openwrt/openwrt #13121: SQM and the performance is too bad for v22.03.5

PerryProPlus, 2023-07-16:
### Describe the bug

Currently I am using my Xiaomi MiWifi Mini to do repeater via wisp mode, I use it with LEDE version 17.01.7.
I am very disappointed with later versions of OpenWrt (due to new LEDE not exist anymore), SQM performance on latest version 22.03.5 is very bad, bufferload when I disable SQM is always at 1000-2000ms, and when I turn on SQM it works, but it only down to 300-400ms, almost unusable due to the high latency, and the router also emits a lot of heat,I don't know what the developers did to make my router work so slow, even the Interface interface takes longer to load, and the software package always fails to download.
I love LEDE 17 so much, I wish developers make such good versions like this, SQM on this version works perfectly, from 1000ms down to 29-45ms When I play games, and the router works very fast and not as hot as the latest version.
MiWiFi Mini OpenWrt developers please improve quality firmware in version 23.05 because I don't think my 128mb ram router can't use OpenWrt as smoothly as LEDE did. I'm very interested in security patches so i try to install new version
I am so sad that OpenWrt is getting worse year by year, I don't have much money to buy new devices due to the expensive cost, this is the most powerful router I have used in 4 years, my internet is only 30mbps (emitted by the landlord and they did not let me connect the lan) .I am not too concerned with expensive devices, please send to development, thank you😔😔😔

### OpenWrt version

22.03.5

### OpenWrt target/subtarget

I don't know

### Device

Xiaomi MiWifi Mini

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-07-18:
check
`tc -s qdisc`
if default fq_codel is applied with "SQM disabled"

To apply it:
`tc qdisc replace dev eth0 root fq_codel`

And examine statistics.
Ideally it should be few requeues and no drops.

Other less CPU intensive queue disciplines are noqueue, pfifo, pfifo_fast, hfsc, which relies on LAN client performing transmit moderation on tail or head or random drops, read tc-???? manuals on a full linux system (in a VM)

PerryProPlus, 2023-07-19:
@brada4 I have tried from version 22 21 19 18 17 15 and I find chaos calmer and lede 17 are perfect for my miwifi, new versions are very slow and have bad performance especially v21 22, I also tried v23 rc2 version but the result is even worse

PerryProPlus, 2023-07-19:
I will buy new router for new version openwrt

PerryProPlus, 2023-07-19:
I will buy new router for new version openwrt 😀

brada4, 2023-07-19:
It is nowhere officially encouraged, but you can try iptables NAT/FILTER via xtables bridge on nftables releases. If you have courage to do so to point out regressions. I need that support to at --notrack in raw table only avail over bridge for example.

PerryProPlus, 2023-07-25:
@brada4 I tried to use the newer openwrt on my miwifi mini for 1 week to send feedback here, I tried the v21 v22 v23rc2 versions (I don't use sqm on this one anymore because I bought a new one), I realized the signal when using client mode on the 2.4GHz band on miwifi in v21 is extremely strong,  It was always at -50dbm and I got speeds above 50Mbps (it was a lot more powerful than the Lede 17 at 30Mbps). But v22 disappointed me with its client mode, which only reached -85 dbm at 17mbps. And this bad thing is also happening with v23 rc2 (only 15mbps), but I was surprised because the response of the luci interface in v23 is dizzyingly fast, it is not slow when moving through settings in the luci interface like v22 and v21 (luci in v23 is much faster than lede 17),  If the Signal Client Mode issue in the V23 is resolved to get the same performance as the V21, I will install the V23 on my Miwifi to use every day.

brada4, 2023-07-25:
current LuCI relies more on client-side javascript in place of OpenWRT-side LUA, that is a reasonable observation.
What is the distance between AP and STA? Do 2 AP-s look similar strength from mobile/laptop in same distance from both?

PerryProPlus, 2023-07-25:
@brada4 I also don't notice how much distance it is, but it is probably about 10m from ap to sta client (3 walls), your 2nd question is a bit hard to understand to me because I am not very good at eng, i can only say v22 and v23 on miwifi have worse speed than my xiaomi phone when using client mode.

PerryProPlus, 2023-07-25:
@brada4 
tested my phone 2m 
miwifi mini (17.3mbps) (client 2.4ghz + ap 5ghz)
modem (41.7mbps) (ap 2.4ghz)

brada4, 2023-07-25:
Probably related to https://github.com/openwrt/openwrt/issues/13153
btw you can have client and AP on (most) mt76 adapters.

PerryProPlus, 2023-07-25:
@brada4 I don't need ap+client at the same time in 2.4ghz because it will probably slow down the wifi speed. And if I turn on ap+ client (2.4ghz) and ap (5ghz) at the same time, my phone can't find 2.4ghz, only 5ghz (the strange thing is that my laptop can still find both 2.4 and 5ghz). 
Ap+ client is something that is very difficult to understand in openwrt, in padavan firmware I have never had this problem

brada4, 2023-07-25:
Check bitrate in `iwinfo` one vs another firmware version. i.e if fresh physical radio limitations damage speed.
Android does not show AP-s  in list it thinks it cannot connect, like not so long ago WPA3 etc, probably soon WEP and so on.

PerryProPlus, 2023-07-25:
@brada4 I set up wpa2, wpa and don't even set password but 2.4ghz does not appear on phone when AP is enabled

brada4, 2023-07-26:
It is to see interface speed of mobile connection - if radio is at maximum speed and signal is all 4 bars, dBm indicator in wifi analyzer is more telling than default Android visualisation. I.e. if signal of "OLD" AP is unnecessarily weakened.