openwrt/openwrt #13124: mtk7621 build syncconfig error

lushangkan, 2023-07-17:
### Describe the bug

Building mt7621 openwrt-23.05 branch error in WSL Ubuntu 22.04

Error log: [log](https://pastebin.com/mNwEGK7G)

### OpenWrt version

openwrt-23.05/57e7188

### OpenWrt target/subtarget

ramips/mt7621

### Device

Xiaomi redmi AC2100

### Image kind

Self-built image

### Diffconfig

```text
https://pastebin.com/DvVNcqMb
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

john-tho, 2023-07-17:
> `NF_CONNTRACK_CHAIN_EVENTS`
> [x] I am reporting an issue for OpenWrt, not an unsupported fork.

This looks like it comes from a downstream fork? In that case, it is not an OpenWrt issue.

lushangkan, 2023-07-17:
Tried to compile in [github action ](https://github.com/lushangkan/openwrt-mt7621/actions/runs/5573313273/jobs/10180420957), but still failed

closed 2023-07-17 by lushangkan