openwrt/openwrt #13128: [23.05-rc2, snapshot] 802.11s (mesh): Could not select hw_mode and channel. (-3)

csharper2005, 2023-07-17:
### Describe the bug

Couldn't start 802.11s (mesh) with the following conditions on OpenWrt SNAPSHOTs and 23.05.0-rc2 (22.03.5 is ok):
- Channel 13 (2.4 GHz)
- HT40/HE40

Error:
`daemon.err wpa_supplicant[1237]: Could not select hw_mode and channel. (-3)`

Full log:
```
Mon Jul 17 16:06:15 2023 kern.info kernel: [ 1290.747460] device phy1-ap0 left promiscuous mode
Mon Jul 17 16:06:15 2023 kern.info kernel: [ 1290.747590] br-lan: port 3(phy1-ap0) entered disabled state
Mon Jul 17 16:06:15 2023 daemon.notice netifd: lan (1755): udhcpc: sending renew to server 192.168.1.129
Mon Jul 17 16:06:15 2023 daemon.notice netifd: lan (1755): udhcpc: lease of 192.168.1.164 obtained from 192.168.1.129, lease time 43200
Mon Jul 17 16:06:15 2023 daemon.notice hostapd: Remove interface 'phy1-ap0'
Mon Jul 17 16:06:15 2023 daemon.notice hostapd: phy1-ap0: interface state ENABLED->DISABLED
Mon Jul 17 16:06:15 2023 daemon.notice hostapd: phy1-ap0: AP-DISABLED
Mon Jul 17 16:06:15 2023 daemon.notice hostapd: phy1-ap0: CTRL-EVENT-TERMINATING
Mon Jul 17 16:06:15 2023 daemon.err hostapd: rmdir[ctrl_interface=/var/run/hostapd]: Permission denied
Mon Jul 17 16:06:15 2023 daemon.notice hostapd: nl80211: deinit ifname=phy1-ap0 disabled_11b_rates=0
Mon Jul 17 16:06:15 2023 daemon.notice hostapd: nl80211: Failed to remove interface phy1-ap0 from bridge br-lan: Invalid argument
Mon Jul 17 16:06:15 2023 kern.info kernel: [ 1290.894309] ath10k_pci 0001:01:00.0: mac flush null vif, drop 0 queues 0xffff
Mon Jul 17 16:06:15 2023 kern.warn kernel: [ 1290.941637] ath10k_pci 0001:01:00.0: peer-unmap-event: unknown peer id 0
Mon Jul 17 16:06:15 2023 kern.warn kernel: [ 1290.941690] ath10k_pci 0001:01:00.0: peer-unmap-event: unknown peer id 0
Mon Jul 17 16:06:16 2023 daemon.notice netifd: Wireless device 'radio1' is now down
Mon Jul 17 16:06:16 2023 daemon.notice hostapd: Configuration file: /var/run/hostapd-phy1.conf (phy phy1-ap0) --> new PHY
Mon Jul 17 16:06:17 2023 daemon.info dnsmasq[1]: read /etc/hosts - 12 names
Mon Jul 17 16:06:17 2023 daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c - 0 names
Mon Jul 17 16:06:22 2023 kern.warn kernel: [ 1298.057920] ath10k_pci 0001:01:00.0: 10.4 wmi init: vdevs: 16  peers: 48  tid: 96
Mon Jul 17 16:06:22 2023 kern.warn kernel: [ 1298.057961] ath10k_pci 0001:01:00.0: msdu-desc: 2500  skid: 32
Mon Jul 17 16:06:22 2023 kern.info kernel: [ 1298.141980] ath10k_pci 0001:01:00.0: wmi print 'P 48/48 V 16 K 144 PH 176 T 186  msdu-desc: 2500  sw-crypt: 0 ct-sta: 0'
Mon Jul 17 16:06:22 2023 kern.info kernel: [ 1298.142837] ath10k_pci 0001:01:00.0: wmi print 'free: 84920 iram: 13156 sram: 11224'
Mon Jul 17 16:06:23 2023 kern.info kernel: [ 1298.551204] ath10k_pci 0001:01:00.0: rts threshold -1
Mon Jul 17 16:06:23 2023 kern.warn kernel: [ 1298.551437] ath10k_pci 0001:01:00.0: Firmware lacks feature flag indicating a retry limit of > 2 is OK, requested limit: 4
Mon Jul 17 16:06:23 2023 kern.info kernel: [ 1298.561930] br-lan: port 3(phy1-ap0) entered blocking state
Mon Jul 17 16:06:23 2023 kern.info kernel: [ 1298.566171] br-lan: port 3(phy1-ap0) entered disabled state
Mon Jul 17 16:06:23 2023 kern.info kernel: [ 1298.571864] device phy1-ap0 entered promiscuous mode
Mon Jul 17 16:06:23 2023 daemon.notice hostapd: phy1-ap0: interface state UNINITIALIZED->COUNTRY_UPDATE
Mon Jul 17 16:06:23 2023 kern.info kernel: [ 1298.894504] IPv6: ADDRCONF(NETDEV_CHANGE): phy1-ap0: link becomes ready
Mon Jul 17 16:06:23 2023 kern.info kernel: [ 1298.894647] br-lan: port 3(phy1-ap0) entered blocking state
Mon Jul 17 16:06:23 2023 kern.info kernel: [ 1298.899938] br-lan: port 3(phy1-ap0) entered forwarding state
Mon Jul 17 16:06:23 2023 daemon.notice hostapd: phy1-ap0: interface state COUNTRY_UPDATE->ENABLED
Mon Jul 17 16:06:23 2023 daemon.notice hostapd: phy1-ap0: AP-ENABLED
Mon Jul 17 16:06:24 2023 kern.info kernel: [ 1300.067542] ath10k_pci 0001:01:00.0: rts threshold -1
Mon Jul 17 16:06:24 2023 kern.info kernel: [ 1300.113942] ath10k_pci 0001:01:00.0: mac flush null vif, drop 0 queues 0xffff
Mon Jul 17 16:06:24 2023 daemon.notice netifd: radio1 (8244): command failed: Link has been severed (-67)
Mon Jul 17 16:06:24 2023 daemon.notice netifd: radio1 (8244): command failed: Link has been severed (-67)
Mon Jul 17 16:06:24 2023 daemon.notice netifd: Wireless device 'radio1' is now up
Mon Jul 17 16:06:24 2023 kern.info kernel: [ 1300.186473] br-lan: port 5(phy1-mesh0) entered blocking state
Mon Jul 17 16:06:24 2023 kern.info kernel: [ 1300.186507] br-lan: port 5(phy1-mesh0) entered disabled state
Mon Jul 17 16:06:24 2023 kern.info kernel: [ 1300.191487] device phy1-mesh0 entered promiscuous mode
Mon Jul 17 16:06:24 2023 daemon.notice netifd: lan (1755): udhcpc: sending renew to server 192.168.1.129
Mon Jul 17 16:06:24 2023 daemon.notice netifd: lan (1755): udhcpc: lease of 192.168.1.164 obtained from 192.168.1.129, lease time 43200
Mon Jul 17 16:06:24 2023 daemon.notice netifd: lan (1755): udhcpc: sending renew to server 192.168.1.129
Mon Jul 17 16:06:24 2023 daemon.notice netifd: lan (1755): udhcpc: lease of 192.168.1.164 obtained from 192.168.1.129, lease time 43200
Mon Jul 17 16:06:25 2023 kern.info kernel: [ 1301.111622] br-lan: port 3(phy1-ap0) entered disabled state
Mon Jul 17 16:06:26 2023 daemon.err wpa_supplicant[1237]: Could not select hw_mode and channel. (-3)
Mon Jul 17 16:06:26 2023 daemon.notice wpa_supplicant[1237]: phy1-mesh0: interface state UNINITIALIZED->DISABLED
Mon Jul 17 16:06:26 2023 daemon.notice wpa_supplicant[1237]: phy1-mesh0: AP-DISABLED
Mon Jul 17 16:06:26 2023 daemon.err wpa_supplicant[1237]: phy1-mesh0: Unable to setup interface.
Mon Jul 17 16:06:26 2023 daemon.err wpa_supplicant[1237]: Failed to initialize hostapd interface for mesh
Mon Jul 17 16:06:26 2023 daemon.err wpa_supplicant[1237]: phy1-mesh0: Failed to init mesh
Mon Jul 17 16:06:26 2023 daemon.notice wpa_supplicant[1237]: phy1-mesh0: leaving mesh
Mon Jul 17 16:06:26 2023 daemon.notice wpa_supplicant[1237]: phy1-mesh0: interface state DISABLED->DISABLED
Mon Jul 17 16:06:26 2023 daemon.notice wpa_supplicant[1237]: phy1-mesh0: AP-DISABLED
Mon Jul 17 16:06:26 2023 daemon.err wpa_supplicant[1237]: hostapd_free_hapd_data: Interface phy1-mesh0 wasn't started
Mon Jul 17 16:06:26 2023 daemon.err wpa_supplicant[1237]: phy1-mesh0: mesh leave error=-107
Mon Jul 17 16:06:26 2023 daemon.err wpa_supplicant[1237]: phy1-mesh0: Could not join mesh
Mon Jul 17 16:06:29 2023 kern.info kernel: [ 1304.677257] br-lan: port 3(phy1-ap0) entered blocking state
Mon Jul 17 16:06:29 2023 kern.info kernel: [ 1304.677296] br-lan: port 3(phy1-ap0) entered forwarding state
Mon Jul 17 16:06:29 2023 daemon.notice netifd: Network device 'phy1-ap0' link is up
```

### OpenWrt version

r23228-cd17d8df2a (23.05.0-rc2), r23579-1c56801dd2 (SNAPSHOT)

### OpenWrt target/subtarget

mediatek/filogic, ramips/mt7621, ipq806x/generic

### Device

Mercusys MR90X v1, TP-Link EC330-G5u v1, Netgear Nighthawk X4S R7800

### Image kind

Official downloaded image

### Steps to reproduce

1. Set up channel 13 and 40 MHz width
2. Add and enable 802.11s node

### Actual behaviour

Mesh doesn't work. Error:
`daemon.err wpa_supplicant[1237]: Could not select hw_mode and channel. (-3)`

### Expected behaviour

Mesh works fine on channel 13 and 40MHz width as it was in release 22.03.5.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-07-17:
"steps to reproduce" does not set regulatory domain, so inability to set Ch13 is among likely possibilities.

csharper2005, 2023-07-18:
> "steps to reproduce" does not set regulatory domain, so inability to set Ch13 is among likely possibilities.

You won't be able to complete step 1 without specifying a suitable regdomain. The problem is with the mesh only. AP works fine.

brada4, 2023-07-18:
Check `iw list` if it has P2P-device caps and more than one P2P-client permitted for that adapter.
(one can complete THAT step with CH13 from command line without specifying country)

csharper2005, 2023-07-18:
> Check `iw list` if it has P2P-device caps and more than one P2P-client permitted for that adapter. (one can complete THAT step with CH13 from command line without specifying country)

Is this what we need?
```
Wiphy phy0
        wiphy index: 0
        max # scan SSIDs: 4
        max scan IEs length: 2200 bytes
        max # sched scan SSIDs: 0
        max # match sets: 0
        Retry short limit: 7
        Retry long limit: 4
        Coverage class: 0 (up to 0m)
        Device supports AP-side u-APSD.
        Device supports T-DLS.
        Available Antennas: TX 0xf RX 0xf
        Configured Antennas: TX 0xf RX 0xf
        Supported interface modes:
                 * IBSS
                 * managed
                 * AP
                 * AP/VLAN
                 * monitor
                 * mesh point
                 * P2P-client
                 * P2P-GO
        Band 1:
                Capabilities: 0x9ff
                        RX LDPC
                        HT20/HT40
                        SM Power Save disabled
                        RX Greenfield
                        RX HT20 SGI
                        RX HT40 SGI
                        TX STBC
                        RX STBC 1-stream
                        Max AMSDU length: 7935 bytes
                        No DSSS/CCK HT40
                Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
                Minimum RX AMPDU time spacing: 4 usec (0x05)
                HT TX/RX MCS rate indexes supported: 0-31
                HE Iftypes: managed
                        HE MAC Capabilities (0x08011a000040):
                                +HTC HE Supported
                                Trigger Frame MAC Padding Duration: 2
                                OM Control
                                Maximum A-MPDU Length Exponent: 3
                                A-MSDU in A-MPDU
                        HE PHY Capabilities: (0x2270ce126d00f3164e3f00):
                                HE40/2.4GHz
                                242 tone RUs/2.4GHz
                                Device Class: 1
                                LDPC Coding in Payload
                                HE SU PPDU with 1x HE-LTF and 0.8us GI
                                NDP with 4x HE-LTF and 3.2us GI
                                STBC Tx <= 80MHz
                                STBC Rx <= 80MHz
                                Full Bandwidth UL MU-MIMO
                                Partial Bandwidth UL MU-MIMO
                                DCM Max Constellation: 2
                                DCM Max Constellation Rx: 2
                                SU Beamformee
                                Beamformee STS <= 80Mhz: 3
                                Beamformee STS > 80Mhz: 3
                                Codebook Size SU Feedback
                                Codebook Size MU Feedback
                                Triggered CQI Feedback
                                Partial Bandwidth Extended Range
                                Partial Bandwidth DL MU-MIMO
                                PPE Threshold Present
[... 173 lines cut ...]
                * [ RRM ]: RRM
                * [ BEACON_RATE_LEGACY ]: legacy beacon rate setting
                * [ BEACON_RATE_HT ]: HT beacon rate setting
                * [ BEACON_RATE_VHT ]: VHT beacon rate setting
                * [ FILS_STA ]: STA FILS (Fast Initial Link Setup)
                * [ CQM_RSSI_LIST ]: multiple CQM_RSSI_THOLD records
                * [ CONTROL_PORT_OVER_NL80211 ]: control port over nl80211
                * [ ACK_SIGNAL_SUPPORT ]: ack signal level support
                * [ TXQS ]: FQ-CoDel-enabled intermediate TXQs
                * [ SCAN_RANDOM_SN ]: use random sequence numbers in scans
                * [ SCAN_MIN_PREQ_CONTENT ]: use probe request with only rate IEs in scans
                * [ CAN_REPLACE_PTK0 ]: can safely replace PTK 0 when rekeying
                * [ AIRTIME_FAIRNESS ]: airtime fairness scheduling
                * [ STA_TX_PWR ]: TX power control per station
                * [ AQL ]: Airtime Queue Limits (AQL)
                * [ CONTROL_PORT_NO_PREAUTH ]: disable pre-auth over nl80211 control port support
                * [ DEL_IBSS_STA ]: deletion of IBSS station support
                * [ SCAN_FREQ_KHZ ]: scan on kHz frequency support
                * [ CONTROL_PORT_OVER_NL80211_TX_STATUS ]: tx status for nl80211 control port support
                * [ FILS_DISCOVERY ]: FILS discovery frame transmission support
                * [ UNSOL_BCAST_PROBE_RESP ]: unsolicated broadcast probe response transmission support
                * [ BEACON_RATE_HE ]: HE beacon rate support (AP/mesh)
                * [ BSS_COLOR ]: BSS coloring support
                * [ RADAR_BACKGROUND ]: Radar background support
```

brada4, 2023-07-18:
> #{ AP, mesh point } <= 16

Should work fine.

Try to scan `iw dev wlan0 scan` and confirm it sees rest of the mesh.

csharper2005, 2023-07-18:
> Try to scan `iw dev wlan0 scan` and confirm it sees rest of the mesh.

ch13/40MHz local, ch13/40MHz remote:
remote point invisible for `iw dev phy0-mesh0 scan`
local mesh and remote mesh has `Could not select hw_mode and channel. (-3)` error

ch13/40MHz local, ch13/20MHz remote:
BSS e0:xx:xx:xx:xx:xx(on phy0-ap0)
        last seen: 173360.397s [boottime]
        TSF: 9625932 usec (0d, 00:00:09)
        freq: 2472
        beacon interval: 100 TUs
        capability: (0x0010)
        signal: -50.00 dBm
        last seen: 80 ms ago
        SSID:
        RSN:     * Version: 1
                 * Group cipher: CCMP
                 * Pairwise ciphers: CCMP
                 * Authentication suites: SAE
                 * Capabilities: 1-PTKSA-RC 1-GTKSA-RC (0x0000)
        HT capabilities:
                Capabilities: 0x1ff
                        RX LDPC
                        HT20/HT40
                        SM Power Save disabled
                        RX Greenfield
                        RX HT20 SGI
                        RX HT40 SGI
                        TX STBC
                        RX STBC 1-stream
                        Max AMSDU length: 3839 bytes
                        No DSSS/CCK HT40
                Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
                Minimum RX AMPDU time spacing: No restriction (0x00)
                HT TX/RX MCS rate indexes supported: 0-15
        HT operation:
                 * primary channel: 13
                 * secondary channel offset: no secondary
                 * STA channel width: 20 MHz
        MESH ID: xxxx
local mesh has `Could not select hw_mode and channel. (-3)` error. remote mesh ok

ch13/40MHz local, ch9/40MHz remote:
`iw dev phy0-mesh0 scan`
```
BSS e0:xx:xx:xx:xx:xx(on phy0-ap0)
        last seen: 172947.865s [boottime]
        TSF: 16998734 usec (0d, 00:00:16)
        freq: 2452
        beacon interval: 100 TUs
        capability: (0x0010)
        signal: -56.00 dBm
        last seen: 650 ms ago
        SSID:
        RSN:     * Version: 1
                 * Group cipher: CCMP
                 * Pairwise ciphers: CCMP
                 * Authentication suites: SAE
                 * Capabilities: 1-PTKSA-RC 1-GTKSA-RC (0x0000)
        HT capabilities:
                Capabilities: 0x1ff
                        RX LDPC
                        HT20/HT40
                        SM Power Save disabled
                        RX Greenfield
                        RX HT20 SGI
                        RX HT40 SGI
                        TX STBC
                        RX STBC 1-stream
                        Max AMSDU length: 3839 bytes
                        No DSSS/CCK HT40
                Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
                Minimum RX AMPDU time spacing: No restriction (0x00)
                HT TX/RX MCS rate indexes supported: 0-15
        HT operation:
                 * primary channel: 9
                 * secondary channel offset: no secondary
                 * STA channel width: 20 MHz
        MESH ID: xxxx
```

csharper2005, 2023-07-18:
ch9/40MHz - ch9/40MHz - mesh works "fine" (using 20 MHz width, see https://github.com/openwrt/openwrt/issues/13112)
ch13/20MHz - ch13/20MHz - also works

brada4, 2023-07-18:
Ahh ohh, by default other channel is above base channel. Go with 20MHz, your neighbours and wifi standards will appreciate.

brada4, 2023-07-18:
https://w1.fi/cgit/hostap/tree/hostapd/hostapd.conf
Default cap is ht40- ,  go figure, it can swap up to down etc

csharper2005, 2023-07-19:
> Ahh ohh, by default other channel is above base channel. Go with 20MHz, your neighbours and wifi standards will appreciate.

What exactly standard do you mean? Please keep in mind that:
1. AP ch13/40 MHz works fine everywhere
2. Mesh  ch13/40 MHz  works fine (22.03.5)
3. Mesh  ch13/40 MHz doesn't work (23.05.0-rc2)

Which exactly paragraph of which standard allows ch13/40 MHz for AP and disallows ch13/40 MHz for mesh?

brada4, 2023-07-19:
The paragraph that other mesh members run AP and secondary channel certainly overlaps.

> Always use 40MHz channels even if the secondary channel overlaps. Using this option does not comply with IEEE 802.11n-2009!

csharper2005, 2023-07-19:
> The paragraph that other mesh members run AP and secondary channel certainly overlaps.

Nevertheless, APs ch13/40 MHz works fine, meshes ch1/40 MHz and ch9/40 MHz work despite of overlapping. Hence, overlapping channels are not a reason.

brada4, 2023-07-19:
MESH CH9 HT40 ? i.e driver somehow flips ht40- default to ht40+ when in mesh mode?

csharper2005, 2023-07-19:
> MESH CH9 HT40 ? i.e driver somehow flips ht40- default to ht40+ when in mesh mode?

In fact - HE20/HT20. AP / Mesh start but HE40/H40 config is ignored:
```
	option htmode 'HE40'
	option noscan '1' 
```
```
phy1-ap0  ESSID: "xxxx"
          Access Point: E2:xx:xx:xx:xx:xx
          Mode: Master  Channel: 9 (2.452 GHz)  HT Mode: HT20
          Center Channel 1: 9 2: unknown
          Tx-Power: 20 dBm  Link Quality: unknown/70
          Signal: unknown  Noise: unknown
          Bit Rate: unknown
          Encryption: WPA3 SAE (CCMP)
          Type: nl80211  HW Mode(s): 802.11b/g/n
          Hardware: 14C3:7662 14C3:7662 [MediaTek MT76x2E]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy1

phy1-mesh0 ESSID: "xxxx"
          Access Point: E0:xx:xx:xx:xx:xx
          Mode: Mesh Point  Channel: 9 (2.452 GHz)  HT Mode: HT20
          Center Channel 1: 9 2: unknown
          Tx-Power: 20 dBm  Link Quality: 48/70
          Signal: -62 dBm  Noise: unknown
          Bit Rate: 13.7 MBit/s
          Encryption: WPA3 SAE (CCMP)
          Type: nl80211  HW Mode(s): 802.11b/g/n
          Hardware: 14C3:7662 14C3:7662 [MediaTek MT76x2E]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy1
```

> https://github.com/openwrt/openwrt/issues/13112 seems related.

Without doubt...

brada4, 2023-07-19:
If you disable HE (via caps lists) and keep only HT? Frankly never tried HT40 as in urban place it was always slower than HT20.

csharper2005, 2023-07-19:
@brada4 it doesn't matter, because we have the same result with TP-Link EC330-G5u v1, Netgear Nighthawk X4S R7800 (n-only devices, without HE capabilities). Only one device from this issue is AX device. 

P.S. I am also in urban place. Nevertheless, 40 MHz increases the throughput ~twice (mt7602, mediatek/filogic). At the same time, I noticed that the mt7603 does not work well at 40 MHz: 40 MHz slower than 20 MHz.

brada4, 2023-07-19:
Yes, i encourage you to disable AX to check if N establishes HT40-

csharper2005, 2023-07-20:
@brada4 nothing has changed.

raenye, 2023-07-27:
I have this problem on channels 10 and 11 too, not just 13.
HT20 ok, HT40 gets the -3 message.
AP is not affected, only mesh.

Channels 1 and 9 only come up as HT20, no matter what `htmode` I set (mesh is HT20 while AP on the same radio is HT40 at the same time).

Device is Xiaomi ax6s (`mt7622`/`mt7915`) whose 2.4 radio is N only. I'm using r23622 snapshot, not 23.05.

csharper2005, 2023-11-19:
Fixed in 23.05.2.

closed 2023-11-19 by csharper2005