openwrt/openwrt #13148: ath10k_pci: mac flush vdev 0 drop 0 queues 0x1 ar->paused: 0x0  arvif->paused: 0x0

brianjmurrell, 2023-07-22:
### Describe the bug

As [reported by multiple people in the forums](https://forum.openwrt.org/t/22-03-4-and-21-02-6-trigger-increased-system-kernel-log-events-on-archer-c7-v2-v4-v5-and-some-others/) since 22.03.4 the `ath10k_pci` driver is being more noisy in the log with:

```
[05/02/2023 15:45:12] ath10k_pci 0000:00:00.0: mac flush vdev 0 drop 0 queues 0x1 ar->paused: 0x0  arvif->paused: 0x0
[05/02/2023 15:45:13] ath10k_pci 0000:00:00.0: mac flush vdev 0 drop 0 queues 0x1 ar->paused: 0x0  arvif->paused: 0x0
```

Some folks have reported those messages go away when they use the non-`ct` firmware, but that's not my experience.  I get those messages with both the `ct` and non-`ct` firmwares.

### OpenWrt version

22.03.05

### OpenWrt target/subtarget

ath79/generic

### Device

TP-Link Archer C7 v2

### Image kind

Self-built image

### Steps to reproduce

Nothing specific.  Just connect wireless clients to an AP of the above type and wait for the messages to show up in the log.

### Actual behaviour

Get log noise:

```
[05/02/2023 15:45:12] ath10k_pci 0000:00:00.0: mac flush vdev 0 drop 0 queues 0x1 ar->paused: 0x0  arvif->paused: 0x0
[05/02/2023 15:45:13] ath10k_pci 0000:00:00.0: mac flush vdev 0 drop 0 queues 0x1 ar->paused: 0x0  arvif->paused: 0x0
```

### Expected behaviour

Shouldn't get that log noise.  Would like to understand what it's saying and why it's starting to emit that message.

### Diffconfig

```text
Mine is an ImageBuilder build, so likely not relevant.
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brianjmurrell, 2023-07-22:
I should add that the above message, in my APs is always preceded by a:

```
hostapd: wlan0: AP-STA-DISCONNECTED aa:ff:11:66:aa:ff
```

message.

mrkiko, 2023-07-22:
I think this started with commit openwrt git commit d54c91bd9ab3c54ee06923eafbd67047816a37e4 .

brianjmurrell, 2023-07-22:
FWIW, I have just noticed there is also a `kmod-ath10k-ct` which I had previously not replaced with `kmod-ath10k` when I replaced the `ct` firmware with the non-`ct` firmware.  I just did that and pushed a new firmware to my routers.  Will have to wait and see if that mitigates the messages.

brada4, 2023-07-22:
It is just noise from security fix that per-station pending packets were discarded after disconnect and will **not** be leaked to next connecting station. Obviously message will go away when everything is considered stable.
Currently non-ct firmware does not implement this flush and will report opposite.

brada4, 2023-07-23:
Message cannot be disabled via `echo 0x0 > /sys/module/ath10k_core/parameters/debug_mask` or respective boot parameter. That is the bug as there should be defined bit in mask to disable this type of message but there is none.

Dmole, 2024-07-05:
```
dmesg | grep -c ath10k_pci
204

uptime
 16:15:02 up 101 days, 21:11,  load average: 0.00, 0.00, 0.00
```

brada4, 2024-07-05:
@Dmole quite idle access point.

olivluca, 2024-09-26:
I see the same message with ath10k_ahb (on an aruba ap303, ipq40xx/generic, openwrt 23.05.2) , but it is harmless, right?

closed 2025-03-31 by openwrt-bot