openwrt/openwrt #13151: ramips/mt7621: cpu performance degredation

rany2, 2023-07-23:
### Describe the bug

A simple iperf3 test running on the router (`iperf3 -s`) would yield inferior results to 21.02.

21.02 would yield 890+ Mbit/s receiving data from the tester, whereas SNAPSHOT would provide the following results.

## Without fix on SNAPSHOT 58c498247b3d06c0ff1967d2f48ba84084a6cf25:

```
root@gw:~# iperf3 -c asus
Connecting to host asus, port 5201
[  5] local 10.146.109.1 port 42976 connected to 10.146.109.9 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  83.8 MBytes   702 Mbits/sec    0    465 KBytes       
[  5]   1.00-2.00   sec  77.8 MBytes   652 Mbits/sec    0    511 KBytes       
[  5]   2.00-3.00   sec  85.5 MBytes   717 Mbits/sec    0    461 KBytes       
[  5]   3.00-4.00   sec  84.4 MBytes   708 Mbits/sec    0    523 KBytes       
[  5]   4.00-5.00   sec  84.8 MBytes   711 Mbits/sec    0    473 KBytes       
[  5]   5.00-6.00   sec  85.6 MBytes   718 Mbits/sec    0    511 KBytes       
[  5]   6.00-7.00   sec  82.4 MBytes   691 Mbits/sec    0    461 KBytes       
[  5]   7.00-8.00   sec  85.0 MBytes   713 Mbits/sec    0    442 KBytes       
[  5]   8.00-9.00   sec  85.0 MBytes   713 Mbits/sec    0    492 KBytes       
[  5]   9.00-10.00  sec  85.0 MBytes   713 Mbits/sec    0    496 KBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   839 MBytes   704 Mbits/sec    0             sender
[  5]   0.00-10.00  sec   835 MBytes   700 Mbits/sec                  receiver

iperf Done.
root@gw:~# iperf3 -c asus -R
Connecting to host asus, port 5201
Reverse mode, remote host asus is sending
[  5] local 10.146.109.1 port 38088 connected to 10.146.109.9 port 5201
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.00   sec  81.9 MBytes   687 Mbits/sec                  
[  5]   1.00-2.00   sec  89.1 MBytes   747 Mbits/sec                  
[  5]   2.00-3.00   sec  86.6 MBytes   726 Mbits/sec                  
[  5]   3.00-4.00   sec  86.7 MBytes   728 Mbits/sec                  
[  5]   4.00-5.00   sec  85.1 MBytes   714 Mbits/sec                  
[  5]   5.00-6.00   sec  88.5 MBytes   742 Mbits/sec                  
[  5]   6.00-7.00   sec  87.6 MBytes   735 Mbits/sec                  
[  5]   7.00-8.00   sec  90.5 MBytes   759 Mbits/sec                  
[  5]   8.00-9.00   sec  87.4 MBytes   733 Mbits/sec                  
[  5]   9.00-10.00  sec  87.0 MBytes   730 Mbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.01  sec   871 MBytes   730 Mbits/sec    0             sender
[  5]   0.00-10.00  sec   870 MBytes   730 Mbits/sec                  receiver

iperf Done.
```

## With fix on 58c498247b3d06c0ff1967d2f48ba84084a6cf25:

```
root@gw:~# iperf3 -c asus
Connecting to host asus, port 5201
[  5] local 10.146.109.1 port 55014 connected to 10.146.109.9 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec   115 MBytes   967 Mbits/sec    0    507 KBytes       
[  5]   1.00-2.00   sec   110 MBytes   923 Mbits/sec    0    592 KBytes       
[  5]   2.00-3.00   sec   102 MBytes   860 Mbits/sec    0    550 KBytes       
[  5]   3.00-4.00   sec   112 MBytes   944 Mbits/sec    0    488 KBytes       
[  5]   4.00-5.00   sec   112 MBytes   944 Mbits/sec    0    565 KBytes       
[  5]   5.00-6.00   sec   112 MBytes   944 Mbits/sec    0    550 KBytes       
[  5]   6.00-7.00   sec   112 MBytes   944 Mbits/sec    0    607 KBytes       
[  5]   7.00-8.00   sec   111 MBytes   933 Mbits/sec    0    580 KBytes       
[  5]   8.00-9.00   sec   112 MBytes   944 Mbits/sec    0    577 KBytes       
[  5]   9.00-10.00  sec   112 MBytes   944 Mbits/sec    0   7.69 KBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec  1.09 GBytes   934 Mbits/sec    0             sender
[  5]   0.00-10.00  sec  1.09 GBytes   932 Mbits/sec                  receiver

iperf Done.

root@gw:~# iperf3 -c asus  -R
Connecting to host asus, port 5201
Reverse mode, remote host asus is sending
[  5] local 10.146.109.1 port 47130 connected to 10.146.109.9 port 5201
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.00   sec  99.7 MBytes   836 Mbits/sec                  
[  5]   1.00-2.00   sec   104 MBytes   870 Mbits/sec                  
[  5]   2.00-3.00   sec   102 MBytes   857 Mbits/sec                  
[  5]   3.00-4.00   sec   103 MBytes   862 Mbits/sec                  
[  5]   4.00-5.00   sec   102 MBytes   854 Mbits/sec                  
[  5]   5.00-6.00   sec   100 MBytes   840 Mbits/sec                  
[  5]   6.00-7.00   sec  98.5 MBytes   826 Mbits/sec                  
[  5]   7.00-8.00   sec  93.4 MBytes   784 Mbits/sec                  
[  5]   8.00-9.00   sec   105 MBytes   880 Mbits/sec                  
[  5]   9.00-10.00  sec   104 MBytes   874 Mbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.01  sec  1012 MBytes   848 Mbits/sec    0             sender
[  5]   0.00-10.00  sec  1011 MBytes   848 Mbits/sec                  receiver

iperf Done.
```

## The Fix:

You may ignore the changes related to removing default CMDLINE,  I just removed it as dtsi already specifies a kernel cmdline which will always override it: [0001-ramips-mt7621-fix-performance-degradation.patch.txt](https://github.com/openwrt/openwrt/files/12139094/0001-ramips-mt7621-fix-performance-degradation.patch.txt). 

Though some of the with some options unsure if they should be removed, though the performance improvement is undeniable. I haven't had enough time to keep making multiple builds to test which fix the performance degradation; but I think I didn't remove anything important/needed for mt7621 platform.

### OpenWrt version

58c498247b3d06c0ff1967d2f48ba84084a6cf25

### OpenWrt target/subtarget

ramips/mt7621

### Device

ASUS RT-AX53U

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-07-23:
Try `--bidir` and compare to 5GHz wifi (which does not bounce 1Gbps onboard MII which usually is a bootleneck even if you have seemingly unrelated network traffic)

rany2, 2023-07-23:
> Try `--bidir` and compare to 5GHz wifi (which does not bounce 1Gbps onboard MII which usually is a bootleneck even if you have seemingly unrelated network traffic)

The point isn't to measure 1Gbps networking performance using iperf3 on the mt7621 device, I just wanted to test CPU/memory performance using a tool like iperf3. I already attached a patch that has solved the issue for me and got away with the performance degradation along with a before/after applying the fix for the iperf3 test.

brada4, 2023-07-23:
Ohh sorry, asus is better known for other iperf3 compatible platforms. You can use localhost, local addresses on interface loop in software same way.

brada4, 2023-07-23:
What you say disabling highmem support speeds up the SoC that is unlikely to approach 4GB RAM?

rany2, 2023-07-23:
> What you say disabling highmem support speeds up the SoC that is unlikely to approach 4GB RAM?

I'm really unsure, I'll test it some more in a bit and maybe come up with revised changes. Though I have seen that disabling CONFIG_BOUNCE helps with performance for some weak ARM SOCs and it is related to HIGHMEM; don't ask me how or why though, I wouldn't know.

rany2, 2023-07-23:
## UPDATE:

- disabling `CONFIG_SCHED_SMT`: **BAD**
- disabling `CONFIG_BOUNCE`: **BAD**
- disabling `CONFIG_BOUNCE` and `CONFIG_HIGHMEM`: **GOOD**

Revised patch with only changes required to fix this issue: [0001-ramips-mt7621-fix-performance-degradation.patch.txt](https://github.com/openwrt/openwrt/files/12140002/0001-ramips-mt7621-fix-performance-degradation.patch.txt)

lukasz1992, 2023-07-24:
Wow, congratulations!

lukasz1992, 2023-07-24:
I am aware of mt7621 devices with 512MB RAM available (like Xiaomi Mi R3P)

Without highmem, we lose 64MB memory, which is not a huge value - imho worth for bringing back better performance.

brada4, 2023-07-24:
You can selectively add options back to particular devices.

closed 2023-07-24 by openwrt-bot

rany2, 2023-07-24:
@lukasz1992 Seems like that concern was resolved with https://github.com/lede-project/source/commit/cd2b74e01e8d5f5b80b82db1cb204c13ed99dd58#diff-0a37e591b3d1a868a65de269b43843c1c11fcb8f5ed2da797c3332a45a7fedbe !

nbd168, 2023-07-24:
I took another look at the code, and it seems that I will have to disable the final 64 MB after all because of the way memory is directly mapped on MIPS.

nbd168, 2023-07-24:
I think upstream would rather see highmem go away entirely eventually, because it is rather complex to deal with in a lot of places. And I guess a lot of upstream developers also don't care much about 32-bit systems anymore.