openwrt/openwrt #13152: nvme should be autoloaded

lu-zero, 2023-07-23:
### Describe the bug

I fixed the extroot support so nvme would work a bit better, but then the nvme module should be autoloaded in order to make so `block` can mount the partition as overlay.

### OpenWrt version

Any

### OpenWrt target/subtarget

any that has nvme support

### Device

any that has nvme support (e.g. bpi-r3)

### Image kind

Official downloaded image

### Steps to reproduce

build the current fstools master, configure extroot to use a nvme partition.

### Actual behaviour

the partition isn't mounted because the nvme module is loaded after.

### Expected behaviour

the nvme module is autoloaded and the partition in the nvme is used as overlay.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

robimarko, 2023-07-23:
NVME is autoloaded:
https://github.com/openwrt/openwrt/blob/main/package/kernel/linux/modules/block.mk#L508C22-L508C22

lu-zero, 2023-07-23:
It is not autoloaded `,1` is missing.

robimarko, 2023-07-23:
That 1 is so it gets loaded in `preinit`, not to autoload it in general.
This isn't really a bug, just an improvement, and easy one to make a PR for.

lu-zero, 2023-07-23:
I'll send over the PR then :)

lu-zero, 2023-08-07:
done in https://github.com/openwrt/openwrt/commit/0fe9a8ca94830270820f7a1eb457f5d5d587125d

closed 2023-08-07 by lu-zero