openwrt/openwrt #13153: [23.05-rc2] MT76 802.11s Mesh 20MHz BW

soxrok2212, 2023-07-24:
### Describe the bug

Since ~May, MT76 has had an issue where a single radio running (1) 802.11s node and (3) or more VAPS causes any STA's max receive bandwidth to be limited to 20MHz.

Remove the 802.11s node OR have 2 or less VAPs and 40 and 80MHz works fine. 

I do think this is related to the problems addressed in:
https://github.com/openwrt/mt76/commit/c50be0b54cddba864637bc0a5b13be2d89494103
https://github.com/openwrt/mt76/commit/d7d7479b00e9a5f4d14ebef6e147a5ce0406962d
https://github.com/openwrt/mt76/commit/1377f586c6f520a6d032053a2e9c0ebc8b277e9b
https://github.com/openwrt/mt76/commit/f704e4f83c6fd21fb39046fa328efb51bee6383b
... as the problem arose exactly at the same time.

For context, this is using wpad-mesh-openssl. The bandwidth across the actual 802.11s link is fine, working at 80MHz as expected. Ping @nbd168 for visibility. Not sure if this is specific to MT76 as I'm unable to test another device, but it happened at the same time as all the other regressions.

### OpenWrt version

r23288-476bf135fc

### OpenWrt target/subtarget

mediatek/filogic

### Device

TP-Link TL-XDR6086

### Image kind

Self-built image

### Steps to reproduce

1. On one single 5GHz radio with 80MHz of bandwidth, enable an 802.11s node (does not even need a peer) and 3 or more VAPs.
2. Observe any STAs on the VAPs are limited to 20MHz bandwidth from the AP with slow speeds (~170mbit/s on a 2x2 device).
3. Remove either one VAP or the 802.11s node.
4. Observe 80MHz bandwidth from the AP to the STA with expected speeds.

### Actual behaviour

STAs on a radio hosting (1) 802.11s node and 3 or more VAPs are limited to 20MHz bandwidth. The mesh node bandwidth is fine.

### Expected behaviour

Bandwidth on a VAP when on a radio with (1) 802.11s node and 3 or more VAPS is not limited to 20MHz.

### Diffconfig

```text
CONFIG_TARGET_mediatek=y
CONFIG_TARGET_mediatek_filogic=y
CONFIG_TARGET_mediatek_filogic_DEVICE_tplink_tl-xdr6086=y
CONFIG_BATMAN_ADV_BATMAN_V=y
CONFIG_BATMAN_ADV_BLA=y
CONFIG_BATMAN_ADV_DAT=y
CONFIG_BATMAN_ADV_MCAST=y
CONFIG_HTOP_LMSENSORS=y
CONFIG_LIBCURL_COOKIES=y
CONFIG_LIBCURL_FILE=y
CONFIG_LIBCURL_FTP=y
CONFIG_LIBCURL_HTTP=y
CONFIG_LIBCURL_MBEDTLS=y
CONFIG_LIBCURL_NGHTTP2=y
CONFIG_LIBCURL_NO_SMB="!"
CONFIG_LIBCURL_PROXY=y
CONFIG_LIBCURL_UNIX_SOCKETS=y
CONFIG_NFS_KERNEL_SERVER_V4=y
CONFIG_OPENSSL_ENGINE=y
CONFIG_OPENSSL_WITH_ASM=y
CONFIG_OPENSSL_WITH_CHACHA_POLY1305=y
CONFIG_OPENSSL_WITH_CMS=y
CONFIG_OPENSSL_WITH_DEPRECATED=y
CONFIG_OPENSSL_WITH_ERROR_MESSAGES=y
CONFIG_OPENSSL_WITH_IDEA=y
CONFIG_OPENSSL_WITH_MDC2=y
CONFIG_OPENSSL_WITH_PSK=y
CONFIG_OPENSSL_WITH_SEED=y
CONFIG_OPENSSL_WITH_SRP=y
CONFIG_OPENSSL_WITH_TLS13=y
CONFIG_OPENSSL_WITH_WHIRLPOOL=y
CONFIG_PACKAGE_adblock=y
CONFIG_PACKAGE_batctl-default=y
CONFIG_PACKAGE_block-mount=y
CONFIG_PACKAGE_ca-certificates=y
CONFIG_PACKAGE_cgi-io=y
CONFIG_PACKAGE_coreutils=y
CONFIG_PACKAGE_coreutils-sort=y
CONFIG_PACKAGE_curl=y
CONFIG_PACKAGE_ddns-scripts=y
CONFIG_PACKAGE_ddns-scripts-services=y
CONFIG_PACKAGE_ethtool-full=y
CONFIG_PACKAGE_hd-idle=y
CONFIG_PACKAGE_htop=y
CONFIG_PACKAGE_iperf3=y
CONFIG_PACKAGE_iptables-mod-ipopt=y
CONFIG_PACKAGE_iptables-nft=y
CONFIG_PACKAGE_kmod-asn1-decoder=y
CONFIG_PACKAGE_kmod-asn1-encoder=y
CONFIG_PACKAGE_kmod-batman-adv=y
CONFIG_PACKAGE_kmod-crypto-arc4=y
CONFIG_PACKAGE_kmod-crypto-cbc=y
CONFIG_PACKAGE_kmod-crypto-cts=y
CONFIG_PACKAGE_kmod-crypto-ecb=y
CONFIG_PACKAGE_kmod-crypto-kpp=y
CONFIG_PACKAGE_kmod-crypto-lib-chacha20=y
CONFIG_PACKAGE_kmod-crypto-lib-chacha20poly1305=y
CONFIG_PACKAGE_kmod-crypto-lib-curve25519=y
CONFIG_PACKAGE_kmod-crypto-lib-poly1305=y
CONFIG_PACKAGE_kmod-crypto-user=y
CONFIG_PACKAGE_kmod-dax=y
CONFIG_PACKAGE_kmod-dm=y
CONFIG_PACKAGE_kmod-dnsresolver=y
CONFIG_PACKAGE_kmod-fs-exportfs=y
CONFIG_PACKAGE_kmod-fs-nfs=y
CONFIG_PACKAGE_kmod-fs-nfs-common=y
CONFIG_PACKAGE_kmod-fs-nfs-common-rpcsec=y
CONFIG_PACKAGE_kmod-fs-nfs-v4=y
CONFIG_PACKAGE_kmod-fs-nfsd=y
CONFIG_PACKAGE_kmod-ifb=y
CONFIG_PACKAGE_kmod-ipt-core=y
CONFIG_PACKAGE_kmod-ipt-ipopt=y
CONFIG_PACKAGE_kmod-keys-encrypted=y
CONFIG_PACKAGE_kmod-keys-trusted=y
CONFIG_PACKAGE_kmod-lib-crc16=y
CONFIG_PACKAGE_kmod-nf-ipt=y
CONFIG_PACKAGE_kmod-nft-compat=y
CONFIG_PACKAGE_kmod-oid-registry=y
CONFIG_PACKAGE_kmod-random-core=y
CONFIG_PACKAGE_kmod-sched-cake=y
CONFIG_PACKAGE_kmod-sched-core=y
CONFIG_PACKAGE_kmod-scsi-core=y
CONFIG_PACKAGE_kmod-tpm=y
CONFIG_PACKAGE_kmod-udptunnel4=y
CONFIG_PACKAGE_kmod-udptunnel6=y
CONFIG_PACKAGE_kmod-usb-storage=y
CONFIG_PACKAGE_kmod-wireguard=y
CONFIG_PACKAGE_libblkid=y
CONFIG_PACKAGE_libcurl=y
CONFIG_PACKAGE_libdevmapper=y
CONFIG_PACKAGE_libiperf3=y
CONFIG_PACKAGE_libiptext=y
CONFIG_PACKAGE_libiptext-nft=y
CONFIG_PACKAGE_libiptext6=y
CONFIG_PACKAGE_libkeyutils=y
CONFIG_PACKAGE_liblua=y
CONFIG_PACKAGE_liblucihttp=y
CONFIG_PACKAGE_liblucihttp-lua=y
CONFIG_PACKAGE_liblucihttp-ucode=y
CONFIG_PACKAGE_libmount=y
CONFIG_PACKAGE_libncurses=y
CONFIG_PACKAGE_libnghttp2=y
CONFIG_PACKAGE_libopenssl=y
CONFIG_PACKAGE_libopenssl-conf=y
CONFIG_PACKAGE_libpcap=y
CONFIG_PACKAGE_libqrencode=y
CONFIG_PACKAGE_librt=y
CONFIG_PACKAGE_libsmartcols=y
CONFIG_PACKAGE_libtirpc=y
CONFIG_PACKAGE_libubus-lua=y
CONFIG_PACKAGE_libustream-mbedtls=m
CONFIG_PACKAGE_libustream-openssl=y
CONFIG_PACKAGE_libuuid=y
CONFIG_PACKAGE_libwolfssl=y
CONFIG_PACKAGE_libwrap=y
CONFIG_PACKAGE_libxtables=y
CONFIG_PACKAGE_lsblk=y
CONFIG_PACKAGE_lua=y
CONFIG_PACKAGE_luci-app-adblock=y
CONFIG_PACKAGE_luci-app-ddns=y
CONFIG_PACKAGE_luci-app-firewall=y
CONFIG_PACKAGE_luci-app-hd-idle=y
CONFIG_PACKAGE_luci-app-sqm=y
CONFIG_PACKAGE_luci-base=y
CONFIG_PACKAGE_luci-lib-base=y
CONFIG_PACKAGE_luci-lib-ip=y
CONFIG_PACKAGE_luci-lib-jsonc=y
CONFIG_PACKAGE_luci-lib-nixio=y
CONFIG_PACKAGE_luci-light=y
CONFIG_PACKAGE_luci-lua-runtime=y
CONFIG_PACKAGE_luci-mod-admin-full=y
CONFIG_PACKAGE_luci-mod-network=y
CONFIG_PACKAGE_luci-mod-status=y
CONFIG_PACKAGE_luci-mod-system=y
CONFIG_PACKAGE_luci-proto-batman-adv=y
CONFIG_PACKAGE_luci-proto-ipv6=y
CONFIG_PACKAGE_luci-proto-ppp=y
CONFIG_PACKAGE_luci-proto-wireguard=y
CONFIG_PACKAGE_luci-ssl-openssl=y
CONFIG_PACKAGE_luci-theme-bootstrap=y
CONFIG_PACKAGE_nfs-kernel-server=y
CONFIG_PACKAGE_nfs-kernel-server-utils=m
CONFIG_PACKAGE_nfs-utils-libs=y
CONFIG_PACKAGE_openssl-util=y
CONFIG_PACKAGE_qrencode=y
CONFIG_PACKAGE_rpcbind=y
CONFIG_PACKAGE_rpcd=y
CONFIG_PACKAGE_rpcd-mod-file=y
CONFIG_PACKAGE_rpcd-mod-iwinfo=y
CONFIG_PACKAGE_rpcd-mod-luci=y
CONFIG_PACKAGE_rpcd-mod-rrdns=y
CONFIG_PACKAGE_rpcd-mod-ucode=y
CONFIG_PACKAGE_sqm-scripts=y
CONFIG_PACKAGE_tc-tiny=y
CONFIG_PACKAGE_tcpdump=y
CONFIG_PACKAGE_terminfo=y
CONFIG_PACKAGE_trusted-firmware-a-mt7986-spim-nand-ddr4=y
CONFIG_PACKAGE_ucode-mod-html=y
CONFIG_PACKAGE_ucode-mod-lua=y
CONFIG_PACKAGE_ucode-mod-math=y
CONFIG_PACKAGE_uhttpd=y
CONFIG_PACKAGE_uhttpd-mod-ubus=y
CONFIG_PACKAGE_wireguard-tools=y
# CONFIG_PACKAGE_wpad-basic-mbedtls is not set
CONFIG_PACKAGE_wpad-mesh-openssl=y
CONFIG_PACKAGE_xtables-nft=y
CONFIG_RPCBIND_LIBWRAP=y
CONFIG_RPCBIND_RMTCALLS=y
CONFIG_WOLFSSL_HAS_NO_HW=y
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

marcust, 2023-08-13:
I see the same thing, assuming that a VAP is just having an access point configured on the same radio. I see it even when I configure a single AP on the radio and enable it, my phone instantly says that Download is like 200MBit, vs. the 1 point something K I see when I disable it. 

Hardware is Netgear WAX206 (on OpenWrt 23.05-SNAPSHOT r23353-9d10944de7) and a Mercusys MR90X (on 23.05-SNAPSHOT r23353-9d10944de7).

soxrok2212, 2023-08-14:
Glad to see its reproducible by others. The 3x VAPs vs 1x may be an artifact of MT7622 vs MT7986, but not sure.

soxrok2212, 2023-08-14:
Looks like latest snapshot seems to have fixed this bug. I need to check the hostapd version now to work around the unable to connect/wrong key issue.

marcust, 2023-08-15:
Just to be clear, that is a different version than I am running currently?

Because this morning I saw that I got full bandwidth at my main router but the reduced when connected to the WiFi of the meshed Netgear, I didn't have time to investigate though why. I power cycle them every night, so I was wondering if this has something to do with it and there is a timing issue in there. Mesh was working perfectly as usual.

soxrok2212, 2023-08-15:
The fix seems to be in latest OpenWrt "rolling" snapshot, not the 23.05 snapshot. I havent checked 23.05 snapshots yet.

marcust, 2023-08-16:
Do you have any indication as to which change fixed that?
What were you running before?

soxrok2212, 2023-08-16:
I don't as of yet. Prior, I was running a 23.05 snapshot from a month or so ago. I took a brief look through the MT76 commit but didn't see anything relevant. Perhaps it was a hostapd issue? There were a good amount of patches recently.

soxrok2212, 2023-08-29:
23.05 snapshots are still "broken". Building from main branch seems to work normally.

marcust, 2023-09-21:
It is fixed in the 23.05 Snapshots for me, I guess due to https://github.com/openwrt/openwrt/commit/9720b094aef89802327683f25824820581fed0b9 

Should be in 23.05-rc4 then.