openwrt/openwrt #13157: DHCP not working properly on the MediaTek MT7915E 5ghz on my Netgear Wax206

FinnVonB, 2023-07-25:
### Describe the bug

Running recent 23.05-rc2 and accessing my wireless networks via  the MediaTek MT7915E 5 GHz radio,  DHCP adresses is given out from only 1 DHCP server no matter which SSID connecting

Maybe it is getting the DHCP from the default-radio ?   

### OpenWrt version

r23228-cd17d8df2a

### OpenWrt target/subtarget

mediatek/mt7622 

### Device

Netgear WAX206

### Image kind

Official downloaded image

### Steps to reproduce

Router (running OpenWrt 22.03.5)  with 2 vlans (LAN -192.168.1.1  &  IOT 192.168.0.1 ) both enabled with DHCP server  -> Netgear Wax206 -> the MediaTekMT7915E Radio setup with -> SSID OpenWrt-LAN & SSID OpenWrt-IOT -> no discrepensies between the 2 setups,  only the attached network differs.
 
Connecting from my phone or any other device to the  OpenWrt-IOT SSID (5GHZ) I get and ip address from the LAN dhcp  not the IOT dhcp

(Disabling the OpenWrt-IOT SSID on (MediaTekMT7915E) I get proper ip addresses from the OpenWrt-IOT 2.4 GHz net (MediaTek MT7622 - so this is my workaround for now)) 

### Actual behaviour

I have 2 SSID's (OpenWrt-LAN and OpenWrt-IOT) on the 5GHz radio on my  Netgear  Wax206 that is setup as an AP. 

When connecting to OpenWrt-LAN I get an DHCP ip address from the LAN server - ok
Connecting to OpenWrt-IOT I also get an ip address from the LAN DHCP server even though my connecting device tells I'm on the OpenWrt-IOT net - connecting devices are phone and Ipads

I have, ofcourse, also a DHCP server running on the IOT and this is working fine and as expected on the 2.4 GHz Radio - on the 2.4 GHz OpenWrt-LAN  gets addresses from the LAN DHCP and OpenWrt-IOT gets ip addresses from the IOT DHCP as expected.

This was working prior to my update to 23.05-rc1 (I am now on 23.05-rc2)  

### Expected behaviour

Get an Ip address from the DHCP server of the net I am connecting to 

### Additional info

Previous to my update to 23.05-rc1 I was running a self-built image from around end February 2023 and in here the DHCP worked fine

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

danpawlik, 2023-07-25:
What is the "network" set for OpenWRT-IOT and could you share the wireless + network config?

FinnVonB, 2023-07-25:
uci show wireless |grep network :

wireless.default_radio0.network='lan'
wireless.default_radio1.network='lan'
wireless.wifinet2.network='IOT'
wireless.wifinet3.network='IOT'

WIRELESS:

config wifi-device 'radio0'
        option type 'mac80211'
        option path 'platform/18000000.wmac'
        option band '2g'
        option cell_density '0'
        option country 'DK'
        option channel '6'
        option htmode 'HT20'

config wifi-iface 'default_radio0'
        option device 'radio0'
        option mode 'ap'
        option encryption 'sae-mixed'
        option key 'X'
        option ssid 'OpenWrt-Lan'
        option network 'lan'

config wifi-device 'radio1'
        option type 'mac80211'
        option path '1a143000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0'
        option band '5g'
        option cell_density '0'
        option country 'DK'
        option channel 'auto'
        option htmode 'HE40'

config wifi-iface 'default_radio1'
        option device 'radio1'
        option mode 'ap'
        option encryption 'sae-mixed'
        option key 'X'
        option ssid 'OpenWrt-Lan'
        option network 'lan'

config wifi-iface 'wifinet2'
        option device 'radio1'
        option mode 'ap'
        option ssid 'OpenWrt-IOT'
        option network 'IOT'
        option key 'X'
        option encryption 'sae-mixed'
        option disabled '1'

config wifi-iface 'wifinet3'
        option device 'radio0'
        option mode 'ap'
        option ssid 'OpenWrt-IOT'
        option key 'X'
        option network 'IOT'
        option encryption 'psk-mixed'

NETWORK:

config interface 'loopback'
        option device 'lo'
        option proto 'static'
        option ipaddr '127.0.0.1'
        option netmask '255.0.0.0'

config globals 'globals'
        option ula_prefix 'fdd7:c1a7:2af6::/48'

config device
        option name 'br-lan'
        option type 'bridge'
        list ports 'lan1'
        list ports 'lan2'
        list ports 'lan3'
        list ports 'lan4'
        option ipv6 '0'

config interface 'lan'
        option device 'br-lan.1'
        option proto 'static'
        option ipaddr '192.168.1.3'
        option netmask '255.255.255.0'
        option gateway '192.168.1.1'
        option delegate '0'

config device
        option name 'eth0'
        option ipv6 '0'

config bridge-vlan
        option device 'br-lan'
        option vlan '1'
        list ports 'lan1:t*'
        list ports 'lan2:u*'
        list ports 'lan3'

config bridge-vlan
        option device 'br-lan'
        option vlan '10'
        list ports 'lan1:t'
        list ports 'lan4'

config interface 'IOT'
        option proto 'dhcp'
        option device 'br-lan.10'
        option delegate '0'
        option peerdns '0'

config device
        option name 'br-lan.1'
        option type '8021q'
        option ifname 'br-lan'
        option vid '1'
        option ipv6 '0'

config device
        option name 'br-lan.10'
        option type '8021q'
        option ifname 'br-lan'
        option vid '10'
        option ipv6 '0'

FinnVonB, 2023-07-25:
Checking the setup I think I may be the culprit - naming the 2.4GHz and the 5GHz with same SSID 
 a 2.4Ghz & 5GHz named OpenWrt-Lan
and 
a 2.4GHz & 5GHz named OpenWrt-IOT

FinnVonB, 2023-07-25:
Now I have renamed 2 SSIDs so I have 4 SSIDs
OpenWrt-Lan (5GHz) ->LAN 
OpenWrt-Lan2.4 (2.4)-> LAN 

OpenWrt-IOT (2.4GHz) -> IOT 
OpenWrt-IOT5.0 (5GHz) -> LAN

Only the SSID named OpenWrt-IOT is giving me an dhcp  ip address in the 192.168.0  (IOT) area
the 3 others is giving dhcp Ip adresses in the 192.168.1 (LAN) area.

Shine-, 2023-07-25:
From a quick look at your config, you're happily mixing VLANs (untagged, tagged, PVID). Also, you're mixing switch-filtered VLANs and software VLANs. More problems are present in your config, but unrelated to this.

The VLAN mixing likely causes the wrong DHCP server to receive the requests from your clients.

You might want to head over to the Wiki pages (e.g. "Converting to DSA" or "DSA Mini-Tutorial") in order to learn how to create a proper VLAN configuration, and to the forums for additional help in case you get lost.

FinnVonB, 2023-07-25:
Okay - point taken - I rest my case and take a review on my devices 
Thanks

closed 2023-07-25 by FinnVonB