openwrt/openwrt #13163: The wpad that came with OpenWrt 23.05.0-rc2 is unremovable.

stanley211, 2023-07-25:
### Describe the bug

While I attempted to remove the wpad that came with OpenWrt 23.05.0-rc2 so I can upgrade it, so I can connect to a WPA2 EAP network, it removes all Luci packages including luci.

### OpenWrt version

OpenWrt 23.05.0-rc2

### OpenWrt target/subtarget

ath79/nand

### Device

glinet_gl-xe300

### Image kind

Official downloaded image

### Steps to reproduce

Luci login, System tabs, Software, Update list, Installed, Filter searching wpad, remove wpad, Luci crashed

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

rany2, 2023-07-25:
Not a bug; Luci requires wpad to be installed and has a dependency on it. What you could instead do is use a tool like `luci-app-attendedsysupgrade` in advanced mode and alter the package list to remove the default wpad and replace it with a wpad with EAP support.

brada4, 2023-07-26:
It was always comfortable to uninstall wpad-small and install wpad-otherssl-mesh without tracking dependencies.

Shine-, 2023-07-26:
> Not a bug; Luci requires wpad to be installed and has a dependency on it. What you could instead do is use a tool like `luci-app-attendedsysupgrade` in advanced mode and alter the package list to remove the default wpad and replace it with a wpad with EAP support.

Not true. LuCI only has a dependency on libustream, so you can't easily replace the SSL library (e.g. mbedTLS to OpenSSL). But wpad can be removed without any dependency issue, and this is in fact the official way in case wpad-basic won't fit your needs (e.g. you require EAP).

Just tried with yesterday's 23.05 snapshot, using LuCI as well as using opkg. Removing wpad works fine. No crash, and LuCI is still accessible afterwards.

@stanley211 could it perhaps be that you were connected via WiFi? Removing wpad will of course break WiFi and disconnect your WiFi client...

stanley211, 2023-07-27:
Yes i was connected to a wifi, but it from the upcoming rc2 version though. it removes a lot of dependencies. This could catches someone by surprised. it camed with basic mesh version of wpad. original version of 23.05 has no problem.
A link to version I'm talking about is https://downloads.openwrt.org/releases/23.05.0-rc2/

Shine-, 2023-07-27:
> Yes i was connected to a wifi, but it from the upcoming rc2 version though. it removes a lot of dependencies.

No, it doesn't. As I already wrote above.

```
BusyBox v1.36.1 (2023-06-26 11:20:39 UTC) built-in shell (ash)

  _______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 OpenWrt 23.05.0-rc2, r23228-cd17d8df2a
 -----------------------------------------------------
=== WARNING! =====================================
There is no root password defined on this device!
Use the "passwd" command to set up a new password
in order to prevent unauthorized SSH logins.
--------------------------------------------------
root@OpenWrt:~# opkg info wpad-basic-mbedtls
Package: wpad-basic-mbedtls
Version: 2023-03-29-bb945b98-1.2
Depends: hostapd-common (= 2023-03-29-bb945b98-1.2), libc, libnl-tiny2023-04-02, hostapd-common, libubus20220615, libmbedtls12
Provides: hostapd, wpa-supplicant
Conflicts: hostapd, hostapd-basic, hostapd-basic-openssl, hostapd-basic-wolfssl, hostapd-basic-mbedtls, hostapd-mini, hostapd-openssl, hostapd-wolfssl, hostapd-mbedtls, wpad, wpad-mesh-openssl, wpad-mesh-wolfssl, wpad-mesh-mbedtls, wpad-basic, wpad-basic-openssl, wpad-basic-wolfssl, wpad, wpad-mesh-openssl, wpad-mesh-wolfssl, wpad-mesh-mbedtls, wpad-basic, wpad-basic-openssl, wpad-basic-wolfssl
Status: install user installed
Architecture: mips_24kc
Installed-Time: 1687778439

root@OpenWrt:~# opkg remove wpad-basic-mbedtls
Removing package wpad-basic-mbedtls from root...
root@OpenWrt:~#
```

stanley211, 2023-08-01:
The problem when i replaced it with the full version of wpad mesh, the newer version required a dependency conflict with a dependency. it refuses to install. But when i removed that dependency, it removes everything in the devices.

brada4, 2023-08-01:
Duplicate of https://github.com/openwrt/openwrt/issues/13082