openwrt/openwrt #13164: Many dnsmasq instances causes OOM kill

bingzhangdai, 2023-07-26:
### Describe the bug

I am running openwrt inside vm. On the host, the physical memory is enough
```
root@hypervisor:~# free -hm
               total        used        free      shared  buff/cache   available
Mem:            31Gi        18Gi       5.3Gi        46Mi       7.9Gi        12Gi
Swap:          8.0Gi          0B       8.0Gi
```
On the openwrt guest, the memory is also sufficient
```
root@OpenWrt:~# free -m
              total        used        free      shared  buff/cache   available
Mem:         500992       91164      369892        5388       39936      362596
Swap:             0           0           0
```
Normally, I can only see one dnsmasq instance
```
root@OpenWrt:~# ps aux | grep dnsmasq
dnsmasq  19483  0.2  5.2  27848 26528 ttyS0    S+   Jul25   1:46 /usr/sbin/dnsmasq -C /var/etc/dnsmasq.conf.cfg01411c -k -x /var/run/dnsmasq/dnsmasq.cfg01411c.pid
root     22754  0.0  0.1   1244   508 pts/0    S+   10:35   0:00 grep dnsmasq
```
However, I saw many oom kill logs in kernel log, one of them is pasted below
```log
Wed Jul 26 05:48:34 2023 kern.warn kernel: [32350.796564] dnsmasq invoked oom-killer: gfp_mask=0x14200ca(GFP_HIGHUSER_MOVABLE), nodemask=(null),  order=0, oom_score_adj=0
Wed Jul 26 05:48:34 2023 kern.warn kernel: [32350.799408] CPU: 2 PID: 20533 Comm: dnsmasq Not tainted 4.14.171 #0
Wed Jul 26 05:48:34 2023 kern.warn kernel: [32350.800252] Hardware name: QEMU Standard PC (i440FX + PIIX, 1996), BIOS rel-1.16.2-0-gea1b7a073390-prebuilt.qemu.org 04/01/2014
Wed Jul 26 05:48:34 2023 kern.warn kernel: [32350.801782] Call Trace:
Wed Jul 26 05:48:34 2023 kern.warn kernel: [32350.802346]  0xffffffff81706faa
Wed Jul 26 05:48:34 2023 kern.warn kernel: [32350.802909]  0xffffffff8110dc3b
Wed Jul 26 05:48:34 2023 kern.warn kernel: [32350.803492]  0xffffffff8110d091
Wed Jul 26 05:48:34 2023 kern.warn kernel: [32350.804059]  ? 0xffffffff8108c74b
Wed Jul 26 05:48:34 2023 kern.warn kernel: [32350.804638]  0xffffffff8110db1e
Wed Jul 26 05:48:34 2023 kern.warn kernel: [32350.805235]  0xffffffff81111bbb
Wed Jul 26 05:48:34 2023 kern.warn kernel: [32350.805774]  0xffffffff81111f2b
Wed Jul 26 05:48:34 2023 kern.warn kernel: [32350.806343]  0xffffffff8112d9bc
Wed Jul 26 05:48:34 2023 kern.warn kernel: [32350.806877]  0xffffffff8112fda9
Wed Jul 26 05:48:34 2023 kern.warn kernel: [32350.807471]  0xffffffff81131701
Wed Jul 26 05:48:34 2023 kern.warn kernel: [32350.807997]  0xffffffff81131807
Wed Jul 26 05:48:34 2023 kern.warn kernel: [32350.808565]  0xffffffff810521fc
Wed Jul 26 05:48:34 2023 kern.warn kernel: [32350.809099]  0xffffffff81052342
Wed Jul 26 05:48:34 2023 kern.warn kernel: [32350.809689]  ? 0xffffffff818015df
Wed Jul 26 05:48:34 2023 kern.warn kernel: [32350.810261]  0xffffffff8104be01
Wed Jul 26 05:48:34 2023 kern.warn kernel: [32350.810821]  0xffffffff818015f5
Wed Jul 26 05:48:34 2023 kern.warn kernel: [32350.811349] RIP: 0033:0x41ded9
Wed Jul 26 05:48:34 2023 kern.warn kernel: [32350.811901] RSP: 002b:00007ffe84eba8b0 EFLAGS: 00010202
Wed Jul 26 05:48:34 2023 kern.warn kernel: [32350.812618] RAX: 000000000240ff20 RBX: 000000000334e7e0 RCX: 00007ff5ec5658a8
Wed Jul 26 05:48:34 2023 kern.warn kernel: [32350.813491] RDX: 0000000000000034 RSI: 0000000000000031 RDI: 0000000000000000
Wed Jul 26 05:48:34 2023 kern.warn kernel: [32350.814393] RBP: 0000000000000031 R08: 0000000000000000 R09: 0000000000000000
Wed Jul 26 05:48:34 2023 kern.warn kernel: [32350.815293] R10: 0000000000000000 R11: 0000000000000246 R12: 0000000000000000
Wed Jul 26 05:48:34 2023 kern.warn kernel: [32350.816185] R13: 0000000064c04331 R14: 0000000000ffffff R15: 00007ff5ec57f480
Wed Jul 26 05:48:34 2023 kern.warn kernel: [32350.817206] Mem-Info:
Wed Jul 26 05:48:34 2023 kern.warn kernel: [32350.817688] active_anon:109190 inactive_anon:18 isolated_anon:0
Wed Jul 26 05:48:34 2023 kern.warn kernel: [32350.817688]  active_file:17 inactive_file:65 isolated_file:0
Wed Jul 26 05:48:34 2023 kern.warn kernel: [32350.817688]  unevictable:0 dirty:0 writeback:0 unstable:0
Wed Jul 26 05:48:34 2023 kern.warn kernel: [32350.817688]  slab_reclaimable:900 slab_unreclaimable:3277
Wed Jul 26 05:48:34 2023 kern.warn kernel: [32350.817688]  mapped:508 shmem:1455 pagetables:398 bounce:0
Wed Jul 26 05:48:34 2023 kern.warn kernel: [32350.817688]  free:4460 free_pcp:254 free_cma:0
Wed Jul 26 05:48:34 2023 kern.warn kernel: [32350.822205] Node 0 active_anon:436760kB inactive_anon:72kB active_file:68kB inactive_file:260kB unevictable:0kB isolated(anon):0kB isolated(file):0kB mapped:2032kB dirty:0kB writeback:0kB shmem:5820kB writeback_tmp:0kB unstable:0kB all_unreclaimable? yes
Wed Jul 26 05:48:34 2023 kern.warn kernel: [32350.825111] DMA free:2392kB min:520kB low:648kB high:776kB active_anon:13472kB inactive_anon:0kB active_file:0kB inactive_file:0kB unevictable:0kB writepending:0kB present:15992kB managed:15908kB mlocked:0kB kernel_stack:0kB pagetables:0kB bounce:0kB free_pcp:0kB local_pcp:0kB free_cma:0kB
Wed Jul 26 05:48:34 2023 kern.warn kernel: [32350.828502] lowmem_reserve[]: 0 473 473 473
Wed Jul 26 05:48:34 2023 kern.warn kernel: [32350.829278] DMA32 free:15448kB min:15860kB low:19824kB high:23788kB active_anon:422832kB inactive_anon:72kB active_file:0kB inactive_file:236kB unevictable:0kB writepending:0kB present:507752kB managed:485084kB mlocked:0kB kernel_stack:1644kB pagetables:1592kB bounce:0kB free_pcp:1016kB local_pcp:144kB free_cma:0kB
Wed Jul 26 05:48:34 2023 kern.warn kernel: [32350.833023] lowmem_reserve[]: 0 0 0 0
Wed Jul 26 05:48:34 2023 kern.warn kernel: [32350.833803] DMA: 2*4kB (M) 1*8kB (M) 1*16kB (M) 1*32kB (M) 0*64kB 0*128kB 1*256kB (M) 0*512kB 0*1024kB 1*2048kB (M) 0*4096kB = 2368kB
Wed Jul 26 05:48:34 2023 kern.warn kernel: [32350.835765] DMA32: 134*4kB (UME) 114*8kB (UE) 150*16kB (UE) 13*32kB (UE) 9*64kB (UE) 17*128kB (U) 13*256kB (UM) 5*512kB (UM) 2*1024kB (U) 0*2048kB 0*4096kB = 14952kB
Wed Jul 26 05:48:34 2023 kern.warn kernel: [32350.837983] 1539 total pagecache pages
Wed Jul 26 05:48:34 2023 kern.warn kernel: [32350.838782] 0 pages in swap cache
Wed Jul 26 05:48:34 2023 kern.warn kernel: [32350.839479] Swap cache stats: add 0, delete 0, find 0/0
Wed Jul 26 05:48:34 2023 kern.warn kernel: [32350.840437] Free swap  = 0kB
Wed Jul 26 05:48:34 2023 kern.warn kernel: [32350.841141] Total swap = 0kB
[... 50 lines cut ...]
I am wondering if the OOM kill actually happens? If yes, why? and also, why is the process 19483 still alive? I cannot explain the log

### OpenWrt version

r10947-65030d81f3

### OpenWrt target/subtarget

x86/64

### Device

QEMU Standard PC (i440FX + PIIX, 1996)

### Image kind

Official downloaded image

### Expected behaviour

No OOM kill in log

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-07-26:
> Wed Jul 26 05:48:34 2023 kern.warn kernel: [32350.817688] active_anon:109190 inactive_anon:18 isolated_anon:0

109190x4k=450MB

The memory usage of this single dnsmasq instance  is genuine cause of OOM.

Two things to diagnose memory leak further:
1/ install zram-swap, it will slow down process allocating for another 50% of RAM, likely helping to understand it.
2/ what is in dnsmasq configuration(s) in /etc/config/dhcp ?

bingzhangdai, 2023-07-26:
> 109190x4k=450MB
>
> The memory usage of this single dnsmasq instance is genuine cause of OOM.
>
> Two things to diagnose memory leak further: 1/ install zram-swap, it will slow down process allocating for [...]

Thanks. I will look at the zram doc and see if it provides some insight. `/etc/config/dhcp` is attached
```
config dnsmasq
        option domainneeded '1'
        option localise_queries '1'
        option rebind_protection '1'
        option rebind_localhost '1'
        option authoritative '1'
        option readethers '1'
        option leasefile '/tmp/dhcp.leases'
        option localservice '1'
        option dnsforwardmax '1000'
        option noresolv '1'
        option local '/moredaisy.local/'
        option domain 'moredaisy.local'
        option confdir '/etc/dnsmasq.d'
        list server '127.0.0.1#5353'
        option cachesize '5000'
        option filterwin2k '1'
        option quietdhcp '1'
```
files under `/etc/dnsmasq.d` are just some simple rules, although they are a lot.
```
root@OpenWrt:~# ls -al /etc/dnsmasq.d
drwxr-xr-x    2 root     root          4096 Jul 22 17:57 .
drwxr-xr-x   25 root     root          4096 Jul 25 22:58 ..
-rw-r--r--    1 root     root       4764017 Jul 19 22:20 accelerated-domains.china.conf
-rw-r--r--    1 dnsmasq  root             0 Jul 26 12:17 adb_list.overall
-rw-r--r--    1 root     root           979 Jul 19 22:19 custom_bypass.conf
-rw-r--r--    1 root     root           181 Jul 22 17:57 custom_domain.conf
-rw-r--r--    1 root     root        444933 Jul 19 22:18 dnsmasq_gfwlist.conf
root@OpenWrt:~# head /etc/dnsmasq.d/custom_bypass.conf
# bypass all *.cn domains
server=/.cn/114.114.114.114
ipset=/.cn/ss_spec_dst_bp

# server=/.com/114.114.114.114
# ipset=/.com/ss_spec_dst_bp

server=/.azure.com/114.114.114.114
ipset=/.azure.com/ss_spec_dst_bp

server=/.azure.net/114.114.114.114
ipset=/.azure.net/ss_spec_dst_bp
```

brada4, 2023-07-26:
How big is ipset at the point dnsmaasq crash

brada4, 2023-07-26:
I got about 50MB dnsmasq in place of 2MB loading [this](https://github.com/felixonmars/dnsmasq-china-list).
I do not routinely browse chinese sites, i suppose you are more into them. So the difference is ipset.

bingzhangdai, 2023-07-27:
> How big is ipset at the point dnsmaasq crash

Do you know how to get the size of ipset? After restarting, all things are refreshed.

I use the [this rule](https://github.com/shadowsocks/luci-app-shadowsocks/wiki/Instruction-of-ss-rules). You could directly look at the graph. It is very intuitive.
The two main rule files are attached. Hope this will help.
[accelerated-domains.china.txt](https://github.com/openwrt/openwrt/files/12178892/accelerated-domains.china.txt)
[dnsmasq_gfwlist.txt](https://github.com/openwrt/openwrt/files/12178890/dnsmasq_gfwlist.txt)
```
root@OpenWrt:/e/dnsmasq.d# ipset list ss_spec_dst_sp | head
Name: ss_spec_dst_sp
Type: hash:net
Revision: 6
Header: family inet hashsize 64 maxelem 65536
Size in memory: 1368
References: 4
Number of entries: 20
Members:
20.239.218.18
192.168.0.0/16
root@OpenWrt:/e/dnsmasq.d# ipset list ss_spec_src_bp | head
Name: ss_spec_src_bp
Type: hash:ip
Revision: 4
Header: family inet hashsize 64 maxelem 65536
Size in memory: 88
References: 2
Number of entries: 0
Members:
root@OpenWrt:/e/dnsmasq.d# ipset list ss_spec_src_fw | head
Name: ss_spec_src_fw
Type: hash:ip
Revision: 4
Header: family inet hashsize 64 maxelem 65536
Size in memory: 88
References: 2
Number of entries: 0
Members:
root@OpenWrt:/e/dnsmasq.d# ipset list ss_spec_src_ac | head
Name: ss_spec_src_ac
Type: hash:ip
Revision: 4
Header: family inet hashsize 64 maxelem 65536
Size in memory: 88
References: 2
Number of entries: 0
Members:
root@OpenWrt:/e/dnsmasq.d# ipset list ss_spec_dst_fw | head
Name: ss_spec_dst_fw
Type: hash:net
Revision: 6
Header: family inet hashsize 64 maxelem 65536
Size in memory: 8152
References: 3
Number of entries: 379
Members:
142.250.199.78
142.250.66.150
root@OpenWrt:/e/dnsmasq.d# ipset list ss_spec_dst_bp | head
Name: ss_spec_dst_bp
Type: hash:net
Revision: 6
Header: family inet hashsize 4096 maxelem 65536
Size in memory: 319320
References: 3
Number of entries: 13759
Members:
103.37.16.0/22
111.192.0.0/12
```
I do not think the ipset will go very large, as the rule can already cover more than 90% cases. Even if it increases, I have more than 300MB free memory.

One possibility is that there are some misconfig on my iptables and ipset, and exceesive ips are added to ipset. But I do not know how to debug that.

bingzhangdai, 2023-07-27:
Here is the memory usage right now.
```
CPU%   MEM%  VIRT  RES     PID USER          TIME+ THR  NI S  R/s W/s  Command
1.7    6.1   36.4M 29.8M 30103 root           0:02 1     0 R    ? ?    /usr/bin/python3 /usr/bin/glances
0.0    5.3   27.2M 25.8M 23910 dnsmasq        4:37 1     0 S    ? ?    /usr/sbin/dnsmasq -C /var/etc/dnsmasq.conf.cfg01411c -k -x /var/run/dnsmasq/dnsmasq.cfg01411c.pid
0.7    2.0   699M  9.99M 17416 root           9:11 7    -5 S    ? ?    ck-client
0.0    1.0   5.10M 4.92M 29849 root           0:00 1     0 S    ? ?    -bash
0.0    0.8   7.31M 3.98M 17414 root           3:56 1    -5 S    ? ?    ss-redir -c /var/etc/shadowsocks.cfg084a8f.json -u -l 2333 --mtu 1500 --fast-open --no-delay
0.0    0.5   4.59M 2.37M  2044 root           0:01 1     0 S    ? ?    /usr/sbin/tcpdump -nn -s0 -l -i eth0 port 53 -C1 -W5 -w /tmp/adb_report.pcap
0.0    0.2   1.79M 772K   1748 root           0:01 1     0 S    ? ?    /sbin/rpcd -s /var/run/ubus.sock -t 30
0.0    0.1   1.05M 612K   2246 root           0:00 1     0 S    ? ?    /usr/sbin/crond -f -c /etc/crontabs -l 5
0.0    0.1   1.14M 584K      1 root           0:01 1     0 S    ? ?    /sbin/procd
0.0    0.1   888K  568K   1842 root           0:00 1     0 S    ? ?    /usr/sbin/dropbear -F -P /var/run/dropbear.1.pid -p 22 -K 300 -T 3
```

bingzhangdai, 2023-08-07:
After some efforts capturing packtes and investigating, I found some clues to what happened,
1. dnsmasq will fork on TCP queries, and it will expand up to 20 instances (this number is hardcoded when compiling dnsmasq). Can verify by running `nc localhost 53` and `ps aux | grep dnsmasq` in another console
2. due to large ipsets, each dnsmasq instance will take around 26MB. 20 instances will definitely cause OOM issue.
3. it is not rare to have multiple TCP connections/queries at the same time under my current network condition
_why not rare? To avoid DNS poisoning, some domain resolution should go through a TCP tunnel to bypass the censorship. However, this tunnel is not very stable by nature. Sometimes, it will lose some packet (usually dropped intentionally by GFW) and TCP will resend after the timeout causing the dnsmasq cannot return very quickly. Hence the client will maintain a relatively long TCP connection with dnsmasq if the client uses TCP to query the domain_

Do you have some suggestions to improve this condition? @brada4 I can come up with some easy but not ideal solutions,
1. redirect TCP queries to another DNS server, let dnsmasq only handle UDP requests
2. add more memory up to 1GB, but I think it is to too sumptuous as there are less than 100MB used now

Thanks very much!

brada4, 2023-08-07:
You can run locally upstream for dnsmasq like unbound or bind, then 1MB dnsmasq can fork as much as it wants.
If you dont use dnsmasq for local dhcp then probably you can use either major DNS server right away.

brada4, 2023-08-07:
Better set unbound to handle dns and blacklist and use dnsmasq backend to serve .lan and 168.192.in-addr.arpa zones for unbound along ton of blacklists in unbound.
I.e

client(tcp/udp)-> unbound
unbound (tcp/udp) -> other dns servers subject to cong
unbound (local zones, udp) -> dnsmasq on localhost53xx

brada4, 2023-08-07:
Oom killer kills biggest process, all other ram users contribute towards invoking it.
dnsmasq does not scale past small home lan or local cache. Red unbound docs carefully - some memory pools are per instance, some per cpu thread