openwrt/openwrt #13179: Accidentally deleted the package libnl-tiny2023-04-02

KoRn0509, 2023-07-28:
### Describe the bug

Accidentally deleted the package libnl-tiny2023-04-02 now I can't log in via ssh, does not accept the password. Also the web interface says that Not Found Xiaomi AX6S 23.05.0-rc2 Is it possible to restore without resetting?

### OpenWrt version

23.05.0-rc2

### OpenWrt target/subtarget

23.05.0-rc2

### Device

Xiaomi AX6S

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-07-29:
You have to reset since "deletion: happened in overlay filesystem the files are intact in the initial root filesystem.

jow-, 2023-07-29:
Use the forums or mailing lists for support questions.

closed 2023-07-29 by jow-