openwrt/openwrt #13181: QCA9987/QCA9989 not detected on master

scr4tchy, 2023-07-28:
### Describe the bug

While on `v23.05.0-rc2` the QCA9889 antenna is recognized properly, on master it does not show up anywhere whatsoever, no errors to be seen except maybe for the one in `lspci` but unsure if relevant.

### OpenWrt version

r23635+6-03b3284805

### OpenWrt target/subtarget

qualcommax/ipq807x

### Device

Xiaomi AX9000

### Image kind

Self-built image

### Steps to reproduce

- Use a QCN 9889 equipped device
- Build master using the snapshot buildinfo 
- Witness the radio not showing up at all

- Reverting to `v23.05.0-rc2` brings back the radio

### Actual behaviour

The radio does not show up

### Expected behaviour

The radio works

### Additional info

```
root@Base:~# lspci -vv
00:00.0 PCI bridge: Qualcomm Device 1002 (prog-if 00 [Normal decode])
        Device tree node: /sys/firmware/devicetree/base/soc/pci@20000000/bridge@0,0
        Control: I/O+ Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR+ FastB2B- DisINTx+
        Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
        Latency: 0
        Interrupt: pin A routed to IRQ 16
        Region 0: Memory at 20220000 (32-bit, non-prefetchable) [size=4K]
        Bus: primary=00, secondary=01, subordinate=ff, sec-latency=0
        I/O behind bridge: 10000-10fff [size=4K] [16-bit]
        Memory behind bridge: 20400000-205fffff [size=2M] [32-bit]
        Prefetchable memory behind bridge: 20600000-207fffff [size=2M] [32-bit]
        Secondary status: 66MHz- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- <SERR- <PERR-
        BridgeCtl: Parity- SERR+ NoISA- VGA- VGA16- MAbort- >Reset- FastB2B-
                PriDiscTmr- SecDiscTmr- DiscTmrStat- DiscTmrSERREn-
        Capabilities: [40] Power Management version 3
                Flags: PMEClk- DSI- D1- D2- AuxCurrent=0mA PME(D0+,D1-,D2-,D3hot+,D3cold+)
                Status: D0 NoSoftRst+ PME-Enable- DSel=0 DScale=0 PME-
        Capabilities: [50] MSI: Enable+ Count=1/32 Maskable+ 64bit+
                Address: 000000000b00a040  Data: 01c0
                Masking: fffffffe  Pending: 00000000
        Capabilities: [70] Express (v2) Root Port (Slot+), MSI 00
                DevCap: MaxPayload 128 bytes, PhantFunc 0
                        ExtTag- RBE+
                DevCtl: CorrErr+ NonFatalErr+ FatalErr+ UnsupReq+
                        RlxdOrd+ ExtTag- PhantFunc- AuxPwr- NoSnoop+
                        MaxPayload 128 bytes, MaxReadReq 512 bytes
                DevSta: CorrErr- NonFatalErr- FatalErr- UnsupReq- AuxPwr+ TransPend-
                LnkCap: Port #0, Speed 8GT/s, Width x1, ASPM not supported
                        ClockPM- Surprise- LLActRep+ BwNot- ASPMOptComp+
                LnkCtl: ASPM Disabled; RCB 128 bytes, Disabled- CommClk-
[... 615 lines cut ...]
CONFIG_PACKAGE_tailscaled=y
CONFIG_PACKAGE_tc-tiny=m
CONFIG_PACKAGE_tcpdump=y
CONFIG_PACKAGE_terminfo=y
CONFIG_PACKAGE_ucode-mod-html=y
CONFIG_PACKAGE_ucode-mod-lua=y
CONFIG_PACKAGE_ucode-mod-math=y
CONFIG_PACKAGE_uhttpd=y
CONFIG_PACKAGE_uhttpd-mod-ubus=y
CONFIG_PACKAGE_unbound-daemon=y
# CONFIG_PACKAGE_wpad-basic-mbedtls is not set
CONFIG_PACKAGE_wpad-openssl=y
CONFIG_PACKAGE_xdp-filter=m
CONFIG_PACKAGE_xdp-loader=m
CONFIG_PACKAGE_xdpdump=m
CONFIG_PACKAGE_zlib=y
CONFIG_REPRODUCIBLE_DEBUG_INFO=y
CONFIG_SDK_LLVM_BPF=y
CONFIG_USE_LLVM_BUILD=y
```
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

robimarko, 2023-07-31:
Thanks for reporting this, in the future please make it clear in the title that this is board related.
It's not a generic issue but rather a PCI driver issue in the recent 6.1 point release due to the:
https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/commit/drivers/pci/controller/dwc/pcie-qcom.c?h=v6.1.42&id=db962c7a711c3393a80a18219960cd54fb33c53d

There is a pending fix upstream that fixes the issue:
https://patchwork.kernel.org/project/linux-arm-msm/patch/20230724063429.3980462-1-quic_srichara@quicinc.com/

Il probably make a PR to add the patch until it gets backported upstream to 6.1 stable.

robimarko, 2023-07-31:
Please give PR https://github.com/openwrt/openwrt/pull/13197 a go

scr4tchy, 2023-08-01:
Sorry for the delay here, my openwrt build environment was unhappy with perl after my latest nix update. Thanks a lot for that @robimarko, it works like a charm. And I had absolutely no idea where to either start troubleshooting or even understanding the issue given the complete lack of logs regarding any potential reason here..

closed 2023-08-01 by scr4tchy