openwrt/openwrt #13184: HW accelerated crypto module cannot be loaded (ccp-crypto)

graysky2, 2023-07-29:
### Describe the bug

`ccp-crypto.ko` provided by kmod-crypto-hw-ccp cannot be loaded

### OpenWrt version

r0+23647-271a54d38f

### OpenWrt target/subtarget

x86/64

### Device

HC Technology.,Ltd. HCAR5000-MI

### Image kind

Self-built image

### Steps to reproduce

Run `modprobe ccp-crypto`

### Actual behaviour

Module is not loaded:
```
# modprobe ccp-crypto
1 module could not be probed
- ccp-crypto
```

### Expected behaviour

Module should be loaded

### Additional info

The system is a Mini PC based on a Ryzen 7 5800U chip.  As a test, I booted into Arch Linux and the ccp-crypto module is loaded fine which rules out the a buggy BIOS on this system.   I confirmed the reported bug using the currently supported 5.15 series of kernels in master but also using the 6.1 series of kernels

I found [this thread](https://forum.openwrt.org/t/apu2-hardware-acceleration-crypto-module-not-loading/49463) but it did not contain a solution.

Booting into Arch Linux:
```
# dmesg | grep -i ccp
[    4.505785] ccp 0000:05:00.2: enabling device (0000 -> 0002)
[    4.519217] ccp 0000:05:00.2: ccp: unable to access the device: you might be running a broken BIOS.
[    4.529376] ccp 0000:05:00.2: tee enabled
[    4.529382] ccp 0000:05:00.2: psp enabled
```

Booting into OpenWrt:
```
# dmesg | grep -i ccp
[    4.098273] ccp 0000:04:00.2: enabling device (0000 -> 0002)
[    4.104660] ccp 0000:04:00.2: ccp: unable to access the device: you might be running a broken BIOS.
[    7.588811] ccp_crypto: Cannot load: there are no available CCPs
[    9.742383] ccp_crypto: Cannot load: there are no available CCPs
[   10.120408] kmodloader: - ccp-crypto - 0
```

### Diffconfig

```text
CONFIG_TARGET_x86=y
CONFIG_TARGET_x86_64=y
CONFIG_TARGET_x86_64_DEVICE_generic=y
CONFIG_DEVEL=y
CONFIG_BUSYBOX_CUSTOM=y
[... 320 lines cut ...]
CONFIG_PACKAGE_zsh=y
CONFIG_PACKAGE_zstd=y
CONFIG_PCRE2_JIT_ENABLED=y
CONFIG_PCRE_JIT_ENABLED=y
CONFIG_PERL_NOCOMMENT=y
CONFIG_PERL_THREADS=y
CONFIG_SAMBA4_SERVER_AVAHI=y
CONFIG_SAMBA4_SERVER_NETBIOS=y
CONFIG_SAMBA4_SERVER_VFS=y
CONFIG_SAMBA4_SERVER_WSDD2=y
CONFIG_TARGET_EXT4_JOURNAL=y
CONFIG_TARGET_EXT4_RESERVED_PCT=1
CONFIG_TARGET_KERNEL_PARTSIZE=32
CONFIG_TARGET_OPTIMIZATION="-march=znver3 -mtune=znver3 -O2 -pipe"
CONFIG_TARGET_OPTIONS=y
CONFIG_TARGET_ROOTFS_PARTSIZE=512
# CONFIG_TARGET_ROOTFS_SQUASHFS is not set
CONFIG_TESTING_KERNEL=y
CONFIG_ZSTD_OPTIMIZE_O3=y
```
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

tmn505, 2023-07-29:
This seems to be not a bug but rather a feature request. No one extended packaging this module for newer AMD systems. The ccp probably needs these devices (psp and tee) to be initialized before talking to them. Looking at Arch Linux config kmod-cpp-crypto will probably need this `CONFIG_CRYPTO_DEV_SP_PSP` enabled and also `CONFIG_AMDTEE` would probably need packaging.

graysky2, 2023-07-29:
Thank you for the insights into this, @tmn505.  I took a stab at enabling these two options and modifying the packaging.  I don't know what I am doing but made these changes: https://github.com/graysky2/openwrt/commits/x86_64-v61-WIP-3

When I booted into the image, I did indeed have `amdtee` loaded but I still got the errors loading the ccp-crypto.  Any thoughts you have are welcomed.

```
# dmesg | grep -i ccp
[    4.092859] ccp 0000:04:00.2: enabling device (0000 -> 0002)
[    4.099246] ccp 0000:04:00.2: ccp: unable to access the device: you might be running a broken BIOS.
[    4.122737] ccp 0000:04:00.2: tee enabled
[    4.127230] ccp 0000:04:00.2: psp enabled
[    7.435607] ccp_crypto: Cannot load: there are no available CCPs
[    9.363215] ccp_crypto: Cannot load: there are no available CCPs
[    9.590618] kmodloader: - ccp-crypto - 0
```

tmn505, 2023-07-29:
There might be some other kernel config symbols needed which I'm not aware of. What's the kernel version of Arch Linux You tested with? Maybe PCI ID is not recognized. Check Your PCI ID if it's in kernel source for 6.1.

graysky2, 2023-07-29:
I was using the linux-lts package from Arch which is currently 6.1.39.  I have 6.1.42 on OpenWrt so it is the best match.

Not sure about the PCI ID you mentioned.  I do see this from `lscpi -v`:
```
04:00.2 Encryption controller: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 10h-1fh) Platform Security Processor
        Subsystem: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 10h-1fh) Platform Security Processor
        Flags: bus master, fast devsel, latency 0, IRQ 59, IOMMU group 14
        Memory at fc900000 (32-bit, non-prefetchable) [size=1M]
        Memory at fcac2000 (32-bit, non-prefetchable) [size=8K]
        Capabilities: [48] Vendor Specific Information: Len=08 <?>
        Capabilities: [50] Power Management version 3
        Capabilities: [64] Express Endpoint, MSI 00
        Capabilities: [a0] MSI: Enable- Count=1/2 Maskable- 64bit+
        Capabilities: [c0] MSI-X: Enable+ Count=2 Masked-
        Capabilities: [100] Vendor Specific Information: ID=0001 Rev=1 Len=010 <?>
        Capabilities: [2a0] Access Control Services
        Kernel driver in use: ccp
```

tmn505, 2023-07-29:
I'm out of ideas, especially since I don't have such hardware. Best bet would be to dig around the time psp and amdtee were introduced, to figure out what is needed on kernel config side. Good Luck.

graysky2, 2023-07-29:
Thanks for your help.  Your suggestions did allow for the dmesg output to match Arch's a bit more (`tee enabled` and `psp enabled`).  Seems like that is progress.  Would be nice to get more verbose output from `modprobe` to see what might be wrong.

tmn505, 2023-07-29:
Post also `lsmod | sort` output of freshly booted Arch Linux, maybe that will give some clues.

graysky2, 2023-07-29:
Looks like:
```
ac97_bus               16384  1 snd_soc_core
aesni_intel           393216  0
amd_sfh                49152  0
amdgpu              10473472  0
bpf_preload            24576  0
ccp                   135168  1 kvm_amd
cec                    81920  1 drm_display_helper
cfg80211             1138688  0
cm32181                16384  0
crc16                  16384  1 ext4
crc32_pclmul           16384  0
crc32c_generic         16384  0
crc32c_intel           24576  3
crct10dif_pclmul       16384  1
cryptd                 24576  2 crypto_simd,ghash_clmulni_intel
crypto_simd            16384  1 aesni_intel
dm_mod                192512  0
drm_buddy              20480  1 amdgpu
drm_display_helper    184320  1 amdgpu
drm_ttm_helper         16384  1 amdgpu
edac_mce_amd           57344  0
ext4                 1032192  1
fuse                  176128  1
gf128mul               16384  1 polyval_generic
ghash_clmulni_intel    16384  0
gpu_sched              49152  1 amdgpu
hid_cmedia             16384  0
i2c_hid                40960  1 i2c_hid_acpi
i2c_hid_acpi           16384  0
i2c_piix4              36864  0
igc                   163840  0
industrialio          110592  1 cm32181
intel_rapl_common      32768  1 intel_rapl_msr
intel_rapl_msr         20480  0
ip6_tables             36864  53 ip6table_filter
ip6t_REJECT            16384  1
ip6t_rt                20480  3
ip6table_filter        16384  1
ip_tables              36864  9 iptable_filter
ipt_REJECT             16384  1
iptable_filter         16384  1
irqbypass              16384  1 kvm
jbd2                  192512  1 ext4
k10temp                16384  0
kvm                  1150976  1 kvm_amd
kvm_amd               176128  0
libcrc32c              16384  1 nf_conntrack
libphy                172032  3 r8169,mdio_devres,realtek
loop                   36864  0
mac_hid                16384  0
mbcache                16384  1 ext4
mc                     77824  1 snd_usb_audio
mdio_devres            16384  1 r8169
nf_conntrack          184320  1 xt_conntrack
nf_defrag_ipv4         16384  1 nf_conntrack
nf_defrag_ipv6         24576  1 nf_conntrack
nf_log_syslog          24576  10
nf_reject_ipv4         16384  1 ipt_REJECT
nf_reject_ipv6         20480  1 ip6t_REJECT
nvme                   61440  1
nvme_common            24576  1 nvme_core
nvme_core             208896  2 nvme
pcspkr                 16384  0
polyval_clmulni        16384  0
polyval_generic        16384  1 polyval_clmulni
r8169                  98304  0
rapl                   16384  0
realtek                36864  1
rfkill                 36864  2 cfg80211
sha512_ssse3           53248  0
snd                   131072  10 snd_seq_device,snd_hwdep,snd_usb_audio,snd_usbmidi_lib,snd_sof,snd_timer,snd_compress,snd_soc_core,snd_pcm,snd_rawmidi
snd_acp3x_pdm_dma      16384  1
snd_acp3x_rn           16384  0
snd_acp_config         16384  4 snd_rn_pci_acp3x,snd_sof_amd_rembrandt,snd_acp_pci,snd_sof_amd_renoir
snd_acp_pci            16384  0
snd_compress           28672  1 snd_soc_core
snd_hwdep              16384  1 snd_usb_audio
snd_intel_dspcfg       36864  1 snd_sof
snd_intel_sdw_acpi     20480  1 snd_intel_dspcfg
snd_pci_acp3x          20480  0
snd_pci_acp5x          20480  0
snd_pci_acp6x          20480  0
snd_pci_ps             20480  0
snd_pcm               172032  10 snd_sof_amd_acp,snd_pci_acp6x,snd_usb_audio,snd_sof,snd_compress,snd_soc_core,snd_sof_utils,snd_acp3x_pdm_dma,snd_pci_ps,snd_pcm_dmaengine
snd_pcm_dmaengine      16384  1 snd_soc_core
snd_rawmidi            49152  1 snd_usbmidi_lib
snd_rn_pci_acp3x       24576  0
snd_rpl_pci_acp6x      20480  0
snd_seq_device         16384  1 snd_rawmidi
snd_soc_acpi           16384  2 snd_sof_amd_acp,snd_acp_config
snd_soc_core          393216  4 snd_acp3x_rn,snd_sof,snd_soc_dmic,snd_acp3x_pdm_dma
snd_soc_dmic           16384  1
snd_sof               335872  2 snd_sof_amd_acp,snd_sof_pci
snd_sof_amd_acp        53248  2 snd_sof_amd_rembrandt,snd_sof_amd_renoir
snd_sof_amd_rembrandt    16384  0
snd_sof_amd_renoir     16384  0
snd_sof_pci            24576  2 snd_sof_amd_rembrandt,snd_sof_amd_renoir
snd_sof_utils          20480  1 snd_sof
snd_timer              49152  1 snd_pcm
snd_usb_audio         397312  0
snd_usbmidi_lib        45056  1 snd_usb_audio
soundcore              16384  1 snd
sp5100_tco             20480  0
ttm                    94208  2 amdgpu,drm_ttm_helper
usbhid                 73728  0
video                  65536  1 amdgpu
wmi                    45056  2 video,wmi_bmof
wmi_bmof               16384  0
x_tables               57344  13 ip6table_filter,xt_conntrack,iptable_filter,xt_LOG,xt_tcpudp,xt_addrtype,ip6t_rt,ip6_tables,ipt_REJECT,ip_tables,xt_limit,xt_hl,ip6t_REJECT
xhci_pci               24576  0
xhci_pci_renesas       24576  1 xhci_pci
xt_LOG                 20480  10
xt_addrtype            16384  4
xt_conntrack           16384  16
xt_hl                  16384  22
xt_limit               16384  13
xt_tcpudp              20480  22
```

tmn505, 2023-07-29:
I don't see ccp_crypto loaded. Post `lsmod | sort` when it's loaded.

graysky2, 2023-07-29:
That is odd.  I repeated it [here](https://gist.github.com/graysky2/f03427b84c32fd48650bbbc68dfcb05d).

graysky2, 2023-07-29:
@tmn505 - I am not convinced that it is working under Arch either... I was confused.  Booting into Arch and running modprobe gives the same without a module getting inserted.

```
# modprobe ccp-crypto
[ 21.073733] ccp_crypto: Cannot load: there are no available CCPs
[ 26.138567] ccp_crypto: Cannot load: there are no available CCPs
```

I think this is a bit of wild goose chase.

tlendacky, 2023-07-31:
Right, if the CCP can't be accessed or there are no queues available, then the ccp-crypto module can't do anything and so it doesn't load.

From the looks of it, the firmware is blocking access (the module issues the "ccp: unable to access the device: you might be running a broken BIOS." message when it reads from the MMIO space and gets all f's), so the CCP can't be used on that processor.

graysky2, 2023-07-31:
I cannot get the module to load on my Ryzen 9 x5950 nor on this Ryzen 7 5800U under Arch Linux.  Have not tried other distros.

tmn505, 2023-08-01:
@tlendacky Thanks for explanation. `ccp: unable to access the device: you might be running a broken BIOS.` bugged me, but I assumed it worked in Arch Linux. That makes it clear. So assuming vendor will release updated UEFI, which will unblock the access, it should work? Or could it be disabled, because some chips regions are faulty?

@graysky2 there's `I2C_DESIGNWARE_AMDPSP` which allows talking to PSP on some platforms using AMD Cezanne SoC, maybe it'll help, but assuming it's "some platform", I don't think it will.

graysky2, 2023-08-05:
Enabling that did not help:
```
# modprobe ccp-crypto
modprobe: ERROR: could not insert 'ccp_crypto': No such device
# dmesg
...
[  +4.373492] ccp_crypto: Cannot load: there are no available CCPs
```
# zgrep -i I2C_DESIGNWARE_AMDPSP /proc/config.gz
CONFIG_I2C_DESIGNWARE_AMDPSP=y
```

tmn505, 2023-08-09:
So I guess we can close this issue as not applicable, since it's hardware/BIOS limitation.

graysky2, 2023-08-09:
Yes!

closed 2023-08-09 by graysky2