openwrt/openwrt #13190: Kernel 6.1 package/kernel/bpf-headers failed to build

CharlesMengCA, 2023-07-30:
### Describe the bug

https://github.com/openwrt/openwrt/blob/main/package/kernel/bpf-headers/Makefile still hard coded as 5.15

```
PKG_NAME:=linux
PKG_PATCHVER:=5.15
```

When building with testing kernel 6.1, if the https://sources.openwrt.org/ doesn't have  the 5.15.123 kernel mirrored, the build will fail.

The curl was trying to use following weird URL to get the v5 kernel 

https://mirrors.mit.edu/kernel/linux/kernel/v6.x/longterm/v5.15.123/linux-5.15.123.tar.xz

### OpenWrt version

reboot-23653-gc4a5fe6cee

### OpenWrt target/subtarget

6.1

### Device

Linksys E8450

### Image kind

Self-built image

### Steps to reproduce

Just build with test kernel 6.1

### Actual behaviour

package/kernel/bpf-headers failed to build

### Expected behaviour

should get the header of kernel 6.1

### Diffconfig

```text
CONFIG_TARGET_mediatek=y
CONFIG_TARGET_mediatek_mt7622=y
CONFIG_TARGET_mediatek_mt7622_DEVICE_linksys_e8450-ubi=y
CONFIG_ALL_KMODS=y
CONFIG_ALL_NONSHARED=y
CONFIG_DEVEL=y
CONFIG_AUTOREMOVE=y
CONFIG_BPF_TOOLCHAIN_BUILD_LLVM=y
# CONFIG_BPF_TOOLCHAIN_NONE is not set
CONFIG_BUILDBOT=y
CONFIG_COLLECT_KERNEL_DEBUG=y
CONFIG_HAS_BPF_TOOLCHAIN=y
CONFIG_IB=y
CONFIG_IMAGEOPT=y
CONFIG_KERNEL_BUILD_DOMAIN="buildhost"
CONFIG_KERNEL_BUILD_USER="builder"
# CONFIG_KERNEL_KALLSYMS is not set
CONFIG_KERNEL_WERROR=y
CONFIG_MAKE_TOOLCHAIN=y
CONFIG_PACKAGE_attendedsysupgrade-common=y
CONFIG_PACKAGE_block-mount=y
CONFIG_PACKAGE_bridger=y
CONFIG_PACKAGE_cgi-io=y
CONFIG_PACKAGE_irqbalance=y
CONFIG_PACKAGE_kmod-asn1-decoder=y
CONFIG_PACKAGE_kmod-crypto-des=y
CONFIG_PACKAGE_kmod-crypto-ecb=y
CONFIG_PACKAGE_kmod-crypto-md4=y
CONFIG_PACKAGE_kmod-crypto-md5=y
CONFIG_PACKAGE_kmod-crypto-sha256=y
CONFIG_PACKAGE_kmod-fs-exfat=y
CONFIG_PACKAGE_kmod-fs-ksmbd=y
CONFIG_PACKAGE_kmod-fs-ntfs3=y
CONFIG_PACKAGE_kmod-fs-smbfs-common=y
CONFIG_PACKAGE_kmod-fs-vfat=y
CONFIG_PACKAGE_kmod-lib-crc-ccitt=m
CONFIG_PACKAGE_kmod-nls-cp437=y
CONFIG_PACKAGE_kmod-nls-iso8859-1=y
CONFIG_PACKAGE_kmod-nls-utf8=y
CONFIG_PACKAGE_kmod-oid-registry=y
# CONFIG_PACKAGE_kmod-ppp is not set
CONFIG_PACKAGE_kmod-sched-act-vlan=y
CONFIG_PACKAGE_kmod-sched-bpf=y
CONFIG_PACKAGE_kmod-sched-core=y
CONFIG_PACKAGE_kmod-sched-flower=y
CONFIG_PACKAGE_kmod-scsi-core=y
CONFIG_PACKAGE_kmod-slhc=m
CONFIG_PACKAGE_kmod-usb-storage=y
CONFIG_PACKAGE_ksmbd-server=y
CONFIG_PACKAGE_libbpf=y
CONFIG_PACKAGE_libelf=y
CONFIG_PACKAGE_libiwinfo-lua=y
CONFIG_PACKAGE_liblua=y
CONFIG_PACKAGE_liblucihttp=y
CONFIG_PACKAGE_liblucihttp-lua=y
CONFIG_PACKAGE_liblucihttp-ucode=y
CONFIG_PACKAGE_libnl-core=y
CONFIG_PACKAGE_libnl-genl=y
CONFIG_PACKAGE_libpcap=m
CONFIG_PACKAGE_libubus-lua=y
CONFIG_PACKAGE_libxdp=m
CONFIG_PACKAGE_lua=y
CONFIG_PACKAGE_luci=y
CONFIG_PACKAGE_luci-app-attendedsysupgrade=y
CONFIG_PACKAGE_luci-app-firewall=y
CONFIG_PACKAGE_luci-app-ksmbd=y
CONFIG_PACKAGE_luci-app-opkg=y
CONFIG_PACKAGE_luci-base=y
CONFIG_PACKAGE_luci-light=y
CONFIG_PACKAGE_luci-mod-admin-full=y
CONFIG_PACKAGE_luci-mod-network=y
CONFIG_PACKAGE_luci-mod-status=y
CONFIG_PACKAGE_luci-mod-system=y
CONFIG_PACKAGE_luci-proto-ipv6=y
CONFIG_PACKAGE_luci-proto-ppp=y
CONFIG_PACKAGE_luci-theme-bootstrap=y
# CONFIG_PACKAGE_ppp is not set
CONFIG_PACKAGE_qosify=m
CONFIG_PACKAGE_rpcd=y
CONFIG_PACKAGE_rpcd-mod-file=y
CONFIG_PACKAGE_rpcd-mod-iwinfo=y
CONFIG_PACKAGE_rpcd-mod-luci=y
CONFIG_PACKAGE_rpcd-mod-rpcsys=y
CONFIG_PACKAGE_rpcd-mod-rrdns=y
CONFIG_PACKAGE_rpcd-mod-ucode=y
CONFIG_PACKAGE_tc-tiny=m
CONFIG_PACKAGE_ucode-mod-html=y
CONFIG_PACKAGE_ucode-mod-math=y
CONFIG_PACKAGE_uhttpd=y
CONFIG_PACKAGE_uhttpd-mod-ubus=y
# CONFIG_PACKAGE_wpad-basic-mbedtls is not set
CONFIG_PACKAGE_wpad-mbedtls=y
CONFIG_PACKAGE_wsdd2=y
CONFIG_PACKAGE_xdp-filter=m
CONFIG_PACKAGE_xdp-loader=m
CONFIG_PACKAGE_xdpdump=m
CONFIG_PACKAGE_zlib=y
CONFIG_REPRODUCIBLE_DEBUG_INFO=y
CONFIG_SDK=y
CONFIG_SDK_LLVM_BPF=y
CONFIG_USE_LLVM_BUILD=y
CONFIG_TESTING_KERNEL=y
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

CharlesMengCA, 2023-07-30:
if PKG_PATCHVER:=5.15 changed to PKG_PATCHVER:=6.1

package/network/config/qosify failed to build.

brada4, 2023-07-31:
Incredible version number you made up here

hnyman, 2023-09-02:
cc @nbd168 

I stumbled into this today:
building for qualcommax/ipq807x with kernel 6.1, and as the bpf-headers Makefile specifies `PKG_PATCHVER:=5.15` while target kernel is already 6.1, it leads to an invalid download link that mixes 6.x and 5.15:
`https://cdn.kernel.org/pub/linux/kernel/v6.x/linux-5.15.129.tar.xz`

Log extract:
```
make[3]: Entering directory '/Openwrt/DL-WRX36/package/kernel/bpf-headers'
mkdir -p /Openwrt/DL-WRX36/dl
SHELL= flock /Openwrt/DL-WRX36/tmp/.linux-5.15.129.tar.xz.flock -c '    /Openwrt/DL-WRX36/scripts/download.pl "/Openwrt/DL-WRX36/dl" "linux-5.15.129.tar.xz" "750ec97ce4f1473e392b367a55eca4ea7a6b1e9e65ca2fb3bbca2eaa64802b66" "" "@KERNEL/linux/kernel/v6.x"    '
+ wget --tries=5 --timeout=20 --output-document=- https://cdn.kernel.org/pub/linux/kernel/v6.x/linux-5.15.129.tar.xz
--2023-09-02 06:17:49--  https://cdn.kernel.org/pub/linux/kernel/v6.x/linux-5.15.129.tar.xz
...
HTTP request sent, awaiting response... 404 Not Found
```

The reason is bpf-headers Makefile using `LINUX_SITE` as URL, and as `LINUX_SITE` gets defined to `.../v6.x/...` due to ipq807x having 6.1, the result is an invalid download URL

https://github.com/openwrt/openwrt/blob/02214ab8dce59ee6b599f8dfdacb0297dc5dcc24/package/kernel/bpf-headers/Makefile#L16-L22

https://github.com/openwrt/openwrt/blob/02214ab8dce59ee6b599f8dfdacb0297dc5dcc24/include/kernel.mk#L68

Not sure if there is any quick solution, but if bpf-headers remains locked to 5.15 (while most targets are already at 6.1), this error will get more frequent.

The error is mitigated by the OpenWrt download source cache: once the phase1 images buildbot has built the newly bumped 5.15.x minor kernel for some remaining 5.15 target, the needed tar file in the OpneWrt source archive, and the failing upstream download is hidden.  (Right now the kernel 5.15.x was bumped just a few hours ago, there has not been a 5.15.129 download, yet.)

commit a15eb1cd (2023-11-06): bpf-headers: Fix download URL with kernel 6.1

This fixes the download of the kernel 5.15 for the bpf-headers when
kernel 6.1 is build for the target.

Even if kernel 6.1 was selected for the target we still use kernel 5.15
for the bpf-headers. The download script tried to download the 5.15
kernel from the 6.x directory on kernel,org and this failed. Define
PKG_SOURCE_URL based on PKG_PATCHVER and not KERNEL_BASE like done in
kernel.mk.

Without this change it tries to download the kernel from this URL:
ttps://cdn.kernel.org/pub/linux/kernel/v6.x/linux-5.15.129.tar.xz

Fixes: #13190
Fixes: #13 [... cut]
PR #13884 (2023-11-06): bpf-headers: Fix download URL with kernel 6.1
This fixes the download of the kernel 5.15 for the bpf-headers when kernel 6.1 is build for the target.

Even if kernel 6.1 was selected for the target we still use kernel 5.15 for the bpf-headers. The download script tried to download the 5.15 kernel from the 6.x directory on kernel,org and this failed. Define PKG_SOURCE_URL based on PKG_PATCHVER and not KERNEL_BASE like done in kernel.mk.

Fixes: #13190
Fixes: #13671
Fixes: #13814
closed 2023-11-06 by openwrt-bot