openwrt/openwrt #13198: Bananapi BPI-R3 freezes randomly

hagier, 2023-07-31:
### Describe the bug

After random time between couple minutes and 24 hours, my router freezes completely. I can see that WiFi LEDs are not blinking anymore and I cannot connect to it using ethernet cable. I tried saving kernel logs between freezes and restarts but I can't see absolutely no errors at all. I tried installing stable releases and SNAPSHOTs but no luck at all. I even started controlling temperature from sensor but I cannot see anything abnormal. I don't know what more can I do to find a source of a problem. I've tried many configurations and solutions for two weeks now and looks like nothing's really changed. 

I decided to open a bug report because I'm not able to find a solution on my own.

System log between crashes (freeze happened around 14:30):
```
Mon Jul 31 12:15:16 2023 kern.info kernel: [   21.052057] IPv6: ADDRCONF(NETDEV_CHANGE): phy0-ap0: link becomes ready
Mon Jul 31 12:15:16 2023 kern.info kernel: [   21.058780] br-lanpriv: port 2(phy0-ap0) entered blocking state
Mon Jul 31 12:15:16 2023 kern.info kernel: [   21.064702] br-lanpriv: port 2(phy0-ap0) entered listening state
Mon Jul 31 12:15:16 2023 daemon.notice netifd: Network device 'phy0-ap0' link is up
Mon Jul 31 12:15:16 2023 kern.info kernel: [   21.162727] br-laniot: port 6(phy0-ap1) entered blocking state
Mon Jul 31 12:15:16 2023 kern.info kernel: [   21.168565] br-laniot: port 6(phy0-ap1) entered disabled state
Mon Jul 31 12:15:16 2023 kern.info kernel: [   21.174680] device phy0-ap1 entered promiscuous mode
Mon Jul 31 12:15:17 2023 daemon.notice netifd: Network device 'phy0-ap1' link is up
Mon Jul 31 12:15:17 2023 daemon.notice netifd: Network device 'phy0-ap1' link is down
Mon Jul 31 12:15:17 2023 kern.info kernel: [   21.272669] br-laniot: port 6(phy0-ap1) entered blocking state
Mon Jul 31 12:15:17 2023 kern.info kernel: [   21.278507] br-laniot: port 6(phy0-ap1) entered listening state
Mon Jul 31 12:15:17 2023 daemon.notice netifd: Network device 'phy0-ap1' link is up
Mon Jul 31 12:15:17 2023 kern.info kernel: [   21.431937] IPv6: ADDRCONF(NETDEV_CHANGE): phy0-ap1: link becomes ready
Mon Jul 31 12:15:17 2023 kern.info kernel: [   21.502160] br-languest: port 2(phy0-ap2) entered blocking state
Mon Jul 31 12:15:17 2023 kern.info kernel: [   21.508176] br-languest: port 2(phy0-ap2) entered disabled state
Mon Jul 31 12:15:17 2023 kern.info kernel: [   21.514434] device phy0-ap2 entered promiscuous mode
Mon Jul 31 14:44:09 2023 kern.info kernel: [   21.581588] br-languest: port 2(phy0-ap2) entered blocking state
Mon Jul 31 14:44:09 2023 kern.info kernel: [   21.587599] br-languest: port 2(phy0-ap2) entered listening state
Mon Jul 31 14:44:09 2023 daemon.notice netifd: Network device 'phy0-ap2' link is up
Mon Jul 31 14:44:09 2023 kern.info kernel: [   21.762001] IPv6: ADDRCONF(NETDEV_CHANGE): phy0-ap2: link becomes ready
Mon Jul 31 14:44:09 2023 kern.info kernel: [   21.911775] br-lanmgmt: port 2(phy0-ap3) entered blocking state
Mon Jul 31 14:44:09 2023 kern.info kernel: [   21.917705] br-lanmgmt: port 2(phy0-ap3) entered disabled state
Mon Jul 31 14:44:09 2023 kern.info kernel: [   21.923889] device phy0-ap3 entered promiscuous mode
Mon Jul 31 14:44:09 2023 kern.info kernel: [   21.931070] br-lanmgmt: port 2(phy0-ap3) entered blocking state
Mon Jul 31 14:44:09 2023 kern.info kernel: [   21.937002] br-lanmgmt: port 2(phy0-ap3) entered listening state
Mon Jul 31 14:44:09 2023 daemon.notice netifd: Network device 'phy0-ap3' link is up
```

Kernel log between crashes (last lines from previous log are recorded hours before freeze):
```
[   48.267590] br-vlan: topology change detected, sending tcn bpdu
[   52.310271] br-lanmgmt: port 1(br-vlan.99) entered learning state
[   52.316439] br-laniot: port 5(br-vlan.50) entered learning state
[   52.322531] br-languest: port 1(br-vlan.20) entered learning state
[   52.328797] br-lanpriv: port 1(br-vlan.10) entered learning state
[   60.630269] br-lanpriv: port 1(br-vlan.10) entered forwarding state
[   60.636542] br-lanpriv: topology change detected, propagating
[   60.642372] br-languest: port 1(br-vlan.20) entered forwarding state
[   60.648742] br-languest: topology change detected, propagating
[   60.654631] br-laniot: port 5(br-vlan.50) entered forwarding state
[   60.660815] br-laniot: topology change detected, propagating
[   60.666525] br-lanmgmt: port 1(br-vlan.99) entered forwarding state
[   60.672794] br-lanmgmt: topology change detected, propagating
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[    0.000000] Linux version 5.15.120 (builder@buildhost) (aarch64-openwrt-linux-musl-gcc (OpenWrt GCC 12.3.0 r23627-dcdcfc1511) 12.3.0, GNU ld (GNU Binutils) 2.40.0) #0 SMP Thu Jul 27 11:46:26 2023
[    0.000000] Machine model: Bananapi BPI-R3
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000040000000-0x00000000bfffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000040000000-0x0000000042ffffff]
[    0.000000]   node   0: [mem 0x0000000043000000-0x000000004302ffff]
[    0.000000]   node   0: [mem 0x0000000043030000-0x000000004fbfffff]
[    0.000000]   node   0: [mem 0x000000004fc00000-0x000000004ffbffff]
[    0.000000]   node   0: [mem 0x000000004ffc0000-0x00000000bfffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000040000000-0x00000000bfffffff]
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.1 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[... 432 lines cut ...]
SNAPSHOT r23643-ca8868a511

### OpenWrt target/subtarget

mediatek/filogic

### Device

Bananapi BPI-R3

### Image kind

Official downloaded image

### Actual behaviour

Freezes happen between ~1 minute to ~24 hours of uptime.

### Additional info

I have a setup with 4 routers. Main router (Banana BPI R3) provides 2.4 GHz WiFi connection for multiple LAN networks and WDS uplink for other routers on 5GHz. Other routers are configured as dumb APs connecting with WDS and providing 2.4GHz WiFi connections. Everything works flawlessly until random freezes. After restarting main router others are connecting without any problem.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

hagier, 2023-07-31:
I should also add that I removed cover from a board it to check if I can prolong time without freeze by providing better cooling but had no luck with that approach also.

gca31, 2023-07-31:
> I should also add that I removed cover from a board it to check if I can prolong time without freeze by providing better cooling but had no luck with that approach also.

When BPI-R3 seems frozen, are you able to read the logs through the "Debug UART" port ?

hagier, 2023-07-31:
I'll try it next time it will freeze. Is there anything specific I should try/check if that would succeed? 

Since it's our main home router and freezes are random I may only have one chance per day for doing it.

gca31, 2023-07-31:
I would try: "wifi down" then "wifi up"

hagier, 2023-08-01:
I managed to catch exception and whatever happens afterwards. Seems like it's trying to recover after crash and getting into crash loop.

```
[ 9213.800209] br-vlan: port 2(mesh0.sta1) entered listening state
[ 9221.986510] br-vlan: port 2(mesh0.sta1) entered learning state
[ 9230.287666] br-vlan: port 2(mesh0.sta1) entered forwarding state
[ 9230.293681] br-vlan: topology change detected, sending tcn bpdu
[34598.542238] br-vlan: port 2(mesh0.sta1) entered disabled state
[34598.548231] br-vlan: topology change detected, propagating
[34598.682136] device mesh0.sta1 left promiscuous mode
[34598.687023] br-vlan: port 2(mesh0.sta1) entered disabled state
[34719.192372] br-vlan: port 2(mesh0.sta1) entered blocking state
[34719.198216] br-vlan: port 2(mesh0.sta1) entered disabled state
[34719.204347] device mesh0.sta1 entered promiscuous mode
[34719.209636] br-vlan: port 2(mesh0.sta1) entered blocking state
[34719.215490] br-vlan: port 2(mesh0.sta1) entered listening state
[34727.363390] br-vlan: port 2(mesh0.sta1) entered learning state
[34735.665088] br-vlan: port 2(mesh0.sta1) entered forwarding state
[34735.671098] br-vlan: topology change detected, sending tcn bpdu
[39787.544550] Unable to handle kernel access to user memory outside uaccess routines at virtual address 0000000000000008
[39787.555257] Mem abort info:
[39787.558067]   ESR = 0x0000000096000005
[39787.561817]   EC = 0x25: DABT (current EL), IL = 32 bits
[39787.567148]   SET = 0, FnV = 0
[39787.570283]   EA = 0, S1PTW = 0
[39787.573442]   FSC = 0x05: level 1 translation fault
[39787.578381] Data abort info:
[39787.581320]   ISV = 0, ISS = 0x00000005
[39787.585195]   CM = 0, WnR = 0
[39787.588187] user pgtable: 4k pages, 39-bit VAs, pgdp=0000000046ce8000
[39787.594665] [0000000000000008] pgd=080000004347f003, p4d=080000004347f003, pud=080000004347f003, pmd=0000000000000000
[39787.605355] Internal error: Oops: 96000005 [#1] SMP
[39787.610227] Modules linked in: pppoe ppp_async nft_fib_inet nf_flow_table_ipv6 nf_flow_table_ipv4 nf_flow_table_inet pppox ppp_generic nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_objref nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_counter nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack_netlink nf_conntrack mt7915e mt76_connac_lib mt76 mac80211 iptable_mangle iptable_filter ipt_REJECT ipt_ECN ip_tables cfg80211 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_ecn xt_dscp xt_comment xt_TCPMSS xt_LOG xt_HL xt_DSCP xt_CLASSIFY x_tables slhc sfp sch_cake nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 mdio_i2c libcrc32c crc_ccitt compat cls_flower act_vlan crypto_safexcel cls_bpf act_bpf sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_route cls_matchall cls_fw cls_flow cls_basic
[39787.610381]  act_skbedit act_mirred act_gact pwm_fan i2c_gpio i2c_algo_bit ifb sha1_generic seqiv md5 des_generic libdes authencesn authenc leds_gpio xhci_plat_hcd xhci_pci xhci_mtk_hcd xhci_hcd gpio_button_hotplug usbcore usb_common
[39787.717889] CPU: 2 PID: 3629 Comm: reader#1 Not tainted 5.15.120 #0
[39787.724139] Hardware name: Bananapi BPI-R3 (DT)
[39787.728652] pstate: 60400005 (nZCv daif +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[39787.735594] pc : sta_set_sinfo+0xa1c/0xc40 [mac80211]
[39787.740670] lr : sta_set_sinfo+0x5bc/0xc40 [mac80211]
[39787.745719] sp : ffffffc00c9635e0
[39787.749017] x29: ffffffc00c9635e0 x28: 0000000000000001 x27: ffffffc00c963dd0
[39787.756134] x26: ffffff800007c880 x25: ffffff800355c900 x24: ffffff8005bda735
[39787.763250] x23: 00000000005ee0ea x22: 0000000000000000 x21: 0000058d1783bf82
[39787.770365] x20: ffffff8005bda000 x19: ffffffc00c963728 x18: 0000000000000000
[39787.777481] x17: 0000000000000000 x16: 0000000000000000 x15: 0000007fb381f420
[39787.784597] x14: 0000000000000000 x13: 0000000000000000 x12: 0000000000000000
[39787.791712] x11: 0000000000000002 x10: 000000000007443b x9 : 0000000000000000
[39787.798827] x8 : ffffffc00c963810 x7 : ffffff8003d00880 x6 : 000000000000003f
[39787.805942] x5 : ffffff8003d003a0 x4 : 0000000000000000 x3 : 0000000000000000
[39787.813057] x2 : ffffffc00c96376c x1 : 0000000000000001 x0 : 0000000000000000
[39787.820172] Call trace:
[39787.822604]  sta_set_sinfo+0xa1c/0xc40 [mac80211]
[39787.827323]  ieee80211_color_change_finish+0x1b64/0x1e80 [mac80211]
[39787.833589]  nl80211_put_sta_rate+0x1030/0x11b4 [cfg80211]
[39787.839071]  netlink_dump+0x104/0x360
[39787.842722]  __netlink_dump_start+0x158/0x2ac
[39787.847062]  genl_family_rcv_msg_dumpit+0x64/0xf4
[39787.851749]  genl_rcv_msg+0x188/0x1c0
[39787.855396]  netlink_rcv_skb+0x58/0x120
[39787.859217]  genl_rcv+0x34/0x50
[39787.862344]  netlink_unicast+0x1f0/0x2ec
[39787.866251]  netlink_sendmsg+0x19c/0x3d0
[39787.870157]  ____sys_sendmsg+0x27c/0x2b0
[39787.874066]  ___sys_sendmsg+0x80/0xf0
[39787.877713]  __sys_sendmsg+0x44/0xa0
[39787.881274]  __arm64_sys_sendmsg+0x20/0x30
[39787.885354]  invoke_syscall.constprop.0+0x4c/0xe0
[39787.890042]  do_el0_svc+0x40/0xd0
[39787.893341]  el0_svc+0x14/0x50
[39787.896382]  el0t_64_sync_handler+0xe0/0x110
[... 435 lines cut ...]
ERROR: can't get kernel image!
* spi-nand0
  - device: spi_nand@1
  - parent: spi@1100a000
  - driver: spi_nand
  - type: NAND flash
  - block size:        0x20000 bytes
  - page size:         0x800 bytes
  - OOB size:          64 bytes
[... cut ...]
  - 0x000000000000-0x000008000000 : "spi-nand0"
          - 0x000000000000-0x000000080000 : "bl2"
          - 0x000000080000-0x000000380000 : "factory"
          - 0x000000380000-0x000000580000 : "fip"
          - 0x000000580000-0x000008000000 : "ubi"
Using ethernet@15100000 device
TFTP from server 192.168.1.254; our IP address is 192.168.1.1
Filename 'openwrt-mediatek-filogic-bananapi_bpi-r3-initramfs-recovery.itb'.
Load address: 0x46000000
Loading: *
Abort
Wrong Image Format for bootm command
ERROR: can't get kernel image!
MT7986>
```

hagier, 2023-08-01:
I'm wondering about  "'spi-nand0' is now active device" part. I have installed OpenWRT to eMMC. I don't know if it would help with diagnostics but figured out that I should add this additional information.

Pins are configured this way now:
[image]

hayzamjs, 2023-08-02:
Is wireless offloading enabled?

hagier, 2023-08-02:
I'm not sure about "wireless offloading". Do you mean Routing/NAT Offloading? If yes, then I tried all possible options (disabled/software/hardware) and it didn't change anything. I also tried enabling and disabling packet steering but no luck with that either.

ne20002, 2023-08-02:
I faced freezes with my BPi-R3 myself after a fresh setup. Actually, I tried a few things and at the moment it runs without problems. What I did is:
* switched to emmc (had problems with NOR (too small) and NAND)
* disabled the crowdsec-firewall-bouncer as it caused high cpu when updating the ipsets (not sure if this caused freezing)
* removed the UPS USB connection and disabled nut (also, it seems my usb on the UPS broke so maybe it's a port issue)
I use sw flow offload and sometimes also enable hw flow offload (for testing, seems to work)
What is wireless ofloading?

ne20002, 2023-08-04:
I just got a freeze again after more than a week stable operation. Two thinks that might be connected:
* it happend when a friend left, she had four devices in my guests network (including one iphone, one itab)
* after restart collectd failed to start due to missing plugin / wrong option enabled (this is fixed now)

hagier, 2023-08-05:
I had some problems with newest SNAPSHOT (couldn't get WiFi to work) and got back to 23.05.0-rc2. Got same problem with freezes as before. Also noticed something similiar to @ne20002 that it often happens when my wife goes downstairs with her laptop and probably is changing APs during that transition. Almost always freezes happened when she left her laptop in the kitchen. 

In the meantime I got radiators for chipsets on a board. I wanted to check if that can be heat related. Before that I tried removing cover but it didn't help at all. After mounting radiators I noticed better temps but freezes still happened. I drilled some holes in the cover and mounted small fan (5V connected to router's USB port). So far we didn't experience any freezes. Temps seems pretty stable as below (last peak is from bufferbloat test). Before all changes I got something like 56-58 degrees. 

[image]

ne20002, 2023-08-19:
I got another freeze. Exactly when an iPhone left the wifi range. This is now a pattern. Freezing of the device happens when an iPhone in guest wifi (2.4 GHz) leaves the wifi range.
It does not happen when an Android device leaves 2.4 GHz guest wifi
It does not happen when an Android device leaves 5 GHz wifi
I now limit wifi for guests to 5Ghz just to see what will happen then.
@dangowrt, is this a known behaviour?

brainstorm, 2023-08-29:
Perhaps related with https://github.com/openwrt/openwrt/issues/12143?

hagier, 2023-08-29:
My trace mentions [mac80211] just like in this comment: https://github.com/openwrt/openwrt/issues/12143#issuecomment-1464886933

Since adding radiators and a fan I had no problems. But I can see some temp/load spikes when a new client access a network.

[image]

[image]

Also it looks like problems are caused by phy1 which is 5 GHz radio. So my "cooling solution" may not be a real solution.
I'm currently using 23.05.0-rc3.

hagier, 2023-09-13:
Freezes happened couple of times since my last post. No temperature spikes observed, also I have now climate control in a room with a router, so it may not be a temperature related problem after all. I'll try to catch more information if that happens again.

jcdutton, 2023-09-23:
Hi.
By looking at the kernel Oops mentioned above, this is obviously a bug in the Linux kernel source code.
I have looked at the source code and the bug would affect all Wifi access points, no matter which manufacturer they are from. I am surprised it has not been noticed by more people.

jcdutton, 2023-10-01:
@hagier 
I think the problem is that the memory / stack is getting corrupted.
If this freeze happens again, please post a new stack track, if it looks different.

dangowrt, 2023-10-02:
@LorenzoBianconi any idea what could be going on here?

lukasz1992, 2023-10-02:
It is not a problem of stack getting corrupted, crash log clearly shows null-pointer dereference (read memory at 0x8), and sta_set_sinfo is responsible for it

jcdutton, 2023-10-02:
@lukasz1992 
[39787.820172] Call trace:
[39787.822604]  sta_set_sinfo+0xa1c/0xc40 [mac80211]
[39787.827323]  ieee80211_color_change_finish+0x1b64/0x1e80 [mac80211]
If you look at the source code, there is no way that "ieee80211_color_change_finish" can call "sta_set_sinfo" so the stack trace is messed up, not displayed correctly.

@hagier 
The only way to make progress with this is to discover the line number within the "sta_set_sinfo" function where the crash happens.
In order to work that out, one needs the exact version of openwrt that was install, together with the "kernel-debug.tar.zst" that matches it:
I.e. from here: https://downloads.openwrt.org/snapshots/targets/mediatek/filogic/
Unfortunately, there is a history of kmods, but not a history of "kernel-debug.tar.zst".

So, I think the only way to diagnose this further, is to install the latest version of openwrt, and at the same time as that download the "kernel-debug.tar.zst"  to match. Then wait for the Oops to happen again.
This would all be a lot easier if openwrt kept the history of the debug versions.

lukasz1992, 2023-10-03:
@jcdutton yes, looks you are right.

And we have very similar tickets:
https://github.com/openwrt/openwrt/issues/12342
https://github.com/openwrt/openwrt/issues/12337 
with even worse calltrace (with recursive call?)

lukasz1992, 2023-10-03:
But @jcdutton another question is why ieee80211_color_change_finish is called?

There are no other callers than mt7915 and mt7996 mcu.c and there is no mt7996_mcu_cca_finish/mt7915_mcu_cca_finish call in call trace. Really interesting...

jcdutton, 2023-10-03:
@lukasz1992 
" ieee80211_color_change_finish" is not actually called. Don't believe the stack trace.
Only the first line of the stack trace is believable
Call trace:
[39787.822604]  sta_set_sinfo+0xa1c/0xc40 [mac80211]

But any further diagnosis of the problem is pretty much impossible because we don't have a matching "kernel-debug.tar.zst" for the kernel version in this bug report.

jcdutton, 2023-10-07:
The kernel Oops has this:
Code: d3441c63 12000c00 8b030ca3 f9409c63 (f9400465)
Extract of the code:
```
/usr/src/linux/net/mac80211/sta_info.c:2451
    d118:       17fffee6        b       ccb0 <sta_set_sinfo+0x6b0>
/usr/src/linux/net/mac80211/sta_info.c:2422
    d11c:       f9402107        ldr     x7, [x8, #64]
    d120:       d3441c42        ubfx    x2, x2, #4, #4
    d124:       12000c00        and     w0, w0, #0xf
    d128:       8b020ce2        add     x2, x7, x2, lsl #3
/usr/src/linux/net/mac80211/sta_info.c:2424
    d12c:       f9409c42        ldr     x2, [x2, #312]
    d130:       f9400447        ldr     x7, [x2, #8]                      <-----   It crashes here
    d134:       b40004c7        cbz     x7, d1cc <sta_set_sinfo+0xbcc>
/usr/src/linux/net/mac80211/sta_info.c:2427
    d138:       52800188        mov     w8, #0xc                        // #12
    d13c:       52800082        mov     w2, #0x4                        // #4
    d140:       9ba81c00        umaddl  x0, w0, w8, x7
    d144:       79400800        ldrh    w0, [x0, #4]
/usr/src/linux/net/mac80211/sta_info.c:2428
    d148:       f10004df        cmp     x6, #0x1
    d14c:       54000080        b.eq    d15c <sta_set_sinfo+0xb5c>  // b.none

```
```

/usr/src/linux/net/mac80211/sta_info.c:2424
       case STA_STATS_RATE_TYPE_LEGACY: {
                struct ieee80211_supported_band *sband;
                u16 brate;
                unsigned int shift;
                int band = STA_STATS_GET(LEGACY_BAND, rate);
                int rate_idx = STA_STATS_GET(LEGACY_IDX, rate);

                sband = local->hw.wiphy->bands[band];

                if (WARN_ON_ONCE(!sband->bitrates))      <------   It crashes here
                        break;

                brate = sband->bitrates[rate_idx].bitrate;
                if (rinfo->bw == RATE_INFO_BW_5)
                        shift = 2;
                else if (rinfo->bw == RATE_INFO_BW_10)
                        shift = 1;
                else
                        shift = 0;
                rinfo->legacy = DIV_ROUND_UP(brate, 1 << shift);
                break;
                }

```

hayzamjs, 2023-10-08:
@jcdutton is there any reliable way to reproduce this issue?

jcdutton, 2023-10-08:
@hayzamjs 
I have not seen the problem with the Oops above.  @hagier  has.
I am just a kernel developer trying to help out.
Based on the line of the code where the Oops happens, setting the wifi access point to LEGACY only might increase the likelihood of this Oops happening.
Alternatively, if you wish to reduce the likelihood of this happening, disable LEGACY on the wifi access point.
I have seen general instability in the BPI-R3, but no Oops yet.

lukasz1992, 2023-10-08:
@jcdutton  
Crash during checking for WARN.

I guess that band is invalid and malformed rate is pushed for calculating stats.

As a workaround I guess that
```if (WARN_ON_ONCE(!sband->bitrates))```
should be replaced with
```if (WARN_ON_ONCE(!sband || !sband->bitrates))```

but this needs to be done probably on some other places.

Solution would be to find where such malformed 'rate' value came from

lxp, 2023-10-08:
I think I have the same issue. While running OpenWrt 23.03-rc3, I had at least 16 crashes.
I am always seeing this oops:
```
[  732.730260] Unable to handle kernel access to user memory outside uaccess routines at virtual address 0000000000000008
[  732.740952] Mem abort info:
[  732.743730]   ESR = 0x0000000096000005
[  732.747462]   EC = 0x25: DABT (current EL), IL = 32 bits
[  732.752762]   SET = 0, FnV = 0
[  732.755801]   EA = 0, S1PTW = 0
[  732.758926]   FSC = 0x05: level 1 translation fault
[  732.763789] Data abort info:
[  732.766654]   ISV = 0, ISS = 0x00000005
[  732.770475]   CM = 0, WnR = 0
[  732.773428] user pgtable: 4k pages, 39-bit VAs, pgdp=0000000045b61000
[  732.779852] [0000000000000008] pgd=0000000000000000, p4d=0000000000000000, pud=0000000000000000
[  732.788531] Internal error: Oops: 96000005 [#1] SMP
[  732.793390] Modules linked in: pppoe ppp_async xt_state xt_conntrack xt_CT wireguard pppox ppp_generic nft_redir nft_nat nft_masq nft_flow_offload nft_fib_inet nft_ct nft_chain_nat nf_nat nf_flow_table_ipv6 nf_flow_table_ipv4 nf_flow_table_inet nf_flow_table nf_conntrack_netlink nf_conntrack mt7915e mt76_connac_lib mt76 mac80211 libchacha20poly1305 ipt_REJECT chacha_neon cfg80211 xt_time xt_tcpudp xt_multiport xt_mark xt_mac xt_limit xt_comment xt_TCPMSS xt_LOG usblp slhc sfp poly1305_neon nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_quota nft_objref nft_numgen nft_log nft_limit nft_hash nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_counter nf_tables nf_reject_ipv6 nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 mdio_i2c macvlan libcurve25519_generic libcrc32c libchacha iptable_mangle iptable_filter ip_tables crc_ccitt compat crypto_safexcel pwm_fan i2c_gpio i2c_algo_bit xt_set x_tables ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet
[  732.793525]  ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ipmac ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6_udp_tunnel udp_tunnel sha1_generic seqiv md5 des_generic libdes authencesn authenc leds_gpio xhci_plat_hcd xhci_pci xhci_mtk_hcd xhci_hcd gpio_button_hotplug usbcore usb_common
[  732.918461] CPU: 2 PID: 2039 Comm: hostapd Not tainted 5.15.127 #0
[  732.924623] Hardware name: Bananapi BPI-R3 (DT)
[  732.929135] pstate: 60400005 (nZCv daif +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[  732.936075] pc : 0xffffffc000b7653c [mac80211@00000000c83f9485+0x83000]
[  732.942677] lr : 0xffffffc000b760dc [mac80211@00000000c83f9485+0x83000]
[  732.949271] sp : ffffffc00c0fb7a0
[  732.952568] x29: ffffffc00c0fb7a0 x28: 0000000000000001 x27: ffffffc00c0fbdd0
[  732.959684] x26: ffffff800007c080 x25: ffffff8006ed4900 x24: ffffff8001114735
[  732.966799] x23: 0000000000137638 x22: ffffff8006ed4900 x21: 0000058d1783bf82
[  732.973914] x20: ffffff8001114000 x19: ffffff800592eb00 x18: 0000000000000000
[  732.981029] x17: 0000000000000c00 x16: ffffffc008e6d000 x15: 0000000000000600
[  732.988144] x14: ffffff8001114e00 x13: ffffff8001114e00 x12: 0000000000000000
[  732.995259] x11: 0000000000000000 x10: ffffff8001114e08 x9 : 0000000000000000
[  733.002374] x8 : ffffff800592ec00 x7 : ffffff80037c0880 x6 : 000000000000003f
[  733.009490] x5 : ffffff80037c03a0 x4 : 0000000000000000 x3 : 0000000000000000
[  733.016604] x2 : ffffff800592eb44 x1 : 0000000000000001 x0 : 0000000000000000
[  733.023720] Call trace:
[  733.026153]  0xffffffc000b7653c [mac80211@00000000c83f9485+0x83000]
[  733.032401]  0xffffffc000b76678 [mac80211@00000000c83f9485+0x83000]
[  733.038648]  0xffffffc000b769ac [mac80211@00000000c83f9485+0x83000]
[  733.044895]  0xffffffc000b8e3e8 [mac80211@00000000c83f9485+0x83000]
[  733.051143]  0xffffffc000b2b974 [cfg80211@00000000eac64e98+0x43000]
[  733.057390]  0xffffffc0086db7a4
[  733.060516]  0xffffffc0086dbc10
[  733.063641]  0xffffffc0086da918
[  733.066766]  0xffffffc0086db0f4
[  733.069891]  0xffffffc0086da0d4
[  733.073016]  0xffffffc0086da36c
[  733.076142]  0xffffffc00863d230
[  733.079267]  0xffffffc00863fa40
[  733.082391]  0xffffffc00863fb74
[  733.085516]  0xffffffc00863fbf0
[  733.088641]  0xffffffc00801f60c
[  733.091766]  0xffffffc00801f6e0
[  733.094891]  0xffffffc008830964
[  733.098016]  0xffffffc008831500
[  733.101140]  0xffffffc0080115cc
[  733.104268] Code: d3441c63 12000c00 8b030ca3 f9409c63 (f9400465) 
[  733.110340] ---[ end trace ef483988745b2cca ]---
[  733.119023] Kernel panic - not syncing: Oops: Fatal exception
```
I tried to analyze it further, but as the kernel does not have symbols and I couldn't find the correct base address of the mac80211 module yet (`00000000c83f9485` from the oops seems to be wrong, `/proc/modules` only shows `0x0000000000000000` even when playing around with `/proc/sys/kernel/kptr_restrict`).
However, I noticed the code is exactly the same as in the oops posted in this issue.

For OpenWrt 23.03-rc3 we still have the kernel-debug.tar.zst, so maybe this call trace still helps?
However, I think we would need to get the correct base address of mac80211 with this kernel. I think it is the same across reboots, because the PC address stays the same across all crashes I have seen.
I also tried to look for the `sta_set_sinfo` symbol in the mac80211 module from https://downloads.openwrt.org/releases/23.05.0-rc3/targets/mediatek/filogic/kernel-debug.tar.zst. It exists but the machine code at this location was o [... cut ...]
I usually see something like this a few seconds (in this case 2 seconds) before the kernel oops:

```
Sat Oct  7 08:53:00 2023 daemon.info hostapd: phy1-ap2: STA ab:cd:ef:ab:cd:ef IEEE 802.11: associated (aid 2)
Sat Oct  7 08:53:00 2023 daemon.notice hostapd: phy0-ap3: Prune association for ab:cd:ef:ab:cd:ef
Sat Oct  7 08:53:00 2023 daemon.notice hostapd: phy0-ap3: AP-STA-DISCONNECTED ab:cd:ef:ab:cd:ef
```

I upgraded to OpenWrt 23.03-rc4 yesterday and haven't seen a crash yet. The crashes usually happend between 3 times a day to once every 4 days.

jcdutton, 2023-10-08:
@lxp 
Just confirming you do have the same problem as the original post.
Your one:
Code: d3441c63 12000c00 8b030ca3 f9409c63 (f9400465)
Original post:
Code: d3441c63 12000c00 8b030ca3 f9409c63 (f9400465)

lxp, 2023-10-08:
@jcdutton Thanks for confirming.

Previously I had legacy_rates disabled. I have now enabled it:
```
wireless.radio0.legacy_rates='1'
```
Let's see, if crashes are now more/less frequent.

Is there anything else, I can do?

jcdutton, 2023-10-08:
I am testing with a patch. Its a bit noisy in the logs, so try to send the logs to another Linux PC using syslog TCP.
My theory at the moment is that somehow band gets above 5 and causes havoc. The patch catches that.
band is stored in such a way that it can have value 0-15, but only 0-5 are valid. If a value of 6-15 appears it will cause havoc.
If something else is happening, there are a few printk in there to try to catch them. 
Compile from source once first, then this file will appear.
openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/backports-6.5/net/mac80211/sta_info.c

Then apply the sta_info.c.diff.txt  patch to it

[sta_info.c.diff.txt](https://github.com/openwrt/openwrt/files/12841530/sta_info.c.diff.txt)

The sort of thing that appears in the log that is normal:
[ 1077.844032] JCD:STA_STATS_RATE_TYPE_LEGACY: rate=16384 0x4000, band=0, rate_idx=0
[ 1077.851524] JCD:STA_STATS_RATE_TYPE_LEGACY: ARRAY_SIZE (bands) 6, NUM_NL80211_BANDS=6
[ 1077.859340] JCD:STA_STATS_RATE_TYPE_LEGACY: sband=00000000792e494b
If "sband" becomes zero, or "band" becomes > 5, it highlights a bug, and will output a useful stack trace, but not crash.

Note, it gets very chatty if you are viewing the wireless page in luci.

jcdutton, 2023-10-12:
@lxp 
Are you able to try my patch?

lxp, 2023-10-14:
@jcdutton
I haven't had time to compile OpenWrt with your patch yet. Usually, I just use the imagebuilder.
In the meantime I had a crash with OpenWrt 23.03-rc4 and `wireless.radio0.legacy_rates='1'`.
However, currently it feels a bit more stable than with OpenWrt 23.03-rc3 and `wireless.radio0.legacy_rates='0'`.

jcdutton, 2023-10-24:
@hagier 
Do you still see this bug. Are you able to try my patch above to see if it helps?

pgauret, 2023-10-24:
Also affected by this issue. I could try the patch but would need a way to quickly test / trigger the bug as the BPI router is in use and the bug happens randomly sometimes only after a few days.

hayzamjs, 2023-10-24:
@jcdutton I've been running with your patch as well as a patch I rolled on my own the last few days (which adds a few more logs) with your patch I can see the following every few seconds but no dumps printed out (yet):

```
Tue Oct 24 06:49:59 2023 kern.debug kernel: [151194.247805] JCD:STA_STATS_RATE_TYPE_LEGACY: rate=16400 0x4010, band=1, rate_idx=0
Tue Oct 24 06:49:59 2023 kern.debug kernel: [151194.255384] JCD:STA_STATS_RATE_TYPE_LEGACY: ARRAY_SIZE (bands) 6, NUM_NL80211_BANDS=6
Tue Oct 24 06:49:59 2023 kern.debug kernel: [151194.263279] JCD:STA_STATS_RATE_TYPE_LEGACY: sband=00000000d48f5de4
```

This particular box is running with `legacy_rates='0'` and I have another box running with `legacy_rates='1'` which also has similar logs but no crashes/printed dumps yet, is there anything more I can do to trigger the crashes more often perhaps?

jcdutton, 2023-10-24:
@pgauret 
The bug appears to be caused when certain devices leave the area or disconnects from the AP.
I have never seen the problem myself, so cannot advise further than that regarding reproducibility.
The patch should prevent the crash/freeze, but it should output some more useful stack trace information to help track down the real cause.
So, you should not get a freeze, but I would ask you to look in the syslog for a stack strace, about once a day, to see if the problem occurred.

jcdutton, 2023-10-24:
@pgauret @hayzamjs 
You might need to set up TCP syslog to a remote host from the BPI-R3, as the BPI cannot store much in the way of logs locally.

hagier, 2023-10-24:
> @hagier Do you still see this bug. Are you able to try my patch above to see if it helps?

Sorry, but I haven't noticed any freezes since my last input in the topic. At the moment I am running version 23.05.0. I changed dump APs locations (I have 3 of them) and maybe it changed connection pattern for clients. Since BPI-R3 is my main router I cannot fiddle too much with it because my wife and I are working remotely.

dangowrt, 2023-10-24:
I finally managed to reproduce this (or a similar) problem on MT7981, but as the Ethernet part of MT7986 is more or less identical it should affect both SoCs: Up on receiving a corrupted Ethernet frame, RX DMA hangs. MediaTek actually reverted to use the old DMA and we may have to do the same:
https://git01.mediatek.com/plugins/gitiles/openwrt/feeds/mtk-openwrt-feeds/+/8ecd51bed4f7b011c2d59cf93627ca598d4e9172
https://git01.mediatek.com/plugins/gitiles/openwrt/feeds/mtk-openwrt-feeds/+/231d3ac3e990b788564973c9a37ea850accce6e4

jcdutton, 2023-10-24:
@dangowrt 
Your post above, is it related to this wifi related bug, or did you mean to post it elsewhere?

pgauret, 2023-10-24:
> @pgauret The bug appears to be caused when certain devices leave the area or disconnects from the AP. I have never seen the problem myself, so cannot advise further than that regarding [...]

If the patch is supposed to prevent the crash I’m ok to try. Need a few days to recompile and setup syslog server, will revert once done.

jcdutton, 2023-10-24:
Some notes on reading stack traces:
[39787.820172] Call trace:
[39787.822604]  sta_set_sinfo+0xa1c/0xc40 [mac80211]
[39787.827323]  ieee80211_color_change_finish+0x1b64/0x1e80 [mac80211]
[39787.833589]  nl80211_put_sta_rate+0x1030/0x11b4 [cfg80211]

The above stack trace looks rubbish (ieee80211_color_change_finish never calls sta_set_sinfo) until you understand what the stack trace function is doing.
The stack trace input has an address in memory.
The stack trace function takes that address, and converts it into a offset of what it thinks is the closest symbol/function name and then an offset.
On the above, looking at objdump of .o files, used during the build, one finds that sta_set_sinfo+0xa1c is still within the sta_set_sinfo function so its name is correct.
But  
ieee80211_color_change_finish+0x1b64   is actually past the end of the ieee80211_color_change_finish function and in fact points to a different function altogether. So the ieee80211_color_change_finish function name is wrong, and one can find the real function name by looking at .o files.
I have not looked into why the name chosen is wrong. There might be a bug in the stack trace function, or not all symbols are available to scan through.

lxp, 2023-10-29:
@jcdutton
Sadly, I still haven't had time to compile OpenWrt with your patch yet.
I am still running OpenWrt 23.03-rc4 and `wireless.radio0.legacy_rates='1'` and I haven't had any more crashes (only one crash in this combination I already mentioned last time).
I just switched the config back to `wireless.radio0.legacy_rates='0'`. Let's see if it again crashes more often in this configuration like with OpenWrt 23.03-rc3.

pgauret, 2023-10-30:
> Compile from source once first, then this file will appear. openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/backports-6.5/net/mac80211/sta_info.c
> 
> Then apply the sta_info.c.diff.txt patch to it

Still trying to get your patch. I’m building against v23.05.0 and could not find the file you refer to, instead I have two occurrences at : 

```
./build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/linux-5.15.134/net/mac80211/sta_info.c
./build_dir/toolchain-aarch64_cortex-a53_gcc-12.3.0_musl/linux-5.15.134/net/mac80211/sta_info.c
```

I patched both files and sysupgrade’d my router, but nothing in the logs, not even the JCD: printk()s. 

Which branch shall I build from ?

lxp, 2023-11-01:
@jcdutton No more crashes to report with my setup until now.
Since today I am running 23.05-snapshot-r23569-6bb3c2eba8 with your patch applied. I reverted my configuration mostly to the one I had with more crashes:
* `wireless.radio0.legacy_rates='0'`
* No collectd running (maybe influences the situation by regularly polling wireless stuff)
* No USB storage mounted (not needed without collectd)
* Case open instead of closed (should result in a bit decreased temperature)

@pgauret You have to do an initial build, so that the backports folder appears. For the 23.05 branch you have to patch the following file:
```
./build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/backports-6.1.24/net/mac80211/sta_info.c
```

lxp, 2023-11-07:
@jcdutton I just checked the log again and I finally got 7 errors with stack traces in total so far.
Here is the first one I saw:

```plain
[201377.381625] JCD:STA_STATS_RATE_TYPE_LEGACY: rate=8196 0x2004, band=0, rate_idx=4
[201377.389119] JCD:STA_STATS_RATE_TYPE_LEGACY: ARRAY_SIZE (bands) 6, NUM_NL80211_BANDS=6
[201377.397025] JCD:STA_STATS_RATE_TYPE_LEGACY: sband=000000006b2822c8
[201474.560620] JCD:STA_STATS_RATE_TYPE_LEGACY: rate=8196 0x2004, band=0, rate_idx=4
[201474.568101] JCD:STA_STATS_RATE_TYPE_LEGACY: ARRAY_SIZE (bands) 6, NUM_NL80211_BANDS=6
[201474.576007] JCD:STA_STATS_RATE_TYPE_LEGACY: sband=0000000000000000
[201474.582274] JCD:STA_STATS_RATE_TYPE_LEGACY: ERROR: sband NULL. Returning
[201474.589050] CPU: 2 PID: 2040 Comm: hostapd Not tainted 5.15.137 #0
[201474.595300] Hardware name: Bananapi BPI-R3 (DT)
[201474.599900] Call trace:
[201474.602418]  dump_backtrace+0x0/0x170
[201474.606159]  show_stack+0x14/0x30
[201474.609546]  dump_stack_lvl+0x60/0x78
[201474.613283]  dump_stack+0x14/0x2c
[201474.616671]  sta_set_sinfo+0xa8c/0xcc0 [mac80211]
[201474.621482]  sta_set_sinfo+0xbd8/0xcc0 [mac80211]
[201474.626271]  sta_info_destroy_addr_bss+0x4c/0x70 [mac80211]
[201474.631927]  ieee80211_color_change_finish+0x1c18/0x1e80 [mac80211]
[201474.638277]  cfg80211_check_station_change+0x1384/0x4700 [cfg80211]
[201474.644623]  genl_family_rcv_msg_doit+0xb4/0x110
[201474.649314]  genl_rcv_msg+0xd0/0x1c0
[201474.652960]  netlink_rcv_skb+0x58/0x120
[201474.656867]  genl_rcv+0x34/0x50
[201474.660080]  netlink_unicast+0x1f0/0x2ec
[201474.664073]  netlink_sendmsg+0x19c/0x3d0
[201474.668068]  ____sys_sendmsg+0x27c/0x2b0
[201474.672063]  ___sys_sendmsg+0x80/0xf0
[201474.675797]  __sys_sendmsg+0x44/0xa0
[201474.679444]  __arm64_sys_sendmsg+0x20/0x30
[201474.683611]  invoke_syscall.constprop.0+0x4c/0xe0
[201474.688387]  do_el0_svc+0x40/0xd0
[201474.691773]  el0_svc+0x14/0x4c
[201474.694902]  el0t_64_sync_handler+0xe0/0x110
[201474.699242]  el0t_64_sync+0x15c/0x160
```

The stack trace is always the same for all errors so far, but the rate parameters are sometimes different.
In total I had two errors with the rate parameters above and four with the following rate parameters:

```plain
[256384.162134] JCD:STA_STATS_RATE_TYPE_LEGACY: rate=8200 0x2008, band=0, rate_idx=8
[256384.169613] JCD:STA_STATS_RATE_TYPE_LEGACY: ARRAY_SIZE (bands) 6, NUM_NL80211_BANDS=6
[256384.177522] JCD:STA_STATS_RATE_TYPE_LEGACY: sband=0000000000000000
[256384.183806] JCD:STA_STATS_RATE_TYPE_LEGACY: ERROR: sband NULL. Returning
```

You can find the corresponding `kernel-debug.tar.zst` here: https://github.com/lxp/openwrt-kernel-debug/raw/master/openwrt-23.05-snapshot-r23569-6bb3c2eba8-mediatek-filogic-bananapi_bpi-r3_kernel-debug.tar.zst

lukasz1992, 2023-11-07:
Thanks, so it looks like condition:

if `(WARN_ON_ONCE(!sband->bitrates))`
in sta_stats_decode_rate

needs to be preceded by

```
if (WARN_ON_ONCE(!sband))
			break;
```

jcdutton, 2023-11-07:
@lxp 
Thank you for the response.
Did the router at least carry on functioning with my patch, instead of crash to a halt?
I have found some other places problems can happen, so will be submitting a patch to upstream when I get a mo.
One thing I wished to check was whether the "band" variable got >= 6 or not, because that could have been another failure mode, but it looks like "band" was a valid value.
Thank you for helping.

lxp, 2023-11-08:
@jcdutton I did not notice any misbehaviour instead of the error messages, so the router is still running since the first boot and is working.
Additionally to your check for `band >= 6`, I also added a check for `band < 0`. However, both checks never failed until now.

Just some things, I found out while browsing the sourcecode:
* band 0 seems to refer to NL80211_BAND_2GHZ
* My guess for the actual stack trace is the following:
  * `sta_stats_decode_rate` (https://elixir.bootlin.com/linux/v6.1.24/source/net/mac80211/sta_info.c#L2356 - where the crash happened)
  * `sta_set_rate_info_rx` (https://elixir.bootlin.com/linux/v6.1.24/source/net/mac80211/sta_info.c#L2389)
  * `sta_set_sinfo` (https://elixir.bootlin.com/linux/v6.1.24/source/net/mac80211/sta_info.c#L2649)
  * `__sta_info_destroy_part2` (https://elixir.bootlin.com/linux/v6.1.24/source/net/mac80211/sta_info.c#L1304)
  * `__sta_info_destroy` (https://elixir.bootlin.com/linux/v6.1.24/source/net/mac80211/sta_info.c#L1324)
  * `sta_info_destroy_addr_bss` (https://elixir.bootlin.com/linux/v6.1.24/source/net/mac80211/sta_info.c#L1350)
* Therefore, it looks like that the crash happens during removing of a station. Shortly before it is completely removed the `sinfo` is collected via `sta_set_sinfo`, which is then used to notify userspace via `cfg80211_del_sta_sinfo` about the deleted station.
* For me it does not seem to be an important information to have an accurate rx rate, at the time when a station is removed. That's why I also think that ignoring that case might be a valid workaround.

Did you figure out, why `local->hw.wiphy->bands[0]` sometimes might be `null`?
My first guess was that the wiphy bands must be somehow corrupted or it is some kind of race-condition.
However, now I think that it might just be that the last rx rate for the station is somehow corrupted.
Could it be that the last rx rate is kept (or not correctly initialized) during roaming between 2 GHz and 5 GHz and if it was a legacy rate then points to the wrong band that does not exist on this phy?

lxp, 2023-11-27:
I think we should apply the hotfix as suggested by @lukasz1992 to avoid unnecessary kernel panics and make BPI-R3 usable also in our setups.
I opened a pull request for the 23.05 branch: https://github.com/openwrt/openwrt/pull/14083

trunneml, 2023-12-31:
I think it is a good idea to add this null check. We should also find out the root cause for sband being null, but this null check is obvious missing and makes sense to me..

essys, 2024-03-27:
Can anyone confirm if a supposed fix was merged into 23.05.3?
I did the upgrade yesterday and the situation didn't change at all. 
My BPI-R3 still regularly freezes and only a power cycle helps.
This happens so randomly, but too often, that the device cannot be used in this state.

pgauret, 2024-03-27:
Same here, upgraded to 23.05.3 yesterday and got a freeze today.

trunneml, 2024-03-28:
@essys @pgauret  20.05.03 contains a similar patch: https://github.com/openwrt/openwrt/blob/v23.05.3/package/kernel/mac80211/patches/subsys/780-avoid-crashing-missing-band.patch

My Banana Pi R3 is not freezing. (But I don't use roaming between 2G and 5G at all). I think your crashes have a different reason.

pgauret, 2024-03-28:
@trunneml indeed I was hoping this patch would stop the crashes. Not the case. I have disabled 2.4Ghz again which helped a lot previously. Let’s see how this holds.

trunneml, 2024-03-28:
@pgauret I had problems with AX under 5Ghz may be try to disable ax under 2G? I‘m using 2G for IOT and 5G for LAN. It is working stable since this patch by disabling AX.