openwrt/openwrt #13199: Kernel Bug Related to Intel Present

MajorMeerkat, 2023-07-31:
### Describe the bug

I found this in my system log. Reading other Linux threads, it seems likely to be part of the linux kernel in use: 5.15

`Mon Jul 31 07:32:45 2023 kern.err kernel: [    0.706042] ACPI BIOS Error (bug): Could not resolve symbol [\_PR.PR00._CPC], AE_NOT_FOUND (20210730/psargs-330)
Mon Jul 31 07:32:45 2023 kern.err kernel: [    0.706889] ACPI Error: Aborting method \_PR.PR01._CPC due to previous error (AE_NOT_FOUND) (20210730/psparse-529)
Mon Jul 31 07:32:45 2023 kern.err kernel: [    0.707793] ACPI BIOS Error (bug): Could not resolve symbol [\_PR.PR00._CPC], AE_NOT_FOUND (20210730/psargs-330)
Mon Jul 31 07:32:45 2023 kern.err kernel: [    0.708628] ACPI Error: Aborting method \_PR.PR02._CPC due to previous error (AE_NOT_FOUND) (20210730/psparse-529)
Mon Jul 31 07:32:45 2023 kern.err kernel: [    0.709527] ACPI BIOS Error (bug): Could not resolve symbol [\_PR.PR00._CPC], AE_NOT_FOUND (20210730/psargs-330)
Mon Jul 31 07:32:45 2023 kern.err kernel: [    0.710362] ACPI Error: Aborting method \_PR.PR03._CPC due to previous error (AE_NOT_FOUND) (20210730/psparse-529)
Mon Jul 31 07:32:45 2023 kern.err kernel: [    0.711262] ACPI BIOS Error (bug): Could not resolve symbol [\_PR.PR00._CPC], AE_NOT_FOUND (20210730/psargs-330)
Mon Jul 31 07:32:45 2023 kern.err kernel: [    0.712096] ACPI Error: Aborting method \_PR.PR04._CPC due to previous error (AE_NOT_FOUND) (20210730/psparse-529)
Mon Jul 31 07:32:45 2023 kern.err kernel: [    0.712993] ACPI BIOS Error (bug): Could not resolve symbol [\_PR.PR00._CPC], AE_NOT_FOUND (20210730/psargs-330)
Mon Jul 31 07:32:45 2023 kern.err kernel: [    0.713827] ACPI Error: Aborting method \_PR.PR05._CPC due to previous error (AE_NOT_FOUND) (20210730/psparse-529)
Mon Jul 31 07:32:45 2023 kern.err kernel: [    0.714726] ACPI BIOS Error (bug): Could not resolve symbol [\_PR.PR00._CPC], AE_NOT_FOUND (20210730/psargs-330)
Mon Jul 31 07:32:45 2023 kern.err kernel: [    0.715560] ACPI Error: Aborting method \_PR.PR06._CPC due to previous error (AE_NOT_FOUND) (20210730/psparse-529)
Mon Jul 31 07:32:45 2023 kern.err kernel: [    0.716458] ACPI BIOS Error (bug): Could not resolve symbol [\_PR.PR00._CPC], AE_NOT_FOUND (20210730/psargs-330)
Mon Jul 31 07:32:45 2023 kern.err kernel: [    0.717297] ACPI Error: Aborting method \_PR.PR07._CPC due to previous error (AE_NOT_FOUND) (20210730/psparse-529)4`

### OpenWrt version

23.05 rc2

### OpenWrt target/subtarget

x86/64

### Device

x86 Generic (ZBOX Nano C1660)

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-07-31:
That is buggy ACPI table in BIOS, boot some major distro from Live USB to check if newer kernel works around it better, and make sure your BIOS ROM is latest. Usually it takes some debugging with acpidump from acpica package and long talks with OEM to get it right.

brada4, 2023-08-03:
CPC means collaborative power control, something of a cpufreq on steroids.

MajorMeerkat, 2023-08-04:
> 

I have checked for any BIOS updates for the device. The last was in 2019, and even then, there is no way to install it other than through Windows. Loading Windows isn't an option on the device.

brada4, 2023-08-04:
Walkthrough guide on fixing acpi table: https://wiki.archlinux.org/title/DSDT
Nothing kernel can do about digging up missing method referred to in ACPI table.

MajorMeerkat, 2023-08-07:
I reverted to 22.03.05 and the issue does not replicate itself there. Is it plausible that it is a Kernel issue?

brada4, 2023-08-07:
> boot some major distro

Sometimes common BIOS bugs get silenced. Please check later kernel from possibly other source.

MajorMeerkat, 2023-08-08:
Hmmm interesting: the bug seems to be mentioned here: https://askubuntu.com/questions/1416198/ubuntu-22-04-acpi-bios-error-bug-could-not-resolve-symbol-errors-on-asus-x7 too.

brada4, 2023-08-08:
If you go to kernel bug tracker you see issue is not finalized, so it will take some time it gets backported to 5.15 series and land into next OpenWRT release.
Essentially overly careful acpi table sanitosation disabling cppc on 2021-23 ?laptops?.

MajorMeerkat, 2023-08-08:
I understand it might take some time before the issue is resolved. I am trying to get some visibility of the issue here, so that it might get resolved along the line. I imagine I am not the only person who will come across the issue either.