openwrt/openwrt #13201: Dns not working with custom port (dual dnsmasq instances)

professor-jonny, 2023-07-31:
### Describe the bug

I attempted to set two Dnsmasq instances with a separate DNS on each instance spanning both interfaces with seperate ports so I could create rules to force clients via rules to use a specific DNS regardless of the instance they connect to.

It seems when setup with multiple instances I can only use port 53 for DNS specifying another port breaks DNS.

### OpenWrt version

commit/4c5d910ef152886ea285f8c1ca50924af03fe55e

### OpenWrt target/subtarget

ipq40x9

### Device

wallystech dr4029

### Image kind

Official downloaded image

### Steps to reproduce

Setup Dual Dnsmasq instances running separate DHCP servers.
setup one Dnsmasq instance's DNS on a custom port and make it span span both interfaces by removing interface and not interface options on that instance.

### Actual behaviour

The DNS only on the instance with port 53 specified works if attached to that interface.
port forwarding to port 53 from the non working instance works.
port forwarding to the custom DNS port reveals broken DNS.

### Expected behaviour

Port forward of DNS queries on custom port should work.

### Additional info

Reverting to port 53 and setting dns to lock to their respective instance causes both DNS's to operate as expected.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-08-02:
What do you do in /etc/config/dhcp and init.d scripts?

professor-jonny, 2023-08-02:
This is what I did with my config sorry if there are errors, I have reverted my config as I could not get it working so I'm going off memory.
I did not play with the init.d scripts.

```
config dnsmasq 'adults_dns'
	option localise_queries '1'
	option rebind_protection '0'
	option local '/adults_lan/'
	option domain 'adults_lan'
	option expandhosts '1'
	option authoritative '1'
	option readethers '1'
	option leasefile '/tmp/adults_lan/dhcp.leases'
	list interface 'adults_lan'
	option localservice '0'
	list notinterface 'kids_lan'
	option confdir '/tmp/adults_lan/dnsmasq.d'
	option quietdhcp '1'
	option filterwin2k '1'
	option allservers '1'
	list server '127.0.0.1#5054'
	list server '127.0.0.1#5053'
	option doh_backup_noresolv '-1'
	option noresolv '1'
	list doh_backup_server '8.8.8.8'
	option boguspriv '0'

config dnsmasq 'kids_dns'
	option localise_queries '1'
	option rebind_protection '0'
	option local '/kids_lan/'
	option domain 'kids_lan'
	option expandhosts '1'
	option authoritative '1'
	option readethers '1'
	option leasefile '/tmp/kids_lan/dhcp.leases'
	option confdir '/tmp/kids_lan/dnsmasq.d'
	option localservice '0'
	list notinterface 'loopback'
	option quietdhcp '1'
	option filterwin2k '1'
	option allservers '1'
	list server '127.0.0.1#5054'
	list server '127.0.0.1#5053'
	option doh_backup_noresolv '-1'
	option noresolv '1'
	list doh_backup_server '8.8.8.8'
	option boguspriv '0'
	option port '5353'

config dhcp 'adults_lan'
	option instance 'adults_dns'
	option interface 'adults_lan'
	option start '100'
	option limit '150'
	option leasetime '12h'
	option force '0'

config dhcp 'kids_lan'
	option instance 'kids_dns'
	option interface 'kids_lan'
	option start '100'
	option leasetime '12h'
	option limit '150'
	option force '0'

```

```
config redirect
        option name 'Divert teenagers DNS'
        option src 'adults_lan'
        option proto 'tcp udp'
        option src_dport '53'
        option dest_port '5353'
        option target 'DNAT'

config redirect
        option name 'Divert kids DNS'
        option src 'kids_lan'
        option proto 'tcp udp'
        option src_dport '53'
        option dest_port '5353'
        option target 'DNAT'

The younger kids' devices connected to the kids_lan all the time and the teenagers and adults connect to adults_lan.

I setup a firewall rule with an ipset with a group of devices with a time component to force the kids_dns to teenage devices when attached to the Adults_lan interface.

The kids_dns is forced with the firewall rule to the alternate port for all devices attached to the Kids_lan interface.

Sadly, it totally broke DNS when attached to the kids_lan interface and port forwarding to 5353 when attached to the adults_lan interface also shows a broken DNS.

Reverting the kids_dns to port 53 and setting the list and list not interface on the Kids_lan after removing the firewall rules fixes DNS resolution on the kids_lan interface which would suggest it did not like the custom port.

```

brada4, 2023-08-03:
Firewall help is in forums.

professor-jonny, 2023-08-03:
> Firewall help is in forums.

It is not a firewall issue as when a device is attached to the kids_lan interface I can forward port 53 to the IP address and port of the other instance and DNS resolution works.

professor-jonny, 2026-03-02:
> For anyone coming across this later, openwrt only runs one dnsmasq instance, regardless of how many config entries there are. That one instance can manage multiple dhcp config sections, but not multiple dnsmasq ones.

I don't think that is correct if, but it seems to work for me and creates different folders for each instance in TMP.

I run multiple instances as they are tied in with httpdns proxy and different instances of AdBlock so I have an adult's instance and a kid's instance with completely different rules etc....

I do admit luci does not support both instances correctly