openwrt/openwrt #13202: sama5d2: toolchain configuration misses -mfpu=vfpv3-d16 or similar

pavlix, 2023-08-01:
### Describe the bug

Sama5D3 doesn't support vfpv3/vfpv4 with 32 registers but only 16 registers. See the datasheet.

https://ww1.microchip.com/downloads/aemDocuments/documents/MPU32/ProductDocuments/DataSheets/SAMA5D3-Series-Data-sheet-DS60001609b.pdf

This seems to be in line with what ARM website says about Cortex A5.

https://developer.arm.com/documentation/100302/0001/introduction/about-the-cortex-a5-fpu

OpenWRT doesn't even seem to use *armv7-a* as the `-march` and even then `-mfpu` is not supposed to be `vpfv4` as that, according to the docs, means 32 double-precision registers which are not supposed to be available.

Building for 32 registers results in user space failure caused by SIGILL most probably when the program attempts to use missing registers.

```
procd: - early -
Kernel panic - not syncing: Attempted to kill init! exitcode=0x00000004
```

That means the following definition doesn't hold at least for *sama5d3* and should probably say `vfpv4-d16` instead of just `vfpv4`. So far I have only tested `vfpv3-d16` and that prevented the SIGILL. This might be specific to *sama5d3* rather than all *sama5* targets.

https://github.com/openwrt/openwrt/blob/main/target/linux/at91/sama5/target.mk

This configuration results in passing `--with-fpu` to the GCC configure script. Using `CPU_SUBTYPE=vfpv4` results in `--with-fpu=vfpv4` that in turn results in SIGILL on *sama5d3*. This applies to the main branch, 23.05.0-rc2 and older releases as well.

### OpenWrt version

23.05.0-rc2

### OpenWrt target/subtarget

at91/sama5

### Device

microchip_sama5d3-xplained

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

ynezz, 2023-08-01:
> This applies to the main branch, 23.05.0-rc2 and older releases as well.

Does that mean, that 22.03 and 21.02 releases doesn't work on this target as well? Out of interest, have you actually run tested those?

ynezz, 2023-08-01:
> and should probably say vfpv4-d16 instead of just vfpv4

Ok, but this change would result in another toolchain and quite huge demand for build/infra resources so if that is really needed, then I would vote to "fix" it by making this subtarget `source-only` as if this target is broken for several years, then clearly there are no users of binary releases?

KanjiMonster, 2023-08-01:
> Ok, but this change would result in another toolchain and quite huge demand for build/infra resources so if that is really needed, then I would vote to "fix" it by making this subtarget [...]

at91/sama5 is the only cortex-a5 user, so switching from vfp4 to vfp4-d16 should not result in more resources needed. Doesn't change the fact that it was broken this long and nobody noticed it until now.

ynezz, 2023-08-01:
> at91/sama5 is the only cortex-a5 user

Thats exactly my point, the only target, clearly without any active userbase, vendor doesn't care.

> so switching from vfp4 to vfp4-d16 should not result in more resources needed

FYI this switch will result in the need for additional `arm_cortex-a5_vfpv4-d16` [builder](https://buildbot.staging.openwrt.org/master/packages/#/builders), so need for additional 3-19 hours of build time (depends on builder hardware specs), numbers taken from the existing [arm_cortex-a5_vfpv4](https://buildbot.staging.openwrt.org/master/packages/#/builders/18).

So just multiply it with the numbers of builds, then number of branches where we're going to backport that fix (23.05 and 22.03?) and you should've complete picture.

ynezz, 2023-08-01:
> Does that mean, that 22.03 and 21.02 releases doesn't work on this target as well? Out of interest, have you actually run tested those?

I mean, that if this target works fine in 21.02 and 22.03, then its likely some kernel/toolchain issue and ideally should be fixed by other means, which wouldn't need additional infra resources.

KanjiMonster, 2023-08-01:
> > at91/sama5 is the only cortex-a5 user
> 
> Thats exactly my point, the only target, clearly without any active userbase, vendor doesn't care.

No, I meant it is the only (sub)target using the cputype `cortex-a5`.

> > so switching from vfp4 to vfp4-d16 should not result in more resources needed
> 
> FYI this switch will result in the need for additional `arm_cortex-a5_vfpv4-d16` [builder](https://buildbot.staging.openwrt.org/master/packages/#/builders), so need for additional 3-24 hours of build time (depends on buildworker hardware specs), numbers taken from the existing [arm_cortex-a5_vfpv4](https://buildbot.staging.openwrt.org/master/packages/#/builders/18).

Which is why at the same time we don't need the `arm_cortex-a5_vfpv4` builder anymore, so we can drop that one => same amount of resources used as before, just now for a different builder.

ynezz, 2023-08-01:
> same amount of resources used as before, just now for a different builder.

Ok, you're right, so I agree, should be fine.

KanjiMonster, 2023-08-01:
> > same amount of resources used as before, just now for a different builder.
> 
> Ok, you're right, so I agree, should be fine.

If we *do* want to drop a builder there might be the chance for `arm_cortex-a7_vfpv4` though - at a first glance, the only user for that is `at91/sama7`, and according to [its datasheet](https://ww1.microchip.com/downloads/aemDocuments/documents/MPU32/ProductDocuments/DataSheets/SAMA7G5-Series-Data-Sheet-DS60001765A.pdf), it does support NEON, so it should be possible to switch it to the already existing `arm_cortex-a7_neon-vfpv4`.

I have no (access to) hardware to verify though.

pavlix, 2023-08-01:
> > This applies to the main branch, 23.05.0-rc2 and older releases as well.
> 
> Does that mean, that 22.03 and 21.02 releases doesn't work on this target as well? Out of interest, have you actually run tested those?

Well. I've been using a patched 21.02 for a while and I can't speak for other users. Before that I used CONFIG_SOFT_FLOAT to work around the issue but that of course requires a freshly-build toolchain as well.

The target device supported by  OpenWRT is SAMA5D3 Xplained:

> The SAMA5D3 Xplained is a fast prototyping and evaluation platform for microprocessor-based design. The board comes with a rich set of ready to use connectivity and storage peripherals and expansion headers for easy customization. A Linux distribution and software package gets you evaluating fast. A USB device connector can be used to power the board as well as programming and debugging it.

It is not a typical OpenWRT-reflashed home router and that's probably why there are no bug reports from OpenWRT image users.

> Which is why at the same time we don't need the `arm_cortex-a5_vfpv4` builder anymore, so we can drop that one => same amount of resources used as before, just now for a different builder.

It does look like `arm_cortex-a5_vfpv4` can safely be dropped. If only classic end users are targeted, maybe `arm_cortex-a5_vfpv4-d16` can indeed be source-only. That still looks like an improvement for the user to know how to build correctly rather than to get a bunch of defunct artifacts including the toolchain.

I will use the toolchain tarball for `-d16` if it is built and published or I will keep building the toolchain tarball myself if it's not.