openwrt/openwrt #13206: Cannot setup wireless interfaces

milouk, 2023-08-01:
### Describe the bug

After a regular attended sysupgrade, the wireless interfaces fail to initialize and hence show as available when connecting to wifi

### OpenWrt version

r23664-11bf820124

### OpenWrt target/subtarget

qualcommax/ipq807x

### Device

Xiaomi AX3600

### Image kind

Official downloaded image

### Additional info

Here is the relative log

```
Tue Aug  1 23:43:59 2023 authpriv.info dropbear[9718]: Child connection from 192.168.1.168:53978
Tue Aug  1 23:43:59 2023 authpriv.notice dropbear[9718]: Pubkey auth succeeded for 'root' with ssh-rsa key SHA256:508evDLxjYw9FeXoxmNkfTRvnBbJhsCAhMnomyFE9k4 from 192.168.1.168:53978
Tue Aug  1 23:44:32 2023 daemon.err uhttpd[2240]: llectors/wifi_stations.lua:33: attempt to index local 'iw' (a nil value)/usr/lib/lua/prometheus-collectors/wifi.lua:19: attempt to index local 'iw' (a nil value)/usr/lib/lua/prometheus-collectors/wifi_stations.lua:33: attempt to index local 'iw' (a nil value)/usr/lib/lua/prometheus-collectors/wifi.lua:19: attempt to index local 'iw' (a nil value)/usr/lib/lua/prometheus-collectors/wifi_stations.lua:33: attempt to index local 'iw' (a nil value)/usr/lib/lua/prometheus-collectors/wifi.lua:19: attempt to index local 'iw' (a nil value)/usr/lib/lua/prometheus-collectors/wifi_stations.lua:33: attempt to index local 'iw' (a nil value)/usr/lib/lua/prometheus-collectors/wifi.lua:19: attempt to index local 'iw' (a nil value)/usr/lib/lua/prometheus-collectors/wifi_stations.lua:33: attempt to index local 'iw' (a nil value)/usr/lib/lua/prometheus-collectors/wifi.lua:19: attempt to index local 'iw' (a nil value)/usr/lib/lua/prometheus-collectors/wifi_stations.lua:33: attempt to index local
Tue Aug  1 23:44:33 2023 daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(br-lan) 192.168.1.157 cc:32:e5:78:e2:c7
Tue Aug  1 23:44:33 2023 daemon.info dnsmasq-dhcp[1]: DHCPACK(br-lan) 192.168.1.157 cc:32:e5:78:e2:c7 TL-WPA8630P
Tue Aug  1 23:45:11 2023 daemon.notice netifd: radio2 (9788): Command failed: ubus call hostapd config_set { "phy": "phy2", "config": "", "prev_config": "/var/run/hostapd-phy2.conf" } (Method not found)
Tue Aug  1 23:45:11 2023 daemon.notice netifd: radio2 (9788): Command failed: ubus call wpa_supplicant config_set { "phy": "phy2", "config": [] } (Method not found)
Tue Aug  1 23:45:11 2023 daemon.notice netifd: radio2 (9788): Failed to open "/usr/share/hostap/wdev.uc": No such file or directory
Tue Aug  1 23:45:11 2023 daemon.notice netifd: Wireless device 'radio2' is now down
Tue Aug  1 23:45:11 2023 daemon.notice netifd: radio2 (9820): Command failed: ubus call wpa_supplicant config_set { "config": [ ], "phy": "phy2", "defer": true } (Method not found)
Tue Aug  1 23:45:11 2023 daemon.notice netifd: radio2 (9820): Interface  setup failed: WPA_SUPPLICANT_FAILED
Tue Aug  1 23:45:11 2023 daemon.notice netifd: radio2 (9820): Failed to parse json data: unexpected end of data
Tue Aug  1 23:45:11 2023 daemon.notice netifd: radio2 (9820): WARNING (wireless_add_process): executable path /usr/sbin/wpad does not match process  path (/proc/exe)
Tue Aug  1 23:45:11 2023 daemon.notice netifd: radio2 (9820): Command failed: ubus call network.wireless notify { "command": 2, "device": "radio2", "data": { "pid": 0, "exe": "\/usr\/sbin\/wpad", "required": true, "keep": true } } (Invalid argument)
Tue Aug  1 23:45:11 2023 daemon.notice netifd: radio2 (9820): Usage: ubus [<options>] <command> [arguments...]
Tue Aug  1 23:45:11 2023 daemon.notice netifd: radio2 (9820): Options:
Tue Aug  1 23:45:11 2023 daemon.notice netifd: radio2 (9820):  -s <socket>:		Set the unix domain socket to connect to
Tue Aug  1 23:45:11 2023 daemon.notice netifd: radio2 (9820):  -t <timeout>:		Set the timeout (in seconds) for a command to complete
Tue Aug  1 23:45:11 2023 daemon.notice netifd: radio2 (9820):  -S:			Use simplified output (for scripts)
Tue Aug  1 23:45:11 2023 daemon.notice netifd: radio2 (9820):  -v:			More verbose output
Tue Aug  1 23:45:11 2023 daemon.notice netifd: radio2 (9820):  -m <type>:		(for monitor): include a specific message type
Tue Aug  1 23:45:11 2023 daemon.notice netifd: radio2 (9820): 			(can be used more than once)
Tue Aug  1 23:45:11 2023 daemon.notice netifd: radio2 (9820):  -M <r|t>		(for monitor): only capture received or transmitted traffic
Tue Aug  1 23:45:11 2023 daemon.notice netifd: radio2 (9820):
Tue Aug  1 23:45:11 2023 daemon.notice netifd: radio2 (9820): Commands:
Tue Aug  1 23:45:11 2023 daemon.notice netifd: radio2 (9820):  - list [<path>]			List objects
Tue Aug  1 23:45:11 2023 daemon.notice netifd: radio2 (9820):  - call <path> <method> [<message>]	Call an object method
Tue Aug  1 23:45:11 2023 daemon.notice netifd: radio2 (9820):  - subscribe <path> [<path>...]	Subscribe to object(s) notifications
Tue Aug  1 23:45:11 2023 daemon.notice netifd: radio2 (9820):  - listen [<path>...]			Listen for events
Tue Aug  1 23:45:11 2023 daemon.notice netifd: radio2 (9820):  - send <type> [<message>]		Send an event
Tue Aug  1 23:45:11 2023 daemon.notice netifd: radio2 (9820):  - wait_for <object> [<object>...]	Wait for multiple objects to appear on ubus
Tue Aug  1 23:45:11 2023 daemon.notice netifd: radio2 (9820):  - monitor				Monitor ubus traffic
Tue Aug  1 23:45:11 2023 daemon.notice netifd: radio2 (9820):
Tue Aug  1 23:45:11 2023 daemon.notice netifd: radio2 (9820): Command failed: ubus call hostapd config_set { "phy": "phy2", "config":"/var/run/hostapd-phy2.conf", "prev_config": "/var/run/hostapd-phy2.conf.prev"} (Method not found)
Tue Aug  1 23:45:11 2023 daemon.notice netifd: radio2 (9820): Device setup failed: HOSTAPD_START_FAILED
Tue Aug  1 23:45:11 2023 daemon.notice netifd: Wireless device 'radio2' set retry=0
Tue Aug  1 23:45:11 2023 daemon.notice netifd: radio2 (9820): Failed to open "/usr/share/hostap/wdev.uc": No such file or directory
Tue Aug  1 23:45:11 2023 daemon.notice netifd: Wireless device 'radio2' is now up
Tue Aug  1 23:45:13 2023 daemon.info dnsmasq[1]: read /etc/hosts - 12 names
Tue Aug  1 23:45:13 2023 daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c - 8 names
Tue Aug  1 23:45:13 2023 daemon.info dnsmasq[1]: read /tmp/hosts/odhcpd - 4 names
Tue Aug  1 23:45:13 2023 daemon.info dnsmasq-dhcp[1]: read /etc/ethers - 0 addresses
Tue Aug  1 23:45:40 2023 daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(br-lan) 192.168.1.157 cc:32:e5:78:e2:c7
Tue Aug  1 23:45:40 2023 daemon.info dnsmasq-dhcp[1]: DHCPACK(br-lan) 192.168.1.157 cc:32:e5:78:e2:c7 TL-WPA8630P
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

khanjui, 2023-08-01:
Same here, probably latest hostapd commits broke it. @nbd168

rany2, 2023-08-01:
Also see my older comment here: https://github.com/openwrt/openwrt/commit/33e4ad767e085f691bf5433d8808b20dfcf63222#commitcomment-123341904 

Seems a different bug here, but it also occurred after I took in those changes.

st7105, 2023-08-02:
Same problem

MisharUser, 2023-08-02:
I also have the same problem

Lyceris-chan, 2023-08-02:
same problem (just so it shows up in my issues tab as well)

kofec, 2023-08-02:
https://github.com/openwrt/openwrt/commit/150e6d28f2659b0614e87cc7822e93f488e11ede

rany2, 2023-08-02:
@kofec not a fix, I still experience https://github.com/openwrt/openwrt/commit/33e4ad767e085f691bf5433d8808b20dfcf63222#commitcomment-123341904. Comment says:

I'm not sure which one of the recent hostapd commits introduced this issue that I'm currently facing on a DBDC chip, but the second radio is almost consistently failing now (about 1 in 3 times). Previously I was on e3dd5ee02bb3acb2022405f67ed7ec854481f8cf which did not have this issue:

```
Tue Aug  1 19:03:35 2023 daemon.notice hostapd: Set new config for phy phy0: /var/run/hostapd-phy0.conf
Tue Aug  1 19:03:35 2023 daemon.notice hostapd: Restart interface for phy phy0
Tue Aug  1 19:03:36 2023 daemon.notice hostapd: Configuration file: data: driver=nl80211 logger_syslog=127 logger_syslog_level=2 logger_stdout=127 logger_stdout_level=2 country_code=PA ieee80211d=1 hw_mode=g supported_rates=60 90 120 180 240 360 480 540 basic_rates=60 120 240 beacon_int=100 chanlist=1-11 ieee80211n=1 ht_coex=0 ht_capab=[LDPC][SHORT-GI-20][SHORT-GI-40][TX-STBC][RX-STBC1][MAX-AMSDU-7935] ieee80211ax=1 he_su_beamformer=1 he_su_beamformee=1 he_mu_beamformer=1 he_bss_color=128 he_spr_sr_control=3 he_default_pe_duration=4 he_rts_threshold=1023 he_mu_edca_qos_info_param_count=0 he_mu_edca_qos_info_q_ack=0 he_mu_edca_qos_info_queue_request=0 he_mu_edca_qos_info_txop_request=0 he_mu_edca_ac_be_aifsn=8 he_mu_edca_ac_be_aci=0 he_mu_edca_ac_be_ecwmin=9 he_mu_edca_ac_be_ecwmax=10 he_mu_edca_ac_be_timer=255 he_mu_edca_ac_bk_aifsn=15 he_mu_edca_ac_bk_aci=1 he_mu_edca_ac_bk_ecwmin=9 he_mu_edca_ac_bk_ecwmax=10 he_mu_edca_ac_bk_timer=255 he_mu_edca_ac_vi_ecwmin=5 he_mu_edca_ac_vi_ecwmax=7 he_mu_edca_ac_vi_aifsn=5 he_mu_edca_ac_
Tue Aug  1 19:03:36 2023 daemon.notice hostapd: eap2ghz: interface state UNINITIALIZED->COUNTRY_UPDATE
Tue Aug  1 19:03:36 2023 daemon.info hostapd: eap2ghz: RADIUS Authentication server 10.146.120.1:1812
Tue Aug  1 19:03:37 2023 daemon.warn hostapd: nl80211: Failed to add interface psk2ghz into bridge br-vlan: Resource busy
Tue Aug  1 19:03:38 2023 daemon.notice hostapd: eap2ghz: interface state COUNTRY_UPDATE->ENABLED
Tue Aug  1 19:03:38 2023 daemon.notice hostapd: eap2ghz: AP-ENABLED
Tue Aug  1 19:03:38 2023 daemon.notice hostapd: netlink: recvfrom failed: No buffer space available
Tue Aug  1 19:03:38 2023 daemon.info hostapd: eap5ghz: RADIUS Authentication server 10.146.120.1:1812
Tue Aug  1 19:03:38 2023 daemon.err hostapd: Failed to initialize QoS Map
Tue Aug  1 19:03:38 2023 daemon.warn hostapd: eap5ghz: Could not connect to kernel driver
Tue Aug  1 19:03:38 2023 daemon.err hostapd: Interface initialization failed
Tue Aug  1 19:03:38 2023 daemon.notice hostapd: eap5ghz: interface state HT_SCAN->DISABLED
Tue Aug  1 19:03:38 2023 daemon.notice hostapd: eap5ghz: AP-DISABLED
Tue Aug  1 19:03:38 2023 daemon.err hostapd: VLAN: vlan_event_receive: recvfrom failed: No buffer space available
```

edrikk, 2023-08-03:
Just updated to [57fda4](https://github.com/openwrt/openwrt/commit/57fda4b743ecf06c217d2670002313d528ff817a) on my AX3600, and the issue is there.  Thought there might be a chance that that [150e6d28](https://github.com/openwrt/openwrt/commit/150e6d28f2659b0614e87cc7822e93f488e11ede) fixed it, but no.

ascendbeing, 2023-08-03:
I wrote this in another issue about this, pasting and do let me know if this works for you:
usually happens when the packages aren't synced up with the snapshot. you can work around this by installing the generic snapshot, compiling your own, or using image builder (but keeping the wpad-basic-mbedtls, dont overwrite with a different wpad or hapd from a feed). the builds may take up to a week for it to build the correct feed.
1-3 days ago there were major changes that the packages haven't been built against. so flash the normal snapshot pre built image. and remember, remove anything that requires full version of wpad out of config/wireless. like 11r/11k/etc options. cause unless they chsnged it having options that require full won't let wpad or hapd run.
I downgraded to 23.05rc2 when this happened to me so I want to know, does this work around the issue for you? 
basically keep wpad-basic-mbedtls from the snapshot. you could extract the files corresponding to it and drop them onto your devices, btw. don't have to reflash if you know how to do it
 there's like 2 different software that can inspect the squashfs images. firmware toolkit is one.

rany2, 2023-08-03:
https://github.com/openwrt/openwrt/commit/b8be20c7e81de2894df8fa2b361c39bc723e4cb5 should resolve this issue for DBDC devices. So far the devices that are reporting this issue after https://github.com/openwrt/openwrt/commit/150e6d28f2659b0614e87cc7822e93f488e11ede are MT7905D (dbdc chip) and Xiaomi AX3600 (also dbdc chip). So seems like this should close up the final regression, though I haven't tested this myself yet.

DazzyWalkman, 2023-08-04:
I think the recent series updates to hostapd justify a release bump to help snapshot image builder users avoid downloading (dated) broken package builds.

milouk, 2023-08-04:
I confirm its fixed, so Im closing this issue

closed 2023-08-04 by milouk

rany2, 2023-08-04:
@nbd168 I think there's still a bug, sometimes when I do a reload of the config after making config changes the second "radio" on a DBDC chip doesn't start up:

```
Fri Aug  4 11:28:59 2023 kern.info kernel: [10017.060302] device eap5ghz left promiscuous mode
Fri Aug  4 11:28:59 2023 kern.info kernel: [10017.065197] br-vlan: port 10(eap5ghz) entered disabled state
Fri Aug  4 11:29:01 2023 daemon.notice hostapd: Set new config for phy phy1: /var/run/hostapd-phy1.conf
Fri Aug  4 11:29:01 2023 daemon.notice hostapd: Reload config for bss 'eap5ghz' on phy 'phy1'
Fri Aug  4 11:29:01 2023 daemon.info hostapd: eap5ghz: RADIUS Authentication server 10.146.120.1:1812
Fri Aug  4 11:29:02 2023 daemon.notice hostapd: Restart interface for phy phy1
Fri Aug  4 11:29:02 2023 daemon.notice hostapd: Remove interface 'eap5ghz'
Fri Aug  4 11:29:02 2023 daemon.notice hostapd: eap5ghz: interface state ENABLED->DISABLED
Fri Aug  4 11:29:02 2023 daemon.notice hostapd: eap5ghz: AP-DISABLED
Fri Aug  4 11:29:02 2023 daemon.notice hostapd: eap5ghz: CTRL-EVENT-TERMINATING
Fri Aug  4 11:29:02 2023 daemon.notice hostapd: nl80211: deinit ifname=eap5ghz disabled_11b_rates=0
Fri Aug  4 11:29:02 2023 daemon.notice hostapd: nl80211: Failed to remove interface eap5ghz from bridge br-vlan: Invalid argument
Fri Aug  4 11:29:03 2023 kern.info kernel: [10020.245434] br-vlan: port 5(eap5ghz) entered blocking state
Fri Aug  4 11:29:03 2023 kern.info kernel: [10020.251222] br-vlan: port 5(eap5ghz) entered disabled state
Fri Aug  4 11:29:03 2023 kern.info kernel: [10020.257231] device eap5ghz entered promiscuous mode
Fri Aug  4 11:29:03 2023 daemon.notice netifd: Network device 'eap5ghz' link is up
Fri Aug  4 11:29:03 2023 daemon.notice netifd: Network device 'eap5ghz' link is down
Fri Aug  4 11:29:03 2023 kern.info kernel: [10020.281370] br-vlan: port 5(eap5ghz) entered blocking state
Fri Aug  4 11:29:03 2023 kern.info kernel: [10020.287005] br-vlan: port 5(eap5ghz) entered forwarding state
Fri Aug  4 11:29:03 2023 daemon.notice hostapd: eap5ghz: interface state UNINITIALIZED->COUNTRY_UPDATE
Fri Aug  4 11:29:03 2023 daemon.notice hostapd: eap5ghz: interface state COUNTRY_UPDATE->HT_SCAN
Fri Aug  4 11:29:03 2023 daemon.err hostapd: hostapd_free_hapd_data: Interface eap5ghz wasn't started
Fri Aug  4 11:29:03 2023 kern.info kernel: [10020.485968] IPv6: ADDRCONF(NETDEV_CHANGE): eap5ghz: link becomes ready
Fri Aug  4 11:29:03 2023 daemon.notice netifd: Network device 'eap5ghz' link is up
Fri Aug  4 11:29:05 2023 daemon.notice hostapd: nl80211: deinit ifname=eap5ghz disabled_11b_rates=0
Fri Aug  4 11:29:05 2023 daemon.notice netifd: Network device 'eap5ghz' link is down
Fri Aug  4 11:29:05 2023 kern.info kernel: [10022.722630] br-vlan: port 5(eap5ghz) entered disabled state
```

rany2, 2023-08-04:
Well, it turns out the same thing sometimes happens for the first radio; the order doesn't matter.