openwrt/openwrt #13210: [IPQ807x] hostapd fails to start at boot for radio1

Soromeister, 2023-08-02:
### Describe the bug

When starting the router, hostapd fails to initialize radio1 with below error:

```
Tue Aug  1 13:44:11 2023 daemon.notice hostapd: Set new config for phy phy1: /var/run/hostapd-phy1.conf
Tue Aug  1 13:44:11 2023 daemon.notice hostapd: Restart interface for phy phy1
Tue Aug  1 13:44:11 2023 daemon.notice hostapd: Error loading config: access to undeclared variable data In find_reusable_wdev(), file /usr/share/hostap/common.uc, line 40, byte 20:   called from function wdev_create (/usr/share/hostap/common.uc:69:46)   called from function iface_restart (/usr/share/hostap/hostapd.uc:65:55)   called from function iface_set_config (/usr/share/hostap/hostapd.uc:212:46)   called from anonymous function (/usr/share/hostap/hostapd.uc:330:33)   `        { wiphy: phyidx });`   Near here ---------------^
Tue Aug  1 13:44:11 2023 daemon.notice netifd: radio1 (2691): Command failed: ubus call hostapd config_set { "phy": "phy1", "config":"/var/run/hostapd-phy1.conf", "prev_config": "/var/run/hostapd-phy1.conf.prev"} (Invalid argument)
Tue Aug  1 13:44:11 2023 daemon.notice netifd: radio1 (2691): Usage: ubus [<options>] <command> [arguments...]
Tue Aug  1 13:44:11 2023 daemon.notice netifd: radio1 (2691): Options:
Tue Aug  1 13:44:11 2023 daemon.notice netifd: radio1 (2691):  -s <socket>:		Set the unix domain socket to connect to
Tue Aug  1 13:44:11 2023 daemon.notice netifd: radio1 (2691):  -t <timeout>:		Set the timeout (in seconds) for a command to complete
Tue Aug  1 13:44:11 2023 daemon.notice netifd: radio1 (2691):  -S:			Use simplified output (for scripts)
Tue Aug  1 13:44:11 2023 daemon.notice netifd: radio1 (2691):  -v:			More verbose output
Tue Aug  1 13:44:11 2023 daemon.notice netifd: radio1 (2691):  -m <type>:		(for monitor): include a specific message type
Tue Aug  1 13:44:11 2023 daemon.notice netifd: radio1 (2691): 			(can be used more than once)
Tue Aug  1 13:44:11 2023 daemon.notice netifd: radio1 (2691):  -M <r|t>		(for monitor): only capture received or transmitted traffic
Tue Aug  1 13:44:11 2023 daemon.notice netifd: radio1 (2691):
Tue Aug  1 13:44:11 2023 daemon.notice netifd: radio1 (2691): Commands:
Tue Aug  1 13:44:11 2023 daemon.notice netifd: radio1 (2691):  - list [<path>]			List objects
Tue Aug  1 13:44:11 2023 daemon.notice netifd: radio1 (2691):  - call <path> <method> [<message>]	Call an object method
Tue Aug  1 13:44:11 2023 daemon.notice netifd: radio1 (2691):  - subscribe <path> [<path>...]	Subscribe to object(s) notifications
Tue Aug  1 13:44:11 2023 daemon.notice netifd: radio1 (2691):  - listen [<path>...]			Listen for events
Tue Aug  1 13:44:11 2023 daemon.notice netifd: radio1 (2691):  - send <type> [<message>]		Send an event
Tue Aug  1 13:44:11 2023 daemon.notice netifd: radio1 (2691):  - wait_for <object> [<object>...]	Wait for multiple objects to appear on ubus
Tue Aug  1 13:44:11 2023 daemon.notice netifd: radio1 (2691):  - monitor				Monitor ubus traffic
Tue Aug  1 13:44:11 2023 daemon.notice netifd: radio1 (2691):
Tue Aug  1 13:44:11 2023 daemon.notice netifd: radio1 (2691): Device setup failed: HOSTAPD_START_FAILED
Tue Aug  1 13:44:11 2023 daemon.notice hostapd: Set new config for phy phy2: /var/run/hostapd-phy2.conf
Tue Aug  1 13:44:11 2023 daemon.notice hostapd: Restart interface for phy phy2
Tue Aug  1 13:44:11 2023 daemon.notice netifd: Wireless device 'radio1' set retry=0
```

Restarting the device from Network > Wireless will successfully restart it.

### OpenWrt version

r23669-4a4e0c636f

### OpenWrt target/subtarget

qualcommax/ipq807x

### Device

Xiaomi AX9000

### Image kind

Official downloaded image

### Steps to reproduce

- Have same build;
- Restart router;
- Observe error in log;
- Restart radio1 from Network > Wireless (inside LuCi);

### Actual behaviour

radio1 fails to start with mentioned error from hostapd

### Expected behaviour

radio1 is supposed to start normally

### Additional info

N/A

### Diffconfig

```text
/bin/ash: ./scripts/diffconfig.sh: not found
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

Lyceris-chan, 2023-08-02:
just updated my mercusys MR70X and same issue here

closed 2023-08-02 by openwrt-bot

PolynomialDivision, 2023-08-02:
Please try again the latest main-branch:
https://github.com/openwrt/openwrt/commit/150e6d28f2659b0614e87cc7822e93f488e11ede

rany2, 2023-08-02:
@asvio I think it's caused by the recent hostapd/netifd changes, I'm experiencing this with mt7915, mt76x2 and ath10k chip as well so it's not specific to your device. Does not occur with e6b72de70367044260a6077bba3a91da463feaab. See https://github.com/openwrt/openwrt/issues/13206 for people with similar issues, your log message matches what I've experienced.

Soromeister, 2023-08-03:
> Please try again the latest main-branch: [150e6d2](https://github.com/openwrt/openwrt/commit/150e6d28f2659b0614e87cc7822e93f488e11ede)

Literally same behavior with build `r23676-57fda4b743` which encompasses [150e6d2](https://github.com/openwrt/openwrt/commit/150e6d28f2659b0614e87cc7822e93f488e11ede).