openwrt/openwrt #13213: mvebu: update to kernel 6.1

lynxis, 2023-08-02:
### Describe the bug

To collect all related bugs and have a discussion of the missing parts for kernel 6.1 for the mvebu target.
The first steps have been made https://github.com/openwrt/openwrt/pull/12938 to add 6.1 as testing kernel.

### OpenWrt version

master

### OpenWrt target/subtarget

mvebu

### Device

all

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

stklcode, 2023-08-02:
For reference, my key takeaways from the initial PR:

Running on a **CZ.NIC Turris Omnia** (cortexa9) with dual MT7915 radios (_mt7915e_ driver) for about a month (currently on 6.1.42) with quite a few packages and configs
=> no obvious regressions from 5.15 so far

Also running on a **Linksys WRT1900ACS** (cortexa9) as testing device.
=> Marvell WiFi drivers (**_mwlwifi_**) do not compile without additional work on the driver. See #12940 or the equivalent upstream proposal (kaloz/mwlwifi#415).

lynxis, 2023-08-03:
I'm running it on cortex a72 / clearfog cn9130 base (with additional patches).

tiagogaspar8, 2023-08-08:
I'm not being able to compile mwlwifi on kernel 6.1:
```
/mnt/openwrt/ss_mod/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/linux-mvebu_cortexa9/mwlwifi-2023-04-29-6a436714/hif/pcie/pcie.c:1296:9: error: unknown type name 'mm_segment_t'
 1296 |         mm_segment_t oldfs;
      |         ^~~~~~~~~~~~
/mnt/openwrt/ss_mod/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/linux-mvebu_cortexa9/mwlwifi-2023-04-29-6a436714/hif/pcie/pcie.c:1305:17: error: implicit declaration of function 'force_uaccess_begin'; did you mean 'user_access_begin'? [-Werror=implicit-function-declaration]
 1305 |         oldfs = force_uaccess_begin();
      |                 ^~~~~~~~~~~~~~~~~~~
      |                 user_access_begin
/mnt/openwrt/ss_mod/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/linux-mvebu_cortexa9/mwlwifi-2023-04-29-6a436714/hif/pcie/pcie.c:1329:9: error: implicit declaration of function 'force_uaccess_end'; did you mean 'file_accessed'? [-Werror=implicit-function-declaration]
 1329 |         force_uaccess_end(oldfs);
      |         ^~~~~~~~~~~~~~~~~
      |         file_accessed
/mnt/openwrt/ss_mod/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/linux-mvebu_cortexa9/mwlwifi-2023-04-29-6a436714/hif/pcie/pcie.c: In function 'pcie_probe':
/mnt/openwrt/ss_mod/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/linux-mvebu_cortexa9/mwlwifi-2023-04-29-6a436714/hif/pcie/pcie.c:1556:14: error: implicit declaration of function 'pci_set_dma_mask' [-Werror=implicit-function-declaration]
 1556 |         rc = pci_set_dma_mask(pdev, DMA_BIT_MASK(32));
      |              ^~~~~~~~~~~~~~~~
cc1: some warnings being treated as errors
make[5]: *** [scripts/Makefile.build:250: /mnt/openwrt/ss_mod/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/linux-mvebu_cortexa9/mwlwifi-2023-04-29-6a436714/hif/pcie/pcie.o] Error 1
make[4]: *** [Makefile:2012: /mnt/openwrt/ss_mod/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/linux-mvebu_cortexa9/mwlwifi-2023-04-29-6a436714] Error 2
make[4]: Leaving directory '/mnt/openwrt/ss_mod/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/linux-mvebu_cortexa9/linux-6.1.42'
make[3]: *** [Makefile:103: /mnt/openwrt/ss_mod/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/linux-mvebu_cortexa9/mwlwifi-2023-04-29-6a436714/.built] Error 2
make[3]: Leaving directory '/mnt/openwrt/ss_mod/package/kernel/mwlwifi'
time: package/kernel/mwlwifi/compile#3.17#0.48#3.71
    ERROR: package/kernel/mwlwifi failed to build.
make[2]: *** [package/Makefile:120: package/kernel/mwlwifi/compile] Error 1
make[2]: Leaving directory '/mnt/openwrt/ss_mod'
make[1]: *** [package/Makefile:114: /mnt/openwrt/ss_mod/staging_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/mnt/openwrt/ss_mod'
make: *** [/mnt/openwrt/ss_mod/include/toplevel.mk:232: world] Error 2
```

Has anyone seen this?

stklcode, 2023-08-08:
Yes, the mwlwifi drivers are not ready yet. See  #12940

tiagogaspar8, 2023-08-08:
Would you believe me if I told you that I actually searched all over the place for this? 🤦
These devices really are a pity...

PalebloodSky, 2023-08-11:
Been testing kernel 6.1 with my WRT32X. Mvebu hasn't worked this well since the OpenWrt 19.07/kernel 4.14 era. VLAN leaks are fixed (between removing that broken hack and new 6.1 fixes), the new mwlwifi driver works better with latency fixed (6.1), SQM performance is excellent (500Mbit down with SQM cake is no problem), DSA performance is great, rock solid stability, etc.

Only thing broken is software flow offloading. But that has been since DSA I think, not a big deal since most people use SQM.

Conforming mvebu is working great on kernel 6.1. Please switch mvebu to kernel to 6.1. Before 23.05 release would be ideal. Thanks to the devs for your efforts.

stklcode, 2023-08-11:
23.05 will be on 5.15 only and IIRC the VLAN leak is already fixed.
It’s in RC2 state already, so highly unlikely a yet experimental kernel update will be backported or anybody even has plans to maintain two kernels in a stable release.

6.1 is target for the next generation, likely 24.xx.

PedroM4, 2023-08-11:
what about the igmp issues that are fixed by removing the ARM-dts-turris-omnia-Add-switch-port-6-node patch? @stklcode did you have time to look into it?

PalebloodSky, 2023-08-11:
@stklcode Yea VLAN is mostly fixed in 5.15, but not entirely. 6.1 fixes the remaining issue as @champtar confirmed here https://github.com/openwrt/openwrt/pull/12878. I confirmed from my tests as well. Nevertheless thanks for confirming the process it's appreciated.

M95D, 2023-10-23:
OpenWrt r24154 with kernel v6.1.57 on mvebu/cortex-a9, Linksys WRT-1900ACv1. I built the kernel with io-uring and there is a problem with samba's uring VFS. I can't access any shares. Samba log:
```make_connection_snum: SMB_VFS_CONNECT for service 'Archive' at '/mnt/Archive' failed: Function not implemented```
If I remove /usr/lib/samba/vfs/io_uring.so, everything starts working again.

Can anyone confirm on a different platform?

trippleflux, 2023-10-23:
@M95D 
Please retry re-compiling Samba4 with my fixes posted in [here](https://forum.openwrt.org/t/samba-4-x-package-support-thread/10841/539). Would it work for your Samba4.

M95D, 2023-10-24:
No change; same error.
The patch was applied according to the build log, but the generated io_uring.so seems to be identical to the old one. I didn't compare the other files.

Neustradamus, 2023-11-18:
For information: Kernel 6.6 is LTS:
6.6 | Greg Kroah-Hartman & Sasha Levin | 2023-10-29 | Dec, 2026
- https://www.kernel.org/category/releases.html

EvilPingu, 2024-01-07:
Hi,

will Software based offloading working again? Currently it is broken i think.

EvilPingu, 2024-02-06:
> > Only thing broken is software flow offloading. But that has been since DSA I think, not a big deal since most people use SQM.
> 
> Are you sure this is broken in 5.15? I am using flow offloading (Turris Omnia - even with upload only SQM) and it seems to work perfectly fine. I get more CPU usage when saturating download without flow offload on. I would consider it quite a regression if that stopped working.

For me yes. If i active it than traffic to WAN is not possible from LAN. It's described here: https://github.com/openwrt/openwrt/pull/9582#issuecomment-1217231499

drujd, 2024-02-07:
> For me yes. If i active it than traffic to WAN is not possible from LAN. It's described here: [#9582 (comment)](https://github.com/openwrt/openwrt/pull/9582#issuecomment-1217231499)

What device do you have? I wonder if this affects only devices that use the mv88e6xxx switch for both LAN and WAN...

ynezz, 2024-02-18:
Done via #14665 (df95595b45267)

closed 2024-02-18 by ynezz