openwrt/openwrt #13216: mipsel_24kc: hostapd-common lacks required .uc files

alaviss, 2023-08-03:
### Describe the bug

Wireless couldn't start on a Xiaomi MiWifi Mini due to `/usr/share/hostap/wdev.uc` not being present in `hostapd-common`.

Logs:

```
Wed Aug  2 11:43:38 2023 daemon.notice netifd: radio0 (3591): Command failed: ubus call hostapd config_set { "phy": "phy0", "config":"/var/run/hostapd-phy0.conf", "prev_config": "/var/run/hostapd-phy0.conf.prev"} (Method not found)
Wed Aug  2 11:43:38 2023 daemon.notice netifd: radio0 (3591): Device setup failed: HOSTAPD_START_FAILED
Wed Aug  2 11:43:38 2023 daemon.notice netifd: Wireless device 'radio0' set retry=0
Wed Aug  2 11:43:38 2023 daemon.notice netifd: radio0 (3591): Failed to open "/usr/share/hostap/wdev.uc": No such file or directory
```

Some commands to verify that files are missing:

On MiWifi Mini (mipsel_24kc):

```
root@miwifi:~# opkg files wpad-openssl
Package wpad-openssl (2023-06-22-599d00be-1.2) is installed on root and has the following files:
/usr/sbin/wpad
/usr/sbin/hostapd
/usr/sbin/wpa_supplicant
root@miwifi:~# opkg files hostapd-common
Package hostapd-common (2023-06-22-599d00be-1.2) is installed on root and has the following files:
/etc/capabilities/wpad.json
/usr/share/acl.d/wpad_acl.json
/lib/netifd/hostapd.sh
/etc/init.d/wpad
/etc/rc.button/wps
/lib/netifd/dhcp-get-server.sh
```

On Linksys E8450 (aarch64_cortex-a53):

```
root@openwrt:~# opkg files wpad-openssl
Package wpad-openssl (2023-06-22-599d00be-1.2) is installed on root and has the following files:
/usr/sbin/wpad
/usr/sbin/hostapd
/etc/init.d/radius
/usr/sbin/hostapd-radius
/usr/share/hostap/wpa_supplicant.uc
/etc/radius/clients
/etc/config/radius
/usr/share/hostap/hostapd.uc
/etc/radius/users
/usr/sbin/wpa_supplicant
/lib/upgrade/keep.d/wpad-openssl
root@openwrt:~# opkg files hostapd-common
Package hostapd-common (2023-06-22-599d00be-1.2) is installed on root and has the following files:
/etc/capabilities/wpad.json
/usr/share/acl.d/wpad_acl.json
/lib/netifd/hostapd.sh
/etc/init.d/wpad
/etc/rc.button/wps
/lib/netifd/dhcp-get-server.sh
/usr/share/hostap/wdev.uc
/usr/share/hostap/common.uc
```

Downloading and unpacking https://downloads.openwrt.org/snapshots/packages/mipsel_24kc/base/hostapd-common_2023-06-22-599d00be-1.2_mipsel_24kc.ipk shows that the files are indeed not present in the package, and not because of a corrupted installation.

### OpenWrt version

r23670-150e6d28f2

### OpenWrt target/subtarget

ramips/mt7620

### Device

Xiaomi MiWiFi Mini

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

ascendbeing, 2023-08-03:
try flashing the normal sysupgrade of the snapshot. or use image builder. don't use ASU. I've had to do stuff like this before; basically 1-3 days ago there were major changes that the packages haven't been built against. so flash the normal snapshot pre built image, because that (and image builder if you know how to use it) with the snapshot images built-in wpa-basic-mbedtls, is then only way to go unless you're feeling adventurous or you build yourself
please let me know if it works for you. and remember remove anything that requires full version of wpad out of config/wireless.
because I downgraded to 23.05rc2 when this happened to me so I want to know does it work for you to do this?

rany2, 2023-08-04:
The file does exist for my mipsel_24kc device. Try a clean build.

alaviss, 2023-08-05:
Yep it's working now