openwrt/openwrt #13217: [IPQ807x] hostapd fails to start at boot for radio1

Soromeister, 2023-08-03:
### Describe the bug

When starting the router, hostapd fails to initialize radio1 with below error:

```
Wed Aug  2 17:06:41 2023 daemon.notice hostapd: Set new config for phy phy1: /var/run/hostapd-phy1.conf
Wed Aug  2 17:06:41 2023 daemon.notice hostapd: Restart interface for phy phy1
Wed Aug  2 17:06:41 2023 daemon.notice hostapd: Error loading config: access to undeclared variable data In find_reusable_wdev(), file /usr/share/hostap/common.uc, line 40, byte 20:   called from function wdev_create (/usr/share/hostap/common.uc:69:46)   called from function iface_restart (/usr/share/hostap/hostapd.uc:65:55)   called from function iface_set_config (/usr/share/hostap/hostapd.uc:212:46)   called from anonymous function (/usr/share/hostap/hostapd.uc:330:33)   `        { wiphy: phyidx });`   Near here ---------------^
Wed Aug  2 17:06:41 2023 daemon.notice netifd: radio1 (2704): Command failed: ubus call hostapd config_set { "phy": "phy1", "config":"/var/run/hostapd-phy1.conf", "prev_config": "/var/run/hostapd-phy1.conf.prev"} (Invalid argument)
Wed Aug  2 17:06:41 2023 daemon.notice netifd: radio1 (2704): Usage: ubus [<options>] <command> [arguments...]
Wed Aug  2 17:06:41 2023 daemon.notice netifd: radio1 (2704): Options:
Wed Aug  2 17:06:41 2023 daemon.notice netifd: radio1 (2704):  -s <socket>:		Set the unix domain socket to connect to
Wed Aug  2 17:06:41 2023 daemon.notice netifd: radio1 (2704):  -t <timeout>:		Set the timeout (in seconds) for a command to complete
Wed Aug  2 17:06:41 2023 daemon.notice netifd: radio1 (2704):  -S:			Use simplified output (for scripts)
Wed Aug  2 17:06:41 2023 daemon.notice netifd: radio1 (2704):  -v:			More verbose output
Wed Aug  2 17:06:41 2023 daemon.notice netifd: radio1 (2704):  -m <type>:		(for monitor): include a specific message type
Wed Aug  2 17:06:41 2023 daemon.notice netifd: radio1 (2704): 			(can be used more than once)
Wed Aug  2 17:06:41 2023 daemon.notice netifd: radio1 (2704):  -M <r|t>		(for monitor): only capture received or transmitted traffic
Wed Aug  2 17:06:41 2023 daemon.notice netifd: radio1 (2704):
Wed Aug  2 17:06:41 2023 daemon.notice netifd: radio1 (2704): Commands:
Wed Aug  2 17:06:41 2023 daemon.notice netifd: radio1 (2704):  - list [<path>]			List objects
Wed Aug  2 17:06:41 2023 daemon.notice netifd: radio1 (2704):  - call <path> <method> [<message>]	Call an object method
Wed Aug  2 17:06:41 2023 daemon.notice netifd: radio1 (2704):  - subscribe <path> [<path>...]	Subscribe to object(s) notifications
Wed Aug  2 17:06:41 2023 daemon.notice netifd: radio1 (2704):  - listen [<path>...]			Listen for events
Wed Aug  2 17:06:41 2023 daemon.notice netifd: radio1 (2704):  - send <type> [<message>]		Send an event
Wed Aug  2 17:06:41 2023 daemon.notice netifd: radio1 (2704):  - wait_for <object> [<object>...]	Wait for multiple objects to appear on ubus
Wed Aug  2 17:06:41 2023 daemon.notice netifd: radio1 (2704):  - monitor				Monitor ubus traffic
Wed Aug  2 17:06:41 2023 daemon.notice netifd: radio1 (2704):
Wed Aug  2 17:06:41 2023 daemon.notice netifd: radio1 (2704): Device setup failed: HOSTAPD_START_FAILED
Wed Aug  2 17:06:41 2023 daemon.notice netifd: Wireless device 'radio1' set retry=0
```

Restarting the device from Network > Wireless will successfully restart it.

This build includes [150e6d2](https://github.com/openwrt/openwrt/commit/150e6d28f2659b0614e87cc7822e93f488e11ede).

### OpenWrt version

r23676-57fda4b743

### OpenWrt target/subtarget

qualcommax/ipq807x

### Device

Xiaomi AX9000

### Image kind

Official downloaded image

### Steps to reproduce

- Have same build;
- Restart router;
- Observe error in log;
- Restart radio1 from Network > Wireless (inside LuCi);

### Actual behaviour

radio1 fails to start with mentioned error from hostapd

### Expected behaviour

radio1 is supposed to start normally

### Additional info

N/A

### Diffconfig

```text
/bin/ash: ./scripts/diffconfig.sh: not found
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

danpawlik, 2023-08-03:
Did you try to remove `/etc/config/wireless` and reboot and check how its working on clean settings?

Soromeister, 2023-08-03:
> Did you try to remove `/etc/config/wireless` and reboot and check how its working on clean settings?

I am using the package `wpad-mbedtls` instead of the standard `wpad-basic-mbedtls` since I require some of the advanced settings. I can certainly attempt to back up the wireless config and try to see whether things improve.

Here is my wireless config for the record:

```
config wifi-device 'radio0'
	option type 'mac80211'
	option path 'soc/10000000.pci/pci0001:00/0001:00:00.0/0001:01:00.0'
	option channel 'auto'
	option channels '100, 104, 108, 112, 116, 120, 124, 128, 132, 136, 140, 144'
	option band '5g'
	option htmode 'VHT80'
	option legacy_rates '0'
	option country 'US'
	option cell_density '3'
	option local_pwr_constraint '0'
	option txpower '35'
	option disabled '1'

config wifi-device 'radio1'
	option type 'mac80211'
	option path 'platform/soc/c000000.wifi'
	option channel 'auto'
	option channels '173, 169, 165, 161, 157, 153, 149'
	option band '5g'
	option htmode 'HE80'
	option legacy_rates '0'
	option he_bss_color '11'
	option he_su_beamformee '1'
	option country 'US'
	option cell_density '3'
	option local_pwr_constraint '0'
	option txpower '35'

config wifi-device 'radio2'
	option type 'mac80211'
	option path 'platform/soc/c000000.wifi+1'
	option channels '13, 11, 6, 1'
	option channel '11'
	option band '2g'
	option htmode 'HE20'
	option he_bss_color '22'
	option he_su_beamformee '1'
	option country 'US'
	option cell_density '3'
	option local_pwr_constraint '0'
	option txpower '15'

config wifi-device 'radio3'
	option type 'mac80211'
	option path 'soc/20000000.pci/pci0000:00/0000:00:00.0/0000:01:00.0'
	option channels '165, 161, 157, 153, 149, 64, 60, 56, 52, 48, 44, 40, 36'
	option channel '36'
	option band '5g'
	option htmode 'HE160'
	option legacy_rates '0'
	option he_bss_color '33'
	option he_su_beamformee '1'
	option country 'US'
	option cell_density '3'
	option local_pwr_constraint '0'
	option txpower '35'

config wifi-iface 'guest_radio0'
	option device 'radio0'
	option network 'lan'
	option mode 'ap'
	option ssid 'router guest'
	option encryption 'sae-mixed'
	option isolate '1'
	option ifname 'r0-guest'
	option key 'myguestpassword'
	option skip_inactivity_poll '1'
	option disassoc_low_ack '0'
	option disabled '1'

config wifi-iface 'default_radio1'
	option device 'radio1'
	option network 'lan'
	option mode 'ap'
	option ssid 'router'
	option encryption 'sae-mixed'
	option isolate '1'
	option ifname 'r1-router'
	option key 'mypassword'
	option skip_inactivity_poll '1'
	option disassoc_low_ack '0'

config wifi-iface 'default_radio2'
	option device 'radio2'
	option network 'lan'
	option mode 'ap'
	option ssid 'router'
	option encryption 'sae-mixed'
	option isolate '1'
	option ifname 'r2-router'
	option key 'mypassword'
	option skip_inactivity_poll '1'
	option disassoc_low_ack '0'

config wifi-iface 'default_radio3'
	option device 'radio3'
	option network 'lan'
	option mode 'ap'
	option ssid 'router (G)'
	option encryption 'sae-mixed'
	option ifname 'r3-router'
	option key 'mypassword'
	option skip_inactivity_poll '1'
	option disassoc_low_ack '0'
```

Soromeister, 2023-08-04:
Fixed by [b8be20c](https://github.com/openwrt/openwrt/commit/b8be20c7e81de2894df8fa2b361c39bc723e4cb5)

closed 2023-08-04 by Soromeister