openwrt/openwrt #13224: Possible Build Regression: act-pedit removed from kernel configs, not built as a module.

barberio, 2023-08-04:
### Describe the bug

At some point between now and 2019, CONFIG_NET_ACT_PEDIT was set to n in kernel configs. This obviously results in it not being included in any build or any module build. Builds prior to 2019 made act-pedit part of the core networking schedule module package.

This is a very useful Traffic Control Action, and is primarily useful in allowing ECN marking on ingress without using a full-fat in software mirred copying of traffic to a virtual device. I would hopefully like to see it put back in, so my traffic control scripts start working again.

### OpenWrt version

r23314-7efec0acca

### OpenWrt target/subtarget

mediatek/mt7622

### Device

Netgear WAX206

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

dave14305, 2023-08-05:
pedit is available in kmod-sched.

https://github.com/openwrt/openwrt/blob/7e7eb5312d7810084547bb54a4b6867c2da08182/package/kernel/linux/modules/netsupport.mk#L1000-L1021

barberio, 2023-08-06:
> pedit is available in kmod-sched.
> 
> https://github.com/openwrt/openwrt/blob/7e7eb5312d7810084547bb54a4b6867c2da08182/package/kernel/linux/modules/netsupport.mk#L1000-L1021

My Apologies. I had a brainfart when typing up the bug report and substituted pedit for skbmod. (I'm having to use pedit instead of skbmod to do ECN marking, which is obviously a problem for IPV6 support.) skbmod is absent entirely from the netsupport modules.

commit e0ea79f3 (2023-08-15): kernel: netsupport: kmod-sched: Add act_skbmod

This adds act_skbmod.ko to kmod-sched.

Fixes: #13224
PR #13282 (2023-08-15): kernel: netsupport: kmod-sched: Add act_skbmod
This adds act_skbmod.ko to kmod-sched.

Fixes: #13224
closed 2023-08-15 by openwrt-bot