openwrt/openwrt #13229: Cannot build firmware for banana pi r3

ChaingTsung, 2023-08-05:
### Describe the bug

- Failed to Build for banana pi r3 firmware

```log
...
DEVICE_ID="bananapi_bpi-r3" SOURCE_DATE_EPOCH=1691102062 FILE_NAME="openwrt-mediatek-filogic-bananapi_bpi-r3-emmc-bl31-uboot.fip" FILE_DIR="/home/yoki/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/tmp" FILE_TYPE="emmc-bl31-uboot.fip" DEVICE_IMG_PREFIX="openwrt-mediatek-filogic-bananapi_bpi-r3" DEVICE_VENDOR="Bananapi" DEVICE_MODEL="BPi-R3" DEVICE_VARIANT="" DEVICE_ALT0_VENDOR="" DEVICE_ALT0_MODEL="" DEVICE_ALT0_VARIANT="" DEVICE_ALT1_VENDOR="" DEVICE_ALT1_MODEL="" DEVICE_ALT1_VARIANT="" DEVICE_ALT2_VENDOR="" DEVICE_ALT2_MODEL="" DEVICE_ALT2_VARIANT="" DEVICE_ALT3_VENDOR="" DEVICE_ALT3_MODEL="" DEVICE_ALT3_VARIANT="" DEVICE_ALT4_VENDOR="" DEVICE_ALT4_MODEL="" DEVICE_ALT4_VARIANT="" DEVICE_ALT5_VENDOR="" DEVICE_ALT5_MODEL="" DEVICE_ALT5_VARIANT="" DEVICE_TITLE="Bananapi BPi-R3" DEVICE_PACKAGES="kmod-hwmon-pwmfan kmod-i2c-gpio kmod-mt7986-firmware kmod-sfp kmod-usb3 e2fsprogs f2fsck mkf2fs mt7986-wo-firmware" TARGET="mediatek" SUBTARGET="filogic" VERSION_NUMBER="SNAPSHOT" VERSION_CODE="r23682-1eb67cb070" SUPPORTED_DEVICES="bananapi,bpi-r3" /home/yoki/openwrt/scripts/json_add_image_info.py /home/yoki/openwrt/build_dir/target-aarch64_cortex-a53_musl/json_info_files/openwrt-mediatek-filogic-bananapi_bpi-r3-emmc-bl31-uboot.fip.json
cat /home/yoki/openwrt/staging_dir/target-aarch64_cortex-a53_musl/image/mt7986-nor-ddr4-bl2.img >> /home/yoki/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/tmp/openwrt-mediatek-filogic-bananapi_bpi-r3-nor-preloader.bin
cp /home/yoki/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/tmp/openwrt-mediatek-filogic-bananapi_bpi-r3-nor-preloader.bin /home/yoki/openwrt/bin/targets/mediatek/filogic/openwrt-mediatek-filogic-bananapi_bpi-r3-nor-preloader.bin
DEVICE_ID="bananapi_bpi-r3" SOURCE_DATE_EPOCH=1691102062 FILE_NAME="openwrt-mediatek-filogic-bananapi_bpi-r3-nor-preloader.bin" FILE_DIR="/home/yoki/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/tmp" FILE_TYPE="nor-preloader.bin" DEVICE_IMG_PREFIX="openwrt-mediatek-filogic-bananapi_bpi-r3" DEVICE_VENDOR="Bananapi" DEVICE_MODEL="BPi-R3" DEVICE_VARIANT="" DEVICE_ALT0_VENDOR="" DEVICE_ALT0_MODEL="" DEVICE_ALT0_VARIANT="" DEVICE_ALT1_VENDOR="" DEVICE_ALT1_MODEL="" DEVICE_ALT1_VARIANT="" DEVICE_ALT2_VENDOR="" DEVICE_ALT2_MODEL="" DEVICE_ALT2_VARIANT="" DEVICE_ALT3_VENDOR="" DEVICE_ALT3_MODEL="" DEVICE_ALT3_VARIANT="" DEVICE_ALT4_VENDOR="" DEVICE_ALT4_MODEL="" DEVICE_ALT4_VARIANT="" DEVICE_ALT5_VENDOR="" DEVICE_ALT5_MODEL="" DEVICE_ALT5_VARIANT="" DEVICE_TITLE="Bananapi BPi-R3" DEVICE_PACKAGES="kmod-hwmon-pwmfan kmod-i2c-gpio kmod-mt7986-firmware kmod-sfp kmod-usb3 e2fsprogs f2fsck mkf2fs mt7986-wo-firmware" TARGET="mediatek" SUBTARGET="filogic" VERSION_NUMBER="SNAPSHOT" VERSION_CODE="r23682-1eb67cb070" SUPPORTED_DEVICES="bananapi,bpi-r3" /home/yoki/openwrt/scripts/json_add_image_info.py /home/yoki/openwrt/build_dir/target-aarch64_cortex-a53_musl/json_info_files/openwrt-mediatek-filogic-bananapi_bpi-r3-nor-preloader.bin.json
cat /home/yoki/openwrt/staging_dir/target-aarch64_cortex-a53_musl/image/mt7986_bananapi_bpi-r3-nor-u-boot.fip >> /home/yoki/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/tmp/openwrt-mediatek-filogic-bananapi_bpi-r3-nor-bl31-uboot.fip
cp /home/yoki/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/tmp/openwrt-mediatek-filogic-bananapi_bpi-r3-nor-bl31-uboot.fip /home/yoki/openwrt/bin/targets/mediatek/filogic/openwrt-mediatek-filogic-bananapi_bpi-r3-nor-bl31-uboot.fip
DEVICE_ID="bananapi_bpi-r3" SOURCE_DATE_EPOCH=1691102062 FILE_NAME="openwrt-mediatek-filogic-bananapi_bpi-r3-nor-bl31-uboot.fip" FILE_DIR="/home/yoki/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/tmp" FILE_TYPE="nor-bl31-uboot.fip" DEVICE_IMG_PREFIX="openwrt-mediatek-filogic-bananapi_bpi-r3" DEVICE_VENDOR="Bananapi" DEVICE_MODEL="BPi-R3" DEVICE_VARIANT="" DEVICE_ALT0_VENDOR="" DEVICE_ALT0_MODEL="" DEVICE_ALT0_VARIANT="" DEVICE_ALT1_VENDOR="" DEVICE_ALT1_MODEL="" DEVICE_ALT1_VARIANT="" DEVICE_ALT2_VENDOR="" DEVICE_ALT2_MODEL="" DEVICE_ALT2_VARIANT="" DEVICE_ALT3_VENDOR="" DEVICE_ALT3_MODEL="" DEVICE_ALT3_VARIANT="" DEVICE_ALT4_VENDOR="" DEVICE_ALT4_MODEL="" DEVICE_ALT4_VARIANT="" DEVICE_ALT5_VENDOR="" DEVICE_ALT5_MODEL="" DEVICE_ALT5_VARIANT="" DEVICE_TITLE="Bananapi BPi-R3" DEVICE_PACKAGES="kmod-hwmon-pwmfan kmod-i2c-gpio kmod-mt7986-firmware kmod-sfp kmod-usb3 e2fsprogs f2fsck mkf2fs mt7986-wo-firmware" TARGET="mediatek" SUBTARGET="filogic" VERSION_NUMBER="SNAPSHOT" VERSION_CODE="r23682-1eb67cb070" SUPPORTED_DEVICES="bananapi,bpi-r3" /home/yoki/openwrt/scripts/json_add_image_info.py /home/yoki/openwrt/build_dir/target-aarch64_cortex-a53_musl/json_info_files/openwrt-mediatek-filogic-bananapi_bpi-r3-nor-bl31-uboot.fip.json
cp /home/yoki/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/tmp/openwrt-mediatek-filogic-bananapi_bpi-r3-sdcard.img.gz /home/yoki/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/tmp/openwrt-mediatek-filogic-bananapi_bpi-r3-sdcard.img.gz.tmp 2>/dev/null || true
ptgen -g -o /home/yoki/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/tmp/openwrt-mediatek-filogic-bananapi_bpi-r3-sdcard.img.gz.tmp -a 1 -l 1024  -H -t 0x83  -N bl2          -r      -p 4079k@17k  -t 0x83 -N ubootenv      -r      -p 512k@4M -t 0x83      -N factory      -r      -p 2M@4608k -t 0xef     -N fip          -r      -p 4M@6656k -N recovery -r      -p 32M@12M  -N install  -r      -p 20M@44M -t 0x2e -N production              -p 4096M@64M
part 17 4079
part 4096 512
part 4608 2048
part 6656 4096
part 12288 32768
part 45056 20480
part 65536 4194304
17408
4176896
4194304
524288
4718592
2097152
6815744
4194304
12582912
33554432
46137344
20971520
67108864
4294967296
cat /home/yoki/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/tmp/openwrt-mediatek-filogic-bananapi_bpi-r3-sdcard.img.gz.tmp >> /home/yoki/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/tmp/openwrt-mediatek-filogic-bananapi_bpi-r3-sdcard.img.gz
rm /home/yoki/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/tmp/openwrt-mediatek-filogic-bananapi_bpi-r3-sdcard.img.gz.tmp
dd if=/home/yoki/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/tmp/openwrt-mediatek-filogic-bananapi_bpi-r3-sdcard.img.gz of=/home/yoki/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/tmp/openwrt-mediatek-filogic-bananapi_bpi-r3-sdcard.img.gz.new bs=17k conv=sync
1+0 records in
1+0 records out
17408 bytes (17 kB, 17 KiB) copied, 0.000334408 s, 52.1 MB/s
mv /home/yoki/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/tmp/openwrt-mediatek-filogic-bananapi_bpi-r3-sdcard.img.gz.new /home/yoki/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/tmp/openwrt-mediatek-filogic-bananapi_bpi-r3-sdcard.img.gz
cat /home/yoki/openwrt/staging_dir/target-aarch64_cortex-a53_musl/image/mt7986-sdmmc-ddr4-bl2.img >> /home/yoki/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/tmp/openwrt-mediatek-filogic-bananapi_bpi-r3-sdcard.img.gz
dd if=/home/yoki/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/tmp/openwrt-mediatek-filogic-bananapi_bpi-r3-sdcard.img.gz of=/home/yoki/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/tmp/openwrt-mediatek-filogic-bananapi_bpi-r3-sdcard.img.gz.new bs=6656k conv=sync
0+1 records in
1+0 records out
6815744 bytes (6.8 MB, 6.5 MiB) copied, 0.0241472 s, 282 MB/s
mv /home/yoki/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/tmp/openwrt-mediatek-filogic-bananapi_bpi-r3-sdcard.img.gz.new /home/yoki/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/tmp/openwrt-mediatek-filogic-bananapi_bpi-r3-sdcard.img.gz
cat /home/yoki/openwrt/staging_dir/target-aarch64_cortex-a53_musl/image/mt7986_bananapi_bpi-r3-sdmmc-u-boot.fip >> /home/yoki/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/tmp/openwrt-mediatek-filogic-bananapi_bpi-r3-sdcard.img.gz
dd if=/home/yoki/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/tmp/openwrt-mediatek-filogic-bananapi_bpi-r3-sdcard.img.gz of=/home/yoki/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/tmp/openwrt- [... cut ...]
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:30: target/linux/install] Error 1
make[2]: Leaving directory '/home/yoki/openwrt'
make[1]: *** [target/Makefile:24: /home/yoki/openwrt/staging_dir/target-aarch64_cortex-a53_musl/stamp/.target_install] Error 2
make[1]: Leaving directory '/home/yoki/openwrt'
make: *** [/home/yoki/openwrt/include/toplevel.mk:232: world] Error 2
yoki@yoki-ubuntu:~/openwrt$ make menuconfig

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

ChaingTsung, 2023-08-05:
I used `make menuconfig` to make the Root Partition increase to 4G.However, error still occurred.

ChaingTsung, 2023-08-05:
The branch of openwrt source is ` main`.

ChaingTsung, 2023-08-07:
[seeds.zip](https://github.com/openwrt/openwrt/files/12273859/seeds.zip)
config file

hauke, 2023-08-14:
Does it build when you use a minimal configuration like:
```
CONFIG_TARGET_mediatek=y
CONFIG_TARGET_mediatek_filogic=y
CONFIG_TARGET_mediatek_filogic_DEVICE_bananapi_bpi-r3=y
```

Maybe there is a bug with triggered by one of your configuration options or just by the size of the image.

SaigyoujiYuyuko233, 2024-02-20:
Exact same error when building for PBI r4.
Disable ramdisk will fix the issue, not sure why
[image]
Reference here: https://forum.banana-pi.org/t/failed-to-build-firmware-for-banana-pi-r3p-on-lede-coode-branch-master/16040/19