openwrt/openwrt #13230: msm8996.dsti problem occurred when compiling 6.1 kernel under ax6！

zxlhhyccc, 2023-08-05:
### Describe the bug

[image]

### OpenWrt version

master branch

### OpenWrt target/subtarget

main

### Device

AX6: 6.1 kernel

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

robimarko, 2023-08-05:
This is completely irrelevant to OpenWrt, that SoC isn't even supported.
Its an upstream issue and just a warning in the end.

zxlhhyccc, 2023-08-06:
> This is completely irrelevant to OpenWrt, that SoC isn't even supported. Its an upstream issue and just a warning in the end.

This warning was not available in the previous 5.15 kernel!

robimarko, 2023-08-06:
But again, its completely irrelevant as the SoC is not used at all, why bother patch it?

zxlhhyccc, 2023-08-06:
> But again, its completely irrelevant as the SoC is not used at all, why bother patch it?

Mainly compilation time to see uncomfortable!

robimarko, 2023-08-06:
Thats not a good reason, we are not going to carry patches for unused SoC-s.

hauke, 2023-08-13:
Please check if this is also a problem in upstream kernel and try to get it fixed there.

This SoC is not used in OpenWrt, so this warning is not relevant for OpenWrt.

closed 2023-08-13 by hauke