openwrt/openwrt #13239: bpi-r3 uboot does not load the eMMC recovery image

lu-zero, 2023-08-07:
### Describe the bug

Using 2023.04 from the 23.05 branch:
```
        ( ( ( OpenWrt ) ) )  [eMMC]       U-Boot 2023.04-OpenWrt-r23312+2-e0c4da1ff0 (Jul 30 2023 - 11:12:40 +0000)

      1. Run default boot command.
      2. Boot system via TFTP.
      3. Boot production system from eMMC.
      4. Boot recovery system from eMMC.
      5. Load production system via TFTP then write to eMMC.
      6. Load recovery system via TFTP then write to eMMC.
      7. Load BL31+U-Boot FIP via TFTP then write to eMMC.
      8. Load BL2 preloader via TFTP then write to eMMC.
      9. Reboot.
      a. Reset all settings to factory defaults.
      0. U-Boot console

  Press UP/DOWN to move, ENTER to select, ESC to quit
```
Installing everything using the options 8, 7, 6 and 5 in sequence.

If I try to run recovery via 4 I get:

```
* spi-nand0
  - device: spi_nand@1
  - parent: spi@1100a000
  - driver: spi_nand
  - type: NAND flash
  - block size:        0x20000 bytes
  - page size:         0x800 bytes
  - OOB size:          64 bytes
  - OOB available:     24 bytes
  - 0x000000000000-0x000008000000 : "spi-nand0"
          - 0x000000000000-0x000000080000 : "bl2"
          - 0x000000080000-0x000000380000 : "factory"
          - 0x000000380000-0x000000580000 : "fip"
          - 0x000000580000-0x000008000000 : "ubi"

MMC read: dev # 0, block # 24576, count 8 ... 8 blocks read: OK
Press ENTER to return to menu
```

And it returns to the uboot menu.

### OpenWrt version

e0c4da1ff0c00f2c4d78025523c38ea6395175d9

### OpenWrt target/subtarget

mediatek/filogic

### Device

Bananapi BPI-R3

### Image kind

Self-built image

### Steps to reproduce

Install by some mean the build, try to boot from the recovery partition.

### Actual behaviour

uboot ask you to press enter and return to the boot menu.

### Expected behaviour

Boot to recovery

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

lu-zero, 2023-08-09:
Tested also with `U-Boot 2023.07.02-OpenWrt-r23690-d198c77764 (Aug 09 2023 - 08:51:46 +0000)` from the snapshot repository. Same behaviour:

- load via tftp succeeds, I can manage to boot recovery after that, after a reboot it fails.

lu-zero, 2023-08-09:
the first time after the tftp update:
```
MT7986> part start mmc 0 $part_recovery part_addr && part size mmc 0 $part_recovery part_size
MT7986> echo $part_addr $part_size
6000 10000
MT7986> imszb $loadaddr image_size && echo ok
ok
MT7986> echo $loadaddr
0x46000000
MT7986> echo $image_size
36b4
```
After booting the image:
```
MT7986> part start mmc 0 $part_recovery part_addr && part size mmc 0 $part_recovery part_size
MT7986> echo $part_addr $part_size
6000 10000
MT7986> echo $loadaddr
0x46000000
MT7986> imszb $loadaddr image_size && echo ok
MT7986>
```
imszb fails.

But if I manually set `image_size` to 36b4 the image correctly boots.
```
MT7986> mmc read $loadaddr 0x$part_addr 0x$image_size
MT7986> setexpr filesize $image_size * 0x200
MT7986> bootm $loadaddr#$bootconf

## Loading kernel from FIT Image at 46000000 ...
   Using 'config-mt7986a-bananapi-bpi-r3' configuration
   Trying 'kernel-1' kernel subimage
     Description:  ARM64 OpenWrt Linux-5.15.123
     Type:         Kernel Image
     Compression:  lzma compressed
     Data Start:   0x460000ec
     Data Size:    3986881 Bytes = 3.8 MiB
     Architecture: AArch64
     OS:           Linux
     Load Address: 0x44000000
     Entry Point:  0x44000000
     Hash algo:    crc32
     Hash value:   bacb5029
     Hash algo:    sha1
     Hash value:   9c55d56cb91cdc966032485048491ffcafe6a0c6
...
```

lu-zero, 2023-08-10:
2023.04-OpenWrt-r23228-cd17d8df2a from the 23.05-rc2 repo seems working fine.

lu-zero, 2023-10-25:
I confirm that's still a problem in 23.05 and reverting 5bd5be03bb5a80338824f7bc9223261a5a14f59d fixes the problem.

hohle, 2023-11-10:
I'm trying to move an installation from NAND to eMMC on a BPI-R3 and the [installation instructions on the wiki](https://openwrt.org/toh/sinovoip/bananapi_bpi_r3_v1.0_v1.1#installation) (to set the boot command to run `emmc_init`) appear to have no effect. Is this the same root cause?