openwrt/openwrt #13242: Samsung A54 on 5G connecting breaks TP Link Archer C7 V2 Wifi

hhasert, 2023-08-07:
### Describe the bug

When the Samsung A54 connects to the 5G wifi DFS channel there is no connection established and the 5G channel completely blocks. In the log you see:

kern.warn kernel: [ 2209.211827] ath10k_pci 0000:00:00.0: too short radar pulse summary (344)

The only way to get it connected is to configure the interface to ignore radar detection, and then it happily connects. I have never seen this behavior with other 5G devices before.

### OpenWrt version

r20134-5f15225c1e

### OpenWrt target/subtarget

ath79/generic

### Device

TP-Link Archer C7 v2

### Image kind

Official downloaded image

### Steps to reproduce

Install OpenWrt 22.03.5 on TP Link Archer C7 V2, configure channel 136 for 5G radio0, configure on WPA2 (strong security). Try to connect with a Samsung S9 (success), try to connect with a Samsung A54 (no success, radar detected, Wifi goes down). The 

### Actual behaviour

Wifi goes down on connect and device has to be rebooted to get things up and running again

### Expected behaviour

Successfull connection (no radar detection)

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

hhasert, 2023-08-07:
On second thought, It might be caused by the phone trying to find nearby devices.... Experimenting

brada4, 2023-08-07:
5G and 5GHz are different things.
What is in your `/etc/config/wireless` namely around regulatory country setting for ath10k radio?

hhasert, 2023-08-07:
`config wifi-device 'radio0'
        option type 'mac80211'
        option path 'pci0000:00/0000:00:00.0'
        option channel '136'
        option band '5g'
        option cell_density '2'
        option htmode 'VHT80'
        option country 'SG'
        option txpower '27'

config wifi-iface 'default_radio0'
        option device 'radio0'
        option network 'lan'
        option mode 'ap'
        option ssid 'Hasert-Boven-5G'
        option key 'KyLiXMythago41'
        option ieee80211w '0'
        option disassoc_low_ack '0'
        option encryption 'psk2+ccmp'
`

brada4, 2023-08-07:
Please set to etsi where you are and auto channel to let DFS and ACS do its job.

It is DFS channel in SG reg dom and DFS is inaccurate outside ETSI on that card.
https://github.com/openwrt/openwrt/issues/13231#issuecomment-1667855411

hhasert, 2023-08-07:
Hmm, I will have a look but my region only supports a very low power setting for that channel, any country you can suggest that would be inside ETSI and still supports some xmit power ?

brada4, 2023-08-07:
It is not radar pulse - your client sent too short radar status report, because it is in different region like easter island.

brada4, 2023-08-07:
Special from Samsung - tap 10x on intelligent wifi version number, new diag menu decodes info elements from nearbya AP-s.

hhasert, 2023-08-08:
> It is not radar pulse - your client sent too short radar status report, because it is in different region like easter island.

I moved the country to NL, is there something I should change on the Samsung A54 ?

brada4, 2023-08-08:
Yes, it has to be in NL, like you see in one of diag screens
```
Country code
AP: NL
STA: NL
```
Anything in EU and closely around will work same.

Also rising debug levels in hostapd.conf may shine some light.
https://w1.fi/cgit/hostap/tree/hostapd/hostapd.conf#n119

Also check samsungs docs - thay say to try connecting in "safe mode" etc.

hhasert, 2023-08-08:
It says AP US and STA NL.  So I need to find where to change that.

brada4, 2023-08-08:
You need to set country code in each radio, i.e twice, for 2.4 and for 5ghz.

hhasert, 2023-08-08:
> You need to set country code in each radio, i.e twice, for 2.4 and for 5ghz.

I did.

hhasert, 2023-08-08:
I set the log level to 0, but nothing new is logged (Luci syslog) and I get the radar pulses after it connects as usual.

brada4, 2023-08-08:
C7 v5 works with galaxy A53 (non-ax device) and shows BE/BE country codes as intended,  and is capable to connect to DE/NL/FR set as AP CC and immediately reflected in intelligent interface after reconnect for hostapd reload.

hhasert, 2023-09-24:
Funny thing is, I have set both interfaces to NL, but if I perform iw reg get, I end up with : 
global
country NL: DFS-ETSI
..
phy#1
country US: DFS-FCC
..
phy#0
country US: DFS-FCC

brada4, 2023-09-24:
Yes. But i get BE/BE in phone's diag menu w same display from reg get.

hhasert, 2023-09-25:
I get NL/NL as well, weird though that the physical interfaces say different, must be a chip/driver thingy then.

brada4, 2023-09-25:
That is normal occurence. It works correctly only with ETSI radar detection https://wireless.wiki.kernel.org/en/users/Drivers/ath10k/configuration
Do you still get short radar report in logs when all software is ETSI?

hhasert, 2023-09-25:
I did not build my own driver, just used the one in the package. The phone reports AP: NL, STA:NL but I still get the same error and suppress it by disabling radar detection.

brada4, 2023-09-25:
There are 2 kinds of drivers - mainline (without -ct) and candelatech (with -ct at the end), same with firmwares. You can try swapping them, though not much hope in that, radar code handling is same.  reporting may differ though. I get genuine radar pulses and forced channel switch every now and then and week or to in uptime landing at CH36 or 44...

hhasert, 2023-09-25:
I am at channel 136 and there is nothing using radar here, so I am fine with disabling radar detection to solve this issue, never see actual radar pulses detected, only the too smaal ones, so I leave it, just need to remind myself every boot or update to run the script to disable it.