openwrt/openwrt #13245: [23.05] Redmi AX6 has issues with 5 GHz band

vadjs, 2023-08-08:
### Describe the bug

There are many of my devices are struggling to connect to 5 GHz band on Redmin AX6 with 23.05.0-rc2 firmware. It causes that any of device that uses WiFi 5/6 is using 2.4 GHz band. It happens in ~80% cases.
Technically 5 GHz band works ind if you turn off the 2.4 GHz band at all, device will connect to 5 GHz and use it. 
It wasn't the case when I had a nightly build(mid May). 

### OpenWrt version

r23228-cd17d8df2a

### OpenWrt target/subtarget

ipq807x/generic

### Device

Redmi AX6

### Image kind

Official downloaded image

### Steps to reproduce

- Setup the WiFi hotspot
- Connect any device using WiFi 5/6(better 5) to WiFi

### Actual behaviour

The device uses only 2.4 GHz band in 80% cases.

### Expected behaviour

By default all devices prefer using 5 GHz band(or both on WiFi 6) which is not the case on the current build.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

raenye, 2023-09-14:
@brada4 I use `dawn`; is `usteer` better?

brada4, 2023-09-14:
usteer is smaller, and you dont need to edit each ap-s configuration past replacing wpad-basic-framework with wpad-framework.