openwrt/openwrt #13248: r8169 ASPM issues on x86

devastgh, 2023-08-09:
### Describe the bug

r8169 has long standing issues with pcie link aspm.  Speed is very slow for rx. Currently the known workarounds are
- enabling flow control on the other end of the cable OR
- disabling aspm in the bios of the device for that pcie link, if the configuration is exposed OR
- disabling aspm in sysfs for the pcie device 

In the current x86 target pcie aspm controls are not exposed in sysfs.
Please add
```
CONFIG_PCIEASPM=y
CONFIG_PCIEASPM_DEFAULT=y
```
options to the x86 target kernel.

### OpenWrt version

r23228-cd17d8df2a

### OpenWrt target/subtarget

x86/64

### Device

Default string Default string

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-08-09:
Could you please add relevant `lspci -vvv` output.

devastgh, 2023-08-10:
> Could you please add relevant `lspci -vvv` output.

`02:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller (rev 15)
        Subsystem: Realtek Semiconductor Co., Ltd. Device 0123
        Control: I/O+ Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx+
        Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
        Latency: 0, Cache Line Size: 64 bytes
        Interrupt: pin A routed to IRQ 19
        Region 0: I/O ports at 3000 [size=256]
        Region 2: Memory at 80404000 (64-bit, non-prefetchable) [size=4K]
        Region 4: Memory at 80400000 (64-bit, non-prefetchable) [size=16K]
        Capabilities: [40] Power Management version 3
                Flags: PMEClk- DSI- D1+ D2+ AuxCurrent=375mA PME(D0+,D1+,D2+,D3hot+,D3cold+)
                Status: D0 NoSoftRst+ PME-Enable- DSel=0 DScale=0 PME-
        Capabilities: [50] MSI: Enable- Count=1/1 Maskable- 64bit+
                Address: 0000000000000000  Data: 0000
        Capabilities: [70] Express (v2) Endpoint, MSI 01
                DevCap: MaxPayload 128 bytes, PhantFunc 0, Latency L0s <512ns, L1 <64us
                        ExtTag- AttnBtn- AttnInd- PwrInd- RBE+ FLReset- SlotPowerLimit 10W
                DevCtl: CorrErr- NonFatalErr- FatalErr- UnsupReq-
                        RlxdOrd+ ExtTag- PhantFunc- AuxPwr- NoSnoop-
                        MaxPayload 128 bytes, MaxReadReq 4096 bytes
                DevSta: CorrErr- NonFatalErr- FatalErr- UnsupReq- AuxPwr+ TransPend+
                LnkCap: Port #0, Speed 2.5GT/s, Width x1, ASPM L0s L1, Exit Latency L0s unlimited, L1 <64us
                        ClockPM+ Surprise- LLActRep- BwNot- ASPMOptComp+
                LnkCtl: ASPM L1 Enabled; RCB 64 bytes, Disabled- CommClk+
                        ExtSynch- ClockPM+ AutWidDis- BWInt- AutBWInt-
                LnkSta: Speed 2.5GT/s, Width x1
                        TrErr- Train- SlotClk+ DLActive- BWMgmt- ABWMgmt-
                DevCap2: Completion Timeout: Range ABCD, TimeoutDis+ NROPrPrP- LTR+
                         10BitTagComp- 10BitTagReq- OBFF Via message/WAKE#, ExtFmt- EETLPPrefix-
                         EmergencyPowerReduction Not Supported, EmergencyPowerReductionInit-
                         FRS- TPHComp- ExtTPHComp-
                         AtomicOpsCap: 32bit- 64bit- 128bitCAS-
                DevCtl2: Completion Timeout: 50us to 50ms, TimeoutDis- LTR+ 10BitTagReq- OBFF Disabled,
                         AtomicOpsCtl: ReqEn-
                LnkCap2: Supported Link Speeds: 2.5GT/s, Crosslink- Retimer- 2Retimers- DRS-
                LnkCtl2: Target Link Speed: 2.5GT/s, EnterCompliance- SpeedDis-
                         Transmit Margin: Normal Operating Range, EnterModifiedCompliance- ComplianceSOS-
                         Compliance Preset/De-emphasis: -6dB de-emphasis, 0dB preshoot
                LnkSta2: Current De-emphasis Level: -6dB, EqualizationComplete- EqualizationPhase1-
                         EqualizationPhase2- EqualizationPhase3- LinkEqualizationRequest-
                         Retimer- 2Retimers- CrosslinkRes: unsupported
        Capabilities: [b0] MSI-X: Enable+ Count=4 Masked-
                Vector table: BAR=4 offset=00000000
                PBA: BAR=4 offset=00000800
        Kernel driver in use: r8169`

brada4, 2023-08-10:
> ASPM L0s L1

Device itself claims to support ASPM.
It may help to not enable and use Linux builtin device blacklist vs one from BIOS.

CONFIG_PCIEASPM_DEFAULT=n

devastgh, 2023-08-10:
> > ASPM L0s L1
> 
> Device itself claims to support ASPM. It may help to not enable and use Linux builtin device blacklist vs one from BIOS.
> 
> CONFIG_PCIEASPM_DEFAULT=n

Well, the device does support ASPM, but it is buggy.
`CONFIG_PCIEASPM_DEFAULT=n` would change the current behaviour. I'm fine with that, but from my understanding, the least intrusive change would be to set that to `y`, as currently with both options not set, it seems like bios aspm settings are honored.

devastgh, 2023-08-10:
> Y there would mean to leave l0 aspm on, linux kernel is likely to have quirks for such a mainstream cards in better shape than BIOS

Sure, it's ok for me either way. The important part is `CONFIG_PCIEASPM=y`
Thanks.

hauke, 2023-08-13:
#13285 activates CONFIG_PCIEASPM also for x86.

commit ff710357 (2023-08-14): x86: Activate CONFIG_PCIEASPM

This activates PCI Express ASPM control in Linux. Without this option it
is completely controlled by the BIOS, now Linux will take over and apply
some workarounds if needed.

Fixes: #13248