openwrt/openwrt #13249: make dirclean kills ./files/

damnms, 2023-08-09:
### Describe the bug

when running make dirclean, it also deletes the ./files folder, what i would not expect as those are my configuration.
i would assume it deletes only built files, downloaded files and such, but not my configuration

### OpenWrt version

r23640-b777685de5

### OpenWrt target/subtarget

mvebu/cortexa9

### Device

SolidRun Clearfog Pro A1

### Image kind

Self-built image

### Steps to reproduce

git clone openwrt 
cd openwrt && mkdir files/etc
make dirclean
ls files

### Actual behaviour

it delets the files folder

### Expected behaviour

it should not delete that folder

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

dave14305, 2023-08-09:
If you run `make V=sc dirclean` do you see any explicit output showing the directory being removed? I don’t see this behavior.

damnms, 2023-08-10:
you are right, i will take a closer look at this this evening. maybe i messed up something :>

closed 2023-08-10 by damnms