openwrt/openwrt #13251: AX210 pci-e wireless card doesn't work

HeHeHeeeeee, 2023-08-09:
### Describe the bug

I use EXSI8 passthrough the AX210 pci-e wireless card, but found it doesn't work.
1.I have downloaded iwlwifi-firmware-ax210, kmod-iwlwifi package and wireless-tools package.
2. ifconfig can't find my wireless card.
3. I turn to Network and found that there is no option named wireless
4. I use official image and download iwlwifi-firmware-ax210, kmod-iwlwifi package, but can't find /etc/config/wireless files. And when I built image myself, there is /etc/config/wireless with no content in it.
5. So what should I do? I can't use WIFI now.

### OpenWrt version

r23228-cd17d8df2a

### OpenWrt target/subtarget

x86/64

### Device

VMware, Inc. VMware20,1

### Image kind

Official downloaded image

### Steps to reproduce

Just boot

### Actual behaviour

dmesg info:

[    0.000000] Linux version 5.15.118 (builder@buildhost) (x86_64-openwrt-linux-musl-gcc (OpenWrt GCC 12.3.0 r23228-cd17d8df2a) 12.3.0, GNU ld (GNU Binutils) 2.40.0) #0 SMP Mon Jun 26 11:20:39 2023
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz root=PARTUUID=150ec62c-2527-2a0b-e2d5-0346afe96102 rootwait console=tty0 console=ttyS0,115200n8 noinitrd
--------------------------------------
etc.....
--------------------------------------
[    3.592697] kmodloader: loading kernel modules from /etc/modules.d/*
[    3.603480] e1000: Intel(R) PRO/1000 Network Driver
[    3.605518] e1000: Copyright (c) 1999-2006 Intel Corporation.
[    3.606880] ixgbe: Intel(R) 10 Gigabit PCI Express Network Driver
[    3.607434] ixgbe: Copyright (c) 1999-2016 Intel Corporation.
[    3.608210] i2c_dev: i2c /dev entries driver
[    3.609129] Loading modules backported from Linux version v6.1.24-0-g0102425ac76b
[    3.609712] Backport generated by backports.git v5.15.92-1-44-gd6ea70fafd36
[    3.610896] e1000e: Intel(R) PRO/1000 Network Driver
[    3.611460] e1000e: Copyright(c) 1999 - 2015 Intel Corporation.
[    3.612558] Intel(R) 2.5G Ethernet Linux Driver
[    3.613101] Copyright(c) 2018 Intel Corporation.
[    3.616582] Intel(R) Wireless WiFi driver for Linux
[    3.618697] iwlwifi 0000:02:01.0: enabling device (0000 -> 0002)
[    3.620940] iwlwifi 0000:02:01.0: Direct firmware load for iwlwifi-ty-a0-gf-a0-72.ucode failed with error -2
[    3.621634] iwlwifi 0000:02:01.0: Falling back to sysfs fallback for: iwlwifi-ty-a0-gf-a0-72.ucode
[    3.631509] PPP generic driver version 2.4.2
[    3.633941] iwlwifi 0000:02:01.0: Direct firmware load for iwlwifi-ty-a0-gf-a0-71.ucode failed with error -2
[    3.634636] iwlwifi 0000:02:01.0: Falling back to sysfs fallback for: iwlwifi-ty-a0-gf-a0-71.ucode
[    3.635498] NET: Registered PF_PPPOX protocol family
[    3.638388] kmodloader: done loading kernel modules from /etc/modules.d/*
[    3.639200] iwlwifi 0000:02:01.0: Direct firmware load for iwlwifi-ty-a0-gf-a0-70.ucode failed with error -2
[    3.639876] iwlwifi 0000:02:01.0: Falling back to sysfs fallback for: iwlwifi-ty-a0-gf-a0-70.ucode
[    3.641561] iwlwifi 0000:02:01.0: Direct firmware load for iwlwifi-ty-a0-gf-a0-69.ucode failed with error -2
[    3.643941] iwlwifi 0000:02:01.0: Falling back to sysfs fallback for: iwlwifi-ty-a0-gf-a0-69.ucode
[    3.645480] iwlwifi 0000:02:01.0: Direct firmware load for iwlwifi-ty-a0-gf-a0-68.ucode failed with error -2
[    3.646327] iwlwifi 0000:02:01.0: Falling back to sysfs fallback for: iwlwifi-ty-a0-gf-a0-68.ucode
[    3.647891] iwlwifi 0000:02:01.0: Direct firmware load for iwlwifi-ty-a0-gf-a0-67.ucode failed with error -2
[    3.648612] iwlwifi 0000:02:01.0: Falling back to sysfs fallback for: iwlwifi-ty-a0-gf-a0-67.ucode
[    3.651750] iwlwifi 0000:02:01.0: api flags index 2 larger than supported by driver
[    3.654249] iwlwifi 0000:02:01.0: TLV_FW_FSEQ_VERSION: FSEQ Version: 0.63.2.2
[    3.656028] iwlwifi 0000:02:01.0: loaded firmware version 66.f1c864e0.0 ty-a0-gf-a0-66.ucode op_mode iwlmvm
[... 107 lines cut ...]
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:7019 errors:0 dropped:0 overruns:0 frame:0
          TX packets:18752 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:872700 (852.2 KiB)  TX bytes:5625785 (5.3 MiB)

eth0      Link encap:Ethernet  HWaddr 00:0C:29:61:EC:9A
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:93933 errors:0 dropped:41 overruns:0 frame:0
          TX packets:18752 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:29468274 (28.1 MiB)  TX bytes:5625785 (5.3 MiB)

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:88 errors:0 dropped:0 overruns:0 frame:0
          TX packets:88 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:9718 (9.4 KiB)  TX bytes:9718 (9.4 KiB)

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-08-10:
This is timeout connecting firmware on the card after that is loaded:
```
 [ 4.991298] iwlwifi 0000:02:01.0: Failed to start RT ucode: -110
```
Likely it has to do with virtualisation reset domains, power save or IRQ virtualisation, PCIE address spaces not mapped in same order as normal host.

Please check web documentation if you are close to being compatible. No wifi adapters supported by vmware at all.

HeHeHeeeeee, 2023-08-10:
> This is timeout connecting firmware on the card after that is loaded:
> 
> ```
>  [ 4.991298] iwlwifi 0000:02:01.0: Failed to start RT ucode: -110
> ```
> 
> Likely it has to do with virtualisation reset domains, power save or IRQ virtualisation, PCIE address spaces not mapped in same order as normal host.
> 
> Please check web documentation if you are close to being compatible. No wifi adapters supported by vmware at all.
> 
> One thing to check is if device brings up or fails same in a supported guest OS - e.g Fedora or Ubuntu and approach "official support channels"

Thanks for your reply, it seems not to do with power save or PCIE mapping.  I have limited knowledge and can't test the other reason may conduce to this situation. I think the most possible reason conduce to this is about EXSI system, then I will try using PVE system to test if I can use it successfully. Thanks again. If other people have the same question, I will appreciate it if you can discuss it in this issue.

brada4, 2023-08-10:
https://www.rdcloudcompute.com/2021/09/manage-esxi-using-wifi-enable-wifi-networking-for-esxi-lab-use-only.html#comment-1466
Problem lies outside vm operating system.

HeHeHeeeeee, 2023-08-11:
> https://www.rdcloudcompute.com/2021/09/manage-esxi-using-wifi-enable-wifi-networking-for-esxi-lab-use-only.html#comment-1466 Problem lies outside vm operating system.

As I mentioned before, I have download kmod-iwlwifi. I tried to update the maximum version which can be detected by openwrt system was iwlwifi-ty-a0-gf-a0-72.ucode, but it had no effect. Scrutinizing the log we can find the system actually find the wifi driver.
[ 4.959420] iwlwifi 0000:02:01.0: Loaded firmware version: 66.f1c864e0.0 ty-a0-gf-a0-66.ucode

HeHeHeeeeee, 2023-08-11:
Then I add AX210 pci-e wireless card to the other virtual machine, Ubuntu Server 22.04 on EXSI8 and find the same error. The kernel version is 5.15.x too, so will the linux kernel version to low conduce this error? But I find the AX210 support minimum kernel version is 5.1.

HeHeHeeeeee, 2023-08-13:
> I sent refetence that it is hit and miss attaching wifi adapter to a vm. Even windows fails to connect some not mentioned adapter, while other one works reliably. With supported guest you are eligible to ask esxii support. May happen they share some trick that works for openwrt.

Thanks for your advice, and I transferred to the PVE platform and found that wireless can be used. It seems that it is indeed an issue with EXSI

closed 2023-08-13 by HeHeHeeeeee

Delitants, 2025-09-28:
Was this card fully enabled for AP mode with 2.4/5/6 bands? Why ticket issue got closed as completed without details?