openwrt/openwrt #13255: Device: MikroTik hEX S (760iGS) doesn't boot 23.05.0-rc1 and 23.05.0-rc2 from TFTP

rom4nik, 2023-08-10:
### Describe the bug

For some reason 23.05.0-rc* images of OpenWrt don't boot via TFTP on this device, meanwhile 22.03.5 boots fine.
When I used 22.03.5 as netbooted initramfs and flashed either of those RCs' sysupgrade images, they were bootable from flash using either of bootloaders (both on 6.44 firmware).

### OpenWrt version

23.05.0-rc1, 23.05.0-rc2

### OpenWrt target/subtarget

ramips/mt7621

### Device

MikroTik RouterBOARD 760iGS

### Image kind

Official downloaded image

### Steps to reproduce

1. Plug PC into eth1/wan, start dnsmasq TFTP server (https://openwrt.org/toh/mikrotik/common#with_dnsmasq)
2. Hold reset button
3. Plug power to router while holding reset
4. After 15s router will download initramfs (visible in dnsmasq log) - stop holding reset, stop dnsmasq and plug PC into eth2/lan1
5. After a few seconds router will beep once

### Actual behaviour

22.03.5: after a few more seconds power LED starts blinking until OpenWrt boots up
23.05.0-rc1 and 23.05.0-rc2: power LED doesn't blink, after 15s router beeps twice and reboots into RouterOS or gets stuck (no responses to ping/DHCP, but eth LEDs work) until factory reset

### Expected behaviour

Both 22.03.5 and 23.05.0-rc* images boot up from TFTP.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

danpawlik, 2023-08-11:
maybe @kempniu can help?

john-tho, 2023-08-11:
Likely we are hitting the RouterBOOT memory region limits for this device, so netboot kernel fails to start.
Could you please detail the file sizes & entry point addresses of the initramfs images that work / don't work.

I have been using the lzma loader for netboot images: https://github.com/john-tho/openwrt/commit/2358bcaae3cd2902464597699aaf52f4fdde577b, but there is still a quirk with it that some images (built from the same state) will fail to start the decompressed kernel.

rom4nik, 2023-08-11:
Sure, I wasn't able to get UART logs yesterday, here are logs from netbooting RC*:

```
RouterBOOT backup booter 6.44

RB760iGS

CPU frequency: 880 MHz
  Memory size: 256 MiB
 Storage size:  16 MiB

Press any key within 2 seconds to enter setup
trying bootp protocol... OK
Got IP address: 192.168.1.162
resolved mac address 8C:16:45:A4:71:90
Gateway: 192.168.1.10
transfer started ........................................... transfer ok, time=2.82s
setting up elf image... kernel out of range
kernel loading failed
..

loading kernel... OK
setting up elf image... OK
jumping to kernel code
[    0.000000] Linux version 5.10.176 (builder@buildhost) (mipsel-openwrt-linux-musl-gcc (OpenWrt GCC 11.2.0 r20134-5f15225c1e) 11.2.0, GNU ld (GNU Binutils) 23
[    0.000000] SoC Type: MediaTek MT7621 ver:1 eco:3
(continues booting from flash)
```
Netbooting 22.03.5:
```
RouterBOOT backup booter 6.44

RB760iGS

CPU frequency: 880 MHz
  Memory size: 256 MiB
 Storage size:  16 MiB

Press any key within 2 seconds to enter setup
trying bootp protocol... OK
Got IP address: 192.168.1.162
resolved mac address 8C:16:45:A4:71:90
Gateway: 192.168.1.10
transfer started ........................................... transfer ok, time=2.85s
setting up elf image... OK
jumping to kernel code
[    0.000000] Linux version 5.10.176 (builder@buildhost) (mipsel-openwrt-linux-musl-gcc (OpenWrt GCC 11.2.0 r20134-5f15225c1e) 11.2.0, GNU ld (GNU Binutils) 23
[    0.000000] SoC Type: MediaTek MT7621 ver:1 eco:3
(continues booting from TFTP)
```

Searching for `kernel out of range` got me wiki page for [MikroTik RB411](https://openwrt.org/toh/mikrotik/rb411) where someone said it's caused by too large TFTP'd image, but that doesn't seem to be the case here:
```
$ ls -l *initramfs*
-rw-r--r-- 1 przemek przemek 5763676 08-05 17:10 openwrt-22.03.5-ramips-mt7621-mikrotik_routerboard-760igs-initramfs-kernel.bin
-rw-r--r-- 1 przemek przemek 5705152 08-10 19:33 openwrt-23.05.0-rc1-ramips-mt7621-mikrotik_routerboard-760igs-initramfs-kernel.bin
-rw-r--r-- 1 przemek przemek 5704704 08-05 16:20 openwrt-23.05.0-rc2-ramips-mt7621-mikrotik_routerboard-760igs-initramfs-kernel.bin
```
As you can see the 22.03.5 initramfs is larger than RCs.

rom4nik, 2023-08-11:
File sizes are above, here are entrypoint addresses:
```
$ readelf -h openwrt-22.03.5-ramips-mt7621-mikrotik_routerboard-760igs-initramfs-kernel.bin | grep Entry
  Entry point address:               0x80b71000
$ readelf -h openwrt-23.05.0-rc1-ramips-mt7621-mikrotik_routerboard-760igs-initramfs-kernel.bin | grep Entry
  Entry point address:               0x80b91000
$ readelf -h openwrt-23.05.0-rc2-ramips-mt7621-mikrotik_routerboard-760igs-initramfs-kernel.bin | grep Entry
  Entry point address:               0x80b91000
```

rom4nik, 2023-08-23:
RC3 boots fine:

```
$ ls -l *rc3*
-rw-r--r-- 1 przemek przemek 5701440 08-23 18:39 openwrt-23.05.0-rc3-ramips-mt7621-mikrotik_routerboard-760igs-initramfs-kernel.bin
$ readelf -h openwrt-23.05.0-rc3-ramips-mt7621-mikrotik_routerboard-760igs-initramfs-kernel.bin | grep Entry
  Entry point address:               0x80b81000
```
I'll close this issue, hopefully stable builds will be netbootable as well. Thanks for the hint about entry point addresses!

muxator, 2023-10-20:
Hi, I confirm that OpenWRT rc3 is netbootable on MikroTik hEX S (760iGS), but **the release version is not**, because its entry point is the same as rc2.

|file|`readelf -h <file> \| grep Entry`| netbootable? |
| --- | --- | --- |
| openwrt-23.05.0-rc2-ramips-mt7621-mikrotik_routerboard-760igs-initramfs-kernel.bin | 0x80b91000 | NO |
| openwrt-23.05.0-rc3-ramips-mt7621-mikrotik_routerboard-760igs-initramfs-kernel.bin | 0x80b81000 | YES |
| openwrt-23.05.0-ramips-mt7621-mikrotik_routerboard-760igs-initramfs-kernel.bin | 0x80b91000 | NO |

I think this information should be reported in the wiki at https://openwrt.org/toh/mikrotik/rb760igs.

Thanks to johnth that helped me at https://forum.openwrt.org/t/mikrotik-hex-s-rb760igs-cant-boot-via-tftp/175012

jebotz, 2024-01-01:
I just ran into this with 23.05.2 (which also entry point at 0x80b91000).  So before seeing this bug report I tried saome older releases, and found that 22.03.1 (entry point at 0x80b61000) works for me, but 22.05.6 does not (entry point at 0x80b81000, which is the same as @muxator reported for 23.05.0-rc3).  So either there is something else going on or slightly different versions of MikroTik's netboot firmware have different maximum entry points?

BlackMage2, 2024-04-24:
the command 'readelf -h <file> | grep Entry' outputs for the file 'openwrt-23.05.0-rc3-ramips-mt7621-mikrotik_routerboard-760igs-initramfs-kernel.bin' nothing. what file should i choose for the hex S?

rom4nik, 2024-04-27:
23.05.0-rc3 still looks good:
```
$ readelf -h openwrt-23.05.0-rc3-ramips-mt7621-mikrotik_routerboard-760igs-initramfs-kernel.bin 
ELF Header:
  Magic:   7f 45 4c 46 01 01 01 00 00 00 00 00 00 00 00 00 
  Class:                             ELF32
  Data:                              2's complement, little endian
  Version:                           1 (current)
  OS/ABI:                            UNIX - System V
  ABI Version:                       0
  Type:                              EXEC (Executable file)
  Machine:                           MIPS R3000
  Version:                           0x1
  Entry point address:               0x80b81000
  Start of program headers:          52 (bytes into file)
  Start of section headers:          5700160 (bytes into file)
  Flags:                             0x70001001, noreorder, o32, mips32r2
  Size of this header:               52 (bytes)
  Size of program headers:           32 (bytes)
  Number of program headers:         2
  Size of section headers:           40 (bytes)
  Number of section headers:         32
  Section header string table index: 31
```

scholzi100, 2026-10-19:
Same board (hEX S / 760iGS, RouterBOOT 6.47.10), same trap: after a failed sysupgrade I burned ~5 hours on netboot recovery. Every TFTP upload completed (verified in packet captures), but the initramfs never executed, no second DHCP, no SSH on any port. I tried 24.10.1, 24.10.8, 25.12.3 and 25.12.6, and every LAN port, before finding this issue.

All of those images have entry points in the 0x80cc1000–0x80d81000 range, so RouterBOOT silently refuses them and falls back to NAND. Netbooting 22.03.5 (0x80b71000) first and then `sysupgrade` to 25.12.6 is the working path.