openwrt/openwrt #13258: Cudy x6 v 2 and Cudy WR3000 no flash Openwrt 23.05 with www

djwujek123, 2023-08-11:
### Describe the bug

Hello.
I am trying to flash the Cudy X6 v2 and Cudy WR3000 routers via the website via the squashfs-sysupgrade.bin file but in both routers the installation fails and I get the error installation failed.

### OpenWrt version

OpenWrt 23.05.0-rc2

### OpenWrt target/subtarget

...

### Device

Cudy X6 v2 or Cudy WR3000

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

danpawlik, 2023-08-11:
You want to move from stock firmware to OpenWRT and it does not work or you want to do upgrade and it does not work?
If the second, why don't you use `auc`  ([Attended Sysupgrade](https://openwrt.org/docs/guide-user/installation/attended.sysupgrade)) ?

ctolzane, 2023-08-12:
As far as I know it's a 2 steps process. First update from cudy original software to cudy openwrt software. Second from cudy openwrt to "genuine" openwrt. Cudy openwrt can be found there:   http://www.cudytech.com/openwrt_software_download

djwujek123, 2023-08-12:
Thank you very much for providing the link with the soft. Am I to understand that I first install OpenWrt from Cuda and then Original OpenWrt with the sysupgrade extension ??

ctolzane, 2023-08-12:
> Thank you very much for providing the link with the soft. Am I to understand that I first install OpenWrt from Cuda and then Original OpenWrt with the sysupgrade extension ??

Yes, you get it right :-)

Djfe, 2023-08-29:
It worked right?
If so then please close this issue.
About Cudy WR3000: it would be very interesting, whether it worked, since that device was added before Cudy offered their signed openwrt version.

The wiki could use installation instructions, I agree.

IHER3, 2023-12-01:
I have the same problem as djwujek123 with a cudy x6 v2.

I've installed version 1.15.39 from http://www.cudytech.com/openwrt_software_download but then I try to install the downloaded version of openwrt (version 23.05) from advanced settings --> firmware and I get the error "File is invalid".

simeononsecurity, 2024-02-21:
I'm attempting this now with WR3000 v1.
I've tried the instructions on the openwrt page at https://openwrt.org/toh/cudy/wr3000_v1
Tried what I learned from the forums with the same steps execept with the image from the image builder rather than the one from the openwrt install page. 

https://forum.openwrt.org/t/support-for-cudy-wr3000/155463/142?page=5
https://firmware-selector.openwrt.org/?version=23.05.0&target=mediatek%2Ffilogic&id=cudy_wr3000-v1

I can get the Cudy OpenWRT image installed however, when I try that, I get a box that I can ssh into but can do nothing else. Doesn't claim to be a r/o filesystem. But it won't let me install any packages and there is no web gui that loads.

http://www.cudytech.com/openwrt_software_download

If I attempt to then flash any of the openwrt official images from that point the system fails to boot at all. The lights on the box just keep flashing and cutting off. No decernable pattern.

Same goes if I try to flash directly the full openwrt images. It won't boot.

So far, the only thing I can do is successfully reflash the Official Cudy Firmware 1.16.2 and 1.16.6.

It's a darn shame, for my purposes the devices are useless to me without full openwrt and ssh access.

Djfe, 2024-02-22:
Please retry with the current firmware
https://firmware-selector.openwrt.org/?version=23.05.2&target=mediatek%2Ffilogic&id=cudy_wr3000-v1

install Cudy OpenWrt, then open the webgui and flags firmware.
choose the official 23.05.2 sysupgrade image and uncheck the checkbox "keep config".
This makes is so that it forgets config on install. Cudy config is incompatible to OpenWrt.

if you used ssh, then you should have done:
sysupgrade -n /tmp/openwrt-23.05.2-mediatek-filogic-cudy_wr3000-v1-squashfs-sysupgrade.bin

-n forgets the config

simeononsecurity, 2024-02-22:
> Please retry with the current firmware https://firmware-selector.openwrt.org/?version=23.05.2&target=mediatek%2Ffilogic&id=cudy_wr3000-v1
>
> install Cudy OpenWrt, then open the webgui and flags [...]

Yeah I got that far already. It was just temperamental. 

> Update... I did absolutely nothing different other than following the openwrt guide, but this time using google chrome over firefox and it worked. I can not believe this is still a thing in 2024.

The series of steps I ended up doing.
Flashed latest Cudy image (at the time 1.16.6), using Google Chrome flash the Cudy Signed OpenWRT Image in the Web GUI, In the Luci upgrade page update using the latest sysupgrade bin (at the time 23.05.2).
I should note that when using the image builder if I made a custom image at all it would fail. I specifically had to use the one from the openwrt instructions page.

Djfe, 2024-02-22:
weird
well now that you are running OpenWrt, installing a custom image from the image builder should work just fine.

> I specifically had to use the one from the openwrt instructions page.

which one?

simeononsecurity, 2024-02-22:
> weird well now that you are running OpenWrt, installing a custom image from the image builder should work just fine.
> 
> > I specifically had to use the one from the openwrt instructions page.
> 
> which one?

On the image selector where it says "Customize installed packages and/or first boot script". That...
I wanted to build an image with the following packages:

```bash
wpad-openssl nano iw-full hostapd-common kmod-mac80211 kmod-cfg80211 kmod-mt76x2u kmod-mt76-connac kmod-mt76-core kmod-mt76-usb kmod-mt76x2-common kmod-mt76x2u kmod-mt7915e kmod-mt7916-firmware mt7981-wo-firmware bridger
```
In addition to what was included by default, obviously accounting for conflicts.

Djfe, 2024-02-22:
> I specifically had to use the one from the openwrt instructions page.

which image of the instructions page did you choose and flash? (which is the image that worked)
https://openwrt.org/toh/cudy/wr3000_v1

it should be the same one as from the firmware selector (unless you choose the custom builder of course)

before you buy more of these you should test the WR3000.
I think the wifi is not stable right now and signal strength is acting weird.
this is stuff that will be fixed eventually. but before you buy several of these, maybe choose the stable X6 instead of WR3000. (if you need them now, else wait for the fix)

simeononsecurity, 2024-02-22:
> > I specifically had to use the one from the openwrt instructions page.
> 
> which image of the instructions page did you choose and flash? (which is the image that worked) https://openwrt.org/toh/cudy/wr3000_v1
Correct, yes

> it should be the same one as from the firmware selector (unless you choose the custom builder of course)

It is the same. Only have issues with the custom builder when I add packages.
https://downloads.openwrt.org/releases/23.05.2/targets/mediatek/filogic/openwrt-23.05.2-mediatek-filogic-cudy_wr3000-v1-squashfs-sysupgrade.bin

> before you buy more of these you should test the WR3000. I think the wifi is not stable right now and signal strength is acting weird. this is stuff that will be fixed eventually. but before you buy [...]

WiFi seemed stable to me as long as I wasn't using 160mhz channels.
I'm optimizing for distance over throughput however.
But you're right. However for only like $30-$40 more I can get the GL.iNet Flint 1, which is pretty comparable, basically identical,  and you don't have to deal with any of these complications. Not to mention it has more ram and onboard flash. Only downside is it gets updates a bit slower than running vanilla OpenWRT. However it still has ssh access and the Luci page. Also it has the capability of implementing tailscale/headscale for remote management of the device. 
It fits my needs better. I just wanted to try this one out. 

Appreciate the help and advice though @Djfe

Djfe, 2024-02-22:
GL.iNet Flint 1 is not supported by OpenWrt.
But if I understand you correctly, then you already know that.
Happy to help.
If the Flint 1 fits your use-case, then great :)
Otherwise: there are definitely alternatives out there, in-case you need them at some point, feel free to ask.

simeononsecurity, 2024-02-22:
> GL.iNet Flint 1 is not supported by OpenWrt. But if I understand you correctly, then you already know that. Happy to help. If the Flint 1 fits your use-case, then great :) Otherwise: there are definitely alternatives out there, in-case you need them at some point, feel free to ask.

It comes with GL.iNets version of OpenWRT on it. Never have had an issue with it. Functions identically except the default web page is customized to their own thing. But they don't restrict you from doing anything you need/want.

Djfe, 2024-02-22:
I'm pretty you might be limited regarding which packages are available and which kernel mods. But apart from that, yes