openwrt/openwrt #13259: GL.iNet MT2500 is not controlling fan correctly

lukasz1992, 2023-08-11:
### Describe the bug

There is a preset inside https://github.com/openwrt/openwrt/blob/main/target/linux/mediatek/files-6.1/arch/arm64/boot/dts/mediatek/mt7981.dtsi#L63 to have 4 steps of fan for MT7981 devices (0, 50, 75 and 100 %).

MT2500 is an example that has a physical fan.

1. I am not sure if reading the temperature works properly (temperature is always between 59 and 61). 
2. Even if assumed that at the moment. Fan however constantly starts and stops - without keeping the margins set in hysteresis parameters.

### OpenWrt version

23.05-SNAPSHOT

### OpenWrt target/subtarget

mediatek/filogic

### Device

GL.iNet GL-MT3000

### Image kind

Official downloaded image

### Steps to reproduce

- start the router

### Actual behaviour

- temperature limits make fan constantly start and stop
- temperature does not seem to be correct, but have no method to test it
- hysteresis for fans is not kept

Apart of that, device is working correctly, did not notice any slowdowns

### Expected behaviour

- I would expect more stable configuration for fans
- I would reconfigure fans in dts from:
```
50% / 75% / 100%
60C / 85C / 115C
```
to
```
20% / 40% / 60% / 80% / 100%
50C / 60C / 70C / 80C / 90C
```
- then one with hardware termometer could confirm if reported temp is valid
- possibly temperature of mt7915 radio could also be included to these maps, but I do not know how

### Additional info

Manufacturer setup 256 maps,
https://github.com/gl-inet/gl-infra-builder/blob/master/patches-mt798x-7.6.6.1/4004-support-MT3000.patch#L151
I am not sure how these are controlled.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

lukasz1992, 2023-08-11:
This is behaving the same for both master and 23.05

@dangowrt do you prefer to have patch prepared for you?

dangowrt, 2023-08-11:
If you believe the fan should be active earlier and/or introduce more steps, please submit a patch.
The stock firmware uses a shell script to control the fan instead of Linux thermal zone.

lukasz1992, 2023-08-11:
Sure, done:

https://github.com/openwrt/openwrt/pull/13264
https://github.com/openwrt/openwrt/pull/13265 - the same for 23.05 branch

willmo, 2025-01-09:
Not that it matters but this can be closed since #13264 right?

closed 2025-01-09 by lukasz1992