openwrt/openwrt #13260: kernel size not a multiple of block size on Ubiquiti Routerstation image

MarkTerMors, 2023-08-11:
### Describe the bug

Trying to upgrade lede 17.01.4 to recent versions of Openwrt (I tried 22.03.5 and 22.03.4 stable releases.)
I get the following error from the Redboot TFTPD:

TFTPD: Incoming connection from 192.168.1.100:48378
Received:  8023994 bytes
Firmware Version: RSx.ar7100.v6.0.0-OpenWrt-r20123-38ccc47687
'kernel' size, 0x00226e22 is not a multiple of block size, 0x00010000! Abort.

and 

TFTPD: Incoming connection from 192.168.1.100:37412
Received:  8024032 bytes
Firmware Version: RSx.ar7100.v6.0.0-OpenWrt-r20134-5f15225c1e
'kernel' size, 0x00226e48 is not a multiple of block size, 0x00010000! Abort.

Trying a sysupgrade from Luci fails silently.
From the command line I get:

root@LEDE:/# sysupgrade -F /tmp/openwrt-22.03.5-ath79-generic-ubnt_routerstation
-squashfs-sysupgrade.bin 
Device routerstation not supported by this image
Supported devices: ubnt,routerstation - Image version mismatch: image 2.0, device 1.0. Please wipe config during upgrade (force required) or reinstall. Reason: Partition design has changed compared to older ver.
Image check 'fwtool_check_image' failed but --force given - will update anyway!
Saving config files...
killall: watchdog: no process killed
Sending TERM to remaining processes ... ubusd logd rpcd netifd odhcpd uhttpd ntpd dnsmasq 
Sending KILL to remaining processes ... 
Switching to ramdisk...
Performing system upgrade...
ash: arithmetic syntax error

### OpenWrt version

22.03.5

### OpenWrt target/subtarget

Ubiquiti Routerstation

### Device

Ubiquiti Routerstation

### Image kind

Official downloaded image

### Steps to reproduce

Try to install current image on a Routerstation with older firmware.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-08-11:
> Please wipe config during upgrade (force required) or reinstall. 

Did it succed after selecting force parameter?

MarkTerMors, 2023-08-12:
No, but upgrading 19.07.10 first got it going again. Currently on 22.03.5 without apparent problems.

tmn505, 2023-08-12:
This is a case of WONTFIX. Along the way the sysupgrade image format changed and 17.01.x doesn't understand it (the first one which did are 19.07.x `ath79` images). The workaround is the procedure You did or using recovery with factory image. Factory image will work regardless of version installed on the device (naturally all setting will reset).
BTW. did You check wiki before upgrading? Maybe I should write warning over there?

MarkTerMors, 2023-08-16:
The images I tried to load by TFTP were Factory images. That's why it was weird that they didn't work.
The sysupgrade path did work in the end though.

brada4, 2023-08-16:
> check wiki 

Indeed one needs to load valid factory image over factory recovery

tmn505, 2023-08-16:
> The images I tried to load by TFTP were Factory images. That's why it was weird that they didn't work.

Thanks for the follow up. I checked it at first with factory from snapshots and it worked without problems. Later I did the 22.03.x factory images and they indeed fail. This is because backported 8e09f9ffc3cc didn't include necessary changes to `target/linux/ath79/image/common-ubnt.mk` and `mkubntimage` didn't accept LZMA loader as an argument. I will send the fix to mailing list. Thank You for the report.

Djfe, 2024-01-21:
this one is fixed now, is it?

commit 7901fec6 (2023-11-20): ath79: image: allow changing kernel option in mkubntimage

Backport didn't include changes to mkubntimage invocation, which allowed
to pass arguments. This in result produces broken routerstation factory
images, which have kernel embedded as kernel partition instead of LZMA
loader.

Fixes: #13260
Fixes: 8e09f9ffc3cc ("ath79: switch some RedBoot based devices to OKLI loader")