openwrt/openwrt #13261: kmod-rtl8812au-ct issues with Alfa device

FabioEight, 2023-08-11:
### Describe the bug

Hello, I'm using kmod-rtl8812au-ct to let OpenWrt recognise my AWUS036AC.
I'm able to see the interface on OpenWrt, but as soon as I click on scan, Luci crashes and I have the following errors in the logs:
```
Fri Aug 11 15:44:52 2023 kern.warn kernel: [ 1914.616255] rtl8812au 1-1.1.2:1.0 wlan0: Current addr:  00 c0 ca 8f 58 64 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Fri Aug 11 15:44:52 2023 kern.warn kernel: [ 1914.629897] rtl8812au 1-1.1.2:1.0 wlan0: Expected addr: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Fri Aug 11 15:44:52 2023 kern.warn kernel: [ 1914.643480] ------------[ cut here ]------------
Fri Aug 11 15:44:53 2023 kern.warn kernel: [ 1914.648099] netdevice: wlan0: Incorrect netdev->dev_addr
Fri Aug 11 15:44:53 2023 kern.warn kernel: [ 1914.653433] WARNING: CPU: 3 PID: 506 at net/core/dev_addr_lists.c:519 dev_addr_check+0x90/0x130
Fri Aug 11 15:44:53 2023 kern.warn kernel: [ 1914.662136] Modules linked in: rt2800usb rt2800lib pppoe ppp_async nft_fib_inet nf_flow_table_inet brcmfmac rtl8812au rt2x00usb rt2x00lib pppox ppp_generic nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_objref nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack mac80211 cfg80211 cdc_ncm cdc_ether usbnet usbhid slhc r8169 r8152 nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 libcrc32c lan78xx hid_generic crc_ccitt compat brcmutil snd_bcm2835(C) hid evdev snd_rawmidi snd_seq_device snd_pcm_oss snd_pcm_dmaengine snd_mixer_oss snd_hwdep snd_compress snd_pcm snd_timer snd soundcore nls_utf8 cmac nls_iso8859_1 nls_cp437 vfat fat realtek microchip
Fri Aug 11 15:44:53 2023 kern.warn kernel: [ 1914.735560] CPU: 3 PID: 506 Comm: rpcd Tainted: G        WC         6.1.42 #0
Fri Aug 11 15:44:53 2023 kern.warn kernel: [ 1914.742690] Hardware name: Raspberry Pi 4 Model B Rev 1.5 (DT)
Fri Aug 11 15:44:53 2023 kern.warn kernel: [ 1914.748513] pstate: 80400005 (Nzcv daif +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
Fri Aug 11 15:44:53 2023 kern.warn kernel: [ 1914.755468] pc : dev_addr_check+0x90/0x130
Fri Aug 11 15:44:53 2023 kern.warn kernel: [ 1914.759557] lr : dev_addr_check+0x90/0x130
Fri Aug 11 15:44:53 2023 kern.warn kernel: [ 1914.763645] sp : ffffffc009c33b10
Fri Aug 11 15:44:53 2023 kern.warn kernel: [ 1914.766949] x29: ffffffc009c33b10 x28: ffffff8043932a00 x27: 0000000000000000
Fri Aug 11 15:44:53 2023 kern.warn kernel: [ 1914.774081] x26: 0000007fcf9f1043 x25: ffffff8043932a10 x24: ffffff8041d0d000
Fri Aug 11 15:44:53 2023 kern.warn kernel: [ 1914.781211] x23: 0000000000001002 x22: 0000000000000041 x21: ffffffc000d0b458
Fri Aug 11 15:44:53 2023 kern.warn kernel: [ 1914.788341] x20: ffffff8041d0d000 x19: ffffff8041d0d000 x18: 0000000000000374
Fri Aug 11 15:44:53 2023 kern.warn kernel: [ 1914.795471] x17: 0000000000000000 x16: 0000000000000000 x15: ffffffc008b4a900
Fri Aug 11 15:44:53 2023 kern.warn kernel: [ 1914.802601] x14: 0000000000000a5c x13: 0000000000000374 x12: 00000000ffffffea
Fri Aug 11 15:44:53 2023 kern.warn kernel: [ 1914.809731] x11: 00000000ffffefff x10: ffffffc008ba2900 x9 : ffffffc008b4a8a8
Fri Aug 11 15:44:53 2023 kern.warn kernel: [ 1914.816860] x8 : 0000000000017fe8 x7 : c0000000ffffefff x6 : 0000000000000001
Fri Aug 11 15:44:53 2023 kern.warn kernel: [ 1914.823990] x5 : ffffff80fb7d2780 x4 : 0000000000000000 x3 : 0000000000000027
Fri Aug 11 15:44:53 2023 kern.warn kernel: [ 1914.831119] x2 : 0000000000000027 x1 : 0000000000000023 x0 : 000000000000002c
Fri Aug 11 15:44:53 2023 kern.warn kernel: [ 1914.838249] Call trace:
Fri Aug 11 15:44:53 2023 kern.warn kernel: [ 1914.840687]  dev_addr_check+0x90/0x130
Fri Aug 11 15:44:53 2023 kern.warn kernel: [ 1914.844428]  __dev_open+0x30/0x1b4
Fri Aug 11 15:44:53 2023 kern.warn kernel: [ 1914.847821]  __dev_change_flags+0x140/0x194
Fri Aug 11 15:44:53 2023 kern.warn kernel: [ 1914.851995]  dev_change_flags+0x24/0x6c
Fri Aug 11 15:44:53 2023 kern.warn kernel: [ 1914.855822]  devinet_ioctl+0x398/0x68c
Fri Aug 11 15:44:53 2023 kern.warn kernel: [ 1914.859565]  inet_ioctl+0x240/0x280
Fri Aug 11 15:44:53 2023 kern.warn kernel: [ 1914.863046]  sock_ioctl+0x2c0/0x3f0
Fri Aug 11 15:44:53 2023 kern.warn kernel: [ 1914.866528]  __arm64_sys_ioctl+0x448/0xbc0
Fri Aug 11 15:44:53 2023 kern.warn kernel: [ 1914.870618]  invoke_syscall.constprop.0+0x5c/0x104
Fri Aug 11 15:44:53 2023 kern.warn kernel: [ 1914.875403]  do_el0_svc+0x50/0x13c
Fri Aug 11 15:44:53 2023 kern.warn kernel: [ 1914.878796]  el0_svc+0x18/0x54
Fri Aug 11 15:44:53 2023 kern.warn kernel: [ 1914.881843]  el0t_64_sync_handler+0xf4/0x120
Fri Aug 11 15:44:53 2023 kern.warn kernel: [ 1914.886104]  el0t_64_sync+0x174/0x178
Fri Aug 11 15:44:53 2023 kern.warn kernel: [ 1914.889759] ---[ end trace 0000000000000000 ]---
```

### OpenWrt version

r23688-afa229038c

### OpenWrt target/subtarget

bcm27xx/bcm2711

### Device

Raspberry Pi 4B

### Image kind

Official downloaded image

### Steps to reproduce

- connect AWUS036AC adapter
- Network > Wireless > select the new adapter and run a scan

### Actual behaviour

- scan never ends
- luci crashes
- have to reboot to restore luci

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

john-tho, 2023-08-11:
Looks like the [OpenWrt patch for `dev_addr`](https://github.com/openwrt/openwrt/commit/a07566ead82a5d33dac5054dd2ade02db9f92e9f) is hitting the Linux runtime checks against directly modifying `dev_addr`:  [d07b26f5bbea ("dev_addr: add a modification check")](https://github.com/torvalds/linux/commit/d07b26f5bbea9ade34dfd6abea7b3ca056c03cd1).
This was addressed differently in https://github.com/openwrt/openwrt/pull/12746: https://github.com/greearb/rtl8812AU_8821AU_linux/commit/fe2afbdde3bd17351caee2ffe46095f14bc6e46d

FabioEight, 2023-08-12:
> Looks like the [OpenWrt patch for `dev_addr`](https://github.com/openwrt/openwrt/commit/a07566ead82a5d33dac5054dd2ade02db9f92e9f) is hitting the Linux runtime checks against directly modifying `dev_addr`: [d07b26f5bbea ("dev_addr: add a modification check")](https://github.com/torvalds/linux/commit/d07b26f5bbea9ade34dfd6abea7b3ca056c03cd1). This was addressed differently in #12746: [greearb/rtl8812AU_8821AU_linux@fe2afbd](https://github.com/greearb/rtl8812AU_8821AU_linux/commit/fe2afbdde3bd17351caee2ffe46095f14bc6e46d)

Sorry I don’t understand this. is it possible to fix/bypass this check or apply the patch in another way?

commit c123e4f0 (2023-08-19): rtl8812au-ct: bump to fix kernel 6.1 compile

 9b2b203 fix usb_recv_tasklet -Wcast-function-type
 a027da5 fix kernel 6.1 80211 link_id
 7a9c802 fix build for kernel 6.1 prandom
 3a3eb24 fix build for kernel 5.17 PDE_DATA
 fe2afbd fix build for kernel 5.17 const netdev->dev_addr
 7275bae fix build for 5.17 kernel complete_and_exit
 c9c2aa7 Update usb_intf.c

revert the upstream 6.1 link_id (depend on kernel version) changes and
force 6.1 link_id

The downstream patch casting const off the direct dev_addr writes
triggers the runtime check from
Linux d07b26f5bbea ("dev_addr: add a modification ch [... cut]
closed 2023-08-19 by openwrt-bot