openwrt/openwrt #13262: Redmi AX6000 random crash (NETDEV WATCHDOG: eth0 (mtk_soc_eth): transmit queue 7 timed out)

tczee36, 2023-08-11:
### Describe the bug

Getting the following crash at least once a week, sometimes more.

Not sure if this is a switch/flow control issue, could find a fix.

WIFI connection continue to function without internet connection. 

Must be manually rebooted to resume internet connection.

`[14933.401104] ------------[ cut here ]------------
[14933.405728] NETDEV WATCHDOG: eth0 (mtk_soc_eth): transmit queue 7 timed out
[14933.412707] WARNING: CPU: 0 PID: 0 at dev_watchdog+0x358/0x380
[14933.418529] Modules linked in: pppoe ppp_async nft_fib_inet nf_flow_table_ipv6 nf_flow_table_ipv4 nf_flow_table_inet pppox ppp_mppe ppp_generic nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_objref nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_fullcone nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_counter nft_chain_nat nf_tables nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_pptp nf_nat_irc nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_nat nf_flow_table nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_pptp nf_conntrack_netlink nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp nf_conntrack_broadcast nf_conntrack_amanda nf_conntrack mt7915e mt76_connac_lib mt76 mac80211 cfg80211 ts_kmp ts_fsm ts_bm slhc nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 macvlan libcrc32c crc_ccitt compat asn1_decoder crypto_safexcel cryptodev sit tunnel4 ip_tunnel crypto_user algif_skcipher
[14933.418667]  algif_rng algif_hash algif_aead af_alg sha1_generic seqiv md5 des_generic libdes authencesn authenc arc4 leds_ws2812b leds_gpio gpio_button_hotplug
[14933.519746] CPU: 0 PID: 0 Comm: swapper/0 Not tainted 5.15.120 #0
[14933.525820] Hardware name: Xiaomi Redmi Router AX6000 (stock layout) (DT)
[14933.532585] pstate: 40400005 (nZcv daif +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[14933.539525] pc : dev_watchdog+0x358/0x380
[14933.543519] lr : dev_watchdog+0x358/0x380
[14933.547513] sp : ffffffc008003d60
[14933.550811] x29: ffffffc008003d60 x28: 0000000000000140 x27: 00000000ffffffff
[14933.557927] x26: 0000000000000000 x25: 0000000000000000 x24: ffffff80012fb4c0
[14933.565043] x23: 0000000000000000 x22: 0000000000000001 x21: ffffffc008f86000
[14933.572159] x20: ffffff80012fb000 x19: 0000000000000007 x18: ffffffc008fa0380
[14933.579274] x17: ffffffc016efd000 x16: ffffffc008000000 x15: 000000000000033f
[14933.586390] x14: 0000000000000115 x13: ffffffc008003a88 x12: ffffffc008ff8380
[14933.593506] x11: 712074696d736e61 x10: ffffffc008ff8380 x9 : 0000000000000000
[14933.600622] x8 : ffffffc008fa0330 x7 : ffffffc008fa0380 x6 : 0000000000000001
[14933.607738] x5 : 0000000000000000 x4 : 0000000000000000 x3 : 0000000000000000
[14933.614853] x2 : ffffff801fe79230 x1 : ffffffc016efd000 x0 : 000000000000003f
[14933.621970] Call trace:
[14933.624402]  dev_watchdog+0x358/0x380
[14933.628048]  call_timer_fn+0x30/0x144
[14933.631699]  __run_timers.part.0+0x240/0x2ec
[14933.635954]  run_timer_softirq+0x38/0x70
[14933.639860]  _stext+0x118/0x348
[14933.642987]  __irq_exit_rcu+0xdc/0xfc
[14933.646635]  irq_exit+0xc/0x1c
[14933.649675]  handle_domain_irq+0x60/0x8c
[14933.653584]  gic_handle_irq+0x50/0x120
[14933.657321]  call_on_irq_stack+0x20/0x30
[14933.661228]  do_interrupt_handler+0x4c/0x54
[14933.665395]  el1_interrupt+0x2c/0x4c
[14933.668958]  el1h_64_irq_handler+0x14/0x20
[14933.673037]  el1h_64_irq+0x78/0x7c
[14933.676423]  arch_cpu_idle+0x14/0x20
[14933.679983]  default_idle_call+0x38/0x124
[14933.683977]  do_idle+0xac/0x110
[14933.687104]  cpu_startup_entry+0x20/0x50
[14933.691010]  rest_init+0xc4/0xd0
[14933.694222]  arch_call_rest_init+0xc/0x14
[14933.698217]  start_kernel+0x618/0x638
[14933.701863]  __primary_switched+0xa0/0xa8
[14933.705857] ---[ end trace f0719674a40d55eb ]---
[14933.710478] mtk_soc_eth 15100000.ethernet eth0: transmit timed out
[14935.395565] mtk_soc_eth 15100000.ethernet eth0: Link is Down
[14935.929138] mtk_soc_eth 15100000.ethernet: warm reset failed
[14935.947866] mtk_soc_eth 15100000.ethernet eth0: configuring for fixed/2500base-x link mode
[14935.956254] mtk_soc_eth 15100000.ethernet eth0: Link is Up - 2.5Gbps/Full - flow control rx/tx`

### OpenWrt version

23.05.0-rc3 (r26804-a3639e76d6) 

### OpenWrt target/subtarget

mediatek/filogic

### Device

Xiaomi Redmi Router AX6000 (stock layout)

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-08-11:
Duplicate of https://github.com/openwrt/openwrt/issues/13122

NovaPixell, 2023-08-15:
I think this bug is given by the ethernet switch drivers in OpenWRT. At the moment I'm using ImmortalWRT with proprietary mediatek drivers, I don't see the bug (kernel 5.4).

NovaPixell, 2023-10-13:
> ### Describe the bug
> Getting the following crash at least once a week, sometimes more.
> 
> Not sure if this is a switch/flow control issue, could not find a fix.
> 
> WIFI connection continue to function without internet connection.
> 
> Must be manually rebooted to resume internet connection.
> 
> ```
> `[14933.401104] ------------[ cut here ]------------
> [14933.405728] NETDEV WATCHDOG: eth0 (mtk_soc_eth): transmit queue 7 timed out
> [14933.412707] WARNING: CPU: 0 PID: 0 at dev_watchdog+0x358/0x380
> [14933.418529] Modules linked in: pppoe ppp_async nft_fib_inet nf_flow_table_ipv6 nf_flow_table_ipv4 nf_flow_table_inet pppox ppp_mppe ppp_generic nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_objref nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_fullcone nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_counter nft_chain_nat nf_tables nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_pptp nf_nat_irc nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_nat nf_flow_table nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_pptp nf_conntrack_netlink nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp nf_conntrack_broadcast nf_conntrack_amanda nf_conntrack mt7915e mt76_connac_lib mt76 mac80211 cfg80211 ts_kmp ts_fsm ts_bm slhc nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 macvlan libcrc32c crc_ccitt compat asn1_decoder crypto_safexcel cryptodev sit tunnel4 ip_tunnel crypto_user algif_skcipher
> [14933.418667]  algif_rng algif_hash algif_aead af_alg sha1_generic seqiv md5 des_generic libdes authencesn authenc arc4 leds_ws2812b leds_gpio gpio_button_hotplug
> [14933.519746] CPU: 0 PID: 0 Comm: swapper/0 Not tainted 5.15.120 #0
> [14933.525820] Hardware name: Xiaomi Redmi Router AX6000 (stock layout) (DT)
> [14933.532585] pstate: 40400005 (nZcv daif +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
> [14933.539525] pc : dev_watchdog+0x358/0x380
> [14933.543519] lr : dev_watchdog+0x358/0x380
> [14933.547513] sp : ffffffc008003d60
> [14933.550811] x29: ffffffc008003d60 x28: 0000000000000140 x27: 00000000ffffffff
> [14933.557927] x26: 0000000000000000 x25: 0000000000000000 x24: ffffff80012fb4c0
> [14933.565043] x23: 0000000000000000 x22: 0000000000000001 x21: ffffffc008f86000
> [14933.572159] x20: ffffff80012fb000 x19: 0000000000000007 x18: ffffffc008fa0380
> [14933.579274] x17: ffffffc016efd000 x16: ffffffc008000000 x15: 000000000000033f
> [14933.586390] x14: 0000000000000115 x13: ffffffc008003a88 x12: ffffffc008ff8380
> [14933.593506] x11: 712074696d736e61 x10: ffffffc008ff8380 x9 : 0000000000000000
> [14933.600622] x8 : ffffffc008fa0330 x7 : ffffffc008fa0380 x6 : 0000000000000001
> [14933.607738] x5 : 0000000000000000 x4 : 0000000000000000 x3 : 0000000000000000
> [14933.614853] x2 : ffffff801fe79230 x1 : ffffffc016efd000 x0 : 000000000000003f
> [14933.621970] Call trace:
> [14933.624402]  dev_watchdog+0x358/0x380
> [14933.628048]  call_timer_fn+0x30/0x144
> [14933.631699]  __run_timers.part.0+0x240/0x2ec
> [14933.635954]  run_timer_softirq+0x38/0x70
> [14933.639860]  _stext+0x118/0x348
> [14933.642987]  __irq_exit_rcu+0xdc/0xfc
> [14933.646635]  irq_exit+0xc/0x1c
> [14933.649675]  handle_domain_irq+0x60/0x8c
> [14933.653584]  gic_handle_irq+0x50/0x120
> [14933.657321]  call_on_irq_stack+0x20/0x30
> [14933.661228]  do_interrupt_handler+0x4c/0x54
> [14933.665395]  el1_interrupt+0x2c/0x4c
> [14933.668958]  el1h_64_irq_handler+0x14/0x20
> [14933.673037]  el1h_64_irq+0x78/0x7c
> [14933.676423]  arch_cpu_idle+0x14/0x20
> [14933.679983]  default_idle_call+0x38/0x124
> [14933.683977]  do_idle+0xac/0x110
> [14933.687104]  cpu_startup_entry+0x20/0x50
> [14933.691010]  rest_init+0xc4/0xd0
> [14933.694222]  arch_call_rest_init+0xc/0x14
> [14933.698217]  start_kernel+0x618/0x638
> [14933.701863]  __primary_switched+0xa0/0xa8
> [14933.705857] ---[ end trace f0719674a40d55eb ]---
> [14933.710478] mtk_soc_eth 15100000.ethernet eth0: transmit timed out
> [14935.395565] mtk_soc_eth 15100000.ethernet eth0: Link is Down
> [14935.929138] mtk_soc_eth 15100000.ethernet: warm reset failed
> [14935.947866] mtk_soc_eth 15100000.ethernet eth0: configuring for fixed/2500base-x link mode
> [14935.956254] mtk_soc_eth 15100000.ethernet eth0: Link is Up - 2.5Gbps/Full - flow control rx/tx`
> ```
> 
> ### OpenWrt version
> 23.05.0-rc3 (r26804-a3639e76d6)
> 
> ### OpenWrt target/subtarget
> mediatek/filogic
> 
> ### Device
> Xiaomi Redmi Router AX6000 (stock layout)
> 
> ### Image kind
> Official downloaded image
> 
> ### Steps to reproduce
> _No response_
> 
> ### Actual behaviour
> _No response_
> 
> ### Expected behaviour
> _No response_
> 
> ### Additional info
> _No response_
> 
> ### Diffconfig
> _No response_
> 
> ### Terms
> * [x]  I am reporting an issue for OpenWrt, not an unsupported fork.

Hi are you still having the problem?
Can you describe what setup you are using? Example: ethernet cable types etc.

tczee36, 2023-11-08:
Ended up resolving the issue with ```
Ethtool -k tso off
```
Hope this help others in the future

brada4, 2023-11-08:
ie tso off, firewall offload on ?

tczee36, 2023-11-08:
> ie tso off, firewall offload on ?

I didnt have firewall enabled. 
TSO was causing the problem for me

alanswanson, 2023-11-09:
@tczee36 For how long a continuous uptime have you had TSO disabled and not seen the error?

This was certainly disabled in the [past](https://github.com/openwrt/openwrt/issues/8660) with [commit](https://git.openwrt.org/?p=openwrt/openwrt.git;a=commitdiff;h=f8b849103dfe1b21a6462837efc9460ca3cca02d) for the old soc_mt7621 driver before the combined mtk_soc_eth driver which was hoped (by non-devs) would fix things. Also it's not [certain](https://forum.openwrt.org/t/mtk-soc-eth-watchdog-timeout-after-r11573/50000/417) to be an actual 100% fix but better than the currently long running issues.

(There was also a [commit](https://git.openwrt.org/?p=openwrt/openwrt.git;a=commitdiff;h=ceb1451c10c1588fd009021c0226823a9959e266) a year ago to fix small frags a year that lock up DMA with, I believe, TSO but there are obviously other issues causing us grief.)

So maybe a PR to reimplement an equivalent commit to disable TSO for mtk_soc_eth?

tczee36, 2023-11-09:
> @tczee36 For how long a continuous uptime have you had TSO disabled and not seen the error?
>
> This was certainly disabled in the [past](https://github.com/openwrt/openwrt/issues/8660) with [...]

Had it running for 3 weeks after turning off tso.

Recently updated to 23.0.5 and go through the ethtool disable tso process again. 

Only few days since upgrade, will see how long this last.

chayleaf, 2023-11-24:
I have BPI-R3, which has similar hardware, running NixOS (another Linux distro) with mainline Linux. For a few months it's been working fine, but now I started sometimes getting this error. This means it's most likely a hardware issue, even if it seems to be relatively common. It looks like the chip used for flow offloading is prone to breakage.

tczee36, 2023-11-25:
My router have been rock solid since my last comment. Uptime 20 days.

sawyer8151, 2023-12-14:
> My router have been rock solid since my last comment. Uptime 20 days.

Helllo, friend!

How did you turn tso off? Сan you write in more detail? Did you install Ethtool from rep?

ChernyaevAN, 2023-12-24:
This two commands does **not helps** for Xiaomi Redmi Router AX6S. I think the problem is the same.

ethtool -k eth0 tso off
ethtool -k br-lan tso off

```
[ 1475.020923] ------------[ cut here ]------------
[ 1475.025575] NETDEV WATCHDOG: eth0 (mtk_soc_eth): transmit queue 4 timed out
[ 1475.032627] WARNING: CPU: 1 PID: 0 at dev_watchdog+0x330/0x33c
[ 1475.038484] Modules linked in: pppoe ppp_async nft_fib_inet nf_flow_table_ipv6 nf_flow_table_ipv4 nf_flow_table_inet pppox ppp_generic nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_objref nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_counter nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack mt7915e mt7615e mt7615_common mt76_connac_lib mt76 mac80211 cfg80211 slhc nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 libcrc32c hwmon crc_ccitt compat seqiv leds_gpio gpio_button_hotplug
[ 1475.094302] CPU: 1 PID: 0 Comm: swapper/1 Not tainted 5.15.144 #0
[ 1475.100388] Hardware name: Xiaomi Redmi Router AX6S (DT)
[ 1475.105690] pstate: 40400005 (nZcv daif +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[ 1475.112644] pc : dev_watchdog+0x330/0x33c
[ 1475.116647] lr : dev_watchdog+0x330/0x33c
[ 1475.120648] sp : ffffffc00800bd90
[ 1475.123953] x29: ffffffc00800bd90 x28: 0000000000000140 x27: 00000000ffffffff
[ 1475.131086] x26: 0000000000000000 x25: 0000000000000001 x24: ffffff800036d4c0
[ 1475.138216] x23: 0000000000000000 x22: 0000000000000001 x21: ffffffc008b06000
[ 1475.145346] x20: ffffff800036d000 x19: 0000000000000004 x18: ffffffc008b1a308
[ 1475.152477] x17: ffffffc00744b000 x16: ffffffc008008000 x15: 000000000000039f
[ 1475.159608] x14: 0000000000000135 x13: ffffffc00800bac8 x12: ffffffc008b72308
[ 1475.166739] x11: 712074696d736e61 x10: ffffffc008b72308 x9 : 0000000000000000
[ 1475.173869] x8 : ffffffc008b1a2b8 x7 : ffffffc008b1a308 x6 : 0000000000000001
[ 1475.181000] x5 : 0000000000000000 x4 : 0000000000000000 x3 : 0000000000000000
[ 1475.188129] x2 : ffffff800ff38090 x1 : ffffffc00744b000 x0 : 000000000000003f
[ 1475.195261] Call trace:
[ 1475.197698]  dev_watchdog+0x330/0x33c
[ 1475.201354]  call_timer_fn.constprop.0+0x20/0x80
[ 1475.205966]  __run_timers.part.0+0x208/0x284
[ 1475.210228]  run_timer_softirq+0x38/0x70
[ 1475.214143]  _stext+0x10c/0x274
[ 1475.217277]  __irq_exit_rcu+0xc0/0xe0
[ 1475.220935]  irq_exit+0xc/0x1c
[ 1475.223984]  handle_domain_irq+0x60/0x8c
[ 1475.227903]  gic_handle_irq+0x64/0x8c
[ 1475.231561]  call_on_irq_stack+0x20/0x34
[ 1475.235477]  do_interrupt_handler+0x4c/0x54
[ 1475.239652]  el1_interrupt+0x2c/0x4c
[ 1475.243222]  el1h_64_irq_handler+0x14/0x20
[ 1475.247311]  el1h_64_irq+0x78/0x7c
[ 1475.250704]  arch_cpu_idle+0x14/0x20
[ 1475.254273]  do_idle+0xc0/0x150
[ 1475.257411]  cpu_startup_entry+0x24/0x60
[ 1475.261327]  secondary_start_kernel+0x130/0x140
[ 1475.265850]  __secondary_switched+0x50/0x54
[ 1475.270026] ---[ end trace c3ab7d01a3e26712 ]---
```

DartBond, 2023-12-30:
Redmi AX6000 looks like the same problem
`
[191406.095926] ------------[ cut here ]------------
[191406.100635] NETDEV WATCHDOG: eth0 (mtk_soc_eth): transmit queue 7 timed out
[191406.107686] WARNING: CPU: 2 PID: 0 at 0xffffffc0086c4014
[191406.113069] Modules linked in: pppoe ppp_async nft_fib_inet nf_flow_table_ipv6 nf_flow_table_ipv4 nf_flow_table_inet pppox ppp_generic nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_objref nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_counter nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack mt7915e mt76_connac_lib mt76 mac80211 cfg80211 slhc nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 libcrc32c crc_ccitt compat crypto_safexcel sha1_generic seqiv md5 des_generic libdes authencesn authenc leds_ws2812b leds_gpio gpio_button_hotplug
[191406.173485] CPU: 2 PID: 0 Comm: swapper/2 Not tainted 5.15.137 #0
[191406.179646] Hardware name: Xiaomi Redmi Router AX6000 (stock layout) (DT)
[191406.186498] pstate: 60400005 (nZCv daif +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[191406.193525] pc : 0xffffffc0086c4014
[191406.197084] lr : 0xffffffc0086c4014
[191406.200641] sp : ffffffc008b7bdb0
[191406.204025] x29: ffffffc008b7bdb0 x28: 0000000000000140 x27: 00000000ffffffff
[191406.211228] x26: 0000000000000000 x25: 0000000000000002 x24: ffffff8000d964c0
[191406.218431] x23: 0000000000000000 x22: 0000000000000001 x21: ffffffc008a36000
[191406.225633] x20: ffffff8000d96000 x19: 0000000000000007 x18: ffffffc008a4a2b8
[191406.232835] x17: ffffffc017496000 x16: ffffffc008b78000 x15: 0000000000000423
[191406.240038] x14: 0000000000000161 x13: ffffffc008b7bae8 x12: ffffffc008aa22b8
[191406.247239] x11: 712074696d736e61 x10: ffffffc008aa22b8 x9 : 0000000000000000
[191406.254441] x8 : ffffffc008a4a268 x7 : ffffffc008a4a2b8 x6 : 0000000000000001
[191406.261643] x5 : 0000000000000000 x4 : 0000000000000000 x3 : 0000000000000000
[191406.268844] x2 : 0000000000000040 x1 : 0000000000000004 x0 : 000000000000003f
[191406.276047] Call trace:
[191406.278565]  0xffffffc0086c4014
[191406.281776]  0xffffffc0080a5cc0
[191406.284988]  0xffffffc0080a6538
[191406.288200]  0xffffffc0080a6e88
[191406.291412]  0xffffffc00801010c
[191406.294623]  0xffffffc00803b360
[191406.297835]  0xffffffc00803b490
[191406.301046]  0xffffffc008089614
[191406.304258]  0xffffffc008422e84
[191406.307470]  0xffffffc00801584c
[191406.310682]  0xffffffc00801614c
[191406.313893]  0xffffffc008833050
[191406.317104]  0xffffffc008833804
[191406.320316]  0xffffffc008011374
[191406.323528]  0xffffffc008833de4
[191406.326740]  0xffffffc008069490
[191406.329951]  0xffffffc0080696a4
[191406.333163]  0xffffffc00801ec00
[191406.336375]  0xffffffc00883c2cc
[191406.339587] ---[ end trace 78a96ace43f58ad5 ]---
`

ChernyaevAN, 2024-01-15:
**I have installed swconfig package and this action solve the problem on Redmi AX6S.**

NovaPixell, 2024-01-15:
> **I have installed swconfig package and this action solve the problem on Redmi AX6S.**

Really? Did you test it?

ChernyaevAN, 2024-01-15:
Yes, I have test it on 3 connected TV's with videos streams and 3 computers. Flow control on switch was **on**.
The same comment was [here](https://github.com/openwrt/mt76/issues/211#issuecomment-1206255299)

NovaPixell, 2024-01-15:
But, how this package fix that issue? Is related?

ChernyaevAN, 2024-01-15:
This package uses other drivers for switch. You can see it [here](https://openwrt.org/docs/techref/swconfig). This drivers do not use DSA. They are legacy like it is mentioned on the site, but they works well in the case of stability and speed issues.

NovaPixell, 2024-01-15:
> This package uses other drivers for switch. You can see it [here](https://openwrt.org/docs/techref/swconfig). This drivers do not use DSA. They are legacy like it is mentioned on the site, but they works well in case of stability and speed issues.

So, Mediatek is supported i mean

brada4, 2024-03-22:
> nft_fullcone

please close the issue, this not support forum for random fork scratched off internet.

tczee36, 2024-03-25:
> The bug appears when the video or other stream starts from the devices connected by wires. It turns off wired devices and WAN connection, WLAN connection stay alive. All settings are default. WAN is DHCP over Ethernet.
> 
> This commands does **not helps** for Xiaomi Redmi Router AX6S. I think the problem is the same.
> 
> ethtool -K eth0 tso off gso off gro off rx off tx off txvlan off hw-tc-offload off ethtool -K br-lan tso off gso off gro off rx off tx off txvlan off hw-tc-offload off ethtool -k eth0 tso off ethtool -k br-lan tso off
> 
> ```
> [ 1475.020923] ------------[ cut here ]------------
> [ 1475.025575] NETDEV WATCHDOG: eth0 (mtk_soc_eth): transmit queue 4 timed out
> [ 1475.032627] WARNING: CPU: 1 PID: 0 at dev_watchdog+0x330/0x33c
> [ 1475.038484] Modules linked in: pppoe ppp_async nft_fib_inet nf_flow_table_ipv6 nf_flow_table_ipv4 nf_flow_table_inet pppox ppp_generic nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_objref nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_counter nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack mt7915e mt7615e mt7615_common mt76_connac_lib mt76 mac80211 cfg80211 slhc nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 libcrc32c hwmon crc_ccitt compat seqiv leds_gpio gpio_button_hotplug
> [ 1475.094302] CPU: 1 PID: 0 Comm: swapper/1 Not tainted 5.15.144 #0
> [ 1475.100388] Hardware name: Xiaomi Redmi Router AX6S (DT)
> [ 1475.105690] pstate: 40400005 (nZcv daif +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
> [ 1475.112644] pc : dev_watchdog+0x330/0x33c
> [ 1475.116647] lr : dev_watchdog+0x330/0x33c
> [ 1475.120648] sp : ffffffc00800bd90
> [ 1475.123953] x29: ffffffc00800bd90 x28: 0000000000000140 x27: 00000000ffffffff
> [ 1475.131086] x26: 0000000000000000 x25: 0000000000000001 x24: ffffff800036d4c0
> [ 1475.138216] x23: 0000000000000000 x22: 0000000000000001 x21: ffffffc008b06000
> [ 1475.145346] x20: ffffff800036d000 x19: 0000000000000004 x18: ffffffc008b1a308
> [ 1475.152477] x17: ffffffc00744b000 x16: ffffffc008008000 x15: 000000000000039f
> [ 1475.159608] x14: 0000000000000135 x13: ffffffc00800bac8 x12: ffffffc008b72308
> [ 1475.166739] x11: 712074696d736e61 x10: ffffffc008b72308 x9 : 0000000000000000
> [ 1475.173869] x8 : ffffffc008b1a2b8 x7 : ffffffc008b1a308 x6 : 0000000000000001
> [ 1475.181000] x5 : 0000000000000000 x4 : 0000000000000000 x3 : 0000000000000000
> [ 1475.188129] x2 : ffffff800ff38090 x1 : ffffffc00744b000 x0 : 000000000000003f
> [ 1475.195261] Call trace:
> [ 1475.197698]  dev_watchdog+0x330/0x33c
> [ 1475.201354]  call_timer_fn.constprop.0+0x20/0x80
> [ 1475.205966]  __run_timers.part.0+0x208/0x284
> [ 1475.210228]  run_timer_softirq+0x38/0x70
> [ 1475.214143]  _stext+0x10c/0x274
> [ 1475.217277]  __irq_exit_rcu+0xc0/0xe0
> [ 1475.220935]  irq_exit+0xc/0x1c
> [ 1475.223984]  handle_domain_irq+0x60/0x8c
> [ 1475.227903]  gic_handle_irq+0x64/0x8c
> [ 1475.231561]  call_on_irq_stack+0x20/0x34
> [ 1475.235477]  do_interrupt_handler+0x4c/0x54
> [ 1475.239652]  el1_interrupt+0x2c/0x4c
> [ 1475.243222]  el1h_64_irq_handler+0x14/0x20
> [ 1475.247311]  el1h_64_irq+0x78/0x7c
> [ 1475.250704]  arch_cpu_idle+0x14/0x20
> [ 1475.254273]  do_idle+0xc0/0x150
> [ 1475.257411]  cpu_startup_entry+0x24/0x60
> [ 1475.261327]  secondary_start_kernel+0x130/0x140
> [ 1475.265850]  __secondary_switched+0x50/0x54
> [ 1475.270026] ---[ end trace c3ab7d01a3e26712 ]---
> ```
> 
> Maybe not the same issue but the same result is on OpenWRT 22.03.6.
> 
> ```
> [  463.124444] ------------[ cut here ]------------
> [  463.129070] NETDEV WATCHDOG: eth0 (mtk_soc_eth): transmit queue 0 timed out
> [  463.136058] WARNING: CPU: 0 PID: 0 at 0xffffffc0106b33d0
> [  463.141360] Modules linked in: pppoe ppp_async nft_fib_inet nf_flow_table_ipv6 nf_flow_table_ipv4 nf_flow_table_inet pppox ppp_generic nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_objref nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_counter nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack mt7915e mt7615e mt7615_common mt76_connac_lib mt76 mac80211 cfg80211 slhc nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_log_ipv6 nf_log_ipv4 nf_log_common nf_defrag_ipv6 nf_defrag_ipv4 libcrc32c hwmon crc_ccitt compat seqiv leds_gpio gpio_button_hotplug
> [  463.199030] CPU: 0 PID: 0 Comm: swapper/0 Not tainted 5.10.201 #0
> [  463.205112] Hardware name: Xiaomi Redmi Router AX6S (DT)
> [  463.210415] pstate: 40000005 (nZcv daif -PAN -UAO -TCO BTYPE=--)
> [  463.216412] pc : 0xffffffc0106b33d0
> [  463.219891] lr : 0xffffffc0106b33d0
> [  463.223369] sp : ffffffc010b3bd80
> [  463.226674] x29: ffffffc010b3bd80 x28: 0000000000000140
> [  463.231979] x27: 00000000ffffffff x26: 0000000000000000
> [  463.237284] x25: 0000000000000000 x24: 0000000000000000
> [  463.242588] x23: ffffff80009bb480 x22: 0000000000000001
> [  463.247893] x21: ffffffc010a06000 x20: ffffff80009bb000
> [  463.253196] x19: 0000000000000000 x18: 0000000000000122
> [  463.258501] x17: 00000000ffffffff x16: 0000000000000000
[... 11 lines cut ...]
> [  463.314942]  0xffffffc01009a410
> [  463.318073]  0xffffffc01009ac88
> [  463.321204]  0xffffffc010000af4
> [  463.324336]  0xffffffc01002eb48
> [  463.327467]  0xffffffc01007e020
> [  463.330598]  0xffffffc010000868
> [  463.333730]  0xffffffc0100023c8
> [  463.336861]  0xffffffc01080d5f8
> [  463.339993]  0xffffffc01005c3c4
> [  463.343125]  0xffffffc01005c5a4
> [  463.346256]  0xffffffc01080dd58
> [  463.349387]  0xffffffc0109908d4
> [  463.352518]  0xffffffc010990d60
> [  463.355650] ---[ end trace 5da9dbc0c8a55abf ]---
> ```

update since my prev post, still up and running no disconnect.

<html><body>

Uptime | 141d 5h 20m 44s
-- | --

</body>
</html>