openwrt/openwrt #13263: [22.03-23.05-rc2+] traffic not properly forwarded through a wan-lan VLAN bridge

TCP3ACK, 2023-08-11:
### Describe the bug

### **Backgroud:**
My ISP transmits iptv multicast data via vlan90 tagged while internet data untagged. So i create vlan wan.90 interface on Newifi-D1 to get both data within one cable. Meanwhile i setup a udpxy server on Router2 to transfer the multicast into unicast stream. So i bridge wan&lan on vlan90 to forward the iptv stream to Router2. 

Here's the demonstration of my network for iptv:
 `ISP(vlan90) ----plug into wan----> Newifi-D1:[br-vlan90(wan.90, lan1.90)] ----->Router2:[vlan90(lan2.90)]--->udpxy`

### **Bug:**
**Since 22.03.0-rc1, the iptv traffic was not able to be properly forwarded through vlan bridge interface** on D1 and the udpxy on Router2 no longer recieved iptv data(Tested official release of 22.03.0-rc1, 22.03.5, 23.05-rc2). But same config working on 21.02.7.

Below is my /etc/config/network, which is **working on 21.02.7** but no luck on 22.03+.

```
config interface 'vlan90'
	option proto 'none'
	option device 'br-vlan90'
	option defaultroute '0'
	option peerdns '0'
	option delegate '0'

config device
	option name 'br-vlan90'
	option type 'bridge'
	list ports 'lan1.90'
	list ports 'wan.90'
```

**PS:** 

- firewall3/4 is **NOT** related to this issue. i stop fw but the issue remains.
- igmp settings may **NOT** be related, the vlan bridge interface on D1 work just like a switch(`option proto 'none'`) with no ip assigned. Actually i think it's layer2 related issue and not layer3.

### OpenWrt version

since 22.03.0-rc1 to 23.05-rc2+

### OpenWrt target/subtarget

ramips/mt7621

### Device

Newifi D1 

### Image kind

Official downloaded image

### Steps to reproduce

described above

### Actual behaviour

traffic not properly forwarded through a wan-lan VLAN bridge

### Expected behaviour

traffic being properly forwarded

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-08-11:
add igmp_snooping 1 option to the bridge.

TCP3ACK, 2023-08-12:
> add igmp_snooping 1 option to the bridge.

`option igmp_snooping '1'` added but still **not** working. Tested on both 22.03.5 and 23.05-rc2 release image. Worked on 21.02.7 either w/wo `igmp_snooping '1'` though. Also `net.ipv4.conf.all.force_igmp_version=2` is always set in /etc/sysctl.conf.

brada4, 2023-08-12:
Please check tcpdump on both interfaces if channel switch makes igmp requests and stream comes back, i.e where packets disappear.
Switch shall work according to your expectation, especially with 2 ports it should forward igmp and streams between both ports without any extra instrumentation.

TCP3ACK, 2023-08-12:
Tcpdump on 22.03+ on Newifi D1, igmp reports sended by udpxy from Router2 vlan90 interface(192.168.90.1) captured, but **no udp stream comes back after igmp report**. Eventually it sended igmp leave report after timeout.
```
root@Newifi:/tmp# tcpdump -n -i br-vlan90 -vvv
tcpdump: listening on br-vlan90, link-type EN10MB (Ethernet), snapshot length 262144 bytes
09:25:21.706714 IP (tos 0x0, ttl 1, id 0, offset 0, flags [DF], proto IGMP (2), length 32, options (RA))
    192.168.1.3 > 224.0.0.1: igmp query v2 [max resp time 10]
09:25:52.301192 IP (tos 0xc0, ttl 1, id 0, offset 0, flags [DF], proto IGMP (2), length 32, options (RA))
    192.168.90.1 > 233.99.65.5: igmp v2 report 233.99.65.5
09:25:56.661227 IP (tos 0xc0, ttl 1, id 0, offset 0, flags [DF], proto IGMP (2), length 32, options (RA))
    192.168.90.1 > 233.99.65.5: igmp v2 report 233.99.65.5
09:25:57.302526 IP (tos 0xc0, ttl 1, id 0, offset 0, flags [DF], proto IGMP (2), length 32, options (RA))
    192.168.90.1 > 224.0.0.2: igmp leave 233.99.65.5
^C
4 packets captured
4 packets received by filter
0 packets dropped by kernel
```

TCP3ACK, 2023-08-12:
Udpxy log at the same time.
```
2023-08-12 09:24:44.681211 EDT	S(7792)	udpxy 1.0-25.1 (prod) standard [Linux 5.10.109 mips]: /usr/bin/udpxy -T -v -S -p 4022 -m lan2.90 -c 20 -l /tmp/udpxy.log -B 65536 
2023-08-12 09:24:44.681735 EDT	S(7792)	Server is starting up, max clients = [20]
2023-08-12 09:24:44.682247 EDT	S(7792)	Setting up listener for [0.0.0.0:4022]
2023-08-12 09:24:44.682687 EDT	S(7792)	Setting low watermark for server socket [8] to [10]
2023-08-12 09:24:44.683090 EDT	S(7792)	Created server socket=[8], backlog=[16]
2023-08-12 09:24:44.683450 EDT	S(7792)	Entering server loop [pselect(2)]
2023-08-12 09:24:44.683790 EDT	S(7792)	Waiting for input from [2] fd's, NO timeout
2023-08-12 09:25:52.261427 EDT	S(7792)	No children exited since last check
2023-08-12 09:25:52.261954 EDT	S(7792)	Got 1 requests
2023-08-12 09:25:52.262224 EDT	S(7792)	Accepting new connection
2023-08-12 09:25:52.262545 EDT	S(7792)	Accepted socket=[9] from 192.168.10.141:53604 n=1/nmax=16
2023-08-12 09:25:52.262803 EDT	S(7792)	Accepting new connection
2023-08-12 09:25:52.263092 EDT	S(7792)	Nothing more to accept
2023-08-12 09:25:52.263335 EDT	S(7792)	accept_requests: Sockets accepted: [1]
2023-08-12 09:25:52.263585 EDT	S(7792)	Waiting for input from [3] fd's, with timeout
2023-08-12 09:25:52.271699 EDT	S(7792)	No children exited since last check
2023-08-12 09:25:52.272172 EDT	S(7792)	Got 1 requests
pre-process sockets [1]: 9
2023-08-12 09:25:52.272443 EDT	S(7792)	acting on accepted socket [9] (1/1)
2023-08-12 09:25:52.272698 EDT	S(7792)	Reading command from socket [9]
2023-08-12 09:25:52.272988 EDT	S(7792)	HTTP buffer [153 bytes] received
GET /rtp/233.99.65.5:5210 HTTP/1.1
User-Agent: libmpv
Accept: */*
Range: bytes=0-
Connection: close
Host: 192.168.10.2:4022
Icy-MetaData: 1

2023-08-12 09:25:52.273261 EDT	S(7792)	Request=[rtp/233.99.65.5:5210], length=[21]
2023-08-12 09:25:52.273514 EDT	S(7792)	Command [rtp] with params [233.99.65.5:5210], tail [] read from socket=[9]
2023-08-12 09:25:52.273768 EDT	S(7792)	udp_relay : new_socket=[9] param=[233.99.65.5:5210]
2023-08-12 09:25:52.274547 EDT	S(7792)	Added client: pid=[7826], maddr=[233.99.65.5], mport=[5210], saddr=[192.168.10.141], sport=[53604]
2023-08-12 09:25:52.274620 EDT	S(7792)	Client process=[7826] started for socket=[9]
2023-08-12 09:25:52.274963 EDT	S(7792)	process_requests: closing accepted socket [9]
2023-08-12 09:25:52.275071 EDT	c(7826)	min socket buffer = [65536], max space to use = [1500], Rmsgs = [1]
2023-08-12 09:25:52.275293 EDT	S(7792)	Processed [1/1] accepted sockets
newly-accepted sockets [1]: -1
2023-08-12 09:25:52.275371 EDT	c(7826)	Setting up multicast listener
2023-08-12 09:25:52.275596 EDT	S(7792)	All accepted sockets processed
2023-08-12 09:25:52.275703 EDT	c(7826)	current receive buffer size is [180224] bytes for socket [6]
2023-08-12 09:25:52.275882 EDT	S(7792)	Waiting for input from [2] fd's, NO timeout
2023-08-12 09:25:52.276093 EDT	c(7826)	multicast-group [ADD]
2023-08-12 09:25:52.276382 EDT	c(7826)	Ifc-bound mcast listener socket=[6]
2023-08-12 09:25:52.276666 EDT	c(7826)	Mcast listener socket=[6] set up
2023-08-12 09:25:52.276926 EDT	c(7826)	min socket buffer = [65536], max space to use = [1500], Rmsgs = [1]
2023-08-12 09:25:52.277177 EDT	c(7826)	Data buffer will hold up to [1] messages
2023-08-12 09:25:52.277429 EDT	c(7826)	RTP (over UDP) stream assumed, no checks
2023-08-12 09:25:52.277746 EDT	c(7826)	socket 6: RCV timeout set to 0 sec, 5 usec
2023-08-12 09:25:52.278012 EDT	c(7826)	socket 6: SEND timeout set to 0 sec, 5 usec
2023-08-12 09:25:52.278269 EDT	c(7826)	current send buffer size is [44800] bytes for socket [9]
2023-08-12 09:25:52.278523 EDT	c(7826)	current receive buffer size is [180224] bytes for socket [6]
2023-08-12 09:25:52.278872 EDT	c(7826)	send buffer size set to [180224] bytes for socket [9]
2023-08-12 09:25:52.279533 EDT	c(7826)	Sent HTTP response code=[200], reason=[OK] to socket=[9]
HTTP/1.1 200 OK
Server: udpxy 1.0-25.1 (prod) standard [Linux 5.10.109 mips]
Content-Type:application/octet-stream

**2023-08-12 09:25:52.279935 EDT	c(7826)	Relaying traffic from socket[6] to socket[9], buffer size=[65536], Rmsgs=[1], pauses=[0]
2023-08-12 09:25:57.298888 EDT	c(7826)	read_buf: socket time-out on read2023-08-12 09:25:57.299469 EDT	c(7826)	read_data - EOF
2023-08-12 09:25:57.299760 EDT	c(7826)	Exited relay loop: received=[-1], sent=[0], quit=[0]**
2023-08-12 09:25:57.300356 EDT	c(7826)	multicast-group [DROP]
2023-08-12 09:25:57.300945 EDT	c(7826)	Mcast listener socket=[6] closed
2023-08-12 09:25:57.301739 EDT	c(7826)	Child process=[7826] exits with rc=[0]
2023-08-12 09:25:57.302821 EDT	S(7792)	*** Caught SIGCHLD (18) ***
2023-08-12 09:25:57.303278 EDT	S(7792)	Waiting on exited children
2023-08-12 09:25:57.303639 EDT	S(7792)	Client [7826] has exited.
2023-08-12 09:25:57.304069 EDT	S(7792)	Deleted client: pid=[7826]
2023-08-12 09:25:57.304336 EDT	S(7792)	Cleaned up 1 children, 0 still running
2023-08-12 09:25:57.304590 EDT	S(7792)	INTERRUPTED, yet will continue.
2023-08-12 09:25:57.304835 EDT	S(7792)	Waiting for input from [2] fd's, NO timeout

```

TCP3ACK, 2023-08-12:
And below is tcpdump on 21.02.7, udp stream immediately flooded in after igmp report, working as expected,

```
root@Newifi:/tmp# tcpdump -n -i br-vlan90 -vvv
09:36:31.957181 IP (tos 0xc0, ttl 1, id 0, offset 0, flags [DF], proto IGMP (2), length 32, options (RA))
    192.168.90.1 > 233.99.65.5: igmp v2 report 233.99.65.5
09:36:31.997867 IP (tos 0xa0, ttl 124, id 0, offset 0, flags [DF], proto UDP (17), length 1356)
    206.71.51.1.1234 > 233.99.65.5.5210: [udp sum ok] UDP, length 1328
09:36:31.999654 IP (tos 0xa0, ttl 124, id 0, offset 0, flags [DF], proto UDP (17), length 1356)
    206.71.51.1.1234 > 233.99.65.5.5210: [udp sum ok] UDP, length 1328
09:36:32.000485 IP (tos 0xa0, ttl 124, id 0, offset 0, flags [DF], proto UDP (17), length 1356)
    206.71.51.1.1234 > 233.99.65.5.5210: [udp sum ok] UDP, length 1328
    ...
    192.168.90.1 > 224.0.0.2: igmp leave 233.99.65.5
```

brada4, 2023-08-12:
`2023-08-12 09:25:57.300356 EDT	c(7826)	multicast-group [DROP]`

Is this just red herring or "old" version does fine?

TCP3ACK, 2023-08-12:
> `2023-08-12 09:25:57.300356 EDT c(7826) multicast-group [DROP]`
> 
> Is this just red herring or "old" version does fine?

timeout and trigger igmp leave report at that moment.
```
09:25:57.302526 IP (tos 0xc0, ttl 1, id 0, offset 0, flags [DF], proto IGMP (2), length 32, options (RA))
    192.168.90.1 > 224.0.0.2: igmp leave 233.99.65.5
```

brada4, 2023-08-12:
Does the join message get ever forwarded to vlan "upstream"?

TCP3ACK, 2023-08-13:
> Does the join message get ever forwarded to vlan "upstream"?

After further investigation, something could go wrong in the wan.90 vlan device in the bridge. It never recieves and thus cannot upload the igmp join/leave reports up to the upstream(ISP modem). However it  can recieve the igmp query v2/v3 message periodically sent by the ISP which i dont need. The lan.90 device is ok, which it recieves the igmp join/leave reports from udpxy. The igmp join/leave report is missing on the wan.90, which normally it should recieve from lan.90 in the same bridge. **In short, the data flow backed to the ISP upwards is interrupted in the wan.90.**

On 21.02 everything works w/o issue. The join report can be transmitted to the ISP and then iptv stream comes in.

brada4, 2023-08-13:
Check if it does not get out untagged or with tag 0

TCP3ACK, 2023-08-14:
Problem Solved. Turns out on 22..03+, creating software vlan (aka wan.90, lan1.90) directly and bridging them together got problem when vlan data sent out(egress). Have to create a clean bridge and use the bridge-vlan-filtering feature to segement the vlan network, which is DSA approach.

But it's strange as Newifi D1 should already move to DSA on 21.02.7 and i dont need the bridge-vlan-filtering feature enabled to work. The vlan egress is fine when just using software vlan directly bridge. Is there any change of DSA between 21.02.7 and 22.03+?

I'll leave the issue open atm but its ok if the Mod decides to close the issue.