openwrt/openwrt #13275: iwinfo: build error with LTO

Jalakas, 2023-08-12:
### Describe the bug

Building package "iwinfo" fails with option "CONFIG_USE_LTO=y"

`[...]
<artificial>:(.text+0x400c): relocation R_MIPS16_26 against `iwinfo_close' cannot be used when making a shared object; recompile with -fPIC
[openwrt]/staging_dir/toolchain-mips_24kc_gcc-12.3.0_musl/lib/gcc/mips-openwrt-linux-musl/12.3.0/../../../../mips-openwrt-linux-musl/bin/ld.bfd: non-dynamic relocations refer to dynamic symbol strcpy
[openwrt]/staging_dir/toolchain-mips_24kc_gcc-12.3.0_musl/lib/gcc/mips-openwrt-linux-musl/12.3.0/../../../../mips-openwrt-linux-musl/bin/ld.bfd: failed to set dynamic section sizes: bad value
collect2: error: ld returned 1 exit status
[...]`

Package builds fine when:
`PKG_BUILD_FLAGS:=no-lto`
is added into makefile "package/network/utils/iwinfo/Makefile"

Similar fix:
https://github.com/openwrt/openwrt/commit/906616d20183bb7be4eb71812ef5e76cb3af56a0

### OpenWrt version

main

### OpenWrt target/subtarget

CONFIG_TARGET_ath79_generic

### Device

CONFIG_TARGET_ath79_generic_DEVICE_tplink_tl-wr842n-v3

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

Jalakas, 2023-10-07:
Fixed in pull request: https://github.com/openwrt/openwrt/pull/13625

closed 2023-12-07 by Jalakas

KaiJan57, 2024-04-25:
The same fix works for `wireless-tools` package, just fyi