openwrt/openwrt #13277: Hyper-V integration degraded in 23.05.0 RCs, hv_utils not loaded

maurice-w, 2023-08-12:
### Describe the bug

Hyper-V integration services are partially non-functional in OpenWrt 23.05.0-rc1/rc2, specifically features which require `hv_utils`. Dynamic memory assignment still works, but neither does heartbeat (no contact) nor shut down (hypervisor can't send shut down command to VM).

Both heartbeat and shut down work fine with 22.03.5 and previous versions.

### OpenWrt version

r23228-cd17d8df2a

### OpenWrt target/subtarget

x86/64

### Device

Microsoft Hyper-V Generation 2

### Image kind

Official downloaded image

### Steps to reproduce

Download [openwrt-23.05.0-rc2-x86-64-generic-squashfs-combined-efi.img.gz](https://downloads.openwrt.org/releases/23.05.0-rc2/targets/x86/64/openwrt-23.05.0-rc2-x86-64-generic-squashfs-combined-efi.img.gz), extract, convert to VHDX (`qemu-img convert -O vhdx`), attach to Hyper-V Gen2 VM, boot.

Initiate VM shut down from Hyper-V (not from within the VM).

### Actual behaviour

Hyper-V reports "Heartbeat: No Contact".

When sending shut down command, Hyper-V reports "Failed to stop: The device is not ready for use. (0x800710DF)". OpenWrt does not shut down.

### Expected behaviour

Hyper-V heartbeat should report "OK (No Application Data)".

Hyper-V shut down command should trigger OpenWrt to shut down and log "hv_utils: Shutdown request received - graceful shutdown initiated".

That's the behaviour in 22.03.5 and previous versions.

### Additional info

I can provide SSH access to live VMs if required for debugging.

These relevant lines from a 22.03.5 system log are missing from a 23.05.0-rc2 system log:

```
fb0: switching to hyperv_fb from simple
hv_utils: Registering HyperV Utility Driver
hv_vmbus: registering driver hv_utils
hv_utils: Heartbeat IC version 3.0
hv_utils: cannot register PTP clock: 0
hv_utils: Shutdown IC version 3.2
hv_utils: TimeSync IC version 4.0
hv_utils: VSS IC version 5.0
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

hauke, 2023-08-13:
Please post the full dmesg output from the openwrt 23.05.0-rc2 image and the 22.03 image booting in Hyper-V.

The driver handling Hyper-V should be compiled into the image:
```
hauke@hauke-t480:~/openwrt/openwrt-23.05$ git grep HYPERV_UTILS
target/linux/x86/64/config-5.15:CONFIG_HYPERV_UTILS=y
target/linux/x86/generic/config-5.15:CONFIG_HYPERV_UTILS=y
```
https://elixir.bootlin.com/linux/v5.15.126/source/drivers/hv/hv_util.c#L245

maurice-w, 2023-08-13:
Thanks for taking a look, @hauke!

> Your should see `Registering HyperV Utility Driver` in OpenWrt 23.05 at bootup.

Nope, unfortunately not. I do see that in 22.03.5, but not in 23.05.0-rc2. No mention of `hv_utils` in the 23.05.0-rc2 dmesg at all.

Full dmesg output from both images booting in the same Hyper-V VM below.

### openwrt-23.05.0-rc2-x86-64-generic-squashfs-combined-efi
```
[    0.000000] Linux version 5.15.118 (builder@buildhost) (x86_64-openwrt-linux-musl-gcc (OpenWrt GCC 12.3.0 r23228-cd17d8df2a) 12.3.0, GNU ld (GNU Binutils) 2.40.0) #0 SMP Mon Jun 26 11:20:39 2023
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz root=PARTUUID=150ec62c-2527-2a0b-e2d5-0346afe96102 rootwait console=tty0 console=ttyS0,115200n8 noinitrd
[    0.000000] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
[    0.000000] x86/fpu: Enabled xstate features 0x3, context size is 576 bytes, using 'compacted' format.
[    0.000000] signal: max sigframe size: 1104
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009ffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000000c0000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x0000000007ff1fff] usable
[    0.000000] BIOS-e820: [mem 0x0000000007ff2000-0x0000000007ff2fff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000007ff3000-0x000000000ec0dfff] usable
[    0.000000] BIOS-e820: [mem 0x000000000ec0e000-0x000000000ec0efff] ACPI data
[    0.000000] BIOS-e820: [mem 0x000000000ec0f000-0x000000000ee79fff] usable
[    0.000000] BIOS-e820: [mem 0x000000000ee7a000-0x000000000ee98fff] ACPI data
[    0.000000] BIOS-e820: [mem 0x000000000ee99000-0x000000000eef1fff] usable
[    0.000000] BIOS-e820: [mem 0x000000000eef2000-0x000000000ef1afff] reserved
[    0.000000] BIOS-e820: [mem 0x000000000ef1b000-0x000000000ff9afff] usable
[    0.000000] BIOS-e820: [mem 0x000000000ff9b000-0x000000000ffcafff] type 20
[    0.000000] BIOS-e820: [mem 0x000000000ffcb000-0x000000000fff2fff] reserved
[    0.000000] BIOS-e820: [mem 0x000000000fff3000-0x000000000fffafff] ACPI data
[    0.000000] BIOS-e820: [mem 0x000000000fffb000-0x000000000fffefff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x000000000ffff000-0x000000000fffffff] usable
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] efi: EFI v2.70 by Microsoft
[    0.000000] efi: ACPI=0xfffa000 ACPI 2.0=0xfffa014 SMBIOS=0xffd8000 SMBIOS 3.0=0xffd6000 MEMATTR=0xf89ba98
[    0.000000] SMBIOS 3.1.0 present.
[    0.000000] DMI: Microsoft Corporation Virtual Machine/Virtual Machine, BIOS Hyper-V UEFI Release v4.1 04/06/2022
[    0.000000] Hypervisor detected: Microsoft Hyper-V
[    0.000000] Hyper-V: privilege flags low 0xae7f, high 0x3b8030, hints 0xe0e24, misc 0xe0bed7b2
[    0.000000] Hyper-V Host Build:22621-10.0-2-0.2134
[    0.000000] Hyper-V: Nested features: 0x0
[    0.000000] Hyper-V: LAPIC Timer Frequency: 0x1e8480
[    0.000000] Hyper-V: Using hypercall for remote TLB flush
[    0.000000] clocksource: hyperv_clocksource_tsc_page: mask: 0xffffffffffffffff max_cycles: 0x24e6a1710, max_idle_ns: 440795202120 ns
[    0.000003] tsc: Detected 1113.602 MHz processor
[    0.000011] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.000013] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.000019] last_pfn = 0x10000 max_arch_pfn = 0x400000000
[    0.000052] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WP  UC- WT
[    0.008221] Secure boot disabled
[    0.008224] ACPI: Early table checksum verification disabled
[    0.008227] ACPI: RSDP 0x000000000FFFA014 000024 (v02 VRTUAL)
[    0.008231] ACPI: XSDT 0x000000000FFF90E8 00005C (v01 VRTUAL MICROSFT 00000001 MSFT 00000001)
[    0.008236] ACPI: FACP 0x000000000FFF8000 000114 (v06 VRTUAL MICROSFT 00000001 MSFT 00000001)
[    0.008241] ACPI: DSDT 0x000000000EE7A000 01E191 (v02 MSFTVM DSDT01   00000001 MSFT 05000000)
[    0.008244] ACPI: FACS 0x000000000FFFE000 000040
[    0.008246] ACPI: OEM0 0x000000000FFF7000 000064 (v01 VRTUAL MICROSFT 00000001 MSFT 00000001)
[    0.008249] ACPI: WAET 0x000000000FFF6000 000028 (v01 VRTUAL MICROSFT 00000001 MSFT 00000001)
[    0.008252] ACPI: APIC 0x000000000FFF5000 000050 (v04 VRTUAL MICROSFT 00000001 MSFT 00000001)
[    0.008254] ACPI: SRAT 0x000000000FFF4000 0002C0 (v02 VRTUAL MICROSFT 00000001 MSFT 00000001)
[    0.008257] ACPI: BGRT 0x000000000FFF3000 000038 (v01 VRTUAL MICROSFT 00000001 MSFT 00000001)
[    0.008259] ACPI: FPDT 0x000000000EC0E000 000034 (v01 VRTUAL MICROSFT 00000001 MSFT 00000001)
[    0.008261] ACPI: Reserving FACP table memory at [mem 0xfff8000-0xfff8113]
[    0.008263] ACPI: Reserving DSDT table memory at [mem 0xee7a000-0xee98190]
[    0.008264] ACPI: Reserving FACS table memory at [mem 0xfffe000-0xfffe03f]
[    0.008265] ACPI: Reserving OEM0 table memory at [mem 0xfff7000-0xfff7063]
[    0.008266] ACPI: Reserving WAET table memory at [mem 0xfff6000-0xfff6027]
[    0.008267] ACPI: Reserving APIC table memory at [mem 0xfff5000-0xfff504f]
[    0.008268] ACPI: Reserving SRAT table memory at [mem 0xfff4000-0xfff42bf]
[... 613 lines cut ...]
[    5.706167] procd: - ubus -
[    5.762102] procd: - init -
[    5.839812] urngd: v1.0.2 started.
[    5.845581] random: crng init done
[    5.846176] random: 25 urandom warning(s) missed due to ratelimiting
[    5.858078] kmodloader: loading kernel modules from /etc/modules.d/*
[... cut ...]
[    5.878911] PPP generic driver version 2.4.2
[    5.880200] NET: Registered protocol family 24
[    5.882121] kmodloader: done loading kernel modules from /etc/modules.d/*
[    8.626484] br-lan: port 1(eth0) entered blocking state
[    8.627311] br-lan: port 1(eth0) entered disabled state
[    8.628186] device eth0 entered promiscuous mode
[    8.629608] br-lan: port 1(eth0) entered blocking state
[    8.630434] br-lan: port 1(eth0) entered forwarding state
[    9.628580] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
```

hauke, 2023-08-13:
#13284 activates the CONFIG_HYPERV_UTILS option and should fix your problem.

maurice-w, 2023-08-14:
Thank you very much! I have subscribed to that PR and will test a snapshot once it has been merged. Will report back.

closed 2023-08-14 by openwrt-bot

maurice-w, 2023-08-20:
Confirmed working in snapshot r23782-ac68fbf526! 👍

commit 54d470ed (2023-08-14): x86: Add virtualization time sync support

This compiles the CONFIG_PTP_1588_CLOCK support into the kernel binary
and activates the drivers for KVM and VMware which allow syncing the
host time with the VM when OpenWrt is running in a VM. With this change
the CONFIG_HYPERV_UTILS driver is now build into the kernel, because it
depends on the PTP framework being compiled in. CONFIG_HYPERV_UTILS was
build as a module, but not packages before.

Fixes: #13277
PR #13284 (2023-08-14): x86: Add virtualization time sync support
This compiles the CONFIG_PTP_1588_CLOCK support into the kernel binary and activates the drivers for KVM and VMware which allow syncing the host time with the VM when OpenWrt is running in a VM. With this change the CONFIG_HYPERV_UTILS driver is now build into the kernel, because it depends on the PTP framework being compiled in. CONFIG_HYPERV_UTILS was build as a module, but not packages before.

Fixes: #13277