openwrt/openwrt #13280: 5G Wifi cannot be started correctly in China

HeHeHeeeeee, 2023-08-13:
### Describe the bug

1. My wireless network card is ax210 pci-e
2. After turning on wireless and setting the 5G signal, it is impossible to find a 5G WiFi when turned on.
3. The 5G frequency band supported by China cannot be found in Luci's wireless interface at all.
4. 2.4G network also has issues, For example, if you turn on auto in the wireless frequency band, you cannot find WiFi, but if you choose to confirm a certain frequency band (1-13), it can be turned on smoothly, Is there a problem with your algorithm?
5. 2.4G and 5G cannot be turned on together. When I reinstall the system, there is only one radio0 when it starts. If I add a radio1 in/etc/config/wireless and find that one of the radio frequencies starts and the other will be knocked out, is it a network card issue or a system issue?

### OpenWrt version

r23691-9194fcb1a4

### OpenWrt target/subtarget

x86/64

### Device

QEMU Standard PC (i440FX + PIIX, 1996)

### Image kind

Self-built image

### Steps to reproduce

1.Enable wifi in the wireless section of the Network tab

### Expected behaviour

I hope the wireless module can become better and better used, after all, the current wireless modules are indeed a bit stretched.

### Additional info

There are no available 5G frequency bands in China

[image]

/etc/config/wireless
（default only radio0）

config wifi-device 'radio0'
        option type 'mac80211'
        option path 'pci0000:00/0000:00:10.0'
        option channel '3'
        option band '2g'
        option htmode 'HE40'
        option txpower '22'
        option cell_density '0'
        option country 'CN'

config wifi-iface 'wifinet0'
        option device 'radio0'
        option mode 'ap'
        option ssid 'OpenWrt2'
        option encryption 'none'
        option network 'lan'

(manual add radio1 but )
config wifi-device 'radio1'
        option type 'mac80211'
        option channel '4'
        option path 'pci0000:00/0000:00:10.0'
        option hwmode '11g'
        option band '5g'
        option htmode 'HE40'
        option txpower '20'
        option cell_density '0'
        option country 'CN'

[... 386 lines cut ...]
CONFIG_PACKAGE_uuidgen=y
CONFIG_PACKAGE_vxlan=y
CONFIG_PACKAGE_wavemon=y
CONFIG_PACKAGE_wifischedule=y
CONFIG_PACKAGE_wireguard-tools=y
CONFIG_PACKAGE_wireless-regdb=y
CONFIG_PACKAGE_wireless-tools=y
CONFIG_PACKAGE_wpad=y
CONFIG_PACKAGE_wsdd2=y
CONFIG_PACKAGE_xtables-nft=y
CONFIG_PACKAGE_zlib=y
CONFIG_SAMBA4_SERVER_AVAHI=y
CONFIG_SAMBA4_SERVER_NETBIOS=y
CONFIG_SAMBA4_SERVER_VFS=y
CONFIG_SAMBA4_SERVER_WSDD2=y
# CONFIG_TARGET_ROOTFS_EXT4FS is not set
CONFIG_WPA_MBO_SUPPORT=y
CONFIG_WPA_MSG_MIN_PRIORITY=3
# CONFIG_PACKAGE_kmod-crypto-kpp is not set
# CONFIG_PACKAGE_kmod-inet-diag is not set
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

coudu, 2023-08-13:
Not a bug, Intel wireless devices cannott be used in AP mode on the 5GHz band and those devices are not DBDC (dual band dual concurent) , it's either 2.4 or 5 GHz at a time, not both together.
Sorry.

pkgadd, 2023-08-13:
Intel is managing the regulatory domain rules entirely inside their proprietary firmware, it listens to the environment to learn where it is located (based on the IEEE 802.11d sent out by the APs in range, this also means that you'll have a problem if one of your neighbours sends out a wrong country code!). There is absolutely nothing that could be done by linux and its drivers, if there really is a problem, you would have to discuss with with Intel engineers via bugzilla.kernel.org (or their own bugtracker(?)), so they can release new firmware blobs.

But there are several other things to keep in mind as well:
- you seem to be running OpenWrt in a virtual machine, which introduces yet another potential problem, as iwlwifi is very -very- *very* sensitive to details of the PCIe bus (even on the bare iron, these cards may not work in all PCIe slots). At least for testing, do test on the real hardware without virtualization - just boot OpenWrt from a USB stick.
- you can replace `/lib/firmware/iwlwifi-ty-a0-gf-a0*` with newer versions, which may (or may not) have fixed this, as iwlwifi isn't really in the prime focus of OpenWrt, the versions might be lagging behind.
- as coudu mentioned, there is no way to use ax210 in AP mode (apart from 2.4 GHz, in a very limited way)
- as coudu mentioned, one radio, one mac, one phy - this card can only tune into a single channel (and band) at a time.

closed 2023-08-14 by hauke

HeHeHeeeeee, 2023-08-14:
> 英特尔完全在其专有固件中管理监管域规则，它侦听环境以了解其位置（基于范围内AP发出的IEEE 802.11d监管提示，这也意味着如果您的邻居之一发送了错误的国家/地区代码，您将遇到问题！linux 及其驱动程序绝对无能为力，如果确实存在问题，您必须通过 bugzilla.kernel.org（或他们自己的错误跟踪器（？））与英特尔工程师讨论，以便他们可以发布新的固件 blob。
> 
> 但还有其他几件事需要记住：
> 
> * 您似乎在虚拟机中运行OpenWrt，这引入了另一个潜在问题，因为iwlwifi对PCIe总线的细节_非常非常敏感_（即使在裸熨斗上，这些卡可能无法在所有PCIe插槽中工作）。至少对于测试，在没有虚拟化的情况下在真实硬件上进行测试 - 只需从USB记忆棒启动OpenWrt即可。
> * 您可以替换为较新的版本，这些版本可能（也可能没有）解决此问题，因为iwlwifi并不是OpenWrt的主要焦点，版本可能会落后。`/lib/firmware/iwlwifi-ty-a0-gf-a0*`
> * 正如 Coudu 提到的，没有办法在 AP 模式下使用 AX210（除了 2.4 GHz，以非常有限的方式）
> * 正如 Coudu 提到的，一个收音机、一台 Mac、一个 PHY - 这张卡一次只能调谐到一个频道（和频段）。

Thank you for your complete explanation. I would like to know how you learned so much knowledge. To be honest, I feel that there are still many things I don't know in computer science. What videos do I need to watch to learn better, or other ways? Also, I would like to know why people like you who maintain open source code do this? What benefits can one gain from such hard work? Don't you have to work?

HeHeHeeeeee, 2023-08-14:
> 我将这些设备上使用的固件更新为[00ce1c0](https://github.com/openwrt/openwrt/commit/00ce1c05877ac38f5ff00c1c2ce3468d5168f76c)中设备支持的最新固件。也许这会有所帮助。
> 
> 谢谢你的详细解释，我现在就关闭这张票。

Thank you, but I have tried it before and it's not very useful. It seems that OpenWrt cannot be used as a WiFi connection device, and the effect is much worse than hard routing.

HeHeHeeeeee, 2023-08-14:
> 这不是一个错误，英特尔无线设备不能在 5GHz 频段上的 AP 模式下使用，并且这些设备不是 DBDC（双频双并发），一次是 2.4 或 5 GHz，而不是两者兼而有之。 不好意思。

Thank you, I understand