openwrt/openwrt #13283: SSID from Wi-Fi list disappears during download

uncharted-user, 2023-08-13:
### Describe the bug

SSID from Wi-Fi list disappears when downloading a large file and takes a few minutes to reappear on the Wi-Fi list. It requires disconnection and reconnection of Wi-Fi to enable internet most of the time. This issue happens on Windows, macOS and, Android. 

### OpenWrt version

OpenWrt 22.03.5 r20134-5f15225c1e

### OpenWrt target/subtarget

ramips/mt76x8

### Device

Netgear R6120

### Image kind

Official downloaded image

### Steps to reproduce

1. Enable all radios.
2. Start downloading any large file such as any ISO (>3 GB),
3. SSID will disappear at any moment.

### Actual behaviour

SSID from Wi-Fi list disappears when downloading a large file and takes a few minutes to reappear on the Wi-Fi list.

### Expected behaviour

A stable Wi-Fi is expected no matter the size or duration of the download. I did not experience any similar issue with the factory firmware.

### Additional info

Disabling one radio reduces the issue significantly.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

khanjui, 2023-08-14:
https://github.com/openwrt/mt76/issues/719#issuecomment-1677257405

hauke, 2023-08-27:
Should be fixed in 76b1e564d202c09d0035315eb6e958a9b0dd4491.

closed 2023-08-27 by hauke