openwrt/openwrt #13289: be2net 10Gbit NIC stops responding

grandiego, 2023-08-14:
### Describe the bug

Fujitsu Futro s930 (AMD GX-424CC 4 core 2.4 GHz x86 64 bit processor, 4 GB RAM), HP NC552SFP dual 10GBit SFP+ pcie card. Openwrt 23.05-rc2. BIOS and NIC firmware upgraded to latest versions.

kmod-be2net module loaded. 
Cisco SFP 1000Base-T plugged in NIC, cat6 cable to my laptop.
Running iperf -s on openwrt and iperf -c on laptop, after several seconds connection stalls, no traffic flows, cannot ping openwrt computer. Detaching and reattaching cable or SFP module makes no difference. I have to reboot openwrt in order to make it responsive again. No messages in dmesg or logread.

If I boot ubuntu 23.04 instead, I can run iperf for hours, no problem.

### OpenWrt version

r23228-cd17d8df2a

### OpenWrt target/subtarget

x86/64

### Device

FUJITSU FUTRO S930

### Image kind

Official downloaded image

### Steps to reproduce

opkg update
opkg install kmod-be2net
<configure NIC, assign IP address, etc>
opkg install iperf
iperf -s

Connect using a patch cable to another computer
run iperf -c <ip of openwrt> 

### Actual behaviour

iperf stalls, (sometimes it takes a couple of runs to trigger the bug), times out and then openwrt no longer responds to ping or is able to send or receive traffic thru be2net NIC.

### Expected behaviour

be2net driver should not freeze NIC

### Additional info

If I run the same test using ubuntu 23.04 instead of OpenWrt 23.05, everything works OK.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

hauke, 2023-08-14:
I do not think that we can help there much from the OpenWrt side, maybe you should send a mail to the Linux netdev mailing list ( netdev@vger.kernel.org ) and ask for help there. OpenWrt does not have any special modifications related to be2net compared to the upstream kernel in OpenWrt.

brada4, 2023-08-15:
- install `irqbalance` , ubuntu has it by default on a SMP system.
- Compare firmware versions in `ethtool -i ethX` -i.e if ubuntu loads some better firmware code.
- - also save boot messages around bringing card up
- x4 PCIe 3.0 is just hair bigger than 40Gbps, try to confirm 2nd port is configured DOWN.
- also save ethtool -k -g -l from ubuntu and compare to openwrt ones.

grandiego, 2023-08-16:
I had irqbalance already installed in openwrt prior to bug report. Second port is down as there's nothing plugged in it.

be2net does not load any firmware, it uses whatever version is in NIC flash.

Boot messages in OpenWrt's dmesg

```
[    7.981686] be2net 0000:01:00.0: FW config: function_mode=0x10003, function_caps=0x7
[    8.401700] be2net 0000:01:00.0: Max: txqs 16, rxqs 17, rss 16, eqs 16, vfs 0
[    8.408958] be2net 0000:01:00.0: Max: uc-macs 30, mc-macs 64, vlans 64
[    8.415893] be2net 0000:01:00.0: enabled 4 MSI-x vector(s) for NIC
[    9.261830] be2net 0000:01:00.0: created 4 TX queue(s)
[    9.561815] be2net 0000:01:00.0: created 5 RX queue(s)
[    9.687511] be2net 0000:01:00.0: FW version is 12.0.1318.0
[    9.694238] be2net 0000:01:00.0: HW Flow control - TX:1 RX:1
[    9.718139] be2net 0000:01:00.0: Emulex OneConnect(be3): PF  port 1
[   10.201814] be2net 0000:01:00.1: FW config: function_mode=0x10003, function_caps=0x7
[   10.561842] be2net 0000:01:00.1: Max: txqs 16, rxqs 17, rss 16, eqs 16, vfs 0
[   10.569081] be2net 0000:01:00.1: Max: uc-macs 30, mc-macs 64, vlans 64
[   10.575987] be2net 0000:01:00.1: enabled 4 MSI-x vector(s) for NIC
[   11.361831] be2net 0000:01:00.1: created 4 TX queue(s)
[   11.661816] be2net 0000:01:00.1: created 5 RX queue(s)
[   11.787487] be2net 0000:01:00.1: FW version is 12.0.1318.0
[   11.794217] be2net 0000:01:00.1: HW Flow control - TX:1 RX:1
[   11.806583] be2net 0000:01:00.1: Emulex OneConnect(be3): PF  port 2
``` 
Ubuntu dmesg shows:

```
[    7.148568] be2net 0000:01:00.0: PCIe error reporting enabled
[    7.449091] be2net 0000:01:00.0: FW config: function_mode=0x10003, function_caps=0x7
[    7.592910] be2net 0000:01:00.0: Max: txqs 16, rxqs 17, rss 16, eqs 16, vfs 0
[    7.592922] be2net 0000:01:00.0: Max: uc-macs 30, mc-macs 64, vlans 64
[    7.593108] be2net 0000:01:00.0: enabled 2 MSI-x vector(s) for NIC
[    7.772887] be2net 0000:01:00.0: created 2 TX queue(s)
[    7.864877] be2net 0000:01:00.0: created 3 RX queue(s)
[    7.925907] be2net 0000:01:00.0: FW version is 12.0.1318.0
[    7.926938] be2net 0000:01:00.0: HW Flow control - TX:1 RX:1
[    7.973519] be2net 0000:01:00.0: Emulex OneConnect(be3): PF  port 1
[    7.974726] be2net 0000:01:00.1: PCIe error reporting enabled
[    8.229022] be2net 0000:01:00.1: FW config: function_mode=0x10003, function_caps=0x7
[    8.373059] be2net 0000:01:00.1: Max: txqs 16, rxqs 17, rss 16, eqs 16, vfs 0
[    8.373084] be2net 0000:01:00.1: Max: uc-macs 30, mc-macs 64, vlans 64
[    8.373350] be2net 0000:01:00.1: enabled 2 MSI-x vector(s) for NIC
[    8.541051] be2net 0000:01:00.1: created 2 TX queue(s)
[    8.613026] be2net 0000:01:00.1: created 3 RX queue(s)
[    8.689949] be2net 0000:01:00.1: FW version is 12.0.1318.0
[    8.691016] be2net 0000:01:00.1: HW Flow control - TX:1 RX:1
[    8.696821] be2net 0000:01:00.1: Emulex OneConnect(be3): PF  port 2
[    8.701923] be2net 0000:01:00.1 enp1s0f1: renamed from eth1
[    8.729497] be2net 0000:01:00.0 enp1s0f0: renamed from eth0
[   35.805023] be2net 0000:01:00.0 enp1s0f0: Link is Down
[   35.856399] be2net 0000:01:00.1 enp1s0f1: Link is Down

``` 
So openwrt creates 4 MSI-x vector(s) for NIC but ubuntu just creates 2. Same for TX/RX queues: 4/5 for OpenWrt and 2/3 for Ubuntu.

Regarding ethtool -k -g -l, they're excatly the same except for ethtool -l output:

OpenWrt
```
Channel parameters for eth0:
Pre-set maximums:
RX:		3
TX:		3
Other:		n/a
Combined:	4
Current hardware settings:
RX:		0
TX:		0
Other:		n/a
Combined:	4

``` 
Ubuntu
```
Channel parameters for enp1s0f0:
Pre-set maximums:
RX:		3
TX:		3
Other:		n/a
Combined:	4
Current hardware settings:
RX:		0
TX:		0
Other:		n/a
Combined:	2

``` 

So I booted OpenWrt and ran

```
ethtool -L eth0 combined 2
ethtool -L eth1 combined 2
``` 
The output
```
[   46.082256] be2net 0000:01:00.0: enabled 2 MSI-x vector(s) for NIC
[   46.502007] be2net 0000:01:00.0: created 2 TX queue(s)
[   46.681991] be2net 0000:01:00.0: created 3 RX queue(s)
[   46.817713] be2net 0000:01:00.0 eth0: Link is Down
[   55.492214] be2net 0000:01:00.1: enabled 2 MSI-x vector(s) for NIC
[   55.912010] be2net 0000:01:00.1: created 2 TX queue(s)
[   56.091994] be2net 0000:01:00.1: created 3 RX queue(s)
[   56.227746] be2net 0000:01:00.1 eth1: Link is Down

```

So now ethtool -l output is the same.  

I ran iperf test again and unfortunately I was able to reproduce the bug. So I decided to try decreasing combined queues from 2 to 1 , but it still fails.

Just for completion, I had a bunch of these NICs running on proxmox 7 using kernel version 5.15.x working flawlessly.

Thanks for your help, anyway

brada4, 2023-08-16:
Try snapshot after aspm supported (if it fails compare power mode with ubuntu)
https://github.com/openwrt/openwrt/issues/13248